linuxboot/heads #533: Linux /boot signing script

itay-grudev, 2019-02-24:
I think it is about time that we solve the really annoying problem of `initrd` mismatch after updates.

We can create a linux package called `heads-sign` that adds an `apt` hook and prompts to sign your `/boot` after an update that involved a kernel or `initrd`. We can add a `grub` hook as well.

Right now, if I update my computer, forget that I updated it and reboot I won't know if the `initrd` (or more rarely kernel) was changed by `apt` or an adversary.

Also given that PureOS is running on top of the Debian Buster repository `initrd` updates are quite common and this issue is really annoying me.

I tried creating the script once, but I messed it up. Unfortunately I don't think I'm not the most qualified person to create a PR for this.

kylerankin, 2019-02-25:
When I first started working on integrating Heads into Librem laptops I also ran into this problem. At first I worked to implement just what you describe--an interactive apt trigger that would launch a GUI and walk you through re-signing files in /boot. The problem, though, is that most distributions are moving away from interactive package installation and into things like Packagekit, which reboot, apply system updates in a restricted, safer environment, and then reboot back into the updated system.

Packagekit is explicitly against any interaction when updating packages--it's supposed to be completely non-interactive. This is tricky to address. The approach I've come up with so far and have worked to get into PureOS is an apt hook that non-interactively creates a special log file in /boot that Heads can refer to. This log file simply makes a note of what files apt has updated (similar to the APT history log file that already exists on the system).

Note that Heads does not *trust* this file. Just that if it exists, it will show information from that file to the user when it detects a change in /boot files, so the user can make an informed decision about whether they think the change is because of those updates.

itay-grudev, 2019-02-26:
That's a bummer and definitely less secure. In fact a classy attack would include a realistically looking log file and a changed `initrd`. Right now it will say `RED - don't do it. Don't give you password` and with that `Hey, your initrd was changed due to system updates, so it's probably OK`.

Do you plan to adopt Packagekit in PureOS?
Because if not or not for now we - a hook would do a really nice job.

kylerankin, 2019-02-26:
PureOS desktop already uses Packagekit when you get update notifications via the GUI.

With respect to your critique on the security of the approach, handling the lifecycle of signing files in /boot on a system that gets periodic kernel updates (or package updates that update initrd, which are many) is a challenging security problem. In the case of *any* notification sent to the user, whether within Heads or within your desktop, an attacker with root access could spoof a notification. On the desktop for instance, they could trick you into signing things during a large package update that didn't include initrd changes by triggering the script anyway (or adding their own hook that adds their rootkit while initrd is being generated but before you sign from the desktop). One can think of a lot of plausible-but-complicated-to-execute scenarios like this.

At least in the case of handling the notification within Heads, you are in a more trusted environment, and only displaying some information from a package log. The attacker who modified the files in /boot (including the package log) still cannot control the contents of the message that Heads displays, they can only control that it could list some plausible-looking packages that might be the cause of the change. The skeptical user would still get the alert and may opt not to re-sign files. It's a better approach that was preceded it, which was to just post a scary warning to the Heads user with no context, for what is more likely than not a false positive triggered by a package update.

itay-grudev, 2019-02-27:
How about the following alternative. We can introduce the following concept:

1. User signed boot (maximum trust)
2. System signed boot (medium trust)

Here is my line of thinking. The updater is already trusted. Right now we trust the generated `initrd` images and the kernel. So if we store a key somewhere on the machine that only `root` has access to, the updater can self-sign the modified files (system signed boot).

The attack vector is gaining root access and stealing the key. But if you can do that, you can also compromise the updater, which in turn can compromise the `initrd` and the kernel. So essentially the security of my proposal is in the worst case equivalent and in the best case superior.

kylerankin, 2019-02-28:
I understand your line of thinking, but I'm not convinced that you get any extra security by having a root-owned key with no passphrase that would sign files in /boot. Any attacker who could set up a root kit could also use this key to sign it. The value of having the user sign these files is that it's something an attacker with root *can't* do as the attacker doesn't have access to the OpenPGP smart card or the user's PIN.

The more I think about it, the more I prefer having these situations handled within the more trusted environment of Heads, instead of within a user's OS. A user who goes through the more common Packagekit-based non-interactive workflow would reboot to update system packages in a very limited single-user environment, immediately reboot, and immediately see the Heads warning and feel comfortable re-signing. Any other instance where they would see that warning would be outside of package updates and rightly make them suspicious.

Your use case of using the command line to update packages is certainly a valid way to do it but it's a method that requires more manual work and expertise. In your case it requires you to reboot immediately whenever you see that the initrd was updated by APT. It also requires OS-specific hooks we'd need to port everywhere. The method I used for PureOS to create a log file in /boot falls back gracefully to a more generic warning in case those hooks aren't in place and the log doesn't exist.

itay-grudev, 2019-03-01:
@kylerankin 

An attacker with a root kit can just modify `update-initramfs` to embed a shady binary inside. And you don't check or sign the integrity of the updater. You trust it implicitly. That is why I believe the security of my approach is equivalent.

On the other hand my approach mitigates the attack where a attacker with no root kit fakes a valid update and tricks the user into signing their otherwise compromised boot.

And keep in mind that I don't want to remove the user signed boot. I just want to complement it. I really like the idea of signing the boot with an external device with a secure element and that gives you the maximum security. I just want to add the ability for the system to sign the files instead of just logging that they were changed.

Heads could still warn the user that the files was changed in _yellow (instead of red)_ when the system modified the files. And red when both the system and the user don't match.

tlaurion, 2019-05-21:
@itay-grudev : I love the idea.

jtmoree-github-com, 2022-05-11:
If I understand the conflict between the two proposals it centers around automation including PackageKit.    What if we give ourselves the ability to manually update the signatures for now?   For those of us not trapped in Packagekit we would apply updates then manually run the sigs update.  

> Your use case of using the command line to update packages is certainly a valid way to do it but it's a method that requires more manual work and expertise. In your case it requires you to reboot immediately whenever you see that the initrd was updated by APT

Do you not reboot after kernel updates?  I reboot immediately after any update that affects sigs because later I won't remember that an update happened.  For example, I rebuilt my system last summer though likely I wasn't compromised.  I couldn't remember the last time I updated and didn't want to take any chances.

tlaurion, 2022-05-11:
One of the approach suggested over Qubes was that the OS signs the kernel, xen and initrd. That unfortunately would not prevent grub config to be lessen, changing xen or kernel command line options between two signed boot options.

The ideal there, for Heads at least, would be that the binaries are detached signed by the distribution and included in the packages being deployed so those detached signatures land on /boot and Heads can verify those.

Heads creates kexec_default.* for selected default boot options. Heads could be improved to show differences between what would be signed next and what was signed prior of permitting a resign there.

Other then that, Qubes is also working on UX notifications. It is a problem for everyone: core system upgrades touching kernel initrd and Xen requires an imminent reboot to stay up to date. The solution there might be as simple as reminding to the user that he needs to reboot.

Under booted OS, it is possible to reuse trusted installer, keys deployed and repository information to validate one last time prior of reboot to make sure that the integrity of those files match what was deployed. For Qubes being Fedora based for the moment, the integrity validation could be done by calling rpm -V on core system packages to notify the user of authenticity and integrity of the updated core packages prior of a reboot.

To synthesise: I think the best approach would be to implement a better UX under OSes to generally inform, and remind, the user that core system packages have been deployed. Deploying addition hooks there to validate integrity prior of rebooting would fix the issue at stake. The point here is that the package manager is the best placed, being already trusted, to validate both integrity and authenticity of deployed package binaries. 

Otherwise, documentation here again could fix the issue as well, but is less intuitive. Heads could suggest an unsafe boot for the user to go back in the OS and run those checks manually prior of signing. Where non-intrusive OS notification reminding a reboot is required when core OS components have been upgraded seems the best avenue to me.

@kylerankin ?