linuxboot/heads #535: Proper way to run memtest from heads?

agent509, 2019-03-14:
I have been unable to run memtest from heads. Every attempt to kexec the memtest kernel has simply left the system frozen at "Starting new kernel." Is it possible to run memtest from within heads? What is the best way to do this? Thanks

tlaurion, 2019-03-17:
@agent509 : what have you tried?
I would expect memtest built as a module and launched directly from the command line to work.

sildur, 2020-08-04:
I have tried every combination I could think of kexec. I found the latest version of memtest86+, v5.31b, works with kexec -f -t elf-x86 memtest works. It seems like it works at least because as soon as it starts, it overwrites the frame buffer, and all I can see is different colors in the screen as it tests patterns. Is there a way to avoid that?

tlaurion, 2020-12-12:
I confirm Heads not being able to kexec into QubesOS memtest from ISO boot, not recognising format.

Seems like the path would be to have the standalone binary dropped onto a usb dongle and launch the application from there.
```
mount-usb
/media/memtest
```

tlaurion, 2020-12-12:
> I have tried every combination I could think of kexec. I found the latest version of memtest86+, v5.31b, works with kexec -f -t elf-x86 memtest works. It seems like it works at least because as soon [...]

Depending on board config, coreboot might not be responsible to init graphics nor framebuffer, which is most often initialized by the linux (i915 gpu driver or others) while most boards initialize graphics first by coreboot then linux driver initializes the framebuffer (boards with FBWHIPTAIL support.)

The kexec appended magic happens at this [line](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-boot#L73), appending arguments (ADDED or REMOVED per board config) prior of attemtping to kexec the passed kernel.

In the case of QubesOS ISO memtest kexec, I have not investigated the issue myself, while elf-x86 seems to need to be explicited.

Let us know your findings.

ryanbarry, 2021-02-07:
I have just run into this as well running Heads v0.2.0-977-ga81ae6e on an x230, have made several attempts and all fail to kexec with the message "Cannot determine the file type of <path to memtest>"

1. from Ubuntu Server 20.10 image (also tried 20.04 and an arch image from 2020) written to a USB drive, I pick USB boot from Heads' menu, select the boot partition of the USB drive, then select the "Test_memory_[/boot/memtest86+.bin]" boot option and get the following:
  
[image]

2. in the pictured recovery shell, try running `kexec -f -t elf-x86 /media/boot/memtest86+.bin` and get the same result, so specifying the type doesn't seem to be helpful
3. download latest memtest86+-5.31b.bin from the source, drop into a freshly-formatted USB drive, get a recovery shell in heads, run `mount-usb` then `kexec -f -t elf-x86 /media/memtest86+-5.31b.bin`... same error results

i even tried all the types kexec supports, just to see if it makes a difference, but no, same result each time. i'm not sure what else to try, i thought maybe i was mistaken about the file type i had downloaded, so in desperation after running this:

```
> file Downloads/memtest86+-5.31b.bin
Downloads/memtest86+-5.31b.bin: DOS/MBR boot sector
```

i just did a `dd if=Downloads/memtest86+-5.31b.bin of=/dev/<usb drive>` and looked at the result, but it's definitely not an image... got an MBR that was all sorts of wrong (1TB partition on 2GB usb disk, for example)

jtmoree-github-com, 2021-10-05:
Got new RAM.  need to test.  Ran into this.  I haven't found any posts of people successfully running memtest86 from kexec--just posts of people having problems.  Most people don't have to run from kexec but since heads is the first thing run from the BIOS we are in a minority who cannot reboot and have bios launch memtest for us.  This means we have very few options to test RAM on this hardware.

Some people get the error listed above and others get a screen that looks like it is doing something but they cannot read any useful information.  I get the error.  If anyone reading this has been able to run memtest86 from kexec and/or heads please post as an FYI.

- https://forums.puri.sm/t/cant-run-memtest/12216
- https://www.reddit.com/r/Purism/comments/i79kpl/running_memtest86_with_pureboot/
- http://lists.infradead.org/pipermail/kexec/2008-October/002849.html
- https://www.mail-archive.com/linuxbios@listman.lanl.gov/msg04075.html

mfc, 2022-05-20:
> Finally!
> 
>     1. Download https://memtest.org/download/v6.00b1/mt86plus_6.00b1_64.grub.iso.zip
> 
>     2. Prepare USB thumb drive (dd iso to thumb drive)
> 
>     3. Go to Heads recovery shell
> 
>     4. `mount-usb` select first partition
> 
>     5. `kexec-f /media/boot/memtest --command-line keyboard=legacy`
> 
>     6. Enjoy

i can confirm this works with latest Heads firmware!

closed 2025-02-13 by tlaurion