linuxboot/heads #538: X220 boots to a black screen and after 1.5 minutes yells at me.

seijinshu, 2019-03-20:
From my prior experience with coreboot, the endless beep signals a RAM error, but it happens with several, several different combinations of RAM and no RAM.

Using the stock configs, built from master branch, and did made sure to extract the blobs.

Using the --ifd --image bios options in flashrom gave me an endless blackscreen with no beep. 

X220 model is the i7 model (which has USB 3), but shouldn't make any difference as it has been tested to work with coreboot.

SebastianMcMillan, 2019-03-20:
I'm also getting this on my X220. I ported heads to my T420, and when I get home tonight I will flash my system and test to see if it's broken, as it's based off of the X220 configs. I also can test it on my X230t to determine if it's a X220 issue or a platform-independent issue, or if there's some sort of error on my end.

SebastianMcMillan, 2019-03-21:
I think your beep error is some sort of user error. Using similar configs as the X220, only modified to target the T420, I get to the heads boot menu. However, I think the delay your experiencing is a heads issue, as I get the issue on my T420 and my X220. I am building heads for my X230t as I am typing this. Refer to #541 for that issue, and I would recommend wiping your copy of the repo, reclone the repo and try building again to fix the error-code. I believe your issue is coreboot being unable to jump to the payload.

seijinshu, 2019-03-21:
I will go ahead and close this issue. I will reopen if I cannot produce a working build, and I will keep an eye on that issue, as that delay will be an issue.

closed 2019-03-21 by seijinshu