linuxboot/heads #540: Call for Collaboration to Apply to NGI "Privacy & Trust Enchancing Technologies" Grant

tlaurion, 2019-03-20:
Grant Information here: https://nlnet.nl/PET/
Form here: https://nlnet.nl/propose/

Framapad for grant proposal collaboration: https://mensuel.framapad.org/p/AQqOdrEBWr

Anyone wants to jump in?

tlaurion, 2019-03-20:
@merge @marmarek @osresearch @kakaroto @mfc : 
Any other improvements you would love to see happening into Heads Heads/QubesOS, integration of better privacy/confidentiality enforcing measures that would benefit all privacy conscious?

merge, 2019-03-21:
what comes to mind, but that's more of a question: how far are we with "forbid software-flashing everywhere except in one root-of-trust place inside the flash"?, for the coreboot side, it started with this discussion: https://coreboot.coreboot.narkive.com/ovqZj0CI/spi-controller-and-lock-bits and I don't know how far that is. In Heads we should use that and maybe, when we read a grub config off disk, think about warning/forbidding kernel parameters like "iomem=relaxed"... thanks.

tlaurion, 2019-03-21:
@merge : #326?

tlaurion, 2019-03-21:
> what comes to mind, but that's more of a question: how far are we with "forbid software-flashing everywhere except in one root-of-trust place inside the flash"?, for the coreboot side, it started [...]

@merge : #326? Never tested it though.

> In Heads we should use that and maybe, when we read a grub config off disk, think about warning/forbidding kernel parameters like "iomem=relaxed"... thanks.

It could be as easy as putting that [here](https://github.com/osresearch/heads/blob/master/boards/x230/x230.config#L31):
export CONFIG_BOOT_KERNEL_REMOVE="iomem=relaxed"

marmarek, 2019-03-21:
> @merge : It could be as easy as putting that [here](https://github.com/osresearch/heads/blob/master/boards/x230/x230.config#L31):
> `export CONFIG_BOOT_KERNEL_REMOVE="iomem=relaxed"`

This isn't viable solution - there are way too many potentially harmful kernel options. If anything, there should be a whitelist of allowed options.

merge, 2019-03-21:
thanks for the links and updates. sorry for the confusion. actually it's quite unrelated and not a good idea to mess with user's kernel commandline so much. All we need for a proper root of trust is (persistent until reboot-)write-protection of the flash as soon as possible __unless__ the user chooses "upgrade" or something. Once we have that, it should really be irrelevant what iomem or others say.

We really need to verify whether that works for the X230's flash chips, and implement a user-friendly solution.

tlaurion, 2019-03-21:
> This isn't viable solution - there are way too many potentially harmful kernel options. If anything, there should be a whitelist of allowed options.

@marmarek: agreed. But:

- Potentially harmful added kernel options arrives into grub configuration per logged in user. 
`/boot` changes are detected [here](https://github.com/osresearch/heads/blob/master/initrd/bin/gui-init#L17), since /boot hashes are generated at each boot and[ validated against user's signed kexec.sig](https://github.com/osresearch/heads/blob/master/initrd/etc/functions#L200), so no file under /boot can change without the user needing to approve the change. IMHO, a whitelist is overkill.

mfc, 2019-03-21:
hey folks, just chipping in that the Qubes proposal I will draft will be focused on Qubes usability (integrating existing efforts that still haven't been reviewed/incorporated into the release) and internationalization/localization/accessibility support. so it will be a bit "conservative" in approach but filling in some very necessary gaps Qubes has.

I don't expect any overlap with these topics so the Heads/Qubes proposal can be as ambitious as desired! :)

SebastianMcMillan, 2019-03-21:
I think that one idea for heads-specific work is to port to more devices. Ideally, support more ThinkPads and as many Chromebooks as feasibly possible. 

On the ThinkPad side of things, since xx20 and xx30 are incredibly similar, it should be rather simple to port kernel configs to T420 (I currently have a PR for that), T430 (someone else seems to be on that), T420s/T430s, T520 and T530. I presume the T410, T510 and X201(s/t) should also be trivial as well (they may even work with the same config to the X230 config that the X220 and the T420 are using currently).

For Chromebooks, the most ideal and easiest way to do this is to collaborate with Mr. Chromebox, which maintains custom coreboot builds for many, many Chromebooks. If we can get the patches used for heads to be merged into mainline coreboot, that effort would be very easy, all that's needed is for someone to update Mr. Chromebox's repo (either via a fork or getting the new base to the main repo) and having a *mostly* universal kernel config for them to use when building the kernel.

Having several different hardware options for users to choose from would not only open heads up to more users, but it could also lead it to being more streamlined than it currently is.

Another thought I just had was to get Heads (and even possibly Qubes) running on ARM devices (namely ARM Chromebooks). ARM-based Chromebooks *may* remove the threat of IME/PSP, unless they also have a hardware level backdoor that I'm presently unaware of.

On the Qubes side of things (independent of Heads), one thing is to get the GUI out of dom0. I've heard chatter about this from various places, but this would be **huge.** 

Another thing worthy of being implemented is proper touch screen and stylus proxy support. I do know that this would be very difficult considering USB input device security is already a problem in itself.

I apologize for this text wall and if it's difficult to read.

tlaurion, 2019-03-22:
> I think that one idea for heads-specific work is to port to more devices. Ideally, support more ThinkPads and as many Chromebooks as feasibly possible.

@SebastianMcMillan : Agreed, with some personal limitations, see below.
> On the ThinkPad side of things, since xx20 and xx30 are incredibly similar, it should be rather simple to port kernel configs to T420 (I currently have a PR for that), T430 (someone else seems to be [...]

Agreed, while again, the most free platforms (binary free of Intel FSP/no AMD PSP, me_cleaner neuterable Intel ME platorms containing only ROMP and BUP modules) should be priorized in that list. This is why I push so much the x230 right now and think that sandy bridge and ivy bridge based laptops should be supported more. This doesn't mean that other platforms should not be added in parallel. My point here is that my own priority is to push most trustaable hardware available today (binary blobfree 4.9 coreboot  (graphic init, native ram init, etc) with Measured boot support, TXT support(optional), TPMv2 support) well supported/tested before attacking other platforms. 

I invite others to collaborate on that and will gladly add those platforms in the points to be covered for the grant but won't do that work myself nor will be able to test those platforms, while inviting the community to do it and be paid for those contributions.
> 
> Another thought I just had was to get Heads (and even possibly Qubes) running on ARM devices (namely ARM Chromebooks).

Follow GSoC progress for [ARM64 here](https://github.com/QubesOS/qubes-doc/pull/772#issuecomment-462172880). I'm personally more interested into [PPC64 port](https://github.com/QubesOS/qubes-issues/issues/4318), for the same reasons evoked above with even more user ownership possibilities ([PowerPC notebook](https://www.powerpc-notebook.org/en/), Power9 completely open designs like the [Talos II/BlackBird and others](https://www.raptorcs.com/content/base/products.html).

> ARM-based Chromebooks _may_ remove the threat of IME/PSP, unless they also have a hardware level backdoor that I'm presently unaware of.

TrustZone is [IME equivalent](https://en.wikipedia.org/wiki/TrustZone#Security_extensions).
This is an interesting file to follow, but not all ARM architectures include [virtualization extension](https://wiki.xen.org/wiki/Xen_ARM_with_Virtualization_Extensions), and from what I saw, the ones that does, for laptops, don't have enough memory to be minimally interesting for QubesOS(8Gb+). But that might change in the future.

> 
> On the Qubes side of things (independent of Heads), one thing is to get the GUI out of dom0. I've heard chatter about this from various places, but this would be **huge.**

It's planned for QubesOS 4.1. Can't find reference right now.
> 
> Another thing worthy of being implemented is proper touch screen and stylus proxy support. I do know that this would be very difficult considering USB input device security is already a problem in itself.
I've roughly tested this myself and it works:
[Scripts](https://github.com/martin-ueding/thinkpad-scripts)
[Tablet support in QubesOS](https://github.com/marmarek/qubes-app-linux-input-proxy/commit/06b20b02efb332793f25f1b3b88a869169896f0c)

> 
> I apologize for this text wall and if it's difficult to read.
No problem. :)

SebastianMcMillan, 2019-03-22:
>TrustZone

How did I forget this? That name makes me shudder every time I hear it. Thankfully, it's not a part of ARM in itself, and that not all current ARM devices have it (yet).

I do agree that testing is very important for a project like this, and that also putting more work more free platforms than others is a better choice (seeing how user trust is the entire point of this project). I wouldn't mind maintaining the T420 and X220.

Another thing I recall reading back in the later half of 2017 was that Libreboot was in the works for the X220 and that it would be released in 2018. Seeing how that never happened, I may try to find where I saw that and track down what caused that to never happen. If someone is able to somehow find a way to get rid of the blobs on the X220, that'll likely be relevant to other Sandy Bridge devices at the very least.

tlaurion, 2019-03-22:
@SebastianMcMillan 

Libreboot on x220 was conditional to the tiny possibility to have it RYF certified if IME could be completely wiped, which, thanks to Trammel and me_cleaner afterward, showed it not possible. 

RYF guidelines and conditions would consequently need to tolerate neutered IME (90Kb BUP and ROMP required modules to fire up main CPU).

I would greatly doubt RYF certification criteria to change. I think the debate happened on Reddit under Leah's name. The x220 product page can be found with web.archive.org for minifree.org.

 

_Sent from my Galaxy S3 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

SebastianMcMillan, 2019-03-22:
That's the picture I seem to be getting as well. I apologize for getting a little off-topic with that thought.

tlaurion, 2019-03-24:
@flammit, @marmarek, @mfc, @osresearch

Anything to else to add? I'll begin working the proposal in the beginning of the week.

Would gladly take reviews/corrections on the framapad link in first post while I go.

tlaurion, 2019-03-31:
Thanks to all that collaborated! The [grant application](https://mensuel.framapad.org/p/AQqOdrEBWr/timeslider#11705) was delivered in time!

tlaurion, 2019-04-30:
Retained. Gonna be considered and evaluated against all other projects in the next 3 weeks before final approval/refusal.

tlaurion, 2019-06-07:
Passed! Last stage!
@mfc @marmarek @blackmaria @merge @osresearch @SebastianMcMillan @kylerankin 

> 
> [...] you applied to the NGI0 PET open call from NLnet. Currently a selection of the projects is with the independent review committee to validate their eligibility, and we are happy to inform you that this includes your project "Integrating measurable security into daily lives" (2019-04-XXX).
> 
> Should your project pass that final hurdle, the selection will be made public and we will contact you to negotiate a Memorandum of Understanding. The final amount of the grant will be determined at that point.
> 
> We will then also need to share some information about the project both with the general audience and with the European Commission. In the interest of time, we ask you to prepare a one paragraph management summary of the project. For examples we refer you to https://NLnet.nl/thema/NGIZeroDiscovery.html and https://NLnet.nl/thema/NGIZeroPET.html

tlaurion, 2019-06-11:
The "[Finish porting Replicant to a newer Android version](https://nlnet.nl/project/ReplicantUpdate/)" fits similar general background behind this grant application.

The idea is to Firmware/boot integrity attestation on more hardware platforms (Heads support for smaller SPI, broader support through VBoot measured boot subconfig: Hedas), ease accessibility and support of such hardware (OEM reownership (Heads), TXT support for AEM(Coreboot/Heads), additional hardening/anonymizing through Salt (QubesOS/Whonix) and new user support/integration (through hidden onion remote AdminVM administration (QubesOS/Whonix).

Anyone has the talent to write such executive summary? @BlackMaria?

BlackMaria, 2019-06-11:
@tlaurion  sure, I will try to write something for the `one paragraph management summary of the project` but I dont understand your point about the similarities with the android text?

IIUC our text only needs to be one paragraph and the examples are provided. 

I will aim for tomorrow evening to write this.

tlaurion, 2019-06-12:
@BlackMaria : I was referring to the tone mainly of the executive summary, since it's about supporting more hardware and setting bases to ease further platform integration (most of the budget would go on that).

BlackMaria, 2019-06-13:
### This is my suggested text for such a proposal

The "Integrating measurable security into daily lives" project's initiative was sparked by the need for usable security made available to the average citizen.  Several projects are contributing a part of this bigger puzzle, QubesOS, coreboot, Heads, me_cleaner, Whonix and others, yet the average person does not have the sophistication to integrate these software projects.  With some effort we can add some missing parts, help the effected projects usability, and facilitate access, cutting-edge developments, currently, only available to the developers and more sophisticated users.  Bringing these projects together will reduce the amount of expertise and effort required to benefit from these project.

tlaurion, 2019-06-14:
Project name changed to "Accessible Security" following Michiel request to ease communication exchanges about the project.

tlaurion, 2019-06-14:
@SebastianMcMillan 
Update in regard of [RYF certification guidelines](https://www.fsf.org/resources/hw/endorsement/criteria). Contacted FSF, waiting for clarifications, since, from what I undertand, Intel ME is considered an exception:
"However, there is an exception for secondary embedded processors. The exception applies to software delivered inside auxiliary and low-level processors and FPGAs, within which software installation is not intended after the user obtains the product. This can include, for instance, microcode inside a processor, firmware built into an I/O device, or the gate pattern of an FPGA. The software in such secondary processors does not count as product software."

SebastianMcMillan, 2019-06-15:
The only issue with that is ME is updatable.

tlaurion, 2019-06-15:
> The only issue with that is ME is updatable.
> […](#)

As is the EC in laptops, the microcode updates for processor (now required....), TPM firmware, hard drive firmware etc. 

The GM45 (x200) was certified in a time microcode updates were a probable prejudice to security, while right now, not having them applied is the prejudice to security. 
The EC in GM45 was upgradeable, still certifiable. 

The question being what are they considering acceptable. The open source SSD/HD firmware is still not readily accessible. But closed source is tolerated on that level. The X200 was certified even with those closed source components.

Everything is upgradable with physical access.

The point being, I think, that once SPI is locked down (#326) or measured (not done actually, but measurements could be extended with something like #493 (complete rom read measurements, or modules measurements) or VBOOT/measured boot in coreboot 4.9+ to include ME integrity checks), what would be RYF certifiable. 

I wrote back an e-mail dissecting differences in vocabulary used to describe Intel ME states. 
Removed(GM45 RYF certified)/Neuteured (ROMP BUP + AltMeDisable bit) /Neutralized or Deactivated (AltMeDisable bit/ HAP bit while modules still being in SPI [kernel syslib and rbe]). 

Waiting for a feedback in currently made differentiation.

tlaurion, 2019-07-04:
@zaolin @marmarek @BlackMaria @mfc @osresearch @kylerankin @adrelanos @merge

Project receives grant :)

mfc, 2019-07-04:
Qubes just received confirmation as well :)

tlaurion, 2019-08-07:
@zaolin @marmarek @BlackMaria @mfc @osresearch @kylerankin @adrelanos @merge
[NGI0-intakedocument.pdf](https://github.com/osresearch/heads/files/3477658/NGI0-intakedocument.pdf)

As stated in this document, it is expected to have smaller deliverable, so that the funds are transmitted directly to you from NLNet, every two weeks upon completion of tasks. The results must be public and reproducible.

Questions?

tlaurion, 2019-08-07:
# Heads/Coreboot specific work #
@zaolin
- [ ] Properly support Intel based, most binary free initialized platforms in coreboot (Ivy Bridge and Sandy Bridge). This implies integrated [Measured boot improvements that landed in Coreboot 4.9+](https://github.com/osresearch/heads/issues/500), which sits atop of VBOOT and change Head's internal logic to take advantage differently of TPM PCRs to have proper and mainstreamed BIOS integrity measured. Technically, it means:
  - [ ] integrate/ test : 
    - [ ] [Add VBOOT support](https://review.coreboot.org/c/coreboot/+/29816) 
    - [ ] [Enable VBOOT support](https://review.coreboot.org/c/coreboot/+/29815/1)
  - [ ] check if fmap is correct
  - [ ] set CMOS flags correctly
- [ ] [Support TXT in Coreboot/Heads, so that [QubesOS Anti-Evil Maid](https://www.qubes-os.org/doc/anti-evil-maid/) can be used with coreboot Open Source Firmware](https://github.com/osresearch/heads/issues/502). **@marmarek said this was not considered a necessity. A discussion should be engaged between you two before going forward.**
- [ ] Integrate write protection, preventing firmware modification from the Operating System through io386 lock down module #326 
- [ ] Upgrade disk encryption (cryptsetup LUKS) support in Heads:
  - [ ] QubesOS dom0 fedora-29/debian-9 based to be able to support [LibremKey-enabled LUKS container unlocking](https://docs.puri.sm/Librem_Key/Getting_Started/User_Manual.html#decrypt-luks-encrypted-drives-with-librem-key) so that Heads can take advantage of this in a generic way.
  - [ ] [Integrate Kali LUKS key slots wipe](https://github.com/DanielG/asmb4/) permitting the user to deny  themselves and others system and data access, basically locking themselves out of the system until a full disk backup is restored. **EDIT: cryptsetup was upgraded to 1.75 with kali patches in [here](https://github.com/osresearch/heads/issues/441#issuecomment-505964330)**
- [ ] [Reduce Heads Whiptail graphical footprint](https://github.com/osresearch/heads/issues/517) to fit a broader range of devices having 4Mb SPI flash, including x200 (Tails perfect Laptop. More generally in 4MB flash, while available in 8MB SPI if vPro) and G505s and probably a lot of others.
- [ ] [Implement forward sealing of firmware upgrades](https://github.com/osresearch/heads/issues/523)

mfc, 2019-08-07:
FYI just had call with NLNet regarding our own grant and to be clear, a timeline is not very important, just the deliverables. you can ask for funds for a completed deliverable (or sub-task) right after you complete it, or months later if that is useful for you. 

Thierry I am assuming since you are the point of contact on this grant you will make clear who is working on which deliverables, get their consent and confirmation on it and the associated costs, and if there are any deliverables missing owners that you will flag that and we can find folks to work on them.

tlaurion, 2019-08-07:
# QubesOS/Whonix related work #
@marmarek @adrelanos
## Create and deploy additional Salt recipes at QubesOS installation/new defaults that ##
- [ ]  Grow QubesOS user base by developing third party assistance services base.
  - [ ]  Develop Whonix (sys-whonix)[ on-demand hidden onion GUI, opening dom0 remote support through SSH](https://github.com/SkypLabs/my-qubes-os-formula/issues/6) and [socat](https://www.whonix.org/wiki/Dev/Qubes#ssh_into_Qubes_dom0). Whonix ticket [here](https://phabricator.whonix.org/T896). @adrelanos: @marmarek provided approximate billing time that got filed for the project. The ideal would be to have a GUI that can manage access to dom0 from sys-whonix, on-demand, when support is necessary. The GUI would permit to copy-paste hidden onion service address and authentication. It could be VNC or other mean. The ideal being the user seeing what is happening to his system while remote support is ongoing. 
- [ ] [Properly handle qube's LVM freed space at all levels](https://github.com/SkypLabs/my-qubes-os-formula/issues/4), reducing catastrophic user experience of running out of of disk space.
- [ ] [Deploy qubes-update-cache ](https://github.com/SkypLabs/my-qubes-os-formula/issues/4)and make update-vm dependent on it, so that packages are downloaded once per TemplateVM types (fedora, debian, whonix-gw, whonix-ws) even if many clones were created and specialized out of them. Keeping one's system up to date then requires only minimal bandwidth. EDIT: since all update sources are now HTTPS, this feature is now questionable.
- [ ] Deploy [good privacy default for NetworkManager](https://github.com/SkypLabs/my-qubes-os-formula/issues/3), forcing randomized mac usage when not told otherwise.

closed 2021-10-31 by tlaurion