linuxboot/heads #543: x220 port not working as it is?

tlaurion, 2019-03-27:
I've seperated x230 linux configs from x220, reverting the changes prior to updating x230 configs to be FBWhiptail compatible in this [branch](https://github.com/tlaurion/heads/tree/x220_tests). 

Basically, resulting in the board config, coreboot and linux configs being like [this](https://github.com/tlaurion/heads/commit/60d03680dd1bfe0e425eb4aa39467e76dd93d70e).

The screen stays black.

Someone has compiled and flashed x220 to their laptop recently?

tlaurion, 2019-03-27:
@Akendo @jgrip @paulmenzel?

SebastianMcMillan, 2019-03-27:
I have, it works. Wait 50 seconds.

tlaurion, 2019-03-27:
Me delay you were talking about?

SebastianMcMillan, 2019-03-27:
@tlaurion

It's actually a TPM delay. And I think I've isolated it to the second measurement.

tlaurion, 2019-04-01:
@SebastianMcMillan after around 50 seconds, from a clean build from your branch, with and without @flammit patch, there is a constant high pitch with lights flashing.

_Sent from my Galaxy S3 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

SebastianMcMillan, 2019-04-01:
@tlaurion

That's a RAM error. Try using a rubber eraser on the RAM contacts, and try every stick until it works. Trust me, I had this issue and it was a pain in the rear.

tlaurion, 2019-04-01:
@SebastianMcMillan flashing back original rom works.

_Sent from my Galaxy S3 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

SebastianMcMillan, 2019-04-01:
@tlaurion

That's expected. Coreboot is picky. VERY VERY picky unlike stock.

techge, 2019-04-01:
I wanted to report this anyway, but did not find the time yet. I will try the waiting thing these days... I just wanted to confirm that I tried multiple times to build and flash on x220 and got a blank screen with heads while coreboot works fine for me.

SebastianMcMillan, 2019-04-01:
@techge wait 50 seconds.

SebastianMcMillan, 2019-04-09:
@tlaurion any update on your problems? I'm currently running a build from this repo's master on my X220, and although I haven't tested it on X220, my repo should, in theory, boot as well since the changes I made to X220 are the exact same changes I made to T420 and it boots fine. Both builds Only issue either of the builds have is the issue above, but they will work, given you get past the X220 being overly picky on RAM init with coreboot.

Currently setting up a new heads environment for my X220 to test my changes.

tlaurion, 2019-04-09:
@SebastianMcMillan I've sent you alle the files through u-root slack. I'm at the same stage.
Tried to debug with beaglebone, but can't get the serial output. Probably a problem of configuration, instructions are not clear. Got output from the x230 coreboot alone with configuration. Will try to dig deeper while removing NO_GFX into coreboot and bisect the changes between defconfig of x230 and heads coreboot config until I get debug and port those changes to the x220 to have something.

I'll swap memory from x230 to x200 and see. If joy happens, i'll update this post, else I'll jump back in this later on.

@Akendo @paulmenzel : Do you have a x220?

tlaurion, 2019-04-09:
My non booting and beeping was related to considered bad memory from coreboot. I now have the 50 seconds delay before Heads boot.

tlaurion, 2019-04-09:
@SebastianMcMillan : Once https://github.com/SebastianMcMillan/heads/pull/1 is merged in, you can close this issue since #541 is the real culprit here. I referred the cbmem PR there.

SebastianMcMillan, 2019-04-09:
It's been merged, you'll need to close as you're the owner of the issue.

closed 2019-04-10 by tlaurion