linuxboot/heads #545: new board porting - min SPI size?

shamen123, 2019-03-31:
Hi folks,
After the success with the X230, Id like to get this on my desktop. Im planning on porting to the ASUS P8H61-M PRO to heads in the future. I have worked on getting the TPM recognized upstream and this is now working under coreboot as of this commit

https://review.coreboot.org/cgit/coreboot.git/commit/?id=690d27faa7a415d16d63409f8d865b1417407a63

Obvs. i will need to backport the change into coreboot 4.9, once Heads gets to 4.9.

my question is - what is the minimum recommended SPI chip size. This mobo has only 4Mb and my memory from the x230 was that this was insufficient for much other than signing the rom using GPG

does heads have a minimum chip size requirements (or even min/max recommends)

tlaurion, 2019-04-01:
@shamen123 

Port to x200 is working on 8mb flash only, the 4mb flash being too small.
 X220 (not working right now) is for 8mb. 

G505s is 4Mb also and is not big enough.

I would say that until some space is gained, 8mb is required.

shamen123, 2019-04-05:
Thanks @tlaurion

closed 2019-04-05 by shamen123