linuxboot/heads #552: pinentry compilation failure

heads105, 2019-04-26:
The only package that doesn't build on Fedora 30 beta is pinentry. If I ./configure in the pinentry using the host configuration, then it builds fine, but it doesn't build under the heads config.

First error is:
pinentry-gtk-2.c:28:28: fatal error: gdk/gdkkeysyms.h: No such file or directory
Followed by a whole set of similar issues once I solved that by hard-coding the gdk location  which natively would be provided by pkg-config --cflags gdk-2.0, I get other similar missing include files.

Should there be a gdk-2.0.pc file or similar in heads/install/lib/pkgconfig/ too?

heads105, 2019-04-26:
Perhaps this is a sufficient fix for modules/pinentry:
 	--disable-pinentry-qt \
+	--disable-pinentry-gtk2 \
+	--disable-pinentry-gnome3 \
 	--disable-pinentry-fltk \

tlaurion, 2019-04-26:
@heads105 : if that works for you, those two are not necessary for pinentry to function under heads and should be removed, as all non required functions to reduce space.

Confirm that it works please!

heads105, 2019-04-26:
@tlaurion Yes, I did make that change and it compiled with no errors. I just flashed the rom on my x230 ok, though haven't done any extensive testing yet.

tlaurion, 2019-04-27:
PR in exchange of credit?:)

heads105, 2019-04-27:
@tlaurion The credit is all yours. I'm kicking myself for not having found heads before. Please just add the two lines if it doesn't negatively affect anyone else which I don't think it should do as ./configure --disable-foo doesn't give any error so it should work even if gtk2 and/or gnome3 aren't even options in pinentry.

tlaurion, 2019-04-27:
@heads105 : nope. Credit goes to @osresearch to have started this project, and then to all Heads contributors!

Welcome aboard! Thanks for the fix!

PR #554 (2019-04-27): Fedora 30 fix for pinentry: remove gtk and gnome3 support. 
TODO: remove all unneeded config options for ALL modules

Fixes #552
closed 2019-04-27 by tlaurion