linuxboot/heads #553: kexec-select-boot not compatible with Fedora

heads105, 2019-04-27:
I just flashed heads on an x230, wiped the partition table of the SSD with a few MB of /dev/zero and installed FC30 beta. When I come to boot I end up with kexec-select-boot calling kexec-parse-boot on the only .cfg file in /boot which is /boot/grub2/grub.cfg that doesn't reference a kernel directly therefore I end up with "No boot options". The kernel and options are in /boot/grub2/grubenv, e.g.

```
# GRUB Environment Block
saved_entry=c565bd7147f94e2186c659605cb6dd8e-5.0.9-301.fc30.x86_64
menu_auto_hide=1
boot_success=1
kernelopts=root=/dev/mapper/fedora-root ro resume=/dev/mapper/fedora-swap rd.lvm.lv=fedora/root rd.luks.uuid=luks-a1559d71-9be2-4501-ab32-67b7059e00c3 rd.lvm.lv=fedora/swap rhgb quiet 
############################################################################################################################################################################################################################################################
```
To be able to boot I created a temporary script that I guess does roughly what kexec-parse-boot does:
```
#!/bin/sh
kexec -l --append="root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.luks.uuid=luks-a1559d71-9be2-4501-ab32-67b7059e00c3 rd.lvm.lv=fedora/swap" --initrd=/boot/initramfs-5.0.9-301.fc30.x86_64.img /boot/vmlinuz-5.0.9-301.fc30.x86_64
kexec -e
```
Though obviously that's insecure.

Any suggestions on a good way to deal with this? I can think of two options:
Create a script that I run after updating the kernel which creates a dummy .cfg file in /boot that kexec-parse-boot can parse.
Fix kexec-parse-boot to cope with Fedora.

tlaurion, 2019-04-27:
@heads105 can you copy grub2.cfg here for reference?

heads105, 2019-04-27:
Actually, I didn't analyse that correctly. It seems the grub OS options are in /boot/loader/entries, e.g.
```
# cat boot/loader/entries/c565bd7147f94e2186c659605cb6dd8e-5.0.0-300.fc30.x86_64.conf
title Fedora (5.0.0-300.fc30.x86_64) 30 (Thirty)
version 5.0.0-300.fc30.x86_64
linux /vmlinuz-5.0.0-300.fc30.x86_64
initrd /initramfs-5.0.0-300.fc30.x86_64.img
options $kernelopts
grub_users $grub_users
grub_arg --unrestricted
grub_class kernel
```
I can work on a patch to handle this if nobody else has come across it yet.
Attached /boot tar with relevant files.

[boot.tar.gz](https://github.com/osresearch/heads/files/3124200/boot.tar.gz)

heads105, 2019-04-28:
Here's a patch for comment. I'm not sure this is the right way to go about it though. Perhaps it might be better to be able to have local scripts that override default behaviour instead?
This patch makes a small change to kexec-select-boot: if $bootdir/loader/entries/ exists then it assumes this is a BLS system and calls a new file kexec-parse-bls (new) instead of kexec-parse-boot.
kexec-parse-bls could be rolled into kexec-parse-boot but it doesn't really fit the state machine model when working on multiple files, so I made it a separate script.

This works on my FC30 system and shouldn't have a problem on any other system that either has an empty $bootdir/loader/entries directory or no directory at all.

[kexec.patch.gz](https://github.com/osresearch/heads/files/3124812/kexec.patch.gz)

ps. don't suppose there's an IRC channel for heads or some other chat system ppl use?

tlaurion, 2019-04-29:
@heads105 
> ps. don't suppose there's an IRC channel for heads or some other chat system ppl use?
Yes: https://u-root.slack.com/

_Sent from my Galaxy S3 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

tlaurion, 2019-04-29:
@heads105 
It would be really awesome if you could clone the heads repo, do your own branch ( git checkout -b yourbranch ) and push the changes into your GitHub, go onto osresearch/heads and propose a PR to be merged into master, saying that it fixes this issue.

That helps collaborative work and doesn't put additional weights on maintainers, which can review, comment and propose changes directly in your code. That is why GitHub actually exists.

Thanks

_Sent from my Galaxy S3 using [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/)_

heads105, 2019-04-29:
@tlaurion I understand, and will definitely do that shortly. Thanks for the slack pointer.

tlaurion, 2019-04-30:
@heads105 tested your patch functionally to validate #551 works with fedora30 on [unencrypted setup, commited](https://github.com/osresearch/heads/pull/551/commits/ab28f31ce04c4afbf58e39a9ceca6a91003c3c74) and it works.

heads105, 2019-04-30:
@tlaurion Not sure I understand. Is that because 551 might conflict with this?

tlaurion, 2019-04-30:
@heads105: sorry wasn't clear.
Master doesn't automatically propose luks devices or lvm.

I tested that the changes i made to that code was compatible with fedora 30 and your patch set. It works :)

heads105, 2019-04-30:
Understood. PR just sent

tlaurion, 2019-05-01:
fixed in #557

tlaurion, 2019-05-01:
@heads105 : had to revert it. see #558

tlaurion, 2019-05-01:
@heads105 
```
[user@dom0 ~]$ ls /boot/loader/entries/*
/boot/loader/entries/b404882d62964cfa8389ef6602f4fc0f-4.14.103-1.pvops.qubes.x86_64.conf
/boot/loader/entries/b404882d62964cfa8389ef6602f4fc0f-4.14.74-1.pvops.qubes.x86_64.conf
/boot/loader/entries/b404882d62964cfa8389ef6602f4fc0f-4.19.29-1.pvops.qubes.x86_64.conf
```
```
[user@dom0 ~]$ cat /boot/loader/entries/b404882d62964cfa8389ef6602f4fc0f-4.14.103-1.pvops.qubes.x86_64.conf
title      Qubes 4.0 (R4.0)
version    4.14.103-1.pvops.qubes.x86_64
machine-id b404882d62964cfa8389ef6602f4fc0f
options    placeholder root=/dev/mapper/qubes_dom0-root ro rd.luks.uuid=luks-caec4f1d-3370-4690-9958-4fa8247d2557 rd.luks.options=discard rd.lvm.lv=qubes_dom0/root rd.lvm.lv=qubes_dom0/swap i915.alpha_support=1 intel_iommu=on rhgb quiet rd.qubes.hide_all_usb intel_iommu=on intel_iommu=igfx_off
linux      /b404882d62964cfa8389ef6602f4fc0f/4.14.103-1.pvops.qubes.x86_64/linux
initrd     /b404882d62964cfa8389ef6602f4fc0f/4.14.103-1.pvops.qubes.x86_64/initrd

```

tlaurion, 2019-05-01:
@heads105 : if the two doesn't exist and one replace the other, then if grub.cfg is found then /boot/lader/entries shouldn't be parsed. Didn't had the time to implement and test the approach.

heads105, 2019-05-01:
@tlaurion Agh. ok, I'll install qubes and figure out the right way to do it.

closed 2019-05-05 by tlaurion

tlaurion, 2019-05-05:
fixed in #559