linuxboot/heads #566: TOTP-blocking PCR mismatch after failed Linux install

aionarap, 2019-05-20:
After:

1. Successfully setting an X230 up to use Qubes, including TOTP at boot, and
2. Attempting and failing to encrypt the Qubes LUKS disk encryption key, but successfully booting with the recovery key, then
2. Successfully booting from a Ubuntu installation USB stick, and
4. Attempting to overwrite the Qubes installation with a fresh, disk-encrypted Ubuntu installation that nevertheless failed after overwriting the /boot partition due to apparent corruption in the install stick and then
5. Re-computing the /boot file hashes and signing them, including resetting the TPM in order to create a new counter due to a forgotten TPM owner password, in an attempt to get the Ubuntu install working anyway, but having the boot fail in the end, and finally
6. Successfully rebooting (with no TPM/TOTP issues), and booting from a Qubes install USB stick, but not going through the installation process, and instead deciding to reboot

I experienced a TPM PCR mismatch on the subsequent boot, after step 6, where the TOTP code could not be unsealed and instead I was given a tamper warning and had to reset the TPM again, scan a new QR code, etc.

It seems to me that the TOTP-blocking PCR mismatch should have occurred after step 5 (resetting the TPM due to lost owner password) if anywhere. Can someone explain to me what happened here?

Did the Qubes installer in step 6, or some corrupt data on the Qubes installation USB stick, isomehow corrupt the HEADS installation, resulting in the PCR mismatch on a subsequent boot?

tlaurion, 2019-05-20:
4 wiped your /boot partition including TPM counter in /boot/kexec_rollback.txt

tlaurion, 2019-05-27:
@aionarap : Please reopen if error is reproducible.

closed 2019-05-27 by tlaurion

aionarap, 2019-06-06:
Haven't tried to reproduce it, but I'm struggling to understand how HEADS works here. 

- wiping the /boot partition results in the TPM failing to unseal?
- I thought the TPM counter was an internal value designed to prevent password-guessing attacks. What does the measured result in kexec_rollback have to do with anything?
- Is it documented anywhere where all the various bits of data are stored, and what is done as part of the TPM unseal process?

tlaurion, 2019-06-06:
@aionarap 
> Haven't tried to reproduce it, but I'm struggling to understand how HEADS works here.
>
>     * wiping the /boot partition results in the TPM failing to unseal?
>
>     * I thought the TPM counter was an [...]

[This is what is documented yet](https://github.com/osresearch/heads-wiki/blob/master/Keys.md#tpm-pcrs). 

The code, referring to rollback. I guess the encountered error happens in selecting default boot (verify_rollback_counter)?
```
user@build-x230-coreboot-48:~/heads$ grep -Rn rollback initrd
grep: initrd/etc/mtab: No such file or directory
initrd/bin/kexec-select-boot:13:valid_rollback="n"
initrd/bin/kexec-select-boot:27:		i) valid_hash="y"; valid_rollback="y" ;;
initrd/bin/kexec-select-boot:29:		f) force_boot="y"; valid_hash="y"; valid_rollback="y" ;;
initrd/bin/kexec-select-boot:67:verify_rollback_counter()
initrd/bin/kexec-select-boot:80:	valid_rollback="y"
initrd/bin/kexec-select-boot:293:	if [ "$CONFIG_BOOT_REQ_ROLLBACK" = "y" -a "$valid_rollback" = "n" ]; then
initrd/bin/kexec-select-boot:294:		die "!!! Missing required rollback counter state"
initrd/bin/kexec-select-boot:331:	TMP_ROLLBACK_FILE="/tmp/kexec/kexec_rollback.txt"
initrd/bin/kexec-select-boot:370:			# in the case of iso boot with a rollback file, do not assume valid
initrd/bin/kexec-select-boot:371:			valid_rollback="n"
initrd/bin/kexec-select-boot:373:			verify_rollback_counter
initrd/bin/gui-init:317:      rm -f /boot/kexec_rollback.txt
initrd/bin/gui-init:320:      check_tpm_counter /boot/kexec_rollback.txt \
initrd/bin/gui-init:327:      sha256sum /tmp/counter-$counter > /boot/kexec_rollback.txt \
initrd/bin/gui-init:328:      || die "Unable to create rollback file"
initrd/bin/kexec-sign-config:7:rollback="n"
initrd/bin/kexec-sign-config:12:		c) counter="$OPTARG"; rollback="y" ;;
initrd/bin/kexec-sign-config:13:		u) update_counter="y"; rollback="y" ;;
initrd/bin/kexec-sign-config:25:if [ "$rollback" = "y" ]; then
initrd/bin/kexec-sign-config:26:	rollback_file="$paramsdir/kexec_rollback.txt"
initrd/bin/kexec-sign-config:34:		check_tpm_counter $rollback_file \
initrd/bin/kexec-sign-config:42:	sha256sum /tmp/counter-$counter > $rollback_file \
initrd/bin/kexec-sign-config:43:	|| die "$paramsdir: Unable to create rollback file"
```