linuxboot/heads #571: Maintain CI to have reproducible roms from different sources

tlaurion, 2019-05-22:
Reproducible builds CI based on Ubuntu docker are not working for a long while now:
https://circleci.com/gh/osresearch/heads
https://gitlab.com/nemanjan00/heads-ci/

Does anyone have a working one?

Fedora-29 based one does, but lacks hooks to build from master commits:
https://gitlab.com/tlaurion/heads

Any help appreciated in fixing any of those problems!

BlackMaria, 2019-05-22:
my $0.25. circleci probably has some restrictions or limits that we are hitting. I cant verify since I don't have access to the account.

1. Space:  it may be related to the amount of disk being used.
1. Threads: tons of threads are spawned when using `make -j 4`. I would suggest removing the -j 4 and see what happens.
   * IMHO we don't really need it to build multi-thread anyway.

NB I have built master in a loop for a weekend and never hit any deadlocks etc.

tlaurion, 2019-10-31:
@osresearch : that would be lovely if fixed. 
Meanwhile, my gitlabci ubuntu builder [works](https://gitlab.com/tlaurion/heads/tree/Ubuntu-PrivacyBeast_X230_QubesOS_Certified)

tlaurion, 2020-01-15:
An temporary fix for this was introduced by @osresearch [here](https://github.com/osresearch/heads/pull/653#pullrequestreview-341233982), using a specific docker image to have bootsrapping tools encompassed in a docker image.

While non-ideal, this works.

tlaurion, 2020-01-17:
Good news, builds are now reproducible again since https://github.com/osresearch/heads/commit/6b485ed89f533f7c264543ac333c36ed0d19d715

Any people experienced enough to propose CircleCI and GitlabCI configurations to be in tree so that CircleCI and GitlabCI could automatically build for differnt hosts and boards upon passed environment variables?  ([CircleCI](https://circleci.com/docs/2.0/env-vars/#using-bash_env-to-set-environment-variables))

- docker image (ex: ubuntu:18.04, defaulting to it)
  - depending on host image type (ex: ubuntu) install proper dependencies (apt, dnf, others...)
- board configuration (ex: x230, defaulting to it)

The community could then subscribe to Heads project over CircleCI and automatically produce default board artifact (x230, qemu) for default host environement (ubuntu), append board rom sha256sum on osresearch's github release page(if built by osresearch), while other builders CIs could infirm/confirm reproducibility across other docker images for a same commit id in some way?

tlaurion, 2020-02-18:
@osresearch : CI builds are failing again from CircleCI, since they changed their log retention policy, I guess. If you come up with a trick to only output failing logs from build/logs/* (only logs modified in the last 5 seconds shown on screen?)

Gusher123, 2020-03-05:
> An temporary fix for this was introduced by @osresearch [here](https://github.com/osresearch/heads/pull/653#pullrequestreview-341233982), using a specific docker image to have bootsrapping tools encompassed to make builds faster, while not permitting easy audit boostrapping dependencies upgrades that could break reproducibility.
> 
> While non-ideal, this works.

@tlaurion @osresearch This temporary fix seems to have broken the CI builds from [Gitlab](https://gitlab.com/tlaurion/heads-ci/-/jobs/404900477) since the januari 15th commit [8e4b109](https://github.com/osresearch/heads/commit/8e4b10922bcfa70eb1891e8933d5e4cb93191c00) (musl-cross-make) to Heads.

tlaurion, 2020-03-06:
@gusher123: https://app.circleci.com/jobs/github/osresearch/heads/

Gusher123, 2020-03-07:
Thanks. That url did not work for me, but https://app.circleci.com/pipelines/github/osresearch/heads did

tlaurion, 2020-03-08:
We still have reproducibility issues on commit https://github.com/osresearch/heads/commit/ad84c38aed84f71522910cf63ad40c3123f04b2e on of those components:

coreboot image differs:
- 1f8405943fdaf6efea02cc07a7cd2f35020baab67d18712035586f5e50cbdbc3 ([CircleCI](https://app.circleci.com/jobs/github/tlaurion/heads/30/parallel-runs/0/steps/0-108))
- ef52c4eef87efa866e960c1e8f4531117a2d652a59245efdb8a3369777c26770 ([GitlabCI](https://gitlab.com/tlaurion/heads/-/jobs/463646725))

Because of:
- bzImage : 
  - 795b307945343c5c7eb0b55a9285bb73f96832d295da9be5e11887db804ebe02 ([CircleCI](https://app.circleci.com/jobs/github/tlaurion/heads/30/parallel-runs/0/steps/0-108))
  - 69961890c7ee425855f8620562b5627584f7b2c5f52132291836445087cc90ef ([GitlabCI](https://gitlab.com/tlaurion/heads/-/jobs/463646725))
Rest is reproducible. @osresearch ?

Other components non-reproducible:
- librem-hotp-verification (To be addressed in #640 since not part of x230 board )

tlaurion, 2020-03-17:
@osresearch
Another issue with the buildsystem happens when [make versions mismatches](https://github.com/osresearch/heads/blob/master/Makefile#L580) on docker CI and cache is already being extracted.

Result is that make is built again, trying to reapply patches which [fails](https://app.circleci.com/pipelines/github/tlaurion/heads/51/workflows/b5239c8b-3395-4f96-a2be-c93c25736be3/jobs/58).

tlaurion, 2020-05-24:
libremkey-hotp-verification issue was fixed under #722

tlaurion, 2020-05-24:
Each commit now produces images on:
[GitlabCI(Debian)](https://app.circleci.com/pipelines/github/osresearch/heads) 
GitlabCI(Fedora-30) (Attaching my instance for automatic builds of master)

tlaurion, 2020-05-24:
@osresearch @BlackMaria @MrChromebox @flammit :

First step accomplished:
- Each commit now triggers two CIs. For [example](https://github.com/osresearch/heads/commit/488d6529e463a9732f653f1d6bb89080f4768237): Click on the green check, and it will show you CircleCI and GitlabCI. Clicking those links shows that vmLinux is now reproducible:

[
795b307945343c5c7eb0b55a9285bb73f96832d295da9be5e11887db804ebe02  /root/project/build/x230-hotp-verification/bzImage](https://app.circleci.com/pipelines/github/osresearch/heads/162/workflows/2d04cd6b-4d3e-457f-97f5-47a3f188e3a4/jobs/170/parallel-runs/0/steps/0-114)
[
69961890c7ee425855f8620562b5627584f7b2c5f52132291836445087cc90ef  /builds/tlaurion/heads-osresearch/build/x230-hotp-verification/bzImage](https://gitlab.com/tlaurion/heads-osresearch/-/jobs/565440344)

While the following are the same:
a2fe6993f933f9a99a8d2b999f400ae57f834bc6443679c45db879c1dcdf3222  /root/project/build/x230-hotp-verification/modules.cpio
46b73363e70cc5e3a398623f6a7e8d4e5fd5fc66122223570f079ac0dab658ec  /builds/tlaurion/heads-osresearch/build/x230-hotp-verification/tools.cpio
980a1e671df217b2fa68896dfc6920695a65ed02c4f7e8135c5f093f50731ed7  /builds/tlaurion/heads-osresearch/build/x230-hotp-verification/heads.cpio

tlaurion, 2020-05-24:
@BlackMaria : How would we make those build systems report their hashes to github for comparison?

tlaurion, 2020-06-09:
@szszszsz :
Update
[osresearch](https://circleci.com/gh/osresearch/heads) and [tlaurion](https://app.circleci.com/pipelines/github/tlaurion/heads) CircleCIs now build on each commit most of the boards actively used by direct community members.

The above points are still missing, while CircleCI still misses a proper checksum statement to validate individual points:
- crossgcc module should be checksumed and build cache remade only when changed.
- all other modules should be checksumed and build cache remade only when changed.
- As of right now, the cache is static (undesired in the long run since not saved when changed)

I'm still struggling with GitlabCI with [space missing](https://gitlab.com/tlaurion/heads/-/jobs/586512967) and didn't found a proper way to fix it yet.

tlaurion, 2020-09-03:
Becoming more urgent to produce ROM images for coreboot 4.12, as seen in #824 and consequentially https://github.com/osresearch/heads/pull/825#issue-479017257

tlaurion, 2020-10-04:
> I'm still struggling with GitlabCI with [space missing](https://gitlab.com/tlaurion/heads/-/jobs/586512967) and didn't found a proper way to fix it yet.

Fixed with https://github.com/tlaurion/heads/tree/GitlabCI_Fix
Now trying to use properly GitlabCI caching system.

tlaurion, 2020-10-04:
Next steps:

- [ ] Implement shellcheck on new scripts (maybe rename all shell scripts to .sh to ease finding them and testing them)

- [ ] [Respect pipeline logic](https://docs.gitlab.com/ee/ci/pipelines/) in both CircleCI and GitlabCI, permitting ROM deployment (release)

- [ ] Have heads versioning tags so [fwupd (LVFS) can be implemented with releasing on github](https://github.com/osresearch/heads/pull/834#issuecomment-699041518). #839 is a good start

Thrilleratplay, 2020-10-09:
@tlaurion 

from #851 

> goal here is to be able to build all boards on the same code base and different build system, auditable with packages being downloaded and caching fastening the process from one build to the next.

That I understood.  The mention of coreboot-4.12 is why I made the suggestion.  Apparently there are quirks with [automake and aclocal](https://github.com/coreboot/coreboot/commit/b0fe89d31bf2b27c77d46c59b10db333328b260e#diff-9e3b6c4d4b2cb5f5064b74c991d084fd) in addition to dependency changes.  It may have been easier to start from there instead of through trial and error with the base Debian image.

> Question here would be: can that docker image be used with current build system and what would be the advantages of it? That would remove main OS as main obstacle for origin build tools. But we cannot say that the project would enforce reproducible builds. We could then say that Heads uses coreboot toolset, and modify Makefile, coreboot to not rely on musl-cross-make, but rely solely on that docket image and have instructions for user to deploy docker on all OSes.

I am still not sure to what extent the ultimate goal is, if it is for anyone to download the Heads source and compile it expecting the result to be the same hash as the CI builds, I do not think this is possible.  Reading through [What's in a build environment? on reproducible-builds.org](https://reproducible-builds.org/docs/perimeter/), `Make board=xxx` would need to ensure the session's variables name of the user running the build, locale, timezone, and other environmental variables are the same without interfering without altering the system wide variables as this could break other running processes on the system.  The user name is the most difficult for a user without full admin privileges.  Playing around with GUIX, I am not sure this would fulfill the requirement either as it is requires the user to install a new package manger with daemon, again requiring admin privileges, to Linux distros that may nor may not support it.  Currently, it does not build using the Arch AUR package as one of the dependencies has a build error.  Even if GUIX was supported on every Linux distro, it would not still exclude Windows and MacOS users from developing.

Although the reproducible-build docs recommend using VMs and containers, I am not saying using a docker container is the solution and am still need to experiment with CircleCI and GitlabCI.

Thrilleratplay, 2020-10-09:
@tlaurion  In addition to `shellcheck`, a further TODO may be bash script unit testing.  I have been using [shellspec](https://github.com/shellspec/shellspec) in [thrilleratplay/coreboot-builder-scripts](https://github.com/Thrilleratplay/coreboot-builder-scripts/tree/master/spec). It isn't perfect but it is a lot better than the other bash testing frameworks I have tried.

Thrilleratplay, 2020-10-19:
@tlaurion Just curious, has [musl.cc](https://musl.cc) been considered an alternative to compiling `musl-cross-make`.  I can understand where it could be rejected as there is some [deviation from musl-cross-make](https://github.com/richfelker/musl-cross-make/issues/99) but I could not find where it was mentioned in any of the Github tickets.

tlaurion, 2020-10-20:
The current considered alternatives, by more knowledgeable people then I am are:
1. coreboot-sdk (with issues on docker image reproducibility (quote needed) : Goal here would be to replace all documentation we have now so that docker could be used on any OS to build heads in a reproducible way.  Sub-instructions could show how to reproduce the docker image itself and be reused, which would be a big plus so Heads don't have to maintain patches to undo upstream work (python2-> python3, gawk, make etc) as all upstream OSes upgrade their build toolstack which have proved to be problematic to even build a same Heads commit years later. If that docker image, specified in Heads for a specific Heads commit could reproduce the same rom output 5 years from now, I thing the goal would be met. That would imply modifying main Makefile and coreboot patches for musl-cross so that the coreboot-sdk musl-cross be used directly, instead of building musl-cross, economizing precious time both on CI builds and users building only for a specific board. One question that rises from this approach, which is untested, is what would happen if the coreboot-sdk for 4.12 is used to build 4.81 roms. No idea.
1. guix-buildstack : This is another approach @daym is playing with. The goal here would be similar to docker above, while guix-toolstack would be an additional layer added on top of any (Linux) based OS layer/docker, permitting in build instructions (and CI) to use a specific guix-toolstack revision in the goal of building the same ROM output. As opposed to the docker image above, we could easily modify the guix-toolstack to fit our needs (let it be LTO-compliant compiler (where LLVM is not interesting yet, from what I've read, but I might be misinformed).
1. Current musl-cross-make approach: musl-cross-make is a "portable" alternative to musl-cross, meaning that what is found after building musl-cross-make and calling an install, which is under heads, under ./crossgcc, is supposed to be portable. For reproducible builds, it means less chances of having hardcoded paths since buildstack is in relative paths. The current approach permits having multiple coreboot version being build, where coreboot is instructed to build, per coreboot configs, to ANY_TOOLCHAIN. That may or may not be the viable solution as opposed to coreboot-sdk or guix-toolchain.
1. musl.cc: I have not invatigated nor understand the clear differences from the quick diagonal reading of 10 minutes I have made on the subject.

The main objective here would be to, IMHO, to distance ourselves the most possible from the host toolchain. For example, even if we use musl-cross-make right now and the abstraction is is supposed to imply, we are still facing kernel reproducibility issues (#734) opened for more then a year after having moved away of musl-cross for musl-cross-make. I am not knowledgeable enough to understand why those changes are coming from, unfortunately. But my intuition is that 1 and 2 are the most desired approaches. 1 because readily available while limiting ourselves for LTO and other optimizations we might want, which 2 would resolve more easily. 1 would be applicable on top of any OS, where 2 would require docker + guix-toolstack.

Thrilleratplay, 2020-10-20:
@tlaurion Thank you for the detailed answer, it was more than I expected.  

From my local testing, I can answer a few questions.  The `coreboot-sdk` docker versions from at least the two years are not able to compile coreboot 4.8, only coreboot 4.12.  Compiling heads on `coreboot-sdk` for x230 requires two additional packages;  `cpio` and `texinfo`.  

I was trying to add `musl-cross-make` already compiled but became lost trying to add an exception to the module to check the `make install` locations.  As the `musl-cross-make` module build directory is 7.5G, adding that to the docker image does not seem reasonable.  This is why I asked about musl.cc.

@daym showed me how to pull his `guix-on-heads` image and planned on trying it later today.  I think it will require docker running `--privlieged`.  This is a potential issue for some end users but I am more concerned with trying to run a privileged image on CircleCI.

tlaurion, 2020-10-25:
Hurray, CircleCI is able to sustain cache for multiple linux and coreboot versions:
```
Found a cache from build 588 at heads-modules-and-patches-zGxbPlRSx6OtBxTpidFp2N538LPES9DQ4z+9sH4XnVA=<no value>
Size: 7.6 GiB
Cached paths:
  * /root/project/packages
  * /root/project/crossgcc
  * /root/project/build

Downloading cache archive...
```
From https://github.com/osresearch/heads/pull/867 built https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/9d19e4e0-a6ab-4d68-bfa1-f47d61b9b760/jobs/588 and reused https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/b191e704-769e-4742-a3af-12fd0a4b7cfe/jobs/589 for the same modules and patches applied.

Now if someone has ideas on having the same musl-cross-make cache being downloaded on different docker images to test reproducibility and push to github releases on tags, that would be the next step in the goal of pushing Heads releases to LVFS!

tlaurion, 2020-10-25:
> I was trying to add `musl-cross-make` already compiled but became lost trying to add an exception to the module to check the `make install` locations. As the `musl-cross-make` module build directory [...]

@Thrilleratplay we see in past CircleCI cache that building a cache, compressing and uploading cache takes a while ([16m45](https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/9d19e4e0-a6ab-4d68-bfa1-f47d61b9b760/jobs/588/parallel-runs/0/steps/0-160)), while downloading it is really fast ([4m21s](https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/b191e704-769e-4742-a3af-12fd0a4b7cfe/jobs/589/parallel-runs/0/steps/0-106))

Building a docker image could be done the same way a cahce is built, when buildstack changes (musl-cross-make or guix-toolstack dependencies) just like in cache.

When rebuilding the same modules/patches, we can see [here](https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/b191e704-769e-4742-a3af-12fd0a4b7cfe/jobs/589) that the build time for each board is reduces at maximum, where it could still be optimized since not everything is considered in the cache right now. For exemple, make and gawk are not cached inside of the musl-cross-make cache, which should probably be added. We see in the first board in line that is being built a build time of [9m22](https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/b191e704-769e-4742-a3af-12fd0a4b7cfe/jobs/589) where subsequent boards builds are just plainly reusing cache ([41s](https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/b191e704-769e-4742-a3af-12fd0a4b7cfe/jobs/589/parallel-runs/0/steps/0-111), [1m23](https://app.circleci.com/pipelines/github/tlaurion/heads/541/workflows/b191e704-769e-4742-a3af-12fd0a4b7cfe/jobs/589/parallel-runs/0/steps/0-115) etc

My belief here is that it might be useless to try to create docker images, where caches could definitely be used for that purposes. If we go the path of guix-toolstack, that would be the same principles as understood, where it could be built from scratch (to test reproducibility where guaranteed) or downloaded as a layer prior of simply building boards, with cross compiler settings set in Makefile and other modules being adjusted to use that toolchain.

@daym @Thrilleratplay @MrChromebox @osresearch: any advice welcome prior of switching build toolstack or going one direction more then the other. Please read https://github.com/osresearch/heads/issues/571#issuecomment-712948834

Thrilleratplay, 2020-11-02:
@tlaurion FYI, when trying to get #872 to build, I found that CircleCI config version 2.1 and version 2 have a lot of incompatibilities but could not find a good document outlining how to migrate between them.  As one of these how `store-artifacts` behaves and is called and that you spent so much time with this, I decided to use shellcheck in Debian.  Be aware of this before updating the config version.

tlaurion, 2020-11-03:
@Thrilleratplay : haaaaaa. Will have to check, I had no idea.

tlaurion, 2020-11-03:
> @tlaurion FYI, when trying to get #872 to build, I found that CircleCI config version 2.1 and version 2 have a lot of incompatibilities but could not find a good document outlining how to migrate [...]

@Thrilleratplay what incompatibilities have you found?

Thrilleratplay, 2020-11-03:
@tlaurion I stopped at `store-articles` but [here is a last build error](https://app.circleci.com/pipelines/github/Thrilleratplay/heads/8/workflows/a773418a-f56a-4c9a-a4f2-69e5b60605e3/jobs/8) I had with CircleCI `version 2.1`.  I think in version 2.1 is was renamed `store_articles`

tlaurion, 2020-11-03:
Seems like local cli validator might be of help here, while 2.1 is in test right now. Interesting what can be done with orbs.
Not a priority though :)

tlaurion, 2020-11-10:
@daym your input on improvements and https://github.com/osresearch/heads/issues/571#issuecomment-716082254 would be more then welcome!

tlaurion, 2020-11-10:
> @daym showed me how to pull his `guix-on-heads` image and planned on trying it later today. I think it will require docker running `--privlieged`. This is a potential issue for some end users but I am more concerned with trying to run a privileged image on CircleCI.

@Thrilleratplay @daym This is not the case anymore? How would we replace musl-cross-make here to depend on guix-toolstack from docker? This is exciting.

Thrilleratplay, 2020-12-28:
> @Thrilleratplay @daym This is not the case anymore? How would we replace musl-cross-make here to depend on guix-toolstack from docker? This is exciting.

@tlaurion Sorry, at the time I was not 100% sure of this and did not want to provide incorrect information.  After becoming more comfortable with Guix in Docker, I can confirm that the `--privilieged` flag is not required when running Guix package manager so long as it is run with the `--disable-chroot` flag in the container.

I will post my current plan in #927.

tlaurion, 2024-05-10:
I think this is fixed by #1660

closed 2024-05-10 by tlaurion