linuxboot/heads #574: TOTP  refreshing

merge, 2019-05-24:
Right now it seems like the TOTP is only refreshed when I move into "Advanced Settings" and back, in order to refresh the main "Heads Boot Menu". We should either
* auto-refresh, if we can do so reliably, or
* add a simple "TOTP refresh" Main Menu entry that we can hit

or am I overlooking something?

merge, 2019-05-24:
Oh, There is a "refresh TOTP code" menu option in advanced settings somewhere. But that takes me back to the very first "Error: GPG keyring empty" Menu. I think that this option should move up the the Main Menu (there is enough room, only 3 entries right now) where the TOTP is actually shown, and __always__ result in showing that (the TOTP) again.

merge, 2019-05-24:
But I guess we want to keep treating the "no GPG" case as an error (and that's fine as soon as we have "auto-generation" to make the "Fix" for users trivially easy!). So only a "refresh" shortcut in the main boot menu is what I'd really want to have. IMO a refresh-button __next to__ the code is crucial for usable TOTP auth.

merge, 2019-05-24:
I totally overlooked that we already have auto-generation + flashing for the GPG key. most of my concers here can be ignored. We have everything except the refresh button next the TOTP code.

kylerankin, 2019-05-24:
I would be OK with a TOTP refresh button on the main menu, I suppose. I'd prefer we keep that main menu as minimal as possible though, for ease-of-use for non-technical people and have everything that doesn't fall into the essential default boot workflow to fall under Advanced Settings.

pahakalle, 2019-05-24:
Could a keyboard button for example R be used to refresh the TOTP code?
This would not crowd the main menu

tlaurion, 2019-05-25:
We could say "press escape to refresh TOTP"

kylerankin, 2019-05-26:
With whiptail you type a key and press enter to exit and process the keystroke. With fbwhiptail you can only use the arrow keys to select an option and press enter. There's no mechanism to process keystrokes outside of it.

tlaurion, 2019-05-27:
> With whiptail you type a key and press enter to exit and process the keystroke. With fbwhiptail you can only use the arrow keys to select an option and press enter. There's no mechanism to process keystrokes outside of it.

As stated [here](https://github.com/osresearch/heads/pull/576#issuecomment-496014609), since we are in main menu and escape exit (ESc key) escapes, redo the loop and respawns the main menu, escape does the TOTP refresh.

commit 186895d4 (2019-05-27): gui-init: move Refresh TOTP/HOTP to the Main Boot Menu

also, rename the current menu entry to being smaller and simpler.

Closes #574
closed 2019-05-27 by kylerankin