linuxboot/heads #583: Detecting USB dynamically breaks booting from msata or other drives

stayinside, 2019-06-15:
In testing the latest commit it seems the changes introduced in #548 [here](https://github.com/osresearch/heads/commit/152689d5d584dfa144b7a725dbd2727490871ef0) breaks some functionality. I was using mount-usb to boot from an internal msata drive on my x230 and that no longer works. The issue is that in line 17 of mount-usb
`stat -c %N /sys/block/sd* | grep usb |`
all other attached storage that isn't a usb is removed. I know there is an issue addressing dual boot already so perhaps this will be a won't fix but allowing non-usb drives has been useful to me. Thanks.

tlaurion, 2019-06-19:
Unfortunately, the device is not USB.
I created a script to boot from sdcard [here](https://github.com/osresearch/heads/pull/551/files#diff-893be823b299bd80d45c36112bbbf1e3). 

And corrected a bug in iso boot here, because of older method artifacts, [here](https://github.com/osresearch/heads/pull/551/commits/9a65931c5707d868a3a14da65e20017646636e8a). 

As a mitigation, you can enter recovery shell:
`mount /dev/sdb1 /media`
call `gui-init`

Since /media is not mounted by mount-usb or mount-sdcard if already mounted, this should work.

@kylerankin : What do we do with that?

kylerankin, 2019-06-19:
This is by design. The goal of that script is to detect and mount USB drives for booting, not detect all possible disks attached to the device. The default boot mechanism exists for booting hard drives on the device. Does that not work for booting internal msata devices? Presumably you could change the Heads configuration to treat that as your boot device and it would scan it.

tlaurion, 2019-06-19:
@kylerankin But then no dual boot with multiple disks.

@stayinside :
A way to do this would be to list all standard block devices in a "mount-secondary.sh" script, and implement the same as [here](https://github.com/osresearch/heads/pull/551/files#diff-9a66e05586b37d4779ee1eb88802e759R259). in gui-init.

This is hackish, because else all scripts need to be duplicated to boot iso and stuff, which depends on usb-init, usb-scan etc to make the magic happen.

tlaurion, 2019-06-20:
@stayinside :

Following precedent post, and if you still have [CONFIG_USB_BOOT_DEV](https://github.com/osresearch/heads/blob/master/boards/x230/x230.config#L34) in board config, pointing to /dev/sdXY

It means that you could simply do from heads recovery:
`mount /media` ( or mount /dev/sdb1 /media if[ /etc/fstab is not populated](https://github.com/osresearch/heads/blob/98598e7f2f27eeb831a66eeb3f50d598c9ff50d0/initrd/init#L94)) 
`usb-init`

tlaurion, 2024-04-16:
That's now media-scan specifying partition but from recovery shell.

closed 2024-04-16 by tlaurion