linuxboot/heads #585: Build fails during libgcrypt on Arch Linux

flawedworld, 2019-06-16:
Attempting to compile heads results in libgcrypt failing to build, it is present and have tested on Debian with same steps and is an issue with the way libgcrypt is installed on Arch Linux. 

Terminal output: https://pastebin.com/9GC168wB

tlaurion, 2019-06-20:
Please do:
```
make BOARD=t430 real.clean
rm -rf build ./build/*
make BOARD=t430 V=1
```
And post back build logs. Never build on arch.

flawedworld, 2019-06-20:
@tlaurion 

Done so exactly as you asked, here's the logs. From newest to oldest.

gpg2.log https://pastebin.com/GiS0U7pj

gpg2.configure.log https://pastebin.com/WdrbPzLm

libgcrypt.log https://pastebin.com/biycZafz

libgcrypt.configure.log https://pastebin.com/bjSRNjz1

tlaurion, 2019-06-24:
Gpg2 logs shows missing glibc dependencies from the errors .

Followed building docs on heads-wiki?
Have you installed arch equivalents of fedora/debian dependencies?

tlaurion, 2019-06-24:
The error is

/home/user/Project/heads_bios/seb/heads/crossgcc/x86_64-linux-musl/lib/gcc/x86_64-linux-musl/5.3.0/../../../../x86_64-linux-musl/bin/ld: warning: libc.so.6, needed by /home/user/Project/heads_bios/seb/heads/install/lib/libgcrypt.so, not found (try using -rpath or -rpath-link)

/home/user/Project/heads_bios/seb/heads/install/lib/libgcrypt.so: undefined reference to `free@GLIBC_2.2.5'

tlaurion, 2019-10-08:
@flawedworld any update?

flawedworld, 2019-10-08:
@tlaurion  So first off, sorry for the lateness, Kinda forgot about this since I'm waiting for this to be brought up to a newer coreboot release. Anyways.

base-devel, gcc-ada, cpio, cmake, iasl, nasm, ccache, perl-digest-sha1 were installed on a fresh Arch install. uuid was sourced from the AUR. All other packages mentioned in the docs were present in either base or base-devel. I no longer run Arch daily FYI.

I was able to compile x230-flash perfectly. Compiling x230 gives me an error during libgpg-error.  https://pastebin.com/u0F6KXwm    https://pastebin.com/i38WXwwr

System was freshly installed, with literally just the above packages. All fully updated.

A potential solution would be the option to compile heads within a docker image, which I might play around with in my downtime. If you need any other logs please let me know.

flawedworld, 2020-05-15:
Closed, see issue #668

closed 2020-05-15 by flawedworld