linuxboot/heads #590: Quest to reduce firmware size

tlaurion, 2019-06-28:
Hello guys.

There is tickets opened to:

-  remove networking ( #564 ), with corresponding PR attempt to remove networking with cleanup ( #572 ) while cryptsetup seemed to depend on CONFIG_NET in the past ( #79 ), where putting `CONFIG_LINUX_E1000E=n` in board config liberates 295688 bytes.
- reduce libslang.so.2 footprint ( #517 ) (
- reduce kernel crypto footprint (what ticket?)

The priorities should be, see below for binaries/libraries footprints:

- [ ] Reduce libgcrypt.so.20 footprint (1145176 uncompressed bytes)
- [ ] Reduce lvm footprint (1599032 uncompressed bytes)
- [ ] Reduce gpg footprint (869568 uncompressed bytes) 
- [ ] libcairo.so.2 footprint (851968 uncompressed bytes)
- [ ] libpixman-1.so.0 footprint (667048 uncompressed bytes)

To help in finding consumed space, here is what I do.
```
make BOARD=x230
cd build/x230/
xz --decompress initrd.cpio.xz
cpio --extract < initrd.cpio
cpio --extract < heads.cpio
cpio --extract < modules.cpio
```
`find . -type f -ls | sort -r -n -k7`

>   789884  12288 -rw-r--r--   1  user     user     12582912 Jun 28 13:00 ./coreboot.rom
>    789885  11912 -rw-r--r--   1  user     user     12212224 Jun 28 13:00 ./initrd.cpio
>    789821  10648 -rw-r--r--   1  user     user     10903552 Jun 28 13:00 ./tools.cpio
>    789880   2868 -rw-r--r--   1  user     user      2936832 Jun 28 12:59 ./bzImage
>    789986   1564 -rwx------   1  user     user      1599032 Jun 28 13:01 ./bin/lvm
>    790106   1120 -rwx------   1  user     user      1145176 Jun 28 13:01 ./lib/libgcrypt.so.20
>    789881    968 -rw-r--r--   1  user     user       989696 Jun 28 12:59 ./modules.cpio
>    789943    852 -rwx------   1  user     user       869568 Jun 28 13:01 ./bin/gpg
>    790103    832 -rwx------   1  user     user       851968 Jun 28 13:01 ./lib/libcairo.so.2
>    790113    652 -rwx------   1  user     user       667048 Jun 28 13:01 ./lib/libpixman-1.so.0
>    790102    556 -rwx------   1  user     user       568680 Jun 28 13:01 ./lib/libc.so
>    789938    480 -rwx------   1  user     user       490640 Jun 28 13:01 ./bin/flashrom
>    789900    476 -rwx------   1  user     user       486024 Jun 28 13:01 ./bin/busybox
>    790018    384 -rwx------   1  user     user       391760 Jun 28 13:01 ./bin/scdaemon
>    789944    352 -rwx------   1  user     user       359960 Jun 28 13:01 ./bin/gpg-agent
>    790117    328 -rwx------   1  user     user       332504 Jun 28 13:01 ./lib/libtpm.so
>    790105    320 -rwx------   1  user     user       324784 Jun 28 13:01 ./lib/libdevmapper.so.1.02
>    789882    316 -rw-r--r--   1  user     user       321024 Jun 28 13:00 ./heads.cpio
>    790109    312 -rwx------   1  user     user       318464 Jun 28 13:01 ./lib/libmbedcrypto.so.0
>    790123    292 -rw-------   1  user     user       295688 Jun 28 13:01 ./lib/modules/e1000e.ko
>    790108    216 -rwx------   1  user     user       217192 Jun 28 13:01 ./lib/libksba.so.8
>    790114    200 -rwx------   1  user     user       201096 Jun 28 13:01 ./lib/libpng16.so.16
>    789925    180 -rwx------   1  user     user       180960 Jun 28 13:01 ./bin/dropbear
>    790037    172 -rwx------   1  user     user       172544 Jun 28 13:01 ./bin/ssh
>    790127    168 -rw-------   1  user     user       170512 Jun 28 13:01 ./lib/modules/mmc_core.ko
>    790131    160 -rw-------   1  user     user       160792 Jun 28 13:01 ./lib/modules/xhci-hcd.ko
>    789960    156 -rwx------   1  user     user       158952 Jun 28 13:01 ./bin/kexec
>    790104    156 -rwx------   1  user     user       155744 Jun 28 13:01 ./lib/libcryptsetup.so.4
>    789924    132 -rwx------   1  user     user       133216 Jun 28 13:01 ./bin/dmsetup
>    790130    132 -rw-------   1  user     user       132520 Jun 28 13:01 ./lib/modules/usb-storage.ko
>    790051    124 -rwx------   1  user     user       122992 Jun 28 13:01 ./bin/tpm
>    790107    120 -rwx------   1  user     user       120024 Jun 28 13:01 ./lib/libgpg-error.so.0
>    790097    108 -rw-------   1  user     user       108800 Jun 28 13:01 ./etc/wordlist_en_eff.txt
>    790121    108 -rwx------   1  user     user       107512 Jun 28 13:01 ./lib/libz.so.1
>    790119     96 -rwx------   1  user     user        98064 Jun 28 13:01 ./lib/libusb-1.0.so.0
>    789983     76 -rwx------   1  user     user        73920 Jun 28 13:01 ./bin/lspci
>    790101     72 -rwx------   1  user     user        73416 Jun 28 13:01 ./lib/libassuan.so.0
>    790124     68 -rw-------   1  user     user        66208 Jun 28 13:01 ./lib/modules/ehci-hcd.ko
>    789914     64 -rwx------   1  user     user        62680 Jun 28 13:01 ./bin/cryptsetup-reencrypt
>    789913     56 -rwx------   1  user     user        55760 Jun 28 13:01 ./bin/cryptsetup
>    790129     52 -rw-------   1  user     user        53136 Jun 28 13:01 ./lib/modules/sdhci.ko
>    790004     52 -rwx------   1  user     user        52520 Jun 28 13:01 ./bin/pinentry-tty
>    790126     52 -rw-------   1  user     user        49352 Jun 28 13:01 ./lib/modules/mmc_block.ko
[... 63 lines cut ...]
>    789905      4 -rwx------   1  user     user          799 Jun 28 13:01 ./bin/cbfs-init
>    790082      4 -rw-------   1  user     user          736 Jun 28 13:01 ./etc/config
>    789997      4 -rwx------   1  user     user          675 Jun 28 13:01 ./bin/network-init-recovery
>    790057      4 -rwx------   1  user     user          661 Jun 28 13:01 ./bin/uefi-init
>    790062      4 -rwx------   1  user     user          634 Jun 28 13:01 ./bin/unseal-totp
>    790074      4 -rwx------   1  user     user          574 Jun 28 13:01 ./bin/x230-flash.init
>    790012      4 -rwx------   1  user     user          366 Jun 28 13:01 ./bin/qubes-measure-luks
>    789939      4 -rwx------   1  user     user          360 Jun 28 13:01 ./bin/flashrom-kgpe-d16-openbmc.sh
>    790092      4 -rwx------   1  user     user          340 Jun 28 13:01 ./etc/keylime-init
>    790072      4 -rwx------   1  user     user          320 Jun 28 13:01 ./bin/wget-measure.sh
>    790013      4 -rwx------   1  user     user          258 Jun 28 13:01 ./bin/reboot
>    790065      4 -rwx------   1  user     user          220 Jun 28 13:01 ./bin/usb-init
>    790008      4 -rwx------   1  user     user          205 Jun 28 13:01 ./bin/poweroff
>    790088      4 -rw-------   1  user     user          197 Jun 28 13:01 ./etc/fstab
>    790093      4 -rw-------   1  user     user          174 Jun 28 13:01 ./etc/motd
>    789946      4 -rwx------   1  user     user          106 Jun 28 13:01 ./bin/gpgv
>    789888      4 -rw-------   1  user     user           73 Jun 28 13:01 ./.gnupg/gpg-agent.conf
>    789920      4 -rwx------   1  user     user           62 Jun 28 13:01 ./bin/diceware-eff.sh
>    790073      4 -rwx------   1  user     user           35 Jun 28 13:01 ./bin/whiptail
>    790096      4 -rw-------   1  user     user           27 Jun 28 13:01 ./etc/shells
>    790095      4 -rw-------   1  user     user           27 Jun 28 13:01 ./etc/passwd
>    790091      4 -rw-------   1  user     user           20 Jun 28 13:01 ./etc/hosts
>    790090      4 -rw-------   1  user     user           10 Jun 28 13:01 ./etc/group
>    789889      4 -rw-------   1  user     user           10 Jun 28 13:01 ./.gnupg/gpg.conf
>

SebastianMcMillan, 2019-07-05:
A few other ideas:

1. Clean up board-specific files from other devices. (Example is flashrom-kgpe-d16-openbmc.sh is present in X230 builds, x230-flash.init being in main X230 or on Chell Chromebook). This change will likely need to happen during the build process.

2. Move flash definitions from flash.sh to the board config files. As we support more boards, this will be critical, as all maintained boards are presently defined in flash.sh, and as this project supports more and more boards, flash.sh will get larger and larger.

These may not be as high priority as other changes, but ones to certainly look at to help reduce firmware size, especially as the project grows.

tlaurion, 2019-07-06:
@SebastianMcMillan : Those are good cleanup ideas, but unfortunately, will not impact the initrd.cpio, heads.cpio nor tools.cpio sizes, those text files being highly compressed. 

The problem lies in binaries, not really compressed under initrd.cpio.xz

Matthew-Bradley, 2019-11-01:
This article may yield some inspiration:
https://lwn.net/Articles/748198/

tlaurion, 2020-03-09:
Solution lies in #307

tlaurion, 2020-03-11:
@zaolin [pointed out](https://github.com/osresearch/heads/pull/568#issuecomment-597005125):
>Update: Getting rid of libgcrypt and replacement for gpg would be a good way to safe 2MB in total.
See, https://sequoia-pgp.org/ as alternative

tlaurion, 2020-03-11:
> > Update: Getting rid of libgcrypt and replacement for gpg would be a good way to safe 2MB in total.
> > See, https://sequoia-pgp.org/ as alternative

@zaolin : Unfortunately, there is [no smartcard support in sequoia-pgp](https://gitlab.com/sequoia-pgp/sequoia/-/issues/114), on which Heads relies on for verified /boot integrity.

zaolin, 2020-03-12:
ask them for support or find someone writing code for it.

tlaurion, 2020-05-03:
Here we go again, since 
- #710 would require newer versions of gpg toolchain, which won't fit in actual x230 board config.
- #709 non neutered ME board configs lack around 300k to fit VBOOT in.

To troubleshoot: 

`make BOARD=x230`
cd build/x230
xz -d initrd.cpio.xz ; for i in initrd.cpio modules.cpio tools.cpio heads.cpio; do cpio -i < $i; done
find . -type f -ls | sort -r -n -k7 |grep -v cpio
   ```
 28192   2956 -rw-r--r--   1 user     user      3023312 May  3 12:25 ./bzImage
    61393   1592 -rwx------   1 user     user      1627856 May  3 12:45 ./bin/lvm
    61507   1132 -rwx------   1 user     user      1156424 May  3 12:45 ./lib/libgcrypt.so.20
    61352    892 -rwx------   1 user     user       911264 May  3 12:45 ./bin/gpg
    61504    740 -rwx------   1 user     user       757232 May  3 12:45 ./lib/libcairo.so.2
    61515    652 -rwx------   1 user     user       666216 May  3 12:45 ./lib/libpixman-1.so.0
    61502    584 -rwx------   1 user     user       596544 May  3 12:45 ./lib/libc.so
    61347    556 -rwx------   1 user     user       568264 May  3 12:45 ./bin/flashrom
    61313    472 -rwx------   1 user     user       483160 May  3 12:45 ./bin/busybox
    61424    400 -rwx------   1 user     user       407784 May  3 12:45 ./bin/scdaemon
    61353    368 -rwx------   1 user     user       376024 May  3 12:45 ./bin/gpg-agent
    61519    332 -rwx------   1 user     user       339304 May  3 12:45 ./lib/libtpm.so
    61506    328 -rwx------   1 user     user       333240 May  3 12:45 ./lib/libdevmapper.so.1.02
    61511    320 -rwx------   1 user     user       325104 May  3 12:45 ./lib/libmbedcrypto.so.0
    61524    300 -rw-------   1 user     user       304272 May  3 12:45 ./lib/modules/e1000e.ko
    61510    224 -rwx------   1 user     user       227696 May  3 12:45 ./lib/libksba.so.8
    61516    204 -rwx------   1 user     user       207912 May  3 12:45 ./lib/libpng16.so.16
    61335    184 -rwx------   1 user     user       184824 May  3 12:45 ./bin/dropbear
    61442    176 -rwx------   1 user     user       176408 May  3 12:45 ./bin/ssh
    61369    168 -rwx------   1 user     user       170984 May  3 12:45 ./bin/kexec
    61528    160 -rw-------   1 user     user       159840 May  3 12:45 ./lib/modules/xhci-hcd.ko
    61505    156 -rwx------   1 user     user       159000 May  3 12:45 ./lib/libcryptsetup.so.4
    61334    136 -rwx------   1 user     user       137464 May  3 12:45 ./bin/dmsetup
    61527    132 -rw-------   1 user     user       132360 May  3 12:45 ./lib/modules/usb-storage.ko
    61509    128 -rwx------   1 user     user       130000 May  3 12:45 ./lib/libgpg-error.so.0
    61456    124 -rwx------   1 user     user       126584 May  3 12:45 ./bin/tpm
    61523    108 -rwx------   1 user     user       108832 May  3 12:45 ./lib/libz.so.1
    61521     96 -rwx------   1 user     user        96896 May  3 12:45 ./lib/libusb-1.0.so.0
    61501     76 -rwx------   1 user     user        76736 May  3 12:45 ./lib/libassuan.so.0
    61390     72 -rwx------   1 user     user        73600 May  3 12:45 ./bin/lspci
    61525     64 -rw-------   1 user     user        64800 May  3 12:45 ./lib/modules/ehci-hcd.ko
    61326     64 -rwx------   1 user     user        62328 May  3 12:45 ./bin/cryptsetup-reencrypt
    61325     60 -rwx------   1 user     user        59144 May  3 12:45 ./bin/cryptsetup
    61514     52 -rwx------   1 user     user        52272 May  3 12:45 ./lib/libpci.so.3.5.4
    61513     52 -rwx------   1 user     user        52272 May  3 12:45 ./lib/libpci.so.3
    61411     52 -rwx------   1 user     user        52200 May  3 12:45 ./bin/pinentry-tty
    61343     52 -rwx------   1 user     user        51616 May  3 12:45 ./bin/fbwhiptail
    61517     48 -rwx------   1 user     user        48088 May  3 12:45 ./lib/libpopt.so.0
    61518     48 -rwx------   1 user     user        47448 May  3 12:45 ./lib/libqrencode.so.3
    61473     32 -rwx------   1 user     user        32560 May  3 12:45 ./bin/veritysetup
    61319     28 -rwx------   1 user     user        27048 May  3 12:45 ./bin/cbmem
    61425     24 -rwx------   1 user     user        22664 May  3 12:45 ./bin/scp
    61349     24 -rwx------   1 user     user        22432 May  3 12:45 ./bin/flashtool
    61488     20 -rw-------   1 user     user        19992 May  3 12:45 ./etc/distro/keys/tails.key
    61495     20 -rw-------   1 user     user        18852 May  3 12:45 ./etc/oem/keys/insurgo.key
    61520     20 -rwx------   1 user     user        18464 May  3 12:45 ./lib/libusb-0.1.so.4
    61317     20 -rwx------   1 user     user        18352 May  3 12:45 ./bin/cbfs
    61461     20 -rwx------   1 user     user        18320 May  3 12:45 ./bin/uefi
    61358     16 -rwx------   1 user     user        14657 May  3 12:45 ./bin/gui-init
    61522     16 -rwx------   1 user     user        14656 May  3 12:45 ./lib/libuuid.so.1
    61512     16 -rwx------   1 user     user        14552 May  3 12:45 ./lib/libnpth.so.0
    61414     16 -rwx------   1 user     user        14200 May  3 12:45 ./bin/poke
    61407     12 -rwx------   1 user     user        12056 May  3 12:45 ./bin/oem-factory-reset
    61486     12 -rw-------   1 user     user        10955 May  3 12:45 ./etc/distro/keys/fedora.key
    61526     12 -rw-------   1 user     user        10728 May  3 12:45 ./lib/modules/ehci-pci.ko
    61529     12 -rw-------   1 user     user        10568 May  3 12:45 ./lib/modules/xhci-pci.ko
[... cut ...]
    61478      4 -rwx------   1 user     user           35 May  3 12:45 ./bin/whiptail
    61497      4 -rw-------   1 user     user           27 May  3 12:45 ./etc/shells
    61496      4 -rw-------   1 user     user           27 May  3 12:45 ./etc/passwd
    61492      4 -rw-------   1 user     user           20 May  3 12:45 ./etc/hosts
    61491      4 -rw-------   1 user     user           10 May  3 12:45 ./etc/group
    61303      4 -rw-------   1 user     user           10 May  3 12:45 ./.gnupg/gpg.conf

```

tlaurion, 2020-05-03:
A [public build](https://app.circleci.com/pipelines/github/tlaurion/heads/135/workflows/a22cb04d-4756-4a85-b651-9221c2810b51/jobs/144) showing that not moving along with this actual ticket or with #703 is breaking x230 support and 12Mb boards altogether, Heads having become too big.

@flammit: This build will fail at coreboot integration of cpios (CBFS region not being big enough), but artifacts will include initird.cpio.xz, heads.cpio, tools.cpio and modules.cpio for others to see the limit and dead end we are now facing with.

Now what?

MrChromebox, 2020-05-03:
@tlaurion is reducing the kernel size not an option, or simply too much effort?

tlaurion, 2020-05-03:
@MrChromebox : my reluctance in attacking Kernel size reduction comes with the conclusions of #453 where some 300k were obtained. Of course, [those recommendations should be investigated](https://github.com/osresearch/heads/issues/590#issuecomment-548672101).

Will reread myself.

tlaurion, 2020-05-25:
@MrChromebox #668 and #710 being resolved, I've been looking at multiple source on kernel debloating to do some state of the art.

It seems that be most interesting article is [here](https://hckuo.github.io/pdfs/cozart.pdf).

Reading.

tlaurion, 2020-05-26:
Playing around with information found under this post to [reduce kernel size](https://lwn.net/Articles/746780/)

user@x230-master:~/heads/build/x230$ git diff > patch
user@x230-master:~/heads/build/x230$ cat patch 
```
diff --git a/config/linux-x230.config b/config/linux-x230.config
index dd5af0c..6cb7ea8 100644
--- a/config/linux-x230.config
+++ b/config/linux-x230.config
@@ -14,6 +14,7 @@ CONFIG_INITRAMFS_SOURCE="../../../blobs/dev.cpio"
 # CONFIG_RD_LZO is not set
 # CONFIG_RD_LZ4 is not set
 CONFIG_CC_OPTIMIZE_FOR_SIZE=y
+# CONFIG_MULTIUSER is not set
 # CONFIG_SGETMASK_SYSCALL is not set
 # CONFIG_SYSFS_SYSCALL is not set
 # CONFIG_BASE_FULL is not set
@@ -25,18 +26,18 @@ CONFIG_CC_OPTIMIZE_FOR_SIZE=y
 # CONFIG_MEMBARRIER is not set
 CONFIG_EMBEDDED=y
 # CONFIG_VM_EVENT_COUNTERS is not set
-# CONFIG_SLUB_DEBUG is not set
 # CONFIG_COMPAT_BRK is not set
+CONFIG_SLOB=y
 CONFIG_JUMP_LABEL=y
 CONFIG_CC_STACKPROTECTOR_STRONG=y
 CONFIG_MODULES=y
+CONFIG_TRIM_UNUSED_KSYMS=y
 # CONFIG_IOSCHED_DEADLINE is not set
 # CONFIG_IOSCHED_CFQ is not set
 CONFIG_SMP=y
 # CONFIG_X86_EXTENDED_PLATFORM is not set
 CONFIG_PROCESSOR_SELECT=y
 # CONFIG_CPU_SUP_CENTAUR is not set
-CONFIG_PREEMPT_VOLUNTARY=y
 CONFIG_X86_REROUTE_FOR_BROKEN_BOOT_IRQS=y
 # CONFIG_X86_MCE_AMD is not set
 # CONFIG_PERF_EVENTS_INTEL_RAPL is not set
@@ -56,7 +57,6 @@ CONFIG_KEXEC_FILE=y
 CONFIG_PHYSICAL_ALIGN=0x1000000
 # CONFIG_MODIFY_LDT_SYSCALL is not set
 # CONFIG_SUSPEND is not set
-CONFIG_ACPI_VIDEO=y
 CONFIG_PCI_MSI=y
 # CONFIG_HT_IRQ is not set
 CONFIG_PCI_IOV=y
@@ -184,9 +184,7 @@ CONFIG_MFD_SYSCON=y
 CONFIG_DRM=y
 CONFIG_DRM_I915=y
 CONFIG_FB_VESA=y
-CONFIG_BACKLIGHT_LCD_SUPPORT=y
 # CONFIG_LCD_CLASS_DEVICE is not set
-CONFIG_BACKLIGHT_CLASS_DEVICE=y
 # CONFIG_BACKLIGHT_GENERIC is not set
 CONFIG_FRAMEBUFFER_CONSOLE=y
 CONFIG_USB=y
@@ -207,9 +205,7 @@ CONFIG_GENERIC_PHY=y
 # CONFIG_DMIID is not set
 CONFIG_GOOGLE_FIRMWARE=y
 CONFIG_GOOGLE_MEMCONSOLE_X86_LEGACY=y
-# CONFIG_EXT2_FS is not set
 CONFIG_EXT4_FS=y
-CONFIG_EXT4_USE_FOR_EXT2=y
 # CONFIG_DNOTIFY is not set
 # CONFIG_INOTIFY_USER is not set
 CONFIG_ISO9660_FS=y
@@ -218,7 +214,6 @@ CONFIG_MSDOS_FS=y
 CONFIG_VFAT_FS=y
 # CONFIG_PROC_SYSCTL is not set
 # CONFIG_PROC_PAGE_MONITOR is not set
-CONFIG_TMPFS=y
 # CONFIG_MISC_FILESYSTEMS is not set
 CONFIG_NLS_DEFAULT="utf8"
 CONFIG_NLS_CODEPAGE_437=y
@@ -238,13 +233,11 @@ CONFIG_STACKTRACE=y
 # CONFIG_DEBUG_BUGVERBOSE is not set
 # CONFIG_RCU_TRACE is not set
 # CONFIG_FTRACE is not set
-# CONFIG_STRICT_DEVMEM is not set
 # CONFIG_X86_VERBOSE_BOOTUP is not set
 # CONFIG_DOUBLEFAULT is not set
 CONFIG_IO_DELAY_0XED=y
 CONFIG_OPTIMIZE_INLINING=y
 # CONFIG_X86_DEBUG_FPU is not set
-CONFIG_HARDENED_USERCOPY=y
 CONFIG_CRYPTO_RSA=m
 CONFIG_CRYPTO_USER=y
 CONFIG_CRYPTO_MCRYPTD=m
```

Raw results
[Before](https://github.com/osresearch/heads/issues/590#issuecomment-623143332):
    `28192   2956 -rw-r--r--   1 user     user      3023312 May  3 12:25 ./bzImage`
Now:
`    31097   2876 -rw-r--r--   1 user     user      2941392 May 26 10:46 ./bzImage`

tlaurion, 2020-05-28:
LTO attempt is continuing under #730 
Any help welcome.

tlaurion, 2020-12-01:
The goal of the 3 different commits linked to this issue is to show different use cases linked to previous discussions over Slack that happened in the past days.

We take #703  [pipeline](https://app.circleci.com/pipelines/github/tlaurion/heads/623/workflows/118fb09b-f2c1-4348-9ca5-cd68615e7f66/jobs/671) as a reference where the [x230-hotp-maximized build output](https://circleci.com/api/v1.1/project/github/tlaurion/heads/671/output/171/0?file=true&allocation-id=5fc57a82e7c04244a5cd4da5-0-build%2F7635F303) will serve as a reference for compressed saved space between feature deactivation. 

This board has has the following [board config](https://github.com/osresearch/heads/blob/1462bee343c5c43f99ab9fdf91b01dc61e9a403e/boards/x230-hotp-maximized/x230-hotp-maximized.config) initially, from which we will deactivate features to compare gained space. 

Initially, this board is fully loaded with 
- FBWHIPTAIL(CAIRO) and dependencies 
- E1000E driver 
- HOTP (For HOTP remote attestation over Nitrokey/Librem Key USB Security dongles).
- NKSTORECLI
- GPG2

Here is an exerpt of used space:
```
"/root/project/build/coreboot-4.8.1/x230-hotp-maximized/cbfstool" "/root/project/build/coreboot-4.8.1/x230-hotp-maximized/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           84708 none
cpu_microcode_blob.bin         0x14c00    microcode       25600 none
fallback/ramstage              0x1b080    stage           81122 none
config                         0x2edc0    raw               768 none
revision                       0x2f100    raw               581 none
cmos_layout.bin                0x2f380    cmos_layout      1804 none
fallback/dsdt.aml              0x2fb00    raw             13646 none
fallback/payload               0x330c0    simple elf    7191492 none
(empty)                        0x70ecc0   null          4524824 none
bootblock                      0xb5f800   bootblock        1968 none
```

- Where  total size of the Heads linux payload itself:
`fallback/payload               0x330c0    simple elf    7191492 none`

- Where free room in defined CBFS region created for the coreboot rom in coreboot config for that board:
`(empty)                        0x70ecc0   null          4524824 none`

Now let's compare.

[https://github.com/tlaurion/heads/commit/ac34b14a816e0beb66313632980f66ab098fdfb5](x230-hotp-maximized: testing compressed gain from removing E1000E and DROPBEAR (User side network tools for https://github.com/osresearch/heads/issues/590)) where [build output](https://circleci.com/api/v1.1/project/github/tlaurion/heads/673/output/171/0?file=true&allocation-id=5fc6ac66cfa4ed571c0e4ab6-0-build%2F78FFD5B3):
```
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           84708 none
cpu_microcode_blob.bin         0x14c00    microcode       25600 none
fallback/ramstage              0x1b080    stage           81123 none
config                         0x2edc0    raw               768 none
revision                       0x2f100    raw               581 none
cmos_layout.bin                0x2f380    cmos_layout      1804 none
fallback/dsdt.aml              0x2fb00    raw             13646 none
fallback/payload               0x330c0    simple elf    7028676 none
(empty)                        0x6e70c0   null          4687640 none
bootblock                      0xb5f800   bootblock        1968 none
```

 
[https://github.com/tlaurion/heads/commit/17cf86e3aa5041932e9dd44f1233d2212825668f](x230-hotp-maximized: reverting entwork tools deactivation. Testing compressed gain from removing FBwhiptail support(Deactivating CAIRO and FBWHIPTAIL) and switching to SLANG and NEWT for console only output without fancyness for https://github.com/osresearch/heads/issues/590)) where [build  output](https://circleci.com/api/v1.1/project/github/tlaurion/heads/674/output/171/0?file=true&allocation-id=5fc6adc87f489d73e9692577-0-build%2F5302E524):
```
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           84708 none
cpu_microcode_blob.bin         0x14c00    microcode       25600 none
fallback/ramstage              0x1b080    stage           81122 none
config                         0x2edc0    raw               768 none
revision                       0x2f100    raw               581 none
cmos_layout.bin                0x2f380    cmos_layout      1804 none
fallback/dsdt.aml              0x2fb00    raw             13646 none
fallback/payload               0x330c0    simple elf    6910916 none
(empty)                        0x6ca4c0   null          4805400 none
bootblock                      0xb5f800   bootblock        1968 none
```

[https://github.com/tlaurion/heads/commit/40033e577f6eb54560efdf885168f6dc731d6604](x230-hotp-maximized: Testing compressed gain from removing Whiptail altogether and switching back to generic-init for https://github.com/osresearch/heads/issues/590)) where [build output](https://circleci.com/api/v1.1/project/github/tlaurion/heads/675/output/171/0?file=true&allocation-id=5fc6aece299c7712308baf11-0-build%2F3EFDB262):
```
[... cut ...]
fallback/ramstage              0x1b080    stage           81110 none
config                         0x2edc0    raw               768 none
revision                       0x2f100    raw               581 none
cmos_layout.bin                0x2f380    cmos_layout      1804 none
fallback/dsdt.aml              0x2fb00    raw             13646 none
fallback/payload               0x330c0    simple elf    6588868 none
(empty)                        0x67bac0   null          5127448 none
bootblock                      0xb5f800   bootblock        1968 none
```

tlaurion, 2020-12-01:
For comparison available in CI, the current [X230-hotp-verification board, with E1000E and DROPBEAR already deactivated build's log](https://circleci.com/api/v1.1/project/github/tlaurion/heads/675/output/142/0?file=true&allocation-id=5fc6aece299c7712308baf11-0-build%2F3EFDB262):
```
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           84708 none
cpu_microcode_blob.bin         0x14c00    microcode       25600 none
fallback/ramstage              0x1b080    stage           81111 none
config                         0x2edc0    raw               570 none
revision                       0x2f040    raw               581 none
cmos_layout.bin                0x2f2c0    cmos_layout      1804 none
fallback/dsdt.aml              0x2fa40    raw             13646 none
fallback/payload               0x33000    simple elf    6860228 none
(empty)                        0x6bde00   null           137688 none
bootblock                      0x6df800   bootblock        1968 none
```

MrChromebox, 2020-12-01:
@tlaurion so basically we'd be splitting the xx20/30 boards into two versions:

1) stock IFD / BIOS region size - reduced capability, no networking, UI, etc
2) 'max' versions which assume modified IFD, cleaned/shrunk ME, etc - full capability

Thrilleratplay, 2020-12-01:
@tlaurion If you are only asking if the current xx20 `CONFIG_CBFS_SIZE` should be set to stock for these boards, I am not sure.  This is roughly 3Mb.  Is this enough for even the most striped down version of Heads?

tlaurion, 2020-12-02:
> @tlaurion so basically we'd be splitting the xx20/30 boards into two versions:
> 
>     1. stock IFD / BIOS region size - reduced capability, no networking, UI, etc
Well, the x230 board config is already reducing its functionalities to fit in stock BIOS region. So yes, no more DROPBEAR nor E1000E as of right now. That baseline was made to have x230-hotp-verification board, which could compile only from x230 version + HOTP.

I'm just letting eveyone know the challenges, and choices that are ahead of us in specializing boards with current force functionnalities since we rely on GPG for key generation, that GNU toolstack is huged and I wasn't able to reduce it further more. The same logic applied with FBWHIPTAIL and its WHIPTAIL equivalents for servers, as detailed in previous posts. I just thought that those extracted binaries were irrelevant for comparison, since what matters is their compressed sizes.

> 
>     2. 'max' versions which assume modified IFD, cleaned/shrunk ME, etc - full capability

Correct.

tlaurion, 2020-12-02:
> @tlaurion If you are only asking if the current xx20 `CONFIG_CBFS_SIZE` should be set to stock for these boards, I am not sure. This is roughly 3Mb. Is this enough for even the most striped down version of Heads? Maybe only have the xx20 maximum versions?

> 
> Also note, that these boards are not supported by 1vyrain and would have needed to have been flashed externally initially.

Right. Forgot about that, but original ROM versio nwas really small for x220 stock.
We can imply that xx20 users are already well aware of those restrictions. Funny enough, the x230 base board limits to 7mb where x200 limits to 7.5mb. Consequently, the base board can differ in base functionalities. Like I said on slack to @flammit which proposed to maintain minimal boards, I will focus on adding fonctionalities and will stray away of the base boards which will probably soon enough require other features to be removed.

As I raised the flag a while ago, it is really difficult to maintain mutliple versions and different toolstacks. More people are welcome, but we need to face the reality that new OSes install with LUKS2 which require new version of cryptsetup part of #893, which will result soon enough into cryptsetup2 module, which will require maximized versions.

Note also that the stalled #709 wil lalso required available space. So there will be a need to minimize the base boards and a choice to continue supporting those from community members. Otherwise, as @flammit said, we wil lbe able to bring back gpg1 still in modules and require users to generate their 4096 bits keys outside and import their public key inside of heads, modify the scripts to valide which version of modules is used etc.... Which means more maintainserhip, not less, on which I do not personally agree from lack of time already.

So this opens the debate on where to go from now for the base boards and who wil ltake the lead into pursuing space reduction or feature reduction when CI builds will start to fail. My only option at that point will be to remove those boards from CI.

Thrilleratplay, 2020-12-02:
@tlaurion Something that needs to be defined is "what is included in the stock builds"?. For the sake of argument, lets say Heads is stripped down to just a Linux payload.  No encryption functionality, no e1000,  module, no dropbear, no fbwhiptail, and maybe no flashrom.  Basically, drops to a shell and the user would run a kexec command to boot from the hard drive or usb or whatever.  Would Heads fit within the 3Mb? If so, how close because if I remember correctly, the Linux 5.x kernel is larger.  With such a limited space, what is the possible longevity of this build if the most fundamental part of Heads will not fit?

Would the stock builds be the same for xx20 and xx30 boards given the significant difference in available space?  I would suspect yes, so a break down of features added to each build would need to be added to the documentation.

The flip side to these questions are "what can be remove?".  How basic can Heads become and still be considered Heads?  While the idea of removing all of the security functionality may sound ludicrous to you, it was something I was going thinking about building for myself.  I miss the days when you flip a switch on a VIC20 and get a prompt almost instantly; security wouldn't be moved to the BIOS, it would not be user friendly, but turning on my computer and getting a prompt in 2 seconds sounds damn sexy to me.  However, this would no longer be Heads at this point.  So a line in the sand must be drawn saying that Heads must include certain functionality to be called Heads if the hardware supports it.  Heads for a X200 does not need to include TPM functionality as the device doesn't support it.

flammit, 2020-12-02:
To give concrete details to this discussion, all you would need to do is disable the UI bits and that gives you enough space for everything else (including cryptsetup2 and gpg2) to fit into a 6.4M payload which fits for standard x230 and ME-shrunked x220 (seems like the only route for that hardware) and 600k of free space to play with on an x230.  (Note: there are other combinations of configs that work as well if you consider gpg1).

If that's not a trade off that is acceptable to you as the x230 hardware owner (no luck for x220 yet), there's a perfectly reasonable alternative by installing the "max" version.

flammit, 2020-12-02:
Just based on my quick tests past last hour, I think if you clean up the kernel config to remove the networking bits (not relevant for laptop boot), you can also fit in the UI bits with at least 300k of space for user config.  It's tight but it doesn't seem like there's really a problem.

Then again if there's a need for even more space for future features/modules, it can be accommodated in the "max" version.

tlaurion, 2020-12-02:
> Just based on my quick tests past last hour, I think if you clean up the kernel config to remove the networking bits (not relevant for laptop boot), you can also fit in the UI bits with at least [...]

@flammit: please revive https://github.com/osresearch/heads/issues/564, that was investigated with some effort in the past and for which conclusions are the original post: 
`remove networking ( #564 ), with corresponding PR attempt to remove networking with cleanup ( #572 ) while cryptsetup seemed to depend on CONFIG_NET and some explicit crypto modules in the past ( #79 ), while putting CONFIG_LINUX_E1000E=n in board config liberates 295688 bytes. @merge : Is that enough?`

tlaurion, 2020-12-02:
> @tlaurion Something that needs to be defined is "what is included in the stock builds"?. For the sake of argument, lets say Heads is stripped down to just a Linux payload. No encryption [...]

I invite you @flammit @Thrilleratplay both to jump into #818. (sonner then later this time :) )

tlaurion, 2020-12-05:
I think the simplest from now on, to check for x220 space limitations, is to follow xx30 boards already in CI, which informs us quite quickly of the compressed CBFS space still available.

For example, busybox 1.32 inclusion PR ( #900 ) CI's [x230 board build](https://app.circleci.com/pipelines/github/tlaurion/heads/633/workflows/bb97022d-5968-4215-9000-40e1d7c7c54f/jobs/681/parallel-runs/0/steps/0-153) informs us that:
```
"/root/project/build/coreboot-4.8.1/x230/cbfstool" "/root/project/build/coreboot-4.8.1/x230/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           85188 none
cpu_microcode_blob.bin         0x14dc0    microcode       25600 none
fallback/ramstage              0x1b240    stage           81881 none
config                         0x2f280    raw               589 none
revision                       0x2f540    raw               581 none
cmos.default                   0x2f7c0    cmos_default      256 none
cmos_layout.bin                0x2f900    cmos_layout      1804 none
fallback/dsdt.aml              0x30080    raw             13646 none
fallback/payload               0x33640    simple elf    6863812 none
(empty)                        0x6bf240   null           131480 none
bootblock                      0x6df400   bootblock        3000 none
```
From this point of time, Heads payload consumes `6863812` and leaves us `131480` to play with, after which the CBFS region of 700000 will be filled. This is interesting to follow and could be used as a deprecation warning and need of changes, since the xx20 has 7500000 and will tolerate more.

TL;DR: when xx30 boards will start to fail building in CI, this will be the sign that some modules will need to be removed even more.

EnigmaticRadical, 2021-04-17:
Apologies is this is patently obvious but isn't the most obvious thing to do is to set the -Os flag in gcc?  Currently the Makefiles all use the -O2 flag.  Using GPG2 as my test, I manually patched all the generated Makefiles using

    find . | grep "Makefile" | grep -v "Makefile." | xargs sed -i 's/O2/Os/g'

and running make, which yielded about 15% space reduction in the gpg2 binary and about 5% in the scdaemon and 5% in gpg-agent.  Is there some limitation (breaking reproducibility?  just really hard to patch the makefiles?) preventhing this?

tlaurion, 2021-11-03:
> find . | grep "Makefile" | grep -v "Makefile." | xargs sed -i 's/O2/Os/g'

@aesrentai excellent insight! Yes, this would require creation of patches under patches/* to patch all related Makefiles after decompresion of the archives and prior of compilation. 

On reproducibility, as far as I know, since musl-cross-make is used to create the final binaries, it should be a magic gain without impact. Note that kernel modules are stripped prior of being injected into modules.cpio and prior en compression of that cpio into modules.cpio.xz

Do you have a PoC of this?

tlaurion, 2022-02-21:
With: 

```
user@heads-tests:~/heads/build/x230-hotp-maximized$ git diff
diff --git a/modules/cairo b/modules/cairo
index 647ed2ca..217137c0 100644
--- a/modules/cairo
+++ b/modules/cairo
@@ -8,7 +8,7 @@ cairo_hash := 8c90f00c500b2299c0a323dd9beead2a00353752b2092ead558139bd67f7bf16
 
 cairo_configure := \
        $(CROSS_TOOLS) \
-       CFLAGS="-DCAIRO_NO_MUTEX=1 -O3"  \
+       CFLAGS="-DCAIRO_NO_MUTEX=1 -O2"  \
        ./configure \
         --host i386-elf-linux \
        --prefix="/" \

```

Before:
`  3434539    740 -rwx------   1 user     user       757232 Feb 21 16:48 ./lib/libcairo.so.2`
After:
`   284396    740 -rwx------   1 user     user       757232 Feb 21 17:08 ./lib/libcairo.so.2`

tlaurion, 2022-02-21:
```
diff --git a/modules/pixman b/modules/pixman
index 65a2e200..b8202672 100644
--- a/modules/pixman
+++ b/modules/pixman
@@ -8,6 +8,7 @@ pixman_hash := 21b6b249b51c6800dc9553b65106e1e37d0e25df942c90531d4c3997aa20a88e
 
 pixman_configure := \
        $(CROSS_TOOLS) \
+       CFLAGS="-O2"  \
        ./configure \
         --host i386-elf-linux \
        --prefix="/" \

```
Before:
`   3434550    652 -rwx------   1 user     user       666216 Feb 21 16:48 ./lib/libpixman-1.so.0`

After:
`   284407    652 -rwx------   1 user     user       666216 Feb 21 17:08 ./lib/libpixman-1.so.0`

tlaurion, 2022-02-21:
```
diff --git a/modules/cairo b/modules/cairo
index 647ed2ca..7fc92331 100644
--- a/modules/cairo
+++ b/modules/cairo
@@ -8,7 +8,7 @@ cairo_hash := 8c90f00c500b2299c0a323dd9beead2a00353752b2092ead558139bd67f7bf16
 
 cairo_configure := \
 	$(CROSS_TOOLS) \
-	CFLAGS="-DCAIRO_NO_MUTEX=1 -O3"  \
+	CFLAGS="-DCAIRO_NO_MUTEX=1 -Os"  \
 	./configure \
         --host i386-elf-linux \
 	--prefix="/" \
diff --git a/modules/libpng b/modules/libpng
index e5c3d718..8debbda4 100644
--- a/modules/libpng
+++ b/modules/libpng
@@ -8,6 +8,7 @@ libpng_hash := 574623a4901a9969080ab4a2df9437026c8a87150dfd5c235e28c94b212964a7
 
 libpng_configure := \
 	$(CROSS_TOOLS) \
+	CFLAGS="-Os" \
 	./configure \
         --host i386-elf-linux \
 	--prefix="/" \
diff --git a/modules/pixman b/modules/pixman
index 65a2e200..e7ec1bd8 100644
--- a/modules/pixman
+++ b/modules/pixman
@@ -8,6 +8,7 @@ pixman_hash := 21b6b249b51c6800dc9553b65106e1e37d0e25df942c90531d4c3997aa20a88e
 
 pixman_configure := \
 	$(CROSS_TOOLS) \
+	CFLAGS="-Os"  \
 	./configure \
         --host i386-elf-linux \
 	--prefix="/" \
diff --git a/modules/tpmtotp b/modules/tpmtotp
index 433df8ce..1ce561d0 100644
--- a/modules/tpmtotp
+++ b/modules/tpmtotp
@@ -13,7 +13,7 @@ tpmtotp_hash := 1082f2b0e4af833e04220dddedcc21a39eb39ee4dc5668bb010e7bcc795c606c
 
 tpmtotp_target := \
 	$(CROSS_TOOLS) \
-	CFLAGS="-I$(INSTALL)/include" \
+	CFLAGS="-I$(INSTALL)/include -Os" \
 	LDFLAGS="-L$(INSTALL)/lib" \
 
 tpmtotp_output := \
diff --git a/modules/zlib b/modules/zlib
index dbdb44e3..b1d2adf5 100644
--- a/modules/zlib
+++ b/modules/zlib
@@ -9,6 +9,7 @@ zlib_hash := c3e5e9fdd5004dcb542feda5ee4f0ff0744628baf8ed2dd5d66f8ca1197cb1a1
 
 zlib_configure := \
 	$(CROSS_TOOLS) \
+	CFLAGS="-Os" \
 	./configure \
 	--prefix="/" \

```

`xz -d initrd.cpio.xz ; for i in initrd.cpio modules.cpio tools.cpio heads.cpio; do cpio -i < $i; done && find . -type f -ls | sort -r -n -k7 |grep -v cpio`

Before :
  ```
3434331  12288 -rw-r--r--   1 user     user     12582912 Feb 21 12:56 ./heads-x230-hotp-maximized-v0.2.1.bis2-35-g3fabfa20-dirty.rom
[... 234 lines cut ...]
   281447      4 -rwx------   1 user     user          661 Feb 21 17:56 ./bin/uefi-init
   281452      4 -rwx------   1 user     user          634 Feb 21 17:56 ./bin/unseal-totp
   281472      4 -rw-------   1 user     user          625 Feb 21 17:56 ./etc/config
   281464      4 -rwx------   1 user     user          574 Feb 21 17:56 ./bin/x230-flash.init
   281431      4 -rwx------   1 user     user          574 Feb 21 17:56 ./bin/t430-flash.init
   281402      4 -rwx------   1 user     user          538 Feb 21 17:56 ./bin/qubes-measure-luks
   281324      4 -rwx------   1 user     user          360 Feb 21 17:56 ./bin/flashrom-kgpe-d16-openbmc.sh
   281462      4 -rwx------   1 user     user          320 Feb 21 17:56 ./bin/wget-measure.sh
   281403      4 -rwx------   1 user     user          258 Feb 21 17:56 ./bin/reboot
   281455      4 -rwx------   1 user     user          220 Feb 21 17:56 ./bin/usb-init
   281397      4 -rwx------   1 user     user          205 Feb 21 17:56 ./bin/poweroff
   281479      4 -rw-------   1 user     user          197 Feb 21 17:56 ./etc/fstab
   281484      4 -rw-------   1 user     user          174 Feb 21 17:56 ./etc/motd
   281332      4 -rwx------   1 user     user          106 Feb 21 17:56 ./bin/gpgv
   284192      4 -rw-------   1 user     user           73 Feb 21 17:08 ./.gnupg/gpg-agent.conf
   281463      4 -rwx------   1 user     user           35 Feb 21 17:56 ./bin/whiptail
   281487      4 -rw-------   1 user     user           27 Feb 21 17:56 ./etc/shells
   281486      4 -rw-------   1 user     user           27 Feb 21 17:56 ./etc/passwd
   281483      4 -rw-------   1 user     user           20 Feb 21 17:56 ./etc/hosts
   284193      4 -rw-------   1 user     user           10 Feb 21 17:08 ./.gnupg/gpg.conf
   281481      4 -rw-------   1 user     user           10 Feb 21 17:56 ./etc/group
   677025      0 -rw-------   1 user     user            0 Feb 21 17:56 ./run/cryptsetup/.placeholder
```

Ok ok.... Its useful!

tlaurion, 2022-02-21:
`user@heads-tests:~/heads/build/x230-hotp-maximized$ xz -d initrd.cpio.xz ; for i in initrd.cpio modules.cpio tools.cpio heads.cpio; do cpio -i < $i; done && find . -type f -ls | sort -r -n -k7`

Before:
```
  3433968  12552 -rw-r--r--   1 user     user     12868096 Feb 21 12:56 ./initrd.cpio
  3434331  12288 -rw-r--r--   1 user     user     12582912 Feb 21 12:56 ./heads-x230-hotp-maximized-v0.2.1.bis2-35-g3fabfa20-dirty.rom
  3421972  11720 -rw-r--r--   1 user     user     11998208 Feb 21 12:42 ./tools.cpio
  3434335   8192 -rw-r--r--   1 user     user      8388608 Feb 21 12:56 ./heads-x230-hotp-maximized-v0.2.1.bis2-35-g3fabfa20-dirty-bottom.rom
  3424621   8192 -rw-r--r--   1 user     user      8388608 Feb 17 12:34 ./heads-x230-hotp-maximized-v0.2.1.bis2-32-g9fd9c0d4-dirty-bottom.rom
  3434336   4096 -rw-r--r--   1 user     user      4194304 Feb 21 12:56 ./heads-x230-hotp-maximized-v0.2.1.bis2-35-g3fabfa20-dirty-top.rom
  3424622   4096 -rw-r--r--   1 user     user      4194304 Feb 17 12:34 ./heads-x230-hotp-maximized-v0.2.1.bis2-32-g9fd9c0d4-dirty-top.rom
  3423309   2956 -rw-r--r--   1 user     user      3023312 Feb 21 12:41 ./bzImage
  3434095   1592 -rwx------   1 user     user      1627856 Feb 21 16:48 ./bin/lvm
  3434542   1136 -rwx------   1 user     user      1160520 Feb 21 16:48 ./lib/libgcrypt.so.20
  3434052    896 -rwx------   1 user     user       915328 Feb 21 16:48 ./bin/gpg
  3434539    740 -rwx------   1 user     user       757232 Feb 21 16:48 ./lib/libcairo.so.2
  3421971    668 -rw-r--r--   1 user     user       684032 Feb 21 12:41 ./modules.cpio
  3434550    652 -rwx------   1 user     user       666216 Feb 21 16:48 ./lib/libpixman-1.so.0
  3434538    584 -rwx------   1 user     user       596544 Feb 21 16:48 ./lib/libc.so
  3434046    568 -rwx------   1 user     user       580568 Feb 21 16:48 ./bin/flashrom
  3434010    496 -rwx------   1 user     user       507776 Feb 21 16:48 ./bin/busybox
  3434540    456 -rwx------   1 user     user       464560 Feb 21 16:48 ./lib/libcryptsetup.so.12
  3434455    408 -rwx------   1 user     user       416040 Feb 21 16:48 ./bin/scdaemon
  3434053    372 -rwx------   1 user     user       380120 Feb 21 16:48 ./bin/gpg-agent
  3434554    332 -rwx------   1 user     user       339304 Feb 21 16:48 ./lib/libtpm.so
  3434541    328 -rwx------   1 user     user       333240 Feb 21 16:48 ./lib/libdevmapper.so.1.02
  3434546    320 -rwx------   1 user     user       325104 Feb 21 16:48 ./lib/libmbedcrypto.so.0
  3434559    300 -rw-------   1 user     user       304272 Feb 21 16:48 ./lib/modules/e1000e.ko
  3434537    268 -rwx------   1 user     user       271040 Feb 21 16:48 ./lib/libblkid.so.1
  3434545    252 -rwx------   1 user     user       256376 Feb 21 16:48 ./lib/libksba.so.8
  3434551    204 -rwx------   1 user     user       207912 Feb 21 16:48 ./lib/libpng16.so.16
  3434337    184 -rw-r--r--   1 user     user       186880 Feb 21 12:56 ./heads.cpio
  3434034    184 -rwx------   1 user     user       184824 Feb 21 16:48 ./bin/dropbear
  3434473    176 -rwx------   1 user     user       176408 Feb 21 16:48 ./bin/ssh
  3434071    168 -rwx------   1 user     user       170984 Feb 21 16:48 ./bin/kexec
  3434563    160 -rw-------   1 user     user       159840 Feb 21 16:48 ./lib/modules/xhci-hcd.ko
  3434033    136 -rwx------   1 user     user       137464 Feb 21 16:48 ./bin/dmsetup
  3434562    132 -rw-------   1 user     user       132360 Feb 21 16:48 ./lib/modules/usb-storage.ko
  3434543    128 -rwx------   1 user     user       130000 Feb 21 16:48 ./lib/libgpg-error.so.0
  3434489    124 -rwx------   1 user     user       126584 Feb 21 16:48 ./bin/tpm
  3434023    116 -rwx------   1 user     user       115192 Feb 21 16:48 ./bin/cryptsetup
  3434558    108 -rwx------   1 user     user       108832 Feb 21 16:48 ./lib/libz.so.1
  3434556     96 -rwx------   1 user     user        96896 Feb 21 16:48 ./lib/libusb-1.0.so.0
  3434024     88 -rwx------   1 user     user        87968 Feb 21 16:48 ./bin/cryptsetup-reencrypt
  3434544     80 -rwx------   1 user     user        81024 Feb 21 16:48 ./lib/libjson-c.so.5
  3434536     76 -rwx------   1 user     user        76736 Feb 21 16:48 ./lib/libassuan.so.0
  3434092     72 -rwx------   1 user     user        73600 Feb 21 16:48 ./bin/lspci
  3434560     64 -rw-------   1 user     user        64800 Feb 21 16:48 ./lib/modules/ehci-hcd.ko
  3434549     52 -rwx------   1 user     user        52272 Feb 21 16:48 ./lib/libpci.so.3.5.4
  3434548     52 -rwx------   1 user     user        52272 Feb 21 16:48 ./lib/libpci.so.3
  3434116     52 -rwx------   1 user     user        52200 Feb 21 16:48 ./bin/pinentry-tty
  3434042     52 -rwx------   1 user     user        51616 Feb 21 16:48 ./bin/fbwhiptail
  3434552     48 -rwx------   1 user     user        48088 Feb 21 16:48 ./lib/libpopt.so.0
  3434553     48 -rwx------   1 user     user        47448 Feb 21 16:48 ./lib/libqrencode.so.3
  3434506     48 -rwx------   1 user     user        45264 Feb 21 16:48 ./bin/veritysetup
  3434068     32 -rwx------   1 user     user        29944 Feb 21 16:48 ./bin/hotp_verification
  3434523     28 -rw-------   1 user     user        27936 Feb 21 16:48 ./etc/distro/keys/tails.key
  3434016     28 -rwx------   1 user     user        27048 Feb 21 16:48 ./bin/cbmem
  3434456     24 -rwx------   1 user     user        22664 Feb 21 16:48 ./bin/scp
[... cut ...]
   281513      4 -rw-------   1 user     user           27 Feb 21 18:19 ./etc/shells
   281512      4 -rw-------   1 user     user           27 Feb 21 18:19 ./etc/passwd
   281509      4 -rw-------   1 user     user           20 Feb 21 18:19 ./etc/hosts
   284193      4 -rw-------   1 user     user           10 Feb 21 17:08 ./.gnupg/gpg.conf
   281507      4 -rw-------   1 user     user           10 Feb 21 18:19 ./etc/group
   281543      0 -rw-------   1 user     user            0 Feb 21 18:19 ./run/cryptsetup/.placeholder

```

tlaurion, 2022-02-22:
Before a3b058de:
```
FMAP REGION: COREBOOT
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           85100 none
cpu_microcode_blob.bin         0x14d80    microcode       26624 none
fallback/ramstage              0x1b600    stage           97721 none
config                         0x33400    raw               790 none
revision                       0x33780    raw               695 none
fallback/dsdt.aml              0x33a80    raw             14615 none
vbt.bin                        0x37400    raw              1433 LZMA (4281 decompressed)
cmos_layout.bin                0x37a00    cmos_layout      1884 none
fallback/postcar               0x381c0    stage           25816 none
fallback/payload               0x3e700    simple elf    7305159 none
(empty)                        0x735f00   null          4365976 none
bootblock                      0xb5fdc0   bootblock       65536 none
```

After [da3e653](https://github.com/tlaurion/heads/commit/da3e653b771fe9cef4c5826e8280c788e390f043):

tlaurion, 2022-07-01:
#1184 freed 313856 bytes.
#1121 freed 424960 bytes.

[Combined](https://github.com/osresearch/heads/pull/1121#issuecomment-1171571652), they free  5148824 - 4360856
787968 bytes (0.7mb)

If anybody can guide into tuning even more, please be my guest.

githubisnonfree, 2022-08-02:
try tcc. you mentioned in the other thread about musl, but tcc is a great c compiler that can produce (usually) smaller binaries (especially when compressed) than gcc or clang, even with -Os etc used.

where gcc is used, try using link-time optimization

also, busybox isn't the smallest posix userland... look at chimera linux, some of freebsd's tools are smaller than busybox's, e.g. freebsd sh is 12k sloc, busybox sh is about 19k last i checked

i've been working on and off porting openbsd userland to linux+musl and stripping it down, their code is even more efficient than freebsd's and often (with features stripped to match busybox) can result in lower amount of code

with bsd-based userland you could probably shave off quite a bit of space used in cbfs. i hope to have something to show later in the year.

githubisnonfree, 2022-08-02:
oh also, try bearssl tls library if you haven't already. super super duper small tls lib, alpine is considering using it in their distro

githubisnonfree, 2022-08-02:
you can mix and match binaries built with tcc or gcc

you could try integrating tcc into your build system, for your portable cross compiler setup that you described earlier. i didn't know heads had that!

githubisnonfree, 2022-08-02:
i should clarify that tcc produces *small* binaries, not *fast* binaries. for speed you want gcc, it's the best one (clang is about as good too). but yeah you want small bins. try tcc!

tlaurion, 2022-08-02:
@githubisnonfree interesting hints.

Yes, @osresearch chose musl-cross then changed it to musl-cross-make as a cross compiler. Enabling LTO was not concluding on my side but things improved a lot there.

Problem as of now is to go back to reproducible builds, totp tool is build against mbetls and everything is linked against musl-cross-make compiled libc.

We could reevaluated the choices here, but opened PRs for passing O3 O2 to Os freed 700kb, while cleaning Linux configurations (which are already minimalist) in another PR freed another 400kb.

With 1mb additional space across all boards using board configured modules, those are the first things to test, which requires t430/w530/x220/t420/t530 board owners which all depend on the shared linux-x230* configurations.

But you got my attention on busybox which is not bash compliant. Busybox is dash and requires people to code with posix style and prevents us to use shellcheck against code base.

I will try to find time to check that up first. But following your work, as you can see with nvmutil :)

tlaurion, 2022-08-02:
> i should clarify that tcc produces small binaries, not fast binaries. for speed you want gcc, it's the best one (clang is about as good too). but yeah you want small bins. try tcc

Thank you. Will see how it can also generalise to Power. But learned that musl musl-cross musl-cross-make was the beat path for safe, small, and generalized upon embedded world.

You have some references to share?

githubisnonfree, 2022-08-02:
btw freebsd sh is also posix-only. it's basically the original bourne shell, rebourne

https://www.freebsd.org/cgi/man.cgi?query=sh&sektion=&n=1

not to be confused with csh which freebsd also has. here is sh, the posix one:

https://cgit.freebsd.org/src/tree/bin/sh

https://github.com/chimera-linux/bsdutils/commit/547ebb8dcfb0c6be55bb728732729cef86834a9a

the sh that i'm currently working on is oksh. someone already ported ksh to linux, so i didn't need to do that:

https://github.com/ibara/oksh

but this is about 19k same as busybox's shell, though with a ton of features. i have a hunch that if i strip down ksh it might be possible to make it smaller than freebsd sh

besides freebsd sh there's also mrsh aiming for full posix, it's *slightly smaller* than freebsd sh but incomplete. freebsd's is currently the best imo

githubisnonfree, 2022-08-02:
as for reference, you mean a website or something?

https://en.wikipedia.org/wiki/Tiny_C_Compiler has some information and the website is https://bellard.org/tcc/ though the author himself doesn't maintain it anymoer, but you can check the mailing list, other people work on it now

githubisnonfree, 2022-08-08:
by the way, the linux kernel can be compiled with tcc

as can musl, and many other things. i'm willing to bet that everything in your build system can be done with tcc

it's possible that you might maybe find a few programs that need minor patching here and there, but tcc is pretty complete. it doesn't have as robust warnings/errors as gcc though, so a thing i normally do is (with -Wall, -Wextra and -Werror set) test code with gcc and clang first, but (if size is a priority) i compile with tcc for what i put into production

in fact, one of the things people talk about with tcc is how quickly it compiles linux

that's another benefit. it only does a single pass, compiling line by line. it compiles very quickly

so in addition to reducing code size, it will reduce compile time aswell

githubisnonfree, 2022-08-08:
try it. just literally modify your build system and/or musl-cross-build to use tcc. try using it as a drop-in replacement, and then see what breaks, if anything. then patch everything up and boot it, see if it works

tcc is a very competent compiler, it's much more "correct" than gcc/clang. the kind of hackiness you see in other compilers simply doesn't exist in tcc. it just compiles your code exactly as written, it doesn't mess around (no optimizer of any kind)

githubisnonfree, 2022-08-08:
coreboot components probably still have to be built with coreboot crossgcc. modifying coreboot to work nicely with tcc might be nice but there's a lot of gcc-ish stuff in their codebase. but everything in your linux distro can probably be built and work nicely with tcc

tlaurion, 2022-08-17:
Consider compressing with `xz -9e` and proper dictionary sizes both initrd.cpio.xz and linux wrapper script. 
Pointers under https://github.com/osresearch/heads/pull/1195#issuecomment-1218046547

tlaurion, 2022-08-17:
Sorry about commit messages above. All were related to -9e, and aimed to see what would pass for all boards and see which changes are actually reducing both kernel and initrd compression. As of now, passing to -9e shows an increase of size, but none of previous commit were just changing to -9e, they were also changing the dictionary size. It is also to note that xz has built-ins default that normally also change the disctionary size, so blunt tests playing with extremes above (dict 1Mib and 100Mib) shows that the defaults are better then random sizes. Nothing else is proven good or wrong at this point, just that changing dict to random sizes is not helping.

- [3e2a543](https://github.com/tlaurion/heads/commit/3e2a543e02687c132beb2c447fa67103c434a8e1) changes only initrd and linux bzImage to use -9e
- [d120023](https://github.com/tlaurion/heads/commit/d120023c7c72e6ef69c9b8a8089d85cb3707f703) changed to xz -9e and dictionnaries sizes to 1Mib (and failed to be packaged for some xx20)
- [fab908a](https://github.com/tlaurion/heads/commit/fab908aaa7fb015ef5026bceb836fcdc314a821b) changed to xz -9e and dictionnaries sizes to 100 Mib (expected to fail packaging for some xx20 as well)

Post will be edited with results when builds are done (those are clean builds if global Makefile is changed, as per master comparison which included CircleCI fix so that CircleCI can now be used to test and compare such builds and have build logs referred directly to proper lines for references). If Makefile is not touched but a patch in modules change outside of coreboot, coreboot + musl-cross-make cache will be reused, speeding up builds.

tlaurion, 2022-08-17:
I may forget about this: but if [fb7a132](https://github.com/tlaurion/heads/commit/fb7a132d09c89896810ffffc159b89d295e824e7) passes and reduce payload's size and increases coreboot's rom free space, we have a win. Once again, passed way too much time testing this and I welcome anyone jumping in this "Quest to reduce firmware size" (payload's size).

druimalban, 2023-05-20:
Some of these comments are fairly general but I think pertinent to the above discussion.

I believe that on platforms with libgfxinit, you can reuse the framebuffer set up by Coreboot, then all you need is the kernel's simplefb and the Google Coreboot framebuffer firmware driver. This no longer requires a patch, it's upstream. This may or may not be an improvement over the current state, but if Linux is doing any graphics initialisation whatsoever currently, there may be a space saving.

The only non-Intel example I can see which you support is the old and quirky KGPE-D16, no longer supported upstream. The suspiciously upbeat Libreboot documentation claims that native graphics initialisation works in text mode, but I've never seen any evidence of this working without a video BIOS on that board.

Nonetheless I suggest that this is fiddling around the edges, the kernel is already pretty small. GPG seems to be the biggest component, and it is difficult to use. I'd personally include a couple of manpages and busybox man!

tlaurion, 2023-05-20:
@druimalban you are right on point with my current experiments with libgfxinit for GMA (Intel only supported in libgfxinit code for native graphic init) and my current status is that its "working" correctly with "simple" hack added into kexec. I have a lot of branches named to test libgfxinit bootsplash, so that older xx30/xx20 platforms shows something (1024x768 framebuffer initialized by libgfxinit and bootplash showing after couple millisecond) up until simplefb drives the framebuffer (still in 1024x768).

Problem there, still with simplefb, is that fbwhiptail won't work writing in single buffer gfx framebuffer and needs work under fbwhiptail. Whiptail (console based menus) works. And then kexec'ing into even tinycore14 for x86_64 works (only provides vesa and the console aquures the passed fb just fine).

Situation is different for qemu q35 coreboot board. Bochs driver in coreboot initializes correctly, shows splashscreen (qemu -vga std) And then bochs DRM takes over the fb correctly. This requires the next kernel to provide bochs DRM as well for next kernel to take over console correctly.

On qemu with bochs providing native init in coreboot, qemu -vga std, and simplefb driving the framebuffer, I'm currently stuck with the fb under qemu not refreshing unless I zoom in zoom out qemu window. This kexec's correctly into next kernel from vesa mode to bochs DRM if provided, but still qemu doesn't refresh the window. Something is wrong there and needs proper troubleshooting to open bug report.

When DRM drivers are removed into kernel, namely i915 fb and DRM, there is a gain of 500kb of compressed firmware space freed under kernel. 

Kernel 5.14+ moved DRM into tiny DRM, and ecosystem is moving forward to remove fb support with kms DRM drivers and is the way forward. Heads could he generalized in using simplefb with native init under coreboot.

Current situation above still works with hacks of kexec so that newer i915drm is passed as VLFB to next kernel call. And libgfxinit enabled for all Intel based gfx and i915drm taking ownership of fb, the user at least gets a splashscreen where he waits for spi to read the payload and execute it (8-10 seconds)

The last case above will probably be the next step, and then moving away of DRM drivers packed in kernel for heads and be replaced by simplefb with fbwhiptail fixed to be able to play with it nicely.

But in the meantime, bumping kernel to at least 5.10 serie is needed for all boards and ideal would be to share the same kernel version so that builds also happen faster for all boards, and for kernel configurations to be properly optimized with more eyes understanding what is happening there in the goal of having only needed options, and board deriving for what is special about them.

Thanks for you input. If you are interested into helping into this let me know! Are you under Matrix? See community page and ping me there, you seem to know a thing or two about what could be going wrong!

tlaurion, 2023-05-20:
> The only non-Intel example I can see which you support is the old and quirky KGPE-D16

There ils dasharo kgpe-d16 supported. Ast driver was good enough last time I tested but unfortunately my CPU/ram is bad and I cannot test. Native gfx init would need to be tested there too with simplefb.

druimalban, 2023-06-08:
> > The only non-Intel example I can see which you support is the old and quirky KGPE-D16
> 
> There is dasharo kgpe-d16 coreboot fork. Ast driver was good enough last time I tested but unfortunately my CPU/ram is bad and I cannot test. Native gfx init would need to be tested there too with simplefb, but yet again, workstation variants (as oppised to server variant) there relies on and/nvidia kernel drivers for which spi space constraints are not really an issue (16mb is plenty there, and 3mdeb sells the connector/spi with wp jumper which is nice).

The Dasharo port of recent Coreboot to the board suffers from faulty automatic fan control, where fans go to full speed and PWM statistics about them become unavailable. The bug occurs approximately 1/3 of the time for me. There is no apparent way to turn it off and it's not clear how it interacts with (working) fan control via openBMC. The last stable Coreboot for the board is 4.11, which is effectively four years old, and has numerous quirks which shan't be fixed, and it isn't fair to bother Coreboot developers about bugs in such an old code-base.

I have suspicions that if using the 3mdeb jumper, ground will be shorted to VCC, assuming that the board's WP pin is pulled to VCC. I do not know of any schematics, so this assumption could be wrong, and the adaptor may be safe to use. Thus I built my own adaptor which breaks out the pins albeit with WP not extending down to the board.

tlaurion, 2023-06-22:
> The Dasharo port of recent Coreboot to the board suffers from faulty automatic fan control, where fans go to full speed and PWM statistics about them become unavailable. The bug occurs approximately [...]

@druimalban : WP is supposed to be pulled to GND per https://docs.dasharo.com/variants/asus_kgpe_d16/spi-wp/ which is what is implemented with jumper and sold at https://shop.3mdeb.com/shop/adapters/flash-chip-adapters/asus-kgpe-d16-flash-chip-adapter/

I would advise into creating a separate issue to track issues you have with kgpe-d16 on dasharo github issues https://github.com/Dasharo/dasharo-issues/issues. I had forgotten about your post here until I needed to come back to the issue at stake and read your post a while ago but since it was irrelevant to issue at stake here, I didn't reply immediately and then it got forgotten.

user@heads-tests-deb12:~/heads/build/x86/linux-5.10.5/linux-x230-maximized$ files=$(find ./ -name built-in.a); size -t $files | sed "s|(TOTALS)|$file|"| sort -nr

tlaurion, 2023-06-22:
Playing around with #1381 getting it to merge, I came up with another oneliner really useful to get to really know the size of everything being packed into the kernel.

Some small notes: Heads builds linux/coreboot in subdirs corresponding to directory name of the linux config file of a board, not in the linux extracted dir. This is the right way, since everything for an arch being compiled as artifacts with musl-cross-make can then be reused in CircleCI to make other boards reuse compiled artifacts without the need fo make distclean or any other cleaning which would require to recompiled everything, not just what differs from one board config to the other through linux config files.

I decided to take a look at current local state of tweaks and built the really now limited t520-maximized board:
`make BOARD=t520-maximized` 
except of coreboot output at the end
```
# Use coreboot.rom, because custom output files might not be processed by cbfstool
"/home/user/heads/build/x86/coreboot-4.19/t520-maximized/cbfstool" "/home/user/heads/build/x86/coreboot-4.19/t520-maximized/coreboot.rom" print
FMAP REGION: COREBOOT
Name                           Offset     Type           Size   Comp
cbfs_master_header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           74024 none
cpu_microcode_blob.bin         0x12240    microcode       26624 none
fallback/ramstage              0x18a80    stage          119765 LZMA (258412 decompressed)
build_info                     0x35ec0    raw               101 none
bootsplash.jpg                 0x35f80    bootsplash      36352 none
fallback/dsdt.aml              0x3edc0    raw             14522 none
vbt.bin                        0x426c0    raw              1389 LZMA (3985 decompressed)
cmos_layout.bin                0x42c80    cmos_layout      2108 none
fallback/postcar               0x43500    stage           22200 none
fallback/payload               0x48c00    simple elf    6997108 none
(empty)                        0x6f50c0   null           879140 none
bootblock                      0x7cbb00   bootblock       17088 none
2023-06-22 12:26:28-04:00 INSTALL   build/x86/coreboot-4.19/t520-maximized/coreboot.rom => build/x86/t520-maximized/heads-t520-maximized-v0.2.0-1590-ge92624e.rom
```

Yes... 879140 on the empty CBFS space means we have 0.8mb left of SPI space. Not so bad still when we consider that that board has 8mb SPI.

Current experiments tweaking linux-x230-maximized.config used to build kernel 5.10.5, exerpt:
```
user@heads-tests-deb12:~/heads/build/x86/linux-5.10.5/linux-x230-maximized$ files=$(find ./ -name built-in.a); size -t $files | sed "s|(TOTALS)|$file|"| sort -nr
19713205	3006613	1105588	23825406	16b8bfe	
 164349	   1338	      0	 165687	  28737	./drivers/gpu/drm/i915/display/intel_display.o (ex ./drivers/gpu/drm/i915/built-in.a)
 164349	   1338	      0	 165687	  28737	./drivers/gpu/drm/i915/display/intel_display.o (ex ./drivers/gpu/drm/built-in.a)
 164349	   1338	      0	 165687	  28737	./drivers/gpu/drm/i915/display/intel_display.o (ex ./drivers/gpu/built-in.a)
 164349	   1338	      0	 165687	  28737	./drivers/gpu/drm/i915/display/intel_display.o (ex ./drivers/built-in.a)
  67591	      0	      0	  67591	  10807	./fs/unicode/utf8-norm.o (ex ./fs/unicode/built-in.a)
  67591	      0	      0	  67591	  10807	./fs/unicode/utf8-norm.o (ex ./fs/built-in.a)
  66834	   4484	    372	  71690	  1180a	./net/core/dev.o (ex ./net/core/built-in.a)
  66834	   4484	    372	  71690	  1180a	./net/core/dev.o (ex ./net/built-in.a)
  66246	    699	     76	  67021	  105cd	./net/core/filter.o (ex ./net/core/built-in.a)
  66246	    699	     76	  67021	  105cd	./net/core/filter.o (ex ./net/built-in.a)
  65530	    545	      0	  66075	  1021b	./drivers/gpu/drm/i915/display/intel_dp.o (ex ./drivers/gpu/drm/i915/built-in.a)
  65530	    545	      0	  66075	  1021b	./drivers/gpu/drm/i915/display/intel_dp.o (ex ./drivers/gpu/drm/built-in.a)
  65530	    545	      0	  66075	  1021b	./drivers/gpu/drm/i915/display/intel_dp.o (ex ./drivers/gpu/built-in.a)
  65530	    545	      0	  66075	  1021b	./drivers/gpu/drm/i915/display/intel_dp.o (ex ./drivers/built-in.a)
  63980	   3166	    956	  68102	  10a06	./kernel/events/core.o (ex ./kernel/events/built-in.a)
  63980	   3166	    956	  68102	  10a06	./kernel/events/core.o (ex ./kernel/built-in.a)
  58165	    812	    591	  59568	   e8b0	./drivers/input/serio/i8042.o (ex ./drivers/input/serio/built-in.a)
  58165	    812	    591	  59568	   e8b0	./drivers/input/serio/i8042.o (ex ./drivers/built-in.a)
  57285	     20	      0	  57305	   dfd9	./drivers/gpu/drm/drm_edid.o (ex ./drivers/gpu/drm/built-in.a)
  57285	     20	      0	  57305	   dfd9	./drivers/gpu/drm/drm_edid.o (ex ./drivers/gpu/built-in.a)
  57285	     20	      0	  57305	   dfd9	./drivers/gpu/drm/drm_edid.o (ex ./drivers/built-in.a)
  56002	    305	      0	  56307	   dbf3	./drivers/gpu/drm/i915/intel_pm.o (ex ./drivers/gpu/drm/i915/built-in.a)
  56002	    305	      0	  56307	   dbf3	./drivers/gpu/drm/i915/intel_pm.o (ex ./drivers/gpu/drm/built-in.a)
  56002	    305	      0	  56307	   dbf3	./drivers/gpu/drm/i915/intel_pm.o (ex ./drivers/gpu/built-in.a)
  56002	    305	      0	  56307	   dbf3	./drivers/gpu/drm/i915/intel_pm.o (ex ./drivers/built-in.a)
  48584	    409	   1032	  50025	   c369	./fs/ext4/super.o (ex ./fs/ext4/built-in.a)
  48584	    409	   1032	  50025	   c369	./fs/ext4/super.o (ex ./fs/built-in.a)
[... cut ...]
  10146	     40	      0	  10186	   27ca	./drivers/gpu/drm/i915/gt/intel_ggtt.o (ex ./drivers/gpu/drm/i915/built-in.a)
  10146	     40	      0	  10186	   27ca	./drivers/gpu/drm/i915/gt/intel_ggtt.o (ex ./drivers/gpu/drm/built-in.a)
  10146	     40	      0	  10186	   27ca	./drivers/gpu/drm/i915/gt/intel_ggtt.o (ex ./drivers/gpu/built-in.a)
  10146	     40	      0	  10186	   27ca	./drivers/gpu/drm/i915/gt/intel_ggtt.o (ex ./drivers/built-in.a)
  10072	     89	     16	  10177	   27c1	./fs/exec.o (ex ./fs/built-in.a)
  10009	     56	      8	  10073	   2759	./fs/libfs.o (ex ./fs/built-in.a)
```

tlaurion, 2023-08-18:
So we got rid of i915drm and freed 500kb in kernel switching to efifb! What next can be reduced?

tlaurion, 2023-09-07:
Also note that each builds produce sizes.txt for easier comparison for a while now. So it's easy to test changes for size comparison across builds to help the continual quest for firmware footprint reduction

closed 2023-11-28 by tlaurion

tlaurion, 2026-08-08:
The smartcard reprovision flow added to the shared initrd (PR #2158) grew the compressed initrd by ~4.5KB. The 8MB flash boards were already at their CBFS limit, so the t420/x220 hotp-maximized variants no longer fit and were moved to unmaintained_boards (see PR #2158).

This issue needs work from board owners: recovering that space (or freeing any ~4.5KB from the shared initrd/firmware) would let these 8MB boards build again. Until then, users of those platforms are assumed to use TOTP, not HOTP, unless proven otherwise.

tlaurion, 2026-09-25:
https://github.com/linuxboot/heads/pull/2214 saves x220 and t420 for a while