linuxboot/heads #598: [Discussion] x220 ThinkPad | Laptop Support Request

WorryTheBirds12, 2019-08-01:
**Discussion**: Running Heads on the ThinkPad x220
**Why?** The x220 is the only decently built modern mechanical keyboard laptop that can run a modern OS. To many users (including me!) this is a big benefit. 

_______________________

Want to support this issue? Post a bounty on it! We accept bounties via [BountySource](https://www.wikipedia.org/wiki/Bountysource).

Kokokokoka, 2019-08-01:
Heads _is_ working on x220.

WorryTheBirds12, 2019-08-01:
@Kokokokoka If this is the case, then I only see the x230 for installing heads:

http://osresearch.net/Installing-Heads

Please point me in the direction of the installation steps.

flammit, 2019-08-01:
ICYMI: With the thinkpad-ec firmware mods, the x230 can use the x220 keyboard.

Kokokokoka, 2019-08-01:
You can follow the build steps of x230, just replace most parts to x220.
One more thing, please, be so kind and read the Coreboot build howto, as you need to extract blobs, cleanup the ME generate ifd decriptor. 
When you've read the wiki (x220 and build how-to), go through the structure of this project. You'll see that there's x220 in the boards directory. Briefly read the main Makefile. you should set board and etc. Yes, this is not easy,
but after having wiki, http://osresearch.net/Installing-Heads, you'll understand how-to build Heads.

SebastianMcMillan, 2019-08-02:
X220 is supported, but due to lack of testers, severe bugs are present.

tlaurion, 2020-02-19:
Functional under https://github.com/osresearch/heads/commit/21faf524b9d791a75f1710ada502f21298b8fa77

closed 2020-02-19 by tlaurion