linuxboot/heads #599: Upgrade Kernel to 4.19.9 LTS

tlaurion, 2019-08-01:
Testing patches applicability... OK.

Seems like `CONFIG_RETPOLINE=n` needs to be deactivated from `config/linux-*` though.
Do we have a problem with that?

tlaurion, 2019-08-01:
As expected, kernel is too big to be inserted as Heads payload.
Will play around... [Quest to reduce firmware size has started](#590) :)

tlaurion, 2019-08-02:
openssl-devel needs to be installed on host system else <openssl/bio.h> isn't found

osresearch, 2019-08-02:
The retpoline and other side-channel fixes don't seem necessary to me.  Am I missing something in the threat model that would allow attacker controlled code on the Heads/LinuxBoot runtime that isn't otherwise a game over for security?

tlaurion, 2019-08-03:
> The retpoline and other side-channel fixes don't seem necessary to me. Am I missing something in the threat model that would allow attacker controlled code on the Heads/LinuxBoot runtime that isn't otherwise a game over for security?

@osresearch : That was also my reasoning.

tlaurion, 2019-08-05:
The urgency in this upgrade is that the MMC SDHCI modules of the old kernel are unreliable.
I've bought 20 sdcards and only 1/4 of them can be reencrypted under Heads, while there is no issue under Qubes sys-usb VM.

I'm still struggling to downsize it under the 3.8M barrier. Working on it.

tlaurion, 2019-08-08:
WIP
[linux-x230-libremkey.txt](https://github.com/osresearch/heads/files/3482400/linux-x230-libremkey.txt)

tlaurion, 2019-08-09:
[Been commited here onward](https://github.com/osresearch/heads/commit/05e8152e9e2bb8154c01660e45b31458f6ea8d2f)

closed 2023-08-14 by tlaurion