linuxboot/heads #604: Release question

Tonux599, 2019-08-18:
With over 2 years, and 646 commits since the last release, when can we expect to see another release? I have an X230 on the way and would love to have some hashes for a reproducible build.

nemanjan00, 2019-08-18:
You can get nightly builds from @tlaurion 

https://gitlab.com/tlaurion/heads-ci/pipelines

tlaurion, 2020-02-15:
Unfortunately, my builds are not reproducible until Heads upstream changes are merged into my branch.

@osresearch's CI is here: https://circleci.com/gh/osresearch
As of today, latest downloadable artifacts are here (example): https://app.circleci.com/jobs/github/osresearch/heads/138/artifacts

tlaurion, 2020-08-01:
Builds are created upon each commit. Problem still present with GitlabCI #727

closed 2020-08-01 by tlaurion

Tonux599, 2020-08-01:
@tlaurion while I respect your decision to close this issue. The question was more intended to raise queries into release milestones and whether or not *.rom (or a checksum file) will be released and signed with your PGP key.

tlaurion, 2020-08-02:
The ROMs are generated per CircleCI. Per PR if commiter has subscribed to circleci and gitlabci, and each commit merged into master is built on circleci.

Those builds already include hashes, both on CI output from public instances and a hashes.txt file for each board that can be generated without blobs.

The question to sign is an interesting one. Publishing the artifacts to github another one. 

If you have insights,  you can help on the CI issues. Note that the kernel is still not reproducible,  and some regressions are still not fixed per passage to musl-cross-make (s3 rebootibg from suspend, keyboard not working on reboot...)