linuxboot/heads #605: # Heads/Coreboot specific work #

zaolin, 2019-08-21:
# Heads/Coreboot specific work #
@zaolin
- [ ] Properly support Intel based, most binary free initialized platforms in coreboot (Ivy Bridge and Sandy Bridge). This implies integrated [Measured boot improvements that landed in Coreboot 4.9+](https://github.com/osresearch/heads/issues/500), which sits atop of VBOOT and change Head's internal logic to take advantage differently of TPM PCRs to have proper and mainstreamed BIOS integrity measured. Technically, it means:
  - [ ] integrate/ test : 
    - [ ] [Add VBOOT support](https://review.coreboot.org/c/coreboot/+/29816) 
    - [ ] [Enable VBOOT support](https://review.coreboot.org/c/coreboot/+/29815/1)
  - [ ] check if fmap is correct
  - [ ] set CMOS flags correctly
- [ ] [Support TXT in Coreboot/Heads, so that [QubesOS Anti-Evil Maid](https://www.qubes-os.org/doc/anti-evil-maid/) can be used with coreboot Open Source Firmware](https://github.com/osresearch/heads/issues/502). **@marmarek said this was not considered a necessity. A discussion should be engaged between you two before going forward.**

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/540#issuecomment-519188762_

zaolin, 2019-08-21:
We will scope that out. Let me come back to you. Regarding Intel TXT @marmarek any input?

zaolin, 2019-08-21:
We need to discuss the security model with flash chip.

tlaurion, 2019-08-23:
@zaolin I'm ready!
Some links to share?

marmarek, 2019-08-23:
AEM in its current form is about boot integrity (xen.gz, vmlinuz, initramfs) only. It is doing that through TXT/DTRM. Heads provide very similar assurance with SRTM + a more convenient verification method (NitroKey with green/red LED or dynamic TOTP). We don't have other usage for TXT in Qubes right now. There are some wild ideas about doing re-attestation after S3, but I think it isn't even properly designed yet.

tlaurion, 2019-08-23:
@marmarek @zaolin
AEM official doc: https://github.com/QubesOS/qubes-antievilmaid

merge, 2019-08-28:
> We need to discuss the security model with flash chip.

Not sure if that's what you mean, but in case it is, this is my view: vboot "RO" and heads will lock the flash (even if we don't yet do that now. should be an easy addition).

zaolin, 2019-08-29:
@merge Yes but updates rely then on heads

merge, 2019-08-30:
> @merge Yes but updates rely then on heads

sure. just as they do already I guess.

tlaurion, 2019-08-30:
@zaolin @merge

Yes. The updates should come from Heads reproducible builds anyway, Nd are updatable from whiptail menus already, keeping modified config and public gpg key from cbfs exports and reinject them prior to flashing with flashrom, as it is now

tlaurion, 2019-08-30:
@zaolin the lockdown part, in heads, would be #326

tlaurion, 2019-10-14:
The general goal here is to broader the range of FSP free boards with neutralized Intel ME under Heads umbrella, while taking advantage of the measured boot support under coreboot VBOOT work that were merged under coreboot 4.9+ while not having been used under heads, which uses an old version of coreboot (4.8.1) since then.

General
- coreboot needs to be upstreamed to latest version under heads
 - As per past debates, it is problematic for Heads reproducible builds to use coreboot master moving target in Heads.
  - commit ID or coreboot release is needed. On which patches applied to coreboot can be added under Heads until next coreboot release.
- Under coreboot 4.8.1, [coreboot was patched to have measured boot working](https://github.com/osresearch/heads/blob/master/patches/coreboot-4.8.1/0000-measuredboot.patch) without VBOOT
 - Note that all other [patches](https://github.com/osresearch/heads/tree/master/patches/coreboot-4.8.1) were upstreamed to coreboot and are not needed in earlier coreboot versions.
- Heads [toolstack](https://github.com/osresearch/tpmtotp) (common for TPMTOTP and HOTP) will need to be modified to [take newer PCR into consideration](https://github.com/osresearch/heads/issues/500#issuecomment-477317163) 
- [FMAP will need to be rewritten/reconsidered](https://github.com/osresearch/heads/issues/562#issue-443299463)

coreboot 4.8.1 specifics
- X230 has an additional 5Mb that could be used by Heads if flashmap was defined correctly after [Intel ME properly neutered](https://libreboot.org/faq.html#intelme). If there was a way to generate FMAP consequently to available space in the 12MB rom, that would be amazing.
- [X220/T420 has a delay of 50 seconds booting heads](https://github.com/osresearch/heads/issues/541). [It seems to be a wait and timeout communicating with Intel ME](https://github.com/osresearch/heads/issues/541#issuecomment-482315521). The same applies to the T420 and all other boards, _**except the x230**_. Why?

Tickets needing to be updated/closed with the work being done here:
https://github.com/osresearch/heads/issues/41#issue-181472484 https://github.com/osresearch/heads/issues/500 https://github.com/osresearch/heads/issues/287 https://github.com/osresearch/heads/issues/562 https://github.com/osresearch/heads/issues/541 https://github.com/osresearch/heads/issues/150 https://github.com/osresearch/heads/issues/15

tlaurion, 2019-10-29:
"Also, keep in mind assembly is a lot more work for the t530, you have to solder a wire to the opposite chip your flashing every single time and so on to bypass flash protection (cs to vcc)"

tlaurion, 2020-07-06:
Work continuing under #562

closed 2022-02-05 by tlaurion