linuxboot/heads #608: X230 suspend

kmeaw, 2019-09-10:
I am having issue with Thinkpad X230 suspend when OS is booted using Heads.

When I try to suspend my laptop, running Linux 5.2.11-arch1-1-ARCH, it disables the backlight (as usual), but instead of suspending after a second enables the backlight again, delays for a few seconds, and then disables the backlight and finally suspends.

On resume the backlight does not enable immediately — I see a delay of a few seconds, then the backlight enables and I see the VT screen, but the CPU fan is running at 100% and the system is immensely slow ­— echoing a single character typed on a keyboard takes about a minute.

No such problem occurs on a vanilla coreboot 563b869 with SeaBIOS and Intel VBT.

kmeaw, 2019-09-10:
Update: the problem does not occur if I extract fallback/payload from the heads ROM file and inject it into vanilla coreboot's ROM, then run the heads payload through the SeaBIOS's boot menu.

merge, 2019-09-23:
yes. suspend and resume on the X230 is simply broken in coreboot 4.8, which is old, but still used in current builds of Heads. This will be fixed when we use coreboot 4.10 or later.

kmeaw, 2019-09-23:
Should I try rebuilding with coreboot 4.10 by just changing version (and hash) inside modules/coreboot ?

merge, 2019-09-23:
nope. it's quite some work to do that (dealing with the Heads-patches on top of coreboot). in fact there've been numerous issues/PRs starting to do that. one needs to take time, start over and finish that work.

tlaurion, 2019-09-24:
I do not see this behavior in QubesOS with 4.8.1 in my branch?

-- Sent from /e/ Mail

gordon-quad, 2019-12-23:
@merge This question might be a little bit out of scope of this issue, but Is there a place where I can check which patches are those and why they are there?
If I understand correctly heads is just linux kernel + initrd, what stops me from just compiling it and flashing into my existing 4.11 coreboot as payload?

tlaurion, 2019-12-24:
> @merge This question might be a little bit out of scope of this issue, but Is there a place where I can check which patches are those and why they are there?
> If I understand correctly heads is just [...]

@gordon-quad : The answer to your question can be found [here](https://github.com/osresearch/heads/tree/master/patches/linux-4.14.62)

If you have questions about this, please open another issue.

tlaurion, 2020-07-06:
Was working in same coreboot version before. Not sure what is happening [here](https://github.com/osresearch/heads/issues/770#issuecomment-654470134)

tlaurion, 2020-07-06:
> @merge This question might be a little bit out of scope of this issue, but Is there a place where I can check which patches are those and why they are there?
> If I understand correctly heads is just [...]

@gordon-quad : That would need coreboot patches or equivalent to take benefit of measured boot, which never landed on coreboot for Ivy/Sandy bridges, which #605 #721 #709 are about.

tlaurion, 2020-08-12:
https://github.com/osresearch/heads/pull/721#issuecomment-673115540

tlaurion, 2020-08-24:
@jans23

tlaurion, 2020-08-26:
@PatrickRudolph do you have a quote to fix this issue?

tlaurion, 2020-09-12:
Tested https://github.com/Thrilleratplay/coreboot-builder-scripts
with updated docker on QubesOS instruction https://gist.github.com/tlaurion/9113983bbdead492735c8438cd14d6cd

Result: no suspend resume problem.

```
user@coreboot-native:~/coreboot-builder-scripts/x230$ cat defconfig-4.12 
CONFIG_VENDOR_LENOVO=y
CONFIG_CBFS_SIZE=0x400000
CONFIG_BOARD_LENOVO_X230=y
CONFIG_GENERIC_LINEAR_FRAMEBUFFER=y
CONFIG_DRIVERS_PS2_KEYBOARD=y
CONFIG_DEFAULT_CONSOLE_LOGLEVEL_8=y
CONFIG_COREINFO_SECONDARY_PAYLOAD=y
CONFIG_MEMTEST_SECONDARY_PAYLOAD=y
CONFIG_NVRAMCUI_SECONDARY_PAYLOAD=y
CONFIG_TINT_SECONDARY_PAYLOAD=y
```

```
user@coreboot-native:~/coreboot-builder-scripts/x230$ cd ..
user@coreboot-native:~/coreboot-builder-scripts$ ls
build.sh  common  flash.sh  LICENSE  README.md  spec  t430  x220  x230
user@coreboot-native:~/coreboot-builder-scripts$ cd ..
user@coreboot-native:~$ ls
coreboot-builder-scripts  Desktop  Documents  Downloads  Music  Pictures  Public  Templates  Videos
user@coreboot-native:~$ git clone^C
(reverse-i-search)`git': git clone https://github.com/Thrilleratplay/coreboot-builder-scripts.^Ct
user@coreboot-native:~$ ls^C
user@coreboot-native:~$ ^C
user@coreboot-native:~$ ls
coreboot-builder-scripts  Desktop  Documents  Downloads  Music  Pictures  Public  Templates  Videos
user@coreboot-native:~$ cd coreboot-builder-scripts/
user@coreboot-native:~/coreboot-builder-scripts$ ./build.sh -t 4.12 x230
Fetching origin
remote: Enumerating objects: 467, done.
remote: Counting objects: 100% (467/467), done.
remote: Compressing objects: 100% (141/141), done.
Receiving objects: 100% (287/287), 39.03 KiB | 974.00 KiB/s, done.
remote: Total 287 (delta 236), reused 194 (delta 143), pack-reused 0
Resolving deltas: 100% (236/236), completed with 158 local objects.
From https://github.com/coreboot/coreboot
   3999aa6cdb..4b58d14fa7  master     -> origin/master
Fetching origin
Fetching origin
HEAD is now at f2c32515ee Documentation/releases: Update for 4.12
Skipping submodule '3rdparty/amd_blobs'
Skipping submodule '3rdparty/blobs'
Skipping submodule '3rdparty/fsp'
Skipping submodule '3rdparty/intel-microcode'
Submodule path '3rdparty/arm-trusted-firmware': checked out '76a08094c53faac3250b845dc12b4884f7e42d84'
remote: Enumerating objects: 126, done.
remote: Counting objects: 100% (126/126), done.
remote: Compressing objects: 100% (104/104), done.
remote: Total 126 (delta 50), reused 98 (delta 22), pack-reused 0
Receiving objects: 100% (126/126), 172.03 KiB | 449.00 KiB/s, done.
Resolving deltas: 100% (50/50), done.
From https://github.com/coreboot/chrome-ec
   4cb500f66..5df4dd14c  firmware-octopus-11297.B     -> origin/firmware-octopus-11297.B
   e26799893..2d6ed727c  firmware-zork-13434.B-master -> origin/firmware-zork-13434.B-master
   248a04dcf..100d6ae9e  master                       -> origin/master
Submodule path '3rdparty/chromeec': checked out '100d6ae9ec1f637f61b414a14c4c9299ce341e09'
Submodule path '3rdparty/libgfxinit': checked out '3318bf26803c77d41b18bef6d7ae4e051b97f9f2'
Submodule path '3rdparty/libhwbase': checked out 'a3edc6ef32d09c566b7ea3555bc753458ec63039'
Submodule path '3rdparty/opensbi': checked out 'c1c7c3ee9eb70e5c25914621f0363bfe244ebc79'
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 46 (delta 37), reused 46 (delta 37), pack-reused 0
Unpacking objects: 100% (46/46), done.
[... 935 lines cut ...]
payload_config                 0x74680    raw              1728 none
payload_revision               0x74d80    raw               236 none
img/tint                       0x74ec0    simple elf      48387 none
img/memtest                    0x80c00    simple elf      47417 none
etc/ps2-keyboard-spinup        0x8c580    raw                 8 none
(empty)                        0x8c5c0    null          3487704 none
bootblock                      0x3dfdc0   bootblock       65536 none

Built lenovo/x230 (ThinkPad X230)

	** WARNING **
coreboot has been built without an Intel Firmware Descriptor.
Never write a complete coreboot.rom without an IFD to your
board's flash chip! You can use flashrom's IFD or layout
parameters to flash only to the BIOS region.

4+0 records in
4+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0294928 s, 142 MB/s
```

```
user@coreboot-native:~/coreboot-builder-scripts$ cat x230/build/coreboot_lenovo-x230-complete.rom.sha256 
42b9909467100ace50760656356ad8f31d2d347df86ea732c8ff119f87bbdc57  /home/coreboot/cb_build/coreboot_lenovo-x230-complete.rom
```

tlaurion, 2020-09-19:
Successfull build per https://github.com/tlaurion/heads/commit/b1886f3bf2be0f2f8ed8dab19da5fec459a34d41
Direct link to ROM to reproduce and comment: https://430-103208611-gh.circle-artifacts.com/0/build/x230-hotp-verification/coreboot.rom

If it works, it means that the maximum space available in CBFS for BIOS is really 0x700000 (I remember I did the calculations somewhere under Heads....)

I'll try to test in next test batch @alex-nitrokey

tlaurion, 2020-09-19:
Explanation is [here](https://github.com/osresearch/heads/issues/667#issuecomment-638971582)

tlaurion, 2020-10-19:
fixed under https://github.com/osresearch/heads/commit/30852aa2739e71ed8c2c889cae91ef3995cfe75a

commit 30852aa2 (2020-10-19): xx30 boards: suspend/resume fix (#838)

* x230-hotp-verification: revert to coreboot "CONFIG_CBFS_SIZE=0x700000" by disabling board "CONFIG_DROPBEAR=y" and "CONFIG_LINUX_E1000E" to save space per @alex-nitrokey test under #770 to fix #608
* x230-htop-verification board: add clarifications on x230 board differences. Fixes #737 #770 #608
* X230 board & coreboot config: Fix x230 board, removing dropbear and e1000e driver.
* t430: board and coreboot config par with x230 to circumvent CBFS linked suspend/resume issues.
closed 2020-10-19 by tlaurion