linuxboot/heads #609: Cross compiler setup - C_INCLUDE_PATH etc

daym, 2019-09-14:
Hi,

we'd like to package heads in the GNU Guix distribution.

There is a slight problem we are working around: Our compilers use the (gcc) environment variables `CPATH`, `C_INCLUDE_PATH`, `CPLUS_INCLUDE_PATH`, `LIBRARY_PATH` and so on in order to specify the location of the system headers (the reason has to do with reproducible builds), see `gcc/incpath.c`. 

Heads builds its own cross compiler (via `musl-cross`) which we have already packaged. However, when Heads later uses that cross compiler, it doesn't unset the environment variables above (those are supported directly by gcc) and thus when using musl-cross's compiler we get a mix of cross and non-cross headers, which isn't good.

We can work around that on our side, but it would be nice if there was a generic mechanism to pass environment variables to a module. Right now, your `CROSS_TOOLS` make variable sometimes gets passed on as environment variable and sometimes as make variable - which isn't good. Also, we are not sure how to UNSET a variable in any case.

daym, 2019-09-14:
I suggest

```
--- a/Makefile.orig     2019-09-14 18:37:04.829332190 +0200
+++ a/Makefile  2019-09-14 18:40:08.090141253 +0200
@@ -348,7 +348,7 @@
                echo "$(MAKE) \
                        -C \"$(build)/$($1_dir)\" \
                        $($1_target)" ;  \
-               $(MAKE) \
+               $($1_build_environment) $(MAKE) \
                        -C "$(build)/$($1_dir)" \
                        $($1_target)  \
        ) \
```

tlaurion, 2020-04-17:
Just saw this. Really interesting.

tlaurion, 2020-04-17:
@jan23?

tlaurion, 2020-05-04:
@daym There is this [PR](https://github.com/osresearch/heads/pull/714) that should resolve your issue. Please give some feeback in your use case.

tlaurion, 2020-05-24:
#714 was merged. 

@daym:Please update if you still have issues, please tag me in.

daym, 2020-05-24:
Thanks! It works much better now.

There are still some weird manual things we have to do:

* busybox has a build failure because it still uses CPATH although I've patched it out using make_environment.  It might be using the wrong compiler (i.e. the host compiler), but I don't know for sure
* When I build BOARD=qemu-linuxboot, coreboot builds and eventually fails because it uses i386 musl cross, which we haven't packaged yet.

tlaurion, 2020-05-29:
@daym :
I've put your[ branch on my github](https://github.com/tlaurion/heads/tree/make-environment) for it to be picked up by CircleCI.
Error logs are out there if you click the "x":
```
make -C "/root/project/build/busybox-1.28.0" CC=/root/project/crossgcc/bin/x86_64-linux-musl-gcc -fdebug-prefix-map=/root/project=heads -gno-record-gcc-switches -D__MUSL__ -I/root/project/install/include -L/root/project/install/lib  CPP=/root/project/crossgcc/bin/x86_64-linux-musl-cpp CXXCPP=/root/project/crossgcc/bin/x86_64-linux-musl-cpp CC_FOR_BUILD= BUILD_CC= HOSTCC= AR=/root/project/crossgcc/bin/x86_64-linux-musl-ar LD=/root/project/crossgcc/bin/x86_64-linux-musl-ld STRIP=/root/project/crossgcc/bin/x86_64-linux-musl-strip NM=/root/project/crossgcc/bin/x86_64-linux-musl-nm OBJCOPY=/root/project/crossgcc/bin/x86_64-linux-musl-objcopy OBJDUMP=/root/project/crossgcc/bin/x86_64-linux-musl-objdump PKG_CONFIG_PATH=/root/project/install/lib/pkgconfig PKG_CONFIG_SYSROOT_DIR=/root/project/install    

make[1]: Entering directory '/root/project/build/busybox-1.28.0'

rm -f .kernelrelease

echo 1.28.0 > .kernelrelease

/root/project/build/busybox-1.28.0/scripts/gen_build_files.sh /root/project/build/busybox-1.28.0 /root/project/build/busybox-1.28.0

make -f scripts/Makefile.build obj=scripts/basic

   -Wp,-MD,scripts/basic/.fixdep.d         -o scripts/basic/fixdep scripts/basic/fixdep.c  

/bin/sh: 1: -Wp,-MD,scripts/basic/.fixdep.d: not found

make[2]: *** [scripts/Makefile.host:104: scripts/basic/fixdep] Error 127

make[1]: *** [Makefile:357: scripts_basic] Error 2

make[1]: Leaving directory '/root/project/build/busybox-1.28.0'

make: *** [Makefile:387: /root/project/build/busybox-1.28.0/.build] Error 1

```

Unfortunately, I cannot merge a build system change if it breaks the current system for another use. If you can help troubleshooting to changes can make it work for both use cases.

tlaurion, 2024-05-10:
I think this is done, where nix is used now as #1661 is merged

closed 2024-05-10 by tlaurion