linuxboot/heads #614: x230 FHD mod not supported

n4ru, 2019-10-23:
Seems the latest version of heads does not support the nitrocaster or KK FHD mod.

Wondering if there's a workaround? The "FHD patch" being reviewed over at coreboot IIRC only works on coreboot 4.9+, and heads is still on 4.8.1.

tlaurion, 2020-03-09:
@n4ru: Would need a x230-fhd specific board, coreboot and probably a linux config?

n4ru, 2020-04-02:
I got it to work by simply taking the vbt from the FHD patch in coreboot, and swapped it with the stock vbt file.

arl4223, 2020-11-11:
n4ru are you able to help we with some more detailed instructions? I have a x230 with the nitrocaster FHD mod and core boot running. But now I would like to switch to heads with a Nitrokey.

tlaurion, 2020-11-13:
@arl4223 
1- what is your working coreboot config? look at referenced blobs in that coreboot config (not sure we can host vbt on github)
2- modify x230 board, coreboot config and linux config under heads.
3- when working, propose a PR for x230-fhd board!

@PatrickRudolph's #709 is still not mergeable. I can't wrap my head around it for the moment to upgrade all (coreboot still supported) boards to 4.12. Help welcome.

tslilc, 2020-11-16:
I have back-ported (really they basically all work verbatim) the patches in the Coreboot merge and created heads-style patches, but they won't do much (anything?) without the VBT blob, and heads does not distribute such blobs. I did not need to use a newer coreboot in the process.

In particular i forewent the external voltage source, bridged J1, and set up early power-on and so on to save a few joules. This all seems to work perfectly with heads, and i've been using it for some time.

I would be happy to share these later when i return home, but again, it depends on the VBT blob.

computer-user123, 2020-12-27:
I am trying my best to follow these instructions, but I am having troubles building coreboot with the x230-FHD mod,
I keep getting errors building.  My skill level is not very high, and I don't know where to begin to solve them.

Any chance one of you kind folks could upload the x230-FHD  "VBT blob" ? (perhaps rapidshare, or something similar?)
I am confident that if I get this file, I can (hopefully) follow tslilc's comments (the one right before mine)

I had heads+qubes working on x230, with a clean qubes install, but after installing the nitrocaster mod, and 1080p lcd, I can no longer boot into OS.   Some live distro's seem to boot ok, booting from usb, but I am having issues booting from HDD.

Thankfully, I was only beginning to learn about qubes, and didn't have anything valuable there that was not backed up!!

Any help for this n00b is greatly appreciated.
Thank you for your time!

Thrilleratplay, 2020-12-27:
@computer-user123 The VBT is available in the [coreboot 230 FHD patch](https://review.coreboot.org/c/coreboot/+/28950).  Although from what I have been told, the VBT does not impact how operating systems display.  I have a nitrocaster mod in my x230 but it runs coreboot, not Heads, and am using coreboot without the patch.  Your issue could be related to video settings in the kernel params but I honestly have no idea.

computer-user123, 2020-12-27:
Thank you Thrilleratplay, 
I did find the VBT file, and I have edited some of the configs, in hopes of building heads properly with the patch,

I am now stuck at building heads.  I can't remember what system I was using, I have tried fresh ubuntu 16.04 / 18.04 / 20.04 VM's without any success.   

I do remember building heads, and successfully flashing with a RPi3 and the chip clip.  I built heads on the x230 machine with ubuntu 18.04 install on the HDD , IIRC.

Here are the errors I am getting while attempting to "make BOARD=x230"
https://pastebin.com/mRs1rYLw

Any help is GREATLY appreciated.  I am now stuck with a machine that has display issues on every boot. :(

tlaurion, 2020-12-27:
You are missing host build tools. From the logs, it seems that you do not have gnat?

computer-user123, 2020-12-27:
@tlaurion , Just installed gnat, it gives the exact same errors.

tlaurion, maybe you could kindly recommend a distro (exact .iso, kernel/version) that you know will 100% work?

Thank you guys, for the quick replies, and the help.

tlaurion, 2020-12-27:
Instructions for building are here, pointing to circleci recipe used to build boards with host dependencies, which builds with Debian-10

http://osresearch.net/Building/

Yes. Documentation needs love.

computer-user123, 2021-01-29:
> I have back-ported (really they basically all work verbatim) the patches in the Coreboot merge and created heads-style patches, but they won't do much (anything?) without the VBT blob, and heads does not distribute such blobs. I did not need to use a newer coreboot in the process.
> 
> In particular i forewent the external voltage source, bridged J1, and set up early power-on and so on to save a few joules. This all seems to work perfectly with heads, and i've been using it for some time.
> 
> I would be happy to share these later when i return home, but again, it depends on the VBT blob.
> 
> EDIT: Here they are, @arl4223 and @tlaurion.
> 
> To the `boards/x230-$variant/x230-$variant.config` file add
> 
> ```
> CONFIG_INTEL_GMA_ADD_VBT_DATA_FILE=y
> CONFIG_INTEL_GMA_VBT_FILE="src/mainboard/$(MAINBOARDDIR)/data.vbt"
> ```
> 
> and put the patched VBT in that location. The other two patches are for matching the modified VBT and for using VCC3P respectively,
> [0062-src-x230-fhd.txt](https://github.com/osresearch/heads/files/5550797/0062-src-x230-fhd.txt)
> [0063-src-x230-fhd-vcc3p.txt](https://github.com/osresearch/heads/files/5550796/0063-src-x230-fhd-vcc3p.txt)
> and they go into the `patches/coreboot-4.8.1` directory.
> 
> To be clear, the modified VBT disconnects the internal LVDS interface which is useful for energy saving and sparing headaches later with OSs that are confused by both displays existing (especially for brightness control purposes). It also allows, as i understand it, ``bios'' graphics to function -- but this point is a little moot as heads boots Linux anyway.

I have tried compiling heads (spec. x230-flash) and (x230) with the changes you mention, and I haven't been able to succeed.  

nothing I have tried will allow internal-edp to display.  I can tell the bios is working, (with 2nd 8mb chip , ME_CLEANED , and 4mb x230-flash.bin flashed.)

I have tried 20x to compile, using "virgin" heads directories, and then applying changes and trying to get a different .bin from the changes. 

Can you please go into more specifics?

so far,  have a "virgin" heads directory, lets say "~/heads/"   ,

To the boards/x230-$variant/x230-$variant.config file add

CONFIG_INTEL_GMA_ADD_VBT_DATA_FILE=y
CONFIG_INTEL_GMA_VBT_FILE="src/mainboard/$(MAINBOARDDIR)/data.vbt"

and put the patched VBT in that location. The other two patches are for matching the modified VBT and for using VCC3P respectively,
0062-src-x230-fhd.txt
0063-src-x230-fhd-vcc3p.txt
and they go into the patches/coreboot-4.8.1 directory.

I do this, and re-compile, with all needed software, and successfully generate a "heads-x230-flash-v0.2.0-1010-g22396fb-dirty-top.rom"  in build/x230-flash ,   and heads-x230-v0.2.0-1010-g22396fb-dirty.rom in build/x230/  

I have tried 10x++  calculating the sha256sum  , and keep thinking that this time I compile, will be the one that works!!!!

computer-user123, 2021-01-29:
I am literate enough to follow directions, and provide needed logs/build files,  but I have tried many times to replicate an x230-flash or x230.(full.bin) , through heads, that works with nitrocaster x230-FHD mod ,

meanwhile, I have a working x230-FHD nitrocaster setup, but with coreboot alone, not heads.

computer-user123, 2021-01-29:
worse case scenario,  (obviously, this goes against the whole concept of heads)  , can someone who has 230-flash (w/fhd) working , please DM me or PM me the top.bin file ?    I have literally tried 20+ times in the last few weeks, and whatever i try, it will "work on the x230"  , but the nitro-caster-enabled display is dead.....    

I have not tried 2ndary VGA output or anything, but the 1080p panel, just wont light up and work, after compiling 20 diff times.
It works with coreboot, and works with orig bios,   im astounded.

computer-user123, 2021-01-29:
Thank you guys in advance,    I know you're gods, and I'm meerely a peasant who wants x230-heads-fhd to work, so this meager peasant can verify his /boot, and know that "ME" isnt watching.....    !

tlaurion, 2021-01-29:
@tonux I remeber that you have such kit right? What is missing here?

tlaurion, 2021-01-29:
Basically just upgrading coreboot version and applying op patch?

tlaurion, 2021-01-29:
Si basically doing a coreboot 4.13 based board config name appened with  fhd and we close this ticket?

computer-user123, 2021-01-29:
@tlaurion ,
Any advice on how to proceed?

I will do anything to get this working,
What steps can I take?

thanks

tslilc, 2021-01-29:
> I am literate enough to follow directions, and provide needed logs/build files, but I have tried many times to replicate an x230-flash or x230.(full.bin) , through heads, that works with nitrocaster [...]

I'm happy to talk through this with you and try to understand what's not working. How would you like to do that?

As far as i can tell, unless you're not building against coreboot-4.8.1, those patches should work without a problem. But even without any of these patches, heads is linux, and so any version of heads should power the eDP lines if the hardware is working. These patches are merely there to disconnect the LVDS in the `bios'.

computer-user123, 2021-01-29:
exact steps followed.

1.start with a fresh (virgin) heads directory (has never been used to build anything)
2.copy 2 patch files "0062-src-src-x230-fhd-.txt" & "0063-src-x230-fhd-vcc3p.txt" to ~/heads/patches/coreboot-4.8.1/  and rename file ends to ".patch"  from ".txt" to match the other files found in the folder.

3. download tarball of https://review.coreboot.org/c/coreboot/+/28950  (source of data.vbt)
-rename data.vbt to fhd-data.vbt and paste into ~/heads/build/coreboot-4.8.1/src/  
**notes: I see the "CONFIG_INTEL_GMA_VBT_FILE="src/mainboard/$(MAINBOARDDIR)/fhd-data.vbt" line you wrote,
I was not sure where the variable $(MAINBOARDDIR) points to,  I would assume that it "auto-fills"  into 
"~/heads/build/coreboot-4.8.1/src/mainboard/lenovo/x230/"
I wanted to avoid getting "lost" with variables.

4. add 2 lines (at top of existing parameters) to 2 files ~/heads/config/coreboot-x230.config & coreboot-x230-flash.config
CONFIG_INTEL_GMA_ADD_VBT_DATA_FILE=y
CONFIG_INTEL_GMA_VBT_FILE="src/fhd-data.vbt"

It is my understanding, that heads for x230 utilizes 3 binaries.
x230-flash.bin (4MB) / coreboot.bin (12MB) / and your individual machine's "me-cleaned-bottom.bin" (8MB)
x230-flash.bin is the top 4mb chip, and prepares the machine to load 12mb "coreboot.bin" (the product of make BOARD=x230),

From what I understand, the bottom 8mb chip is first flashed with the clean-me-bottom.bin,  then flash x230-flash.bin on top 4mb chip,
then heads recovery shell or command prompt should open, then at that time, you can plug in a fat32 usb drive, and flash the full 12mb coreboot.bin/coreboot.rom
I think that heads will selectively overwrite the bottom 8mb chip, while keeping the me-cleaned firmware portion, (and possibly ifd/me/gbe regions??? )

I originally built heads x230 and x230-flash properly, around 1 year ago or more.  I was successfull and it worked fine,
but then I got the nitro-caster "daughter" board soldered in, and upgraded to a quality 12.5" 1080p IPS display.
After that, I was able to use it somewhat, but certain linux distro wouldnt boot, the secondary display, or whatever caused a lot of problems.

Since I could no longer reliably use heads, I went back to just straight coreboot x230-fhd, and was successful.
I would like to go back to heads, (with FHD support) to make use of my yubi security token.

I am going to flash these files, just freshly built following the steps outlined in this post.
783979ac5670d02b7fcac9b9e194be17981a2f19351430ab572d6764cc4680ad  build/x230/heads-x230-v0.2.0-1010-g22396fb-dirty.rom
0ab8c4e913a78ed0ff27d426bb50a13de24d0d4b0f1d3f01d562c7e0595f4ff2  build/x230-flash/heads-x230-flash-v0.2.0-1010-g22396fb-dirty-top.rom

I am confident that the system is actually booting, I am not getting error beeps, and the think-light and KB backlights work, 
I unfortunately don't have a VGA cable handy to try to troubleshoot.  I am sure the bin's I keep producing are working, just the display remains blank.
It lights up, but doesnt show anything, just blank.

I will report back, and hopefully she works this time! :)
Thank you VERY much for the assistance... 

-- Did not work.  1080p screen on EDP lights up, but nothing is displayed.

computer-user123, 2021-01-30:
I just built again,
from virgin directory, new heads never built with.
1.copy 2 patch files as is, to heads/patches/coreboot-4.8.1/
2. add 
CONFIG_INTEL_GMA_ADD_VBT_DATA_FILE=y
CONFIG_INTEL_GMA_VBT_FILE="src/mainboard/$(MAINBOARDDIR)/fhd-data.vbt"
to config/coreboot-x230-$variant.config
3. rename data.vbt (from https://review.coreboot.org/c/coreboot/+/28950) to fhd-data.vbt  
4. copy fhd-data.vbt to /heads/build/coreboot-4.8.1/src/mainboard/lenovo/x230/ 

I have tried this just now, and its not working.  
I tried to take the top.bin from the coreboot.rom 12mb file (output of make x230) using the command
dd if=coreboot.rom of=top.rom bs=1M skip=8  
this was not successful.

^^^^ should all the other smaller files be copied from the tarbar to this directory?

the sha256 of the files built are:
765f0555ff5d5e85859ec3df76982afe243a452e569b6ae233928fb540d26eb6  top.bin
699559798da07ac0abad0d8a5976af976e6355553ac2cac5768096120f471756  coreboot.rom

I really appreciate all the help.  Thank you guys.
I have been banging my head at the wall, and have literally flashed 20x , and it still wont show anything on the screen.
Anything helps.

tlaurion, 2021-01-30:
OP specifies that coreboot 4.9+ is required. There is a PR for the x230 over coreboot 4.13 for the x230-maximized board. That PR should be useful integrate the changes here and tested with the mod.

Then that having be proved successful,  create a new board config.

The blobs cannot be hosted here, but downloaded from coreboot got repo from a static commit and measured is a blob download script, needed to be called prior ofbuilding that board so the blob is in path.

The parrot board PR had implemented such logic to download ME from coreboot blobs repo. I invite you to do the same to automate the process and make this board available to others.

tlaurion, 2021-01-31:
Let's see if I can help here...

4.13's PR #949.

This issue points to https://github.com/osresearch/heads/issues/614#issuecomment-751489575 coreboot review. 
Patch is here: https://review.coreboot.org/changes/coreboot~28950/revisions/12/patch?zip

So instead of explaining, let's do it.

```
cd heads
git remote add tlaurion-github https://github.com/tlaurion/heads
git checkout tlaurion-github/x230_coreboot-4_13
git reset --hard
wget https://review.coreboot.org/changes/coreboot~28950/revisions/12/patch?zip -O x230-fhd.zip
unzip x230-fhd.zip 
mv 0ed94cb.diff patches/coreboot-4.13/0002-x230-fhd-variant.patch
rm build/coreboot-4.13/ -rf
make BOARD=x230-maximized-fhd CPUS=2
(ctrl-c after coreboot extraction)
cd build/coreboot-4.13/
cp ../../config/coreboot-x230-maximized.config .config
make menuconfig
(change board to x230-fhd, exit and save (under .config))
make savedefconfig
(reduce changeset to minimal against defconfig for that board)
mv defconfig ../../config/coreboot-x230-maximized-fhd.config
cp -r boards/x230-maximized/ boards/x230-maximized-fhd/
mv boards/x230-maximized-fhd/x230-maximized.config boards/x230-maximized-fhd/x230-maximized-fhd.config 
vim boards/x230-maximized-fhd/x230-maximized-fhd.config
(Change description of board)
vim .circleci/config.yaml
(duplicate similar CI section, modify statements to refelct new board name, save)
git checkout -b x230-maximized-fhd
git add .circleci/config.yaml
git add boards/x230-maximized-fhd/x230-maximized-fhd.config 
git add config/coreboot-x230-maximized-fhd.config 
git add patches/coreboot-4.13/0002-x230-fhd-variant.patch 
git commit -m "x230-maximized-fhd: add fhd mod into coreboot x320"
git push  tlaurion-github
```

Discussion should continue under PR that will refer here as the result of the above.

tlaurion, 2021-02-09:
@n4ru @computer-user123 @tslilc : #967 was tested successfull by @Tonux599 https://github.com/osresearch/heads/pull/967#issuecomment-775634408

Please report back here your stories (you can click green checkmarks of latest commit of the PR to go to CircleCI's build, go to artifacts, and download top and bottom roms to externally reflash).

The next commit will be to include this board into normal CircleCI builds if reported as functional here.

closed 2023-02-09 by tlaurion