linuxboot/heads #616: Support models without FSP and Intel ME neutered + deactivated

tlaurion, 2019-10-24:
Here is the list of additional ThinkPads that could be supported in coreboot 4.11 with measured boot on top of VBOOT, thanks to[ NlNet grant and 9elements](#605).

Vote in comments please, in this github issue.

For the moment:
X230, X220, T530 are in scope.

Sandy/Ivy bridge Lenovo laptops with 12MiB flash (4 + 8):
    * T530
    * W530
    * X230
    * T430
    * X131e
    * X1 carbon gen1
    * S230u

Sandy/Ivy bridge Lenovo laptops with 8MiB flash:
    * T520
    * W520
    * X220
    * X1

Sandy/Ivy bridge Lenovo laptops with 16MiB flash:
    * T430s
    * T431s

SebastianMcMillan, 2019-10-24:
Anything about the T420?

tlaurion, 2019-10-24:
It's a 8+4?

tlaurion, 2019-10-24:
@SebastianMcMillan : How is the chassis and general quality of te model?

SebastianMcMillan, 2019-10-24:
I'd argue that the chassis and general quality of the T420 is just as good, if not better, than the X220/X230.

siro20, 2019-10-25:
The T440p makes of the non public MRC.bin for raminit. You have to extract it from chromeos coreboot images.

tlaurion, 2019-10-29:
> I'd argue that the chassis and general quality of the T420 is just as good, if not better, than the X220/X230.
> 
> And yes, it is listed under secondary importance in #605.
> 
> Edit: it may be too early to tell, but will T440p be considered as well?

Put in first importance under #605.

tlaurion, 2019-10-29:
"Also, keep in mind assembly is a lot more work for the t530, you have to solder a wire to the opposite chip your flashing every single time and so on to bypass flash protection (cs to vcc)"

seitokaichou, 2019-11-08:
Voting for the T430.

techge, 2019-11-12:
Voting for the x220 (though I am not sure if necessary as seemingly already in scope?). Would help testing as soon as #617 is fixed...

pgera, 2019-11-19:
Pointing out some nice properties in favour of t530:
- Native wide gamut full hd display 
- Quad core CPUs up to i7-3940xm
- 16 GB RAM
- Extra slice battery possible

This makes it the most powerful free-ish laptop we have today. RAM can be 32 GB with w530, but that comes with the nvidia card which has other issues.

That-Random-Guy, 2019-11-24:
Sorry to bother... a lot of this is way over my head. After hearing about Purism and their coreboot/heads implementation, I was wondering if this thing would ever work on laptops like the Dell XPS? Are there plans to make this work on any laptop or is that not how this works? My intent is to run Linux on an XPS and use Purism's Librem Key to enable the tamper protection feature like how Purism sets it up on their laptops.

seitokaichou, 2019-11-24:
@That-Random-Guy no, only a subset of the devices supported by coreboot.

tlaurion, 2020-08-01:
#709 #721 related.

closed 2020-08-01 by tlaurion