linuxboot/heads #617: musl-cross-git patches do not apply successfully

sellerie98, 2019-10-25:
I just tried to build HEADS for the x230 (non-flash), but it fails trying to apply the HEADS patches to the HEAD of musl-cross-git. I tried building it on Fedora 30 on x86_64.

Log:
```
make[1]: Verzeichnis „/run/media/sellerie/Storage/git/heads/build/coreboot-4.8.1“ wird verlassen
#echo '******* Building crossgcc-arm (this might take a while) ******'
#make -C "/run/media/sellerie/Storage/git/heads/build/coreboot-4.8.1" crossgcc-arm
2019-10-25 22:16:20+02:00 CONFIG coreboot
git clone https://github.com/GregorR/musl-cross "/run/media/sellerie/Storage/git/heads/build/musl-cross-git"
Klone nach '/run/media/sellerie/Storage/git/heads/build/musl-cross-git' ...
remote: Enumerating objects: 73, done.
remote: Counting objects: 100% (73/73), done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 869 (delta 25), reused 68 (delta 23), pack-reused 796
Empfange Objekte: 100% (869/869), 323.01 KiB | 1.10 MiB/s, Fertig.
Löse Unterschiede auf: 100% (524/524), Fertig.
cd /run/media/sellerie/Storage/git/heads/build/musl-cross-git && git submodule update --init --checkout
if [ -r patches/musl-cross.patch ]; then ( cd /run/media/sellerie/Storage/git/heads/build/musl-cross-git ; patch -p1 ) < patches/musl-cross.patch || exit 1 ; fi
patching file config.sh
Hunk #2 FAILED at 22.
1 out of 2 hunks FAILED -- saving rejects to file config.sh.rej
```

himu007, 2019-10-25:
Got the same error on Ubuntu 18.04
```
make[2]: Leaving directory '/home/user/heads/build/coreboot-4.8.1'
#echo '******* Building crossgcc-arm (this might take a while) ******'
#/home/user/heads/build/make-4.2.1/make -C "/home/user/heads/build/coreboot-4.8.1" crossgcc-arm
2019-10-25 22:49:56+01:00 CONFIG coreboot
git clone https://github.com/GregorR/musl-cross "/home/user/heads/build/musl-cross-git"
Cloning into '/home/user/heads/build/musl-cross-git'...
remote: Enumerating objects: 73, done.
remote: Counting objects: 100% (73/73), done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 869 (delta 25), reused 68 (delta 23), pack-reused 796
Receiving objects: 100% (869/869), 323.01 KiB | 1.13 MiB/s, done.
Resolving deltas: 100% (524/524), done.
cd /home/user/heads/build/musl-cross-git && git submodule update --init --checkout
if [ -r patches/musl-cross.patch ]; then ( cd /home/user/heads/build/musl-cross-git ; patch -p1 ) < patches/musl-cross.patch || exit 1 ; fi
patching file config.sh
Hunk #2 FAILED at 22.
1 out of 2 hunks FAILED -- saving rejects to file config.sh.rej
make[1]: *** [Makefile:379: /home/user/heads/build/musl-cross-git/.canary] Error 1
make[1]: Leaving directory '/home/user/heads'
Makefile:584: recipe for target 'all' failed
make: *** [all] Error 2
```

osresearch, 2019-10-29:
The musl-cross-git dependency should be a specific tag, rather than checking out head.  Looks like they've added some new code to `config.sh` that is breaking the build (and also potentially making things non reproducible).

osresearch, 2019-10-29:
Looks like even with our updated patch `musl-cross` still fails to build due to missing hashes in their tree. I opened an issue https://github.com/GregorR/musl-cross/issues/75

I'm also working on an updated patch that will make it easier to pin to specific git commits.

tlaurion, 2019-11-28:
--strip 1
is resulting in microcode updates not being found in path by coreboot:
```
==> ./build/log/coreboot.log <==
    CC         ramstage/arch/x86/wakeup.o
    CC         ramstage/acpi/sata.o
    CP         ramstage/arch/x86/memlayout.ld
    CP         ramstage/lib/program.ld
    CC         generated/ramstage.o
    CC         cbfs/fallback/ramstage.debug
Microcode error: 3rdparty/blobs/cpu/intel/model_206ax/microcode.bin does not exist
Microcode error: 3rdparty/blobs/cpu/intel/model_306ax/microcode.bin does not exist
make[1]: *** [src/cpu/Makefile.inc:40: x230-libremkey/cpu_microcode_blob.bin] Error 1
make[1]: Leaving directory '/heads/build/coreboot-4.8.1'

Questioning the policy of pushing microcode updates in coreboot since we are not in a rolling release anyway. Removing the dependency fixes the problem.  [Playing here](https://gitlab.com/tlaurion/heads/compare/heads-ci...PrivacyBeast_X230_QubesOS_Certified-musl-fix)

```

tlaurion, 2019-12-08:
Fixed with https://github.com/osresearch/heads/commit/8af849cadc684259865123195b86317ecfa53528

tlaurion, 2019-12-10:
The patches apply successfully. 

Resulting binaries crash dump. (Eg. lvm in my tests).

@osresearch proposed to base Heads on https://github.com/richfelker/musl-cross-make/ instead.
@rofl0r ?

osresearch, 2020-01-07:
This seems to be working now in head after the tag pinning merge. The musl-cross-make suggestion is now in #650 .

closed 2020-01-07 by osresearch