linuxboot/heads #619: Makefile points to wrong musl gcc binary

sellerie98, 2019-11-01:
The Makefile points to the wrong gcc binary:

```
make[1]: Entering directory '/media/Storage/git/heads/build/linux-4.14.62/linux-x230'
../scripts/gcc-version.sh: line 26: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: No such file or directory
../scripts/gcc-version.sh: line 27: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: No such file or directory
../scripts/gcc-version.sh: line 29: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: No such file or directory
../scripts/gcc-version.sh: line 26: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: No such file or directory
../scripts/gcc-version.sh: line 27: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: No such file or directory
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
  CHK     include/config/kernel.release
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
  Using .. as source for kernel
make[3]: /media/Storage/git/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make[3]: *** [/media/Storage/git/heads/build/linux-4.14.62/Makefile:1110: prepare-compiler-check] Error 1
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:24: __sub-make] Error 2
make[1]: Leaving directory '/media/Storage/git/heads/build/linux-4.14.62/linux-x230'
make: *** [Makefile:378: /media/Storage/git/heads/build/linux-4.14.62/linux-x230/.build] Error 1
```

tlaurion, 2019-11-01:
@osresearch

osresearch, 2019-11-02:
It's fixed in https://github.com/osresearch/heads/commit/e5038e6adf9133a51c80b9699e552a68167db63b

That could be cherry picked and applied to master

tlaurion, 2019-12-06:
@sellerie98  should be fixed with latest commit 8af849c

Please confirm and close ticket

tlaurion, 2020-03-09:
@sellerie98 please reopen if still problematic

closed 2020-03-09 by tlaurion