linuxboot/heads #621: Help Thinkpad Roms

alelopezperez, 2019-11-08:
Can someone give me a builded rom for thinkpad T400 and T420.
I've been trying to build heads and I cant; each probblem i solve just leaves another one.

I was trying to find a forum to post this but this is the only means of communication to ask about this.

tlaurion, 2019-11-08:
T400 is not supported for the moment.

T420 has a pull request but laptop will hang for 50 seconds at each boot and needs development. You can ask for ROM to the developer which created the pull request. Search for issue on the t420 to see details.

The build system has a problem with newer musl, which has its seperate issue which needs resolution to be able to build ROMs.

-- Sent from /e/ Mail

alelopezperez, 2019-11-08:
with which thinkpad i'll ge the best performance??

techge, 2019-11-12:
what do you mean by performance? x230 is probably best supported regarding heads

closed 2019-11-14 by alelopezperez