linuxboot/heads #630: Heads fails to build at linux stage (unable to initialize decompress status for section .debug_info)

EDmitry, 2019-11-24:
I am using Debian 10 running in Qubes, fails same way in Fedora 30. I am using master branch.

-----
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    ../scripts/checksyscalls.sh
  DESCEND  objtool
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/compile.h
  LD [M]  crypto/rsa_generic.o
/home/user/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-ld: crypto/rsa.o: unable to initialize decompress status for section .debug_info
/home/user/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-ld: crypto/rsa.o: unable to initialize decompress status for section .debug_info
/home/user/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-ld: crypto/rsa.o: unable to initialize decompress status for section .debug_info
/home/user/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-ld: crypto/rsa.o: unable to initialize decompress status for section .debug_info
crypto/rsa.o: file not recognized: File format not recognized
make[4]: *** [../scripts/Makefile.build:574: crypto/rsa_generic.o] Error 1
make[3]: *** [/home/user/heads/build/linux-4.14.62/Makefile:1035: crypto] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:24: __sub-make] Error 2
make[1]: Leaving directory '/home/user/heads/build/linux-4.14.62/linux-qemu'
make: *** [Makefile:378: /home/user/heads/build/linux-4.14.62/linux-qemu/.build] Error 1

EDmitry, 2019-11-24:
I am using libelf-dev-175. Do I need to somehow downgrade that to be compatible with binutils that heads is using?

bertcubley, 2019-11-29:
I'm having the same issue on Debian 10.

EDmitry, 2019-12-02:
I had to use Debian 9 system with libelf-dev downgraded to 174. Then it built just fine.

tlaurion, 2019-12-06:
@EDmitry @bertcubley should be fixed with latest commit https://github.com/osresearch/heads/commit/8af849cadc684259865123195b86317ecfa53528

Please confirm and close ticket

EDmitry, 2019-12-28:
That fixed it for me. Thank you!

closed 2019-12-28 by EDmitry