linuxboot/heads #631: Having trouble building Heads because bad fetch urls

fjUSoF3CCTtX, 2019-11-24:
Using this as the build guide `http://osresearch.net/Building.html`, which says to `git clone https://github.com/osresearch/heads` on a x230 running Debian 10.

The make process is failing to complete because many of the links in the build files don't work. I have edited the files to use functioning urls based on the log outputs when the make process fails. For example, I had to replace `GMP_ARCHIVE="https://ftpmirror.gnu.org/gmp/gmp-${GMP_Version}.tar.xz"` with `GMP_ARCHIVE="https://ftp.gnu.org/gnu/gmp/gmp-${GMP_Version}.tar.xz` in `heads/build/coreboot-4.8.1/util/crossgcc/buildgcc`. But, I have stalled out with `heads/build/musl-cross-git/defs.sh`. The fetch for `http://git.savannah.gnu.org/gitweb/?=config.get;a=blob_plain;f=` doesn't work because `http://git.savannah.gnu.org` appears to be non-fuctional. What a potential replacement url might be is not transparent to me, so I am stuck for the moment.

Is there something I am missing? Or is there a more up-to-date set of files for building Heads?

tlaurion, 2019-12-06:
@EDmitry @bertcubley should be fixed with latest commit https://github.com/osresearch/heads/commit/8af849cadc684259865123195b86317ecfa53528

Please confirm and close ticket

osresearch, 2020-01-09:
coreboot's cross compiler is no longer used as of #653, musl-cross-make has updated URLs and verifies the hashes as well.

closed 2020-01-09 by osresearch