linuxboot/heads #636: make fails with error: implicit declaration of function 'tlcl_measure'

AdmerStroh, 2019-12-03:
```
make
2019-12-03 18:37:19+01:00 CONFIG coreboot
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/net/ethernet/intel/e1000/e1000.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/ata/libata.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/ata/ahci.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/ata/libahci.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/usb/host/ehci-hcd.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/usb/host/ehci-pci.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/usb/host/xhci-hcd.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/usb/host/xhci-pci.ko
2019-12-03 18:37:19+01:00 INSTALL-MODULE drivers/usb/storage/usb-storage.ko
2019-12-03 18:37:19+01:00 SYMLINK bin/busybox
2019-12-03 18:37:22+01:00 INSTALL-BIN build/kexec-tools-2.0.16/build/sbin/kexec
2019-12-03 18:37:23+01:00 INSTALL-BIN build/tpmtotp-git/totp
2019-12-03 18:37:23+01:00 INSTALL-BIN build/tpmtotp-git/hotp
2019-12-03 18:37:23+01:00 INSTALL-BIN build/tpmtotp-git/base32
2019-12-03 18:37:23+01:00 INSTALL-BIN build/tpmtotp-git/qrenc
2019-12-03 18:37:23+01:00 INSTALL-BIN build/tpmtotp-git/util/tpm
2019-12-03 18:37:23+01:00 INSTALL-BIN build/pciutils-3.5.4/lspci
2019-12-03 18:37:23+01:00 INSTALL-BIN build/flashrom-1.0/flashrom
2019-12-03 18:37:23+01:00 INSTALL-BIN build/cryptsetup-1.7.3/src/.libs/cryptsetup
2019-12-03 18:37:23+01:00 INSTALL-BIN build/cryptsetup-1.7.3/src/.libs/cryptsetup-reencrypt
2019-12-03 18:37:23+01:00 INSTALL-BIN build/cryptsetup-1.7.3/src/.libs/veritysetup
2019-12-03 18:37:23+01:00 INSTALL-BIN build/gnupg-2.2.10/g10/gpg
2019-12-03 18:37:23+01:00 INSTALL-BIN build/gnupg-2.2.10/agent/gpg-agent
2019-12-03 18:37:23+01:00 INSTALL-BIN build/gnupg-2.2.10/scd/scdaemon
2019-12-03 18:37:24+01:00 INSTALL-BIN build/pinentry-1.1.0/tty/pinentry-tty
2019-12-03 18:37:24+01:00 INSTALL-BIN build/LVM2.2.02.168/tools/dmsetup
2019-12-03 18:37:24+01:00 INSTALL-BIN build/LVM2.2.02.168/tools/lvm
2019-12-03 18:37:24+01:00 INSTALL-BIN build/dropbear-2016.74/ssh
2019-12-03 18:37:24+01:00 INSTALL-BIN build/dropbear-2016.74/scp
2019-12-03 18:37:24+01:00 INSTALL-BIN build/dropbear-2016.74/dropbear
2019-12-03 18:37:24+01:00 INSTALL-BIN build/flashtools-0.0.1/flashtool
2019-12-03 18:37:24+01:00 INSTALL-BIN build/flashtools-0.0.1/peek
2019-12-03 18:37:24+01:00 INSTALL-BIN build/flashtools-0.0.1/poke
2019-12-03 18:37:24+01:00 INSTALL-BIN build/flashtools-0.0.1/cbfs
2019-12-03 18:37:24+01:00 INSTALL-BIN build/flashtools-0.0.1/uefi
2019-12-03 18:37:24+01:00 INSTALL-BIN build/newt-0.52.20/whiptail
2019-12-03 18:37:24+01:00 INSTALL-BIN build/msrtools-git/wrmsr
2019-12-03 18:37:24+01:00 INSTALL-BIN build/msrtools-git/rdmsr
2019-12-03 18:37:24+01:00 INSTALL-BIN build/msrtools-git/cpuid
2019-12-03 18:37:25+01:00 INSTALL-BIN build/coreboot-4.11/util/cbmem/cbmem
2019-12-03 18:37:25+01:00 INSTALL-LIB build/LVM2.2.02.168/libdm/libdevmapper.so.1.02
2019-12-03 18:37:25+01:00 INSTALL-LIB build/popt-1.16/.libs/libpopt.so.0
2019-12-03 18:37:25+01:00 INSTALL-LIB build/cryptsetup-1.7.3/lib/.libs/libcryptsetup.so.4
2019-12-03 18:37:25+01:00 INSTALL-LIB build/mbedtls-2.4.2/library/libmbedcrypto.so.0
2019-12-03 18:37:25+01:00 INSTALL-LIB build/npth-1.6/src/.libs/libnpth.so.0
2019-12-03 18:37:25+01:00 INSTALL-LIB build/libgcrypt-1.8.3/src/.libs/libgcrypt.so.20
2019-12-03 18:37:25+01:00 INSTALL-LIB build/qrencode-3.4.4/.libs/libqrencode.so.3
2019-12-03 18:37:25+01:00 INSTALL-LIB build/pciutils-3.5.4/lib/libpci.so.3.5.4
2019-12-03 18:37:25+01:00 INSTALL-LIB build/pciutils-3.5.4/../../install/lib/libpci.so.3
2019-12-03 18:37:25+01:00 INSTALL-LIB build/libksba-1.3.5/src/.libs/libksba.so.8
2019-12-03 18:37:25+01:00 INSTALL-LIB build/libassuan-2.5.1/src/.libs/libassuan.so.0
2019-12-03 18:37:25+01:00 INSTALL-LIB build/slang-2.3.1a/src/elfobjs/libslang.so.2
2019-12-03 18:37:25+01:00 INSTALL-LIB build/libusb-1.0.21/libusb/.libs/libusb-1.0.so.0
2019-12-03 18:37:25+01:00 INSTALL-LIB build/libgpg-error-1.32/src/.libs/libgpg-error.so.0
2019-12-03 18:37:25+01:00 INSTALL-LIB build/musl-1.1.15/lib/libc.so
2019-12-03 18:37:25+01:00 INSTALL-LIB build/newt-0.52.20/libnewt.so.0.52
2019-12-03 18:37:25+01:00 INSTALL-LIB build/tpmtotp-git/libtpm/libtpm.so
2019-12-03 18:37:25+01:00 INSTALL-LIB build/libusb-compat-0.1.5/libusb/.libs/libusb-0.1.so.4
2019-12-03 18:37:25+01:00 INSTALL-LIB build/util-linux-2.29.2/.libs/libuuid.so.1
2019-12-03 18:37:25+01:00 INSTALL-LIB build/zlib-1.2.11/libz.so.1
2019-12-03 18:37:25+01:00 INSTALL boards/qemu-coreboot/qemu-coreboot.config
2019-12-03 18:37:26+01:00 HASH bf761112c22f2b53b6c6ac4ebaa7880a5de5b3ed dirty qemu-coreboot
2019-12-03 18:37:26+01:00 CPIO      build/qemu-coreboot/tools.cpio
2019-12-03 18:37:26+01:00 UNCHANGED build/qemu-coreboot/tools.cpio
3ce0f2b7ac4aae65e97eb83a3b9ec32967f9b29f822df74c7ba4d7a3a103d316  /home/richard/heads/build/qemu-coreboot/tools.cpio
2019-12-03 18:37:26+01:00 HASHES    build/qemu-coreboot/tools.cpio
2019-12-03 18:37:26+01:00 CPIO      build/qemu-coreboot/heads.cpio
2019-12-03 18:37:26+01:00 UNCHANGED build/qemu-coreboot/heads.cpio
5f34bc728f192ba31c0560b0c42da493741df46851948055f1fde18db7054928  /home/richard/heads/build/qemu-coreboot/heads.cpio
2019-12-03 18:37:26+01:00 HASHES    build/qemu-coreboot/heads.cpio
2019-12-03 18:37:26+01:00 MAKE coreboot
tail /home/richard/heads/build/log/coreboot.log
-----
    CC         ramstage/lib/compute_ip_checksum.o
    CC         ramstage/lib/coreboot_table.o
    CC         ramstage/lib/crc_byte.o
    CC         ramstage/lib/delay.o
    CC         ramstage/lib/dimm_info_util.o
    CC         ramstage/lib/edid.o
    CC         ramstage/lib/edid_fill_fb.o
    CC         ramstage/lib/fallback_boot.o
    CC         ramstage/lib/fmap.o
    CC         ramstage/lib/gcc.o
    CC         ramstage/lib/halt.o
    CC         ramstage/lib/hardwaremain.o
src/lib/hardwaremain.c: In function 'platform_segment_loaded':
src/lib/hardwaremain.c:557:3: error: implicit declaration of function 'tlcl_measure'; did you mean 'tlcl_resume'? [-Werror=implicit-function-declaration]
   tlcl_measure(3, (const void *) start, size);
   ^~~~~~~~~~~~
   tlcl_resume
cc1: all warnings being treated as errors
make[1]: *** [Makefile:365: qemu-coreboot/ramstage/lib/hardwaremain.o] Fehler 1
make[1]: Verzeichnis „/home/richard/heads/build/coreboot-4.11“ wird verlassen
make: *** [Makefile:377: /home/richard/heads/build/coreboot-4.11/qemu-coreboot/.build] Fehler 1

```

merge, 2019-12-04:
(via private email) you said you use https://github.com/osresearch/heads/pull/568 so this doesn't apply to Heads in general, but to that pull request then.

but as it looks, you build Heads for qemu here. is this intentional? I doubt that this is supported right now actually. So: does building the master branch succeed for you?

AdmerStroh, 2019-12-04:
I forgot to change the board in the Makefile. Now everything compiled and installed just fine.

Thanks!

closed 2019-12-04 by AdmerStroh