linuxboot/heads #639: [Discussion] Consequences of losing TPM owner password

alex-nitrokey, 2019-12-19:
The quite old [keys.md](https://github.com/osresearch/heads-wiki/blob/master/Keys.md#tpm-owner-password) draft does ask an interesting question:

> Are there any consequences of an attacker controlling [the TPM owner] key?

As we have recently seen the new [oem-factory-reset](https://github.com/osresearch/heads/pull/606), I was wondering if it does any harm to leave the TPM owner password to the default.

I understand that this is definitely an issue if using the LUKS key on TPM, but what about the other stuff? Can an attacker access the OTP secret seed? Or anything else? Or can she only replace such information?

tlaurion, 2020-01-01:
@alex-nitrokey :
Resetting TPM is [required only to set the counter properly](https://github.com/osresearch/heads/blob/master/initrd/etc/functions#L164).

It is impossible to access secrets from the recovery console, [measurements being extended before accessing the recovery shell, and secrets being wiped](https://github.com/osresearch/heads/blob/master/initrd/etc/functions#L13).

alex-nitrokey, 2020-01-02:
I am not sure, if I made my question clear. What I wanted to know is, what are the consequences if an attacker would know the TPM password of my heads device? What could possibly go wrong? I could not find an information about this.

To my understanding she could not break the firmware verification which is ensured by the OTP secret during boot.

I am sorry, if you already answered this question. It is just that I did not understand you response.

tlaurion, 2020-03-09:
@alex-nitrokey sorry for the late response.

The attacker would be able to reseal measurements. If those changed, the OTP would be different, same applying to HOTP verification using the same measurements. So you are right.

The TPM ownership passphrase is required only to set initial nvram. Look in code:
`grep -rni tpm ./initrd`

Does that answer your question?

alex-nitrokey, 2020-03-16:
That makes it very clear. Now I also see the confusion :) Thank you very much for the clarification!

closed 2020-03-16 by alex-nitrokey

jtmoree-github-com, 2021-02-10:
@tlaurion  Can the secrets be recovered if I have the TPM owner password?  Can I extract data from the TPM?  I'm thinking of the use case where I need to troubleshoot in the recovery shell.  This also applies to OP question about attack vectors if the password is compromised.

tlaurion, 2021-02-11:
@jtmoree-github-com The TPM is not keeping any state other of what is sealed into NV space under setting a default boot option and setting a Disk Unlock Key passphrase released from it, if measurements from TPM + passphrase is good. 

> Is the secrets can be recovered by the TPM owner passphrase? 

@jtmoree-github-com 
No. [The owner passphrase is used to set presence](https://trmm.net/Glitter), nothing else, as referred previously. 
Please ask questions on things you are not understanding.