linuxboot/heads #642: Build fails compiling libgpg-error

petermn2, 2019-12-26:
I have tried building from clean clones of osresearch/heads on three different machines, and consistently get the following failure:

make[1]: Entering directory '/home/petermn2/heads/build/libgpg-error-1.32'
Making install in m4
make[2]: Entering directory '/home/petermn2/heads/build/libgpg-error-1.32/m4'
make[3]: Entering directory '/home/petermn2/heads/build/libgpg-error-1.32/m4'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/petermn2/heads/build/libgpg-error-1.32/m4'
make[2]: Leaving directory '/home/petermn2/heads/build/libgpg-error-1.32/m4'
Making install in src
make[2]: Entering directory '/home/petermn2/heads/build/libgpg-error-1.32/src'
gawk -f ./mkerrnos.awk ./errnos.in >code-to-errno.h
gawk: ./mkerrnos.awk:86: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkerrcodes1.awk ./errnos.in >_mkerrcodes.h
gawk: ./mkerrcodes1.awk:84: warning: regexp escape sequence `\#' is not a known regexp operator
/home/petermn2/heads/install/bin/musl-gcc -fdebug-prefix-map=/home/petermn2/heads=heads -gno-record-gcc-switches -D__MUSL__  -E   -P _mkerrcodes.h | grep GPG_ERR_ | \
               gawk -f ./mkerrcodes.awk >mkerrcodes.h
gawk: ./mkerrcodes.awk:88: warning: regexp escape sequence `\#' is not a known regexp operator
rm _mkerrcodes.h
gcc -I. -I. -o mkerrcodes ./mkerrcodes.c
./mkerrcodes | gawk -f ./mkerrcodes2.awk >code-from-errno.h
gawk: ./mkerrcodes2.awk:94: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
	./err-sources.h.in >err-sources-sym.h
gawk: ./mkstrtable.awk:113: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
	./err-codes.h.in >err-codes-sym.h
gawk: ./mkstrtable.awk:113: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
	-v prefix=GPG_ERR_ -v namespace=errnos_ \
	./errnos.in >errnos-sym.h
gawk: fatal: cannot use gawk builtin `namespace' as variable name
make[2]: *** [Makefile:1451: errnos-sym.h] Error 2
make[2]: Leaving directory '/home/petermn2/heads/build/libgpg-error-1.32/src'
make[1]: *** [Makefile:499: install-recursive] Error 1
make[1]: Leaving directory '/home/petermn2/heads/build/libgpg-error-1.32'
tail /home/petermn2/heads/build/log/libgpg-error.log
-----
               gawk -f ./mkerrcodes.awk >mkerrcodes.h
gawk: ./mkerrcodes.awk:88: warning: regexp escape sequence `\#' is not a known regexp operator
rm _mkerrcodes.h
gcc -I. -I. -o mkerrcodes ./mkerrcodes.c
./mkerrcodes | gawk -f ./mkerrcodes2.awk >code-from-errno.h
gawk: ./mkerrcodes2.awk:94: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
	./err-sources.h.in >err-sources-sym.h
gawk: ./mkstrtable.awk:113: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
	./err-codes.h.in >err-codes-sym.h
gawk: ./mkstrtable.awk:113: warning: regexp escape sequence `\#' is not a known regexp operator
gawk -f ./mkstrtable.awk -v textidx=2 -v nogettext=1 \
	-v prefix=GPG_ERR_ -v namespace=errnos_ \
	./errnos.in >errnos-sym.h
gawk: fatal: cannot use gawk builtin `namespace' as variable name
make[2]: *** [Makefile:1451: errnos-sym.h] Error 2
make[2]: Leaving directory '/home/petermn2/heads/build/libgpg-error-1.32/src'
make[1]: *** [Makefile:499: install-recursive] Error 1
make[1]: Leaving directory '/home/petermn2/heads/build/libgpg-error-1.32'
make: *** [Makefile:379: /home/petermn2/heads/build/libgpg-error-1.32/.build] Error 1
[libgpg-error.log](https://github.com/osresearch/heads/files/4003278/libgpg-error.log)
[libgpg-error.configure.log](https://github.com/osresearch/heads/files/4003279/libgpg-error.configure.log)

flammit, 2019-12-26:
I saw the same problem trying to compile on Fedora 31. Sounds like you have gawk 5 installed on the host machine.  If you downgrade to 4 or use a distro with an older gawk, it should work.

petermn2, 2019-12-26:
> I saw the same problem trying to compile on Fedora 31. Sounds like you have gawk 5 installed on the host machine. If you downgrade to 4 or use a distro with an older gawk, it should work

Thanks, 
I am using gawk 5 on all three machines. I'll downgrade and give it a try.

petermn2, 2019-12-26:
That was it. I built it fine in a Fedora 29 VM on my Qubes desktop.

Thanks again flammit!

closed 2019-12-26 by petermn2

osresearch, 2020-01-07:
Fixed in libgpg-error 1.37 (released last month): https://dev.gnupg.org/T4459