linuxboot/heads #643: The motd banner is printed twice

EDmitry, 2019-12-28:
The banner is printed two times on my machine. I see in the code:

```
	cat /etc/motd
	cat /etc/motd > /dev/tty0
```

Is this on purpose?

flammit, 2019-12-31:
I believe there were/are still code paths that had multiple serial devices (older keylime tests, kgpe-d16 in server mode), but the banner should probably be changed to handle it appropriately.

EDmitry, 2020-01-01:
Also I see that it was colorful at some point, then was removed by that commit adding support for Dell R630 server, then restored without any colors. Any objection to getting those colors back? :)

osresearch, 2020-01-07:
It was a debugging feature when we didn't know which serial port went where during board bringup... We can probably turn it off (and maybe bring back some colors)

tlaurion, 2022-02-05:
Fixed in master since a while