linuxboot/heads #644: Qubes Xen is chatty after installing heads during load

EDmitry, 2019-12-28:
When loading Qubes I can see Xen logging mixed with dom0's linux startup messages. This is the default kexec generated from grub:

`Qubes, with Xen hypervisor|xen|kernel /xen-4.8.5-12.fc25.gz placeholder console=none dom0_mem=min:1024M dom0_mem=max:4096M iommu=no-igfx ucode=scan smt=off |module /vmlinuz-4.19.82-1.pvops.qubes.x86_64 placeholder root=/dev/mapper/qubes_dom0-root ro rd.luks.uuid=luks-0e27084e-fcc2-4c66-822c-3bf3e09a1d18 rd.lvm.lv=qubes_dom0/root rd.lvm.lv=qubes_dom0/swap i915.alpha_support=1 plymouth.ignore-serial-consoles quiet rd.qubes.hide_all_usb|module /initramfs-4.19.82-1.pvops.qubes.x86_64.img`

`xl info` from dom0 shows commandline arguments correctly set for xen:

`xen_commandline        : placeholder console=none dom0_mem=min:1024M dom0_mem=max:4096M iommu=no-igfx ucode=scan smt=off no-real-mode reboot=no vga=current`

I thought `console=none` was the option that is supposed to suppress xen's logging. Is anything else different between the way grub and heads load the OS?

tlaurion, 2019-12-28:
Board configs have add and remove statements. Example here:
https://github.com/osresearch/heads/blob/master/boards/x230/x230.config#L31

tlaurion, 2019-12-28:
But the exact line sent to kexec is shown from heads before actually kexec'ing.

Apart from that, Heads is supposed to pass Xen arguments without modifying them for a while now.

EDmitry, 2019-12-30:
@tlaurion do you see the same behavior on your x230 with Qubes? I have rhgb option removed for dom0 kernel, but you would probably need to hit esc during Qubes boot to see the logging that I'm talking about. So far it looks like all the options are supplied to Xen correctly, but there clearly has to be some difference, I just don't know what it is yet.

tlaurion, 2020-01-01:
@EDmitry Mixed output like? Picture?

EDmitry, 2020-01-01:
@tlaurion actually, that seems to be the Qubes Linux kernel logging. Heads removes "quiet" option from the linux kernel parameters be default for x230. Why does it do that?

tlaurion, 2020-01-01:
> @tlaurion actually, that seems to be the Qubes Linux kernel logging. Heads removes "quiet" option from the linux kernel parameters by default for x230. Why does it do that?

@EDmitry To be more chatty. I [haven't tested other levels of verbosity](https://gitlab.com/tlaurion/heads/blob/PrivacyBeast_X230_QubesOS_Certified/boards/x230-libremkey/x230-libremkey.config#L37). You propose removing it so it goes quiet?

EDmitry, 2020-01-01:
I’d say let grub options (or default config) specify whether ‘quiet’ option is needed. Why would heads override and remove it forcefully?

closed 2024-04-16 by tlaurion