linuxboot/heads #645: Running kexec-save-key wiped the only slot remaining

EDmitry, 2019-12-28:
After running `kexec-save-key -p /boot/ /dev/sda2` the only slot generated by Qubes installation was wiped, and the new key wasn't inserted. Verified with luksDump that all slots are disabled after running the command. I see that from the code it supposed to wipe the slot 1, but I think the reason my main key was in slot 1 is because I change the LUKS password before. That probably inserted a new key in slot 1 and deleted from 0. 

I think `kexec-seal-key` should verify that it's not removing the only slot available.

flammit, 2019-12-31:
Yeah that's a good catch.  We probably should be using luksRemoveKey instead.

tlaurion, 2020-01-01:
> Yeah that's a good catch. We probably should be using luksRemoveKey instead.

@flammit: instead of [killslot](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-seal-key#L66)?

tlaurion, 2020-01-01:
> After running `kexec-save-key -p /boot/ /dev/sda2` the only slot generated by Qubes installation was wiped, and the new key wasn't inserted. Verified with luksDump that all slots are disabled after [...]

@EDmitry : When you change the password for a keyslot without specifying which one to play with, cryptsetup creates a new slot and deletes the old one. So you had noting in slot 0 anymore, and save-key does't validate if slot 0 exists, kills slot 1 and save new key in 1 as instructed. 

Any proposition?

EDmitry, 2020-01-01:
@tlaurion yes, the password change was needed because of a suspicion that my password was exposed to couch surfing. The drive didn’t have to be re-encrypted.

I think just checking the amount of slots remaining should be enough. The script should abort if there is only one slot remaining at #1.

Jetba, 2021-01-14:
Came across this as well while setting up my T430.
After installing Debian 10 with defaults using LUKS+LVM and setting up TPM Disk encryption keys I noticed that there is no support for LUKS version 2. Downgrading LUKS to version 1 also moved my key to slot 1, which caused this issue during setup.

A rare occurance but got me confused for a second 😄 One fast Debian reinstall later and making sure I was using slot 0 solved everything.

tlaurion, 2023-08-30:
Not sure why I cannot tag this issue from #1482, but that PR fixes this. Closing prematurely otherwise won't be closed automatically (Bad Github)

closed 2023-08-30 by tlaurion

tlaurion, 2023-09-01:
@EDmitry @Jetba fixed under #1482 
[image]