linuxboot/heads #661: Moving from std coreboot rom to heads X230

skollrc, 2020-01-24:
Hi all, 

I'm curruntly trying to move from standard coreboot ROM to heads. Builds (x230 & x230-flash) works well but there is something I don't understand: The 12mb ROM does not contain me or gbe.
I have a coreboot rom with cleaned me, what will happend if I flash my laptop with the 12mb ROM?
Does the std build (make BOARD=x230) create a ROM with an IFD that considerd a cleaned ME? Does it just ignore the ME and GBE regions during flash?

MrChromebox, 2020-01-24:
the Heads build does not contain the IFD, ME, or GBE regions.  Simply flash with `flashrom -w coreboot_12mb.rom --ifd -i BIOS` to avoid touching those regions

skollrc, 2020-01-24:
Ok, thx for the quick answer.
So why doing a 12MB rom?

MrChromebox, 2020-01-24:
> So why doing a 12MB rom?

sorry, I meant that to indicate you should use the x230 target, not x230-flash. Since you already have coreboot flashed, and (presumably) can flash internally, there's no need to use the -flash version (which is meant to just facilitate flashing the "full" x230 build when flashing a non-corebooted device)

skollrc, 2020-01-24:
I understood this part of your answer. But the 8mb chip is normaly used for ME/GBE/IFD right? Does heads use space on both chips?

MrChromebox, 2020-01-24:
the 8MB chip contains the IFD, ME, GBE, and the bottom part of the BIOS region. The 4MB chip contains only the top of the BIOS region. If your payload is small enough to fit into 4MB (ie, using SeaBIOS or Tianocore) then you can use a 4MB CBFS and the extra part of the 8MB chip isn't used. But Heads needs more than 4MB currently, so will use a ~7MB CBFS inside the BIOS region.

All of this is somewhat academic when flashing internally though, since all you have to do is use the `--ifd -i BIOS` parameters and flashrom will handle the rest. And that's exactly what Heads does when updating the firmware from within itself.

skollrc, 2020-01-24:
Ok, now I understand better :-) thx for this explanation!

skollrc, 2020-02-03:
I have a little problem with --idf option. Flashrom doesn't recognize it.
In the man page there is nothing about this. Is there any other way to flash ?

MrChromebox, 2020-02-04:
--ifd (not idf) tells flashrom to get the layout from the Intel Flash Descriptor, -i bios tells it to flash the  BIOS region from the flash descriptor

skollrc, 2020-02-04:
Yes sorry --ifd (miss writing) but it doesn't work. I tried --ifd, -i ifd, -i fd. Is there a particular version of flashrom to use?

MrChromebox, 2020-02-04:
you must be using an ancient version of flashrom. Use 1.0 min, 1.1 preferred

skollrc, 2020-02-05:
Thx! default version on debian is 0.99. It works perfectly now :-)

closed 2020-02-05 by skollrc