linuxboot/heads #664: Doc/Wiki - GPG key compatibility list

skollrc, 2020-01-27:
Hi all, 

Sorry if my questions look too simple but there still things I do not understand:
Is there any GPG keys compatibility list? Is the Nitrokey U2F compatible with heads?
Is specificaly this key: https://docs.puri.sm/PureBoot/Heads.html compatible? 
Is there any limitation or keys that are know not working with heads?

tlaurion, 2020-02-05:
Any gpg2 supported keys would provide GPG required functionalities.
HOTP compliant USB security keys, providing both GPG features and HOTP measurements are the LibremKey, Nitrokey Pro/Nitrokey Storage as of now. 

If @Yubico (and other advertising HOTP USB security Keys) wants to do the required work to make their keys supported inside of Heads by contributing upstream to the [verification library](https://github.com/Nitrokey/nitrokey-hotp-verification), I do not see why support would not be extended.

Note that the https://github.com/osresearch/heads/issues/641 https://github.com/osresearch/heads/issues/665 are not reproducible/not relevant anymore since https://github.com/osresearch/heads/commit/92e706bf1bd6affdbf28f99d9de0ed513ceaff48 unless proven otherwise.

tlaurion, 2020-02-05:
Please publish relevant Questions/Pull Requests (PR) at https://github.com/osresearch/heads-wiki in the future.

closed 2020-08-01 by tlaurion