linuxboot/heads #667: No screen output after flashing to a T430

nickfrostatx, 2020-01-30:
I tried twice to flash Heads onto a T430. 

- Once using the `x230` config built off of master
- Once with a modified config where I told coreboot my board was `THINKPAD_T430` instead of x230: https://github.com/nickfrostatx/heads/compare/ac71f295a..f2039a74

In both cases, it never gives any screen output, just a black screen. I left it on for 30 minutes or so, and still no output.

x230-flash works just fine. I think I'm only using the upper 4MB of the 12MB ROM image by using `--ifd --image bios`. Maybe that's where I'm going wrong?

The steps I followed (in case I've been doing it wrong):

1. Attach a chip clip to the 4MB BIOS chip
2. Write `x230-flash.rom` to the chip
3. Plug in a flash drive with the 12MB `coreboot.rom` that I built from `make BOARD=x230`
3. Boot the machine
4. `mount-usb`
5. `flashrom --force --noverify-all --programmer internal:laptop=force_I_want_a_brick --ifd --image bios /media/coreboot.rom`
6. Reboot. I didn't set up GPG keys or the TPM or anything

flawedworld, 2020-01-30:
Please see #656 and #580 . The port is done pretty much, we need to just fix GPG issues which are due to upstream it seems.

tlaurion, 2020-01-30:
@flawedworld : GPG_TTY temporary fix is in master now (until busybox tty command works again so that value to /dev/console is not hardcoded)

nickfrostatx, 2020-01-30:
@flawedworld I flashed a build off of #656. I'd expect to get an error message about GPG, but I don't get any graphics output at all.

flawedworld, 2020-01-31:
@nickfrostatx Are you using a T430 with Nvidia GPU or only the iGPU?

nickfrost-asana, 2020-02-01:
This is the iGPU

flawedworld, 2020-05-15:
@nickfrostatx Please update us after trying the latest commits in #656

tlaurion, 2020-05-20:
@nickfrostatx #656 is merged. Please tag me here if you still experience the same issue.

openfinesse, 2020-05-31:
@tlaurion I'm not sure if @nickfrostatx ever figured it out, but I have tried to flash to the t430 multiple times and it always ends up with no video output after the reboot.

Steps:
1. Externally flash 8MB unlocked and me cleaned image.
2. Flash t430-flash.rom image to the 4MB chip
3. Boot to heads shell, mount usb, and add GPG key to 12MB image.
4. Flash 12MB coreboot file (from /build/t430/
5. Reboot with "reboot" command.

The laptop seems to reboot but there is no video output after this point at all. (Tried MiniDP incase)

I've attached the images I built if thats of any use.
[t430-images.zip](https://github.com/osresearch/heads/files/4707097/t430-images.zip)
Please let me know if I should be doing something differently.

SebastianMcMillan, 2020-05-31:
@openfinesse 
Do you have an Intel or nvidia model?

openfinesse, 2020-05-31:
Intel

flawedworld, 2020-05-31:
@openfinesse can you try but instead of typing in reboot you just hold down power then turn on again? I am going to check hashes in a second please give me a mo.

flawedworld, 2020-05-31:
Building cleanly @ commit 78053b9ec1a2030f479b1b2ba7e57f9b0e0ab681 with make BOARD=t430 produces a SHA256 of coreboot.rom being:
`fc79e3de048865d52897ddf2a718df550c9ab42d88280d3cce6d7edfd3da2670`

Your SHA256 of the coreboot.rom in the attached zip file is:
`f7361916361e2619348fb37dd0dafba451df507887fca3efb159a20841f6b715`

You may wish to try another system or try with a clean tree.

flawedworld, 2020-05-31:
Adding on to that, building cleanly @ commit 78053b9ec1a2030f479b1b2ba7e57f9b0e0ab681 with make BOARD=t430-flash produces a SHA256 of t430-flash.rom being:
`1a607dc2c0e3ef1a081c3da02e043bf6eff3ac50c2a808467887cd15d94eeb6b`

Your SHA256 of t430-flash.rom in the attached zip is:
`909e2f5cbd84573630d01f3c0a77acb79393ee84f9172a39b2e2b578295378d1`

lsafd, 2020-06-01:
First off I would like to say that I am a relatively new user so please excuse the mistakes that I will probably make.

I am flashing to a x230 and having the exact same issue as @openfinesse.

Steps:
1. Flash x230-flash.rom image to the 4MB chip.l
2. Externally flash 8MB unlocked and me cleaned image.
3. Boot to heads shell, run "mount-usb".
4. Run "flash.sh -c /media/coreboot.rom" to flash the 12MB coreboot.rom image (from /build/x230-hotp-verification/).
5. Run "reboot" to reboot.

I end up with a blank screen, but the light for the power button turns on and off as expected. Attached is the coreboot.rom file.

[rom_file.zip](https://github.com/osresearch/heads/files/4709026/rom_file.zip)

flawedworld, 2020-06-03:
@lsafd Your coreboot.rom file has a SHA256 of:
`2a5877ed2909b4660b8f74a78821f40afa6cafb0186f68672fec7a83cf210ba3`

Building at 78053b9ec1a2030f479b1b2ba7e57f9b0e0ab681 produces:
`84ed5c262d78eb14fa5ae8bf44acd6e670852b3e0b2ddf062864e76a1aecc8e5`

Try on a clean tree or another system.

lsafd, 2020-06-03:
I will do a clean build when I can, but I am currently having problems with #733.

tlaurion, 2020-06-03:
> Building cleanly @ commit [78053b9](https://github.com/osresearch/heads/commit/78053b9ec1a2030f479b1b2ba7e57f9b0e0ab681) with make BOARD=t430 produces a SHA256 of coreboot.rom being:
> `fc79e3de048865d52897ddf2a718df550c9ab42d88280d3cce6d7edfd3da2670`
> 
> Your SHA256 of the coreboot.rom in the attached zip file is:
> `f7361916361e2619348fb37dd0dafba451df507887fca3efb159a20841f6b715`
> 
> You may wish to try another system or try with a clean tree.

Kernel is not reproducible. So hash wont be the same until #571 is fixed.

tlaurion, 2020-06-03:
Reopening issue

tlaurion, 2020-06-03:
Let me know which tickets to close and which to let open, a lot exist for t430.

SebastianMcMillan, 2020-06-03:
I think the issue title "T430 build does not boot" (adjust for X230 if applicable) is a better fit, since I don't think this is a video issue.

lsafd, 2020-06-03:
Alright, I re-cloned the git and built `x230-hotp-verification` again. The resulting `coreboot.rom` file has the exact same sha256.

Is there any new information I could provide or new strategies I could try. I could post images of me running `flash.sh` if that would help. I am just unsure where to go from here.

tlaurion, 2020-06-03:
@flawedworld is latest commit working for you? (nothing changed related to t40 per se)

tlaurion, 2020-06-03:
> Alright, I re-cloned the git and built `x230-hotp-verification` again. The resulting `coreboot.rom` file has the exact same sha256.
>
> Is there any new information I could provide or new strategies I [...]

@lsafd : i'm confused. You are talking about x230 and not t430?

lsafd, 2020-06-03:
Sorry I didn't understand I will open another issue. Should I delete my previous comments as they were all related to the x230?

tlaurion, 2020-06-03:
They have really common base, but this issue is linked to @openfinesse issue which is a T430 not booting (with potential screen issue)

tlaurion, 2020-06-04:
@flawedworld @lsafd @SebastianMcMillan @openfinesse @nickfrostatx 

I did a quick review of the following facts.

- xx30 (t430 and x230 and others, x230 tested only) have the following space available for BIOS region in IFD per output of ifdtool -x
```
Found Region Section
FLREG0:    0x00000000
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
FLREG1:    0x0bff0500
  Flash Region 1 (BIOS): 00500000 - 00bfffff 
FLREG2:    0x04ff0003
  Flash Region 2 (Intel ME): 00003000 - 004fffff 
FLREG3:    0x00020001
  Flash Region 3 (GbE): 00001000 - 00002fff 
FLREG4:    0x00001fff
  Flash Region 4 (Platform Data): 00fff000 - 00000fff (unused)

```

We are interested in the following line:
`Flash Region 1 (BIOS): 00500000 - 00bfffff `
so `00bfffff - 00500000 = 6FFFFF` (70000, which **was** the correct value of initial x230 board, while I do not understand why and how t430 could have a 80000 value. This seems to be the reason of the no screen output aka no boot from coreboot.

What we could do here is take some of the #307 (the IFD) and put it in blobs, considering that me was cleaned and outputs:
```
The ME region can be reduced up to:
 00003000:0001afff me
```
So 0001afff - 00003000 = 17FFF (decimal: 98303)

Meaning that the original region:
```
FLREG2:    0x04ff0003
  Flash Region 2 (Intel ME): 00003000 - 004fffff 
```
now having became
```
FLREG2:    0x04ff0003
  Flash Region 2 (Intel ME): 00003000 - 0001afff
```
 
So we passed from 
 `004fffff - 00003000 = 4FCFFF` (dec: 5230591)
to dec: 98303
so gained 5230591 - 98303 = 5132288 

Which we can pass to the BIOS region:
```
FLREG1:    0x0bff0500
  Flash Region 1 (BIOS): 00500000 - 00bfffff 
```
Which we remember was hex: 70000 
to Dec: 7340032 + 5132288 = 12472320
To hex: BE5000 (MAXIMAL THEORIC CBFS REGION VALUE IN COREBOOT CONFIG FOR FULLY NEUTERED ME)

I made the error of replacing 70000 to 80000 in x230-hotp-verification based on considered working t430 coreboot config. This doesn't seem to be valid.

Reconsidering what to do from here for the x230-hotp-verification platform, which doesn't compile if we go back to CBFS_REGION of 70000.

flawedworld, 2020-06-04:
> @flawedworld is latest commit working for you? (nothing changed related to t430 per se)

Works on commit 78053b9ec1a2030f479b1b2ba7e57f9b0e0ab681 for me, I am using fully stripped out and shrunk ME

tlaurion, 2020-06-05:
> > @flawedworld is latest commit working for you? (nothing changed related to t430 per se)
> 
> Works on commit [78053b9](https://github.com/osresearch/heads/commit/78053b9ec1a2030f479b1b2ba7e57f9b0e0ab681) for me, I am using fully stripped out and shrunk ME

@flawedworld : so without libremkey-hotp-verification support.

bwachter, 2020-07-17:
I got a t430 to play with a few weeks ago, and ran into the same issue. 9719510 is the first revision which works for me, though it seems it's not a fully clean build - I've only removed the t430 folder, and got a working image.

I've since removed everything in build, which results in an image with a different checksum, and a black screen when flashed. So far I haven't been able to get a working image with a clean build.

tlaurion, 2020-08-31:
@Trapacid @flawedworld @lsafd @SebastianMcMillan : What is the right CBFS_REGION to be in master?
Is https://github.com/osresearch/heads-wiki/blob/master/Clean-the-ME-firmware.md followed?

We seem to not move forward on this which creates duplicated like #815. 
Board owners, please follow up.

SebastianMcMillan, 2020-08-31:
I would not be the right person to ask about this since I don't have a
T430. However, it should be using an identical config as the X230, so
cross-reference to the X230 config.

bwachter, 2020-08-31:
I've tried the same CBFS_REGION setting as for x230 when debugging the black screen. I have no idea what made the one I currently use boot - but it wasnt the CBFS size change.

Trapacid, 2020-08-31:
I've tried to follow https://github.com/corna/me_cleaner/wiki/External-flashing on shrinking the regions, but I don't know what to do with the shrunken ME and IFD bin files besides putting them in heads/build/coreboot-4.8.1/3rdparty/blobs/mainboard/lenovo/t430 and running make BOARD=t430 and t430-flash in heads directory.

Tried flashing that coreboot.rom from build/t430 internally and just gave a black screen after reboot.

Running me_cleaner to shrink the compiled coreboot.rom prints some python error about 'entries' not being defined and running me_cleaner on the regular ME blobs from the original 12mb dump prints the same thing.

CBFS size in t430 config is 0x800000 while size in x230 is in 0x700000.

And just in case, my T430 does not have an NVIDIA card.

tlaurion, 2020-08-31:
@Trapacid CNRS should be equal as in x230 so 710000 and me cleaner instructions from heads-wiki should be applied on 8MB spi flash dump and reflashed.

Any improvement on doc and t430 PR should follow please

tlaurion, 2020-09-01:
Instructions for testing here: https://github.com/osresearch/heads/issues/815#issuecomment-684097153

tlaurion, 2020-09-02:
Fixed

closed 2020-09-02 by tlaurion