linuxboot/heads #668: libgpg-error build fails with newer gawk (Upstream Issue)

flawedworld, 2020-01-30:
Compilation fails upon libgpg-error when using gawk 5.0+ OpenWRT fixed this using commit https://github.com/openwrt/packages/pull/9371/commits/9afde0d39ae364c327063d36b188465058056a86 and this should give insight into the fixes needed until upstream resolves this.

Log: https://pastebin.com/raw/F8987WAg

Tested on Fedora 31 fully updated 30/01/2019
Gawk version which failed: 5.0.1 (gawk-5.0.1-5.fc31.x86_64.rpm)
Solution was to forcibly install Gawk from Fedora 30 - Gawk 4.2.1 (gawk-4.2.1-6.fc30.x86_64.rpm)
`rpm -Uvh http://download-ib01.fedoraproject.org/pub/fedora/linux/releases/30/Everything/x86_64/os/Packages/g/gawk-4.2.1-6.fc30.x86_64.rpm --force`

tlaurion, 2020-02-05:
@flawedworld :
It would require Heads to: 

- Not depend on the host provided gawk util provided by the host system (provided as a module on which other dependent modules would depend on, deployed under install
- or patching relevant modules requiring gawk to patch with provided fix by https://github.com/openwrt/packages/commit/9afde0d39ae364c327063d36b188465058056a86

Up for the task?

tlaurion, 2020-05-03:
https://dev.gnupg.org/T4827
https://dev.gnupg.org/T4459

Solution seems to upgrade gpg related modules. Attacking
(Linked to #710 )

MrChromebox, 2020-05-03:
looks like we can either update to libgpg-error 1.37 (in which the issue is fixed) or cherry pick the fix / apply as a patch (ref: https://git.gnupg.org/cgi-bin/gitweb.cgi?p=libgpg-error.git;a=commit;h=7865041c77f4f7005282f10f9b6666b19072fbdf)

tlaurion, 2020-05-04:
[Build on debian Buulseye worked](https://app.circleci.com/pipelines/github/tlaurion/heads/139/workflows/2bce64e8-cd49-4041-97f2-05f27c93447c/jobs/149).

PR #714 (2020-05-22): GPG toolstack upgrade to latest available versions (Fixes Gawk issue)
This includes to be reverted CircleCi config to [test current build over debian bullseye which was successful](https://app.circleci.com/pipelines/github/tlaurion/heads/144/workflows/4542964a-8a00-4951-858b-093f4de9ff39/jobs/154), while resulting build is currently untested.

Fixes #668 and problem in #710 PR to include Heads into Guix facing gawk issues

Todo: push additional commit to reverse that changeset once reviewed.

Reviewers: 
I repeat here message in [commit](https://github.com/tlaurion/heads/commit/b52ed555220aabd [... cut]
commit 7ce12fe6 (2020-09-07): Add gawk module to use if the host is running a different major version (fixes #668) (#811)
closed 2020-05-22 by tlaurion