linuxboot/heads #671: New script to decrypt LUKS

PowerPress, 2020-02-02:
Does this script invalidate the protection provided by heads and the TPM?

https://github.com/cryptolok/AES-REX

tlaurion, 2020-02-03:
I'm not sure I understand the scope of your question reviewing both projects (ram and register inspection separate tools).

The ram one could find the unlock key/disk recovery key in memory even though the paper specifies that luks enforces anti-forensic measures (untested).

The register one would imply that the tool using AES is launched by the tool which relies on crypto libraries from dbg which is not provided by heads and launched under heads recovery, which invalidates TPM measurements by extending them prior of launching the recovery shell.

@PowerPress: What is your specific concern?

PowerPress, 2020-02-03:
I was concerned this would be able to pull secrets from the TPM and bypass the protections. So since heads does not support dbg this is not vulnerable?

tlaurion, 2020-02-03:
The TPM stores the disk unlocking secret in its [nvram space from here](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-seal-key#L105)

If you are talking about the TOTP secret that is generated when sealing in the form of a [QR code](https://github.com/osresearch/heads/blob/master/initrd/bin/seal-totp#L89), it is also stored in [TPM nvram space from here](https://github.com/osresearch/heads/blob/master/initrd/bin/seal-totp#L35) and then read from [TPM nvram space from here](https://github.com/osresearch/heads/blob/master/initrd/bin/unseal-totp#L9). Note that those secrets, stored temporarily in tmpfs, are shredded in memory right after their usage so they are not present in ram.

The same applies to the process of [sealing](https://github.com/osresearch/heads/blob/master/initrd/bin/seal-libremkey#L19) and [unsealing](https://github.com/osresearch/heads/blob/master/initrd/bin/unseal-hotp#L39) with HOTP.

If that answer addresses your concern, please close this issue or specify more precisely your concern.

closed 2020-02-03 by PowerPress