linuxboot/heads #672: kexec can't boot qubes 4.1 iso

0rb677, 2020-02-13:
Hi @tlaurion any chance to boot from heads? Did you try 4.1 iso? 
I use master branch.

[image]
[image]

MrChromebox, 2020-02-13:
where are you seeing Qubes 4.1? 4.0.3 is the latest release

0rb677, 2020-02-13:
@MrChromebox 

laptop : x220
install media: https://openqa.qubes-os.org/tests/5493/asset/iso/Qubes-4.1-20200113-x86_64.iso
firmware: heads - boot fails.

laptop : x220-tablet
install media: https://openqa.qubes-os.org/tests/5493/asset/iso/Qubes-4.1-20200113-x86_64.iso
Rufus MBR/Legacy method dd (only legacy mode, uefi stuck before Anaconda start)
bios: modified 1.46 (unlocked RAM and whitelist wifi) NOT coreboot (http://www.mcdonnelltech.com/X220_v1.46_Modified_BIOS.zip) - boot success

MrChromebox, 2020-02-13:
any reason you're using a nightly build instead of the 4.0.3 release? Can you test that so we know if it's version-specific?

0rb677, 2020-02-13:
> any reason you're using a nightly build instead of the 4.0.3 release? Can you test that so we know if it's version-specific?

Yes. fedora-25 on dom0 is totally outdated and not supported.
Qubes 4.1 runs fedora-31, many new features, include installation.

MrChromebox, 2020-02-13:
I'll test both here as well and see if I can replicate on a Librem

tlaurion, 2020-02-13:
I confirm install works on my branch (not main Heads) for latest official QubesOS release.
Unfortunately, I just own x230 for testing. Will put that in my todo list but i'm quite busy right now.

Please check in other issues for t420 and x220 users and ping them here.
We might as well open an issue to track who tests what hardware. That would be really helpful.

MrChromebox, 2020-02-13:
Installers for both Qubes 4.0.3 and 4.1-20200113 boot fine on a Librem 13v4 running Heads master (bcf522cb) + the handful of branding changes from Purism's tree. Given the display artifacts, there's at least an IOMMU config issue at play, perhaps more. I'd start by adding `intel_iommu=igfx_off` to the x220's coreboot config `CONFIG_LINUX_COMMAND_LINE` param, which both the x230 and Librems already use. That param only affects the Heads payload, not any kexec'd kernels.

tlaurion, 2020-02-13:
See [here](https://github.com/osresearch/heads/pull/578/files#diff-d8b4fe91f08f8822cbab99f7ea3eedf7R23)

tlaurion, 2020-02-13:
Personally, I would remove the quiet there for all the boards.

MrChromebox, 2020-02-13:
> Personally, I would remove the quiet there for all the boards

I'll push a PR after testing

MrChromebox, 2020-02-13:
> Personally, I would remove the quiet there for all the boards.

hmm, it's pretty verbose w/o 'quiet' and not in a particularly useful way

tlaurion, 2020-02-14:
Agreed. Tested too. Nope!

0rb677, 2020-02-14:
@MrChromebox @tlaurion thanks, i just woke up, now i will reflash my laptop today.

0rb677, 2020-02-14:
Also changed chip to Winbond W25Q64.V and added iomem=relaxed

tlaurion, 2020-02-15:
@0rb677 I wouldn't recommend.
https://github.com/QubesOS/qubes-doc/pull/828

Fedora-25 on dom0 is not problematic, all needed packages are backported and all your actions should be in AppVMs, not dom0 anyway.

I would stick to 4.0.3 if I were you.

0rb677, 2020-02-15:
@tlaurion i disagree, because i use tilling window managers instead DE.

0rb677, 2020-02-15:
@0rb677 i use tilling managers instead DE. so, dom0 need tons of tweaks/scripts for good usability. 
I don't see anything criminal.  Dom0 is isolated and has toolchain. But mostly devel libs are outdated or none at all. Qubes is an experimental platform in general and thats cool. you have no problem watching movies on naked hypervisor if you want.

0rb677, 2020-02-15:
@tlaurion sad news.
Also build is complete. If you want i can flash and test 4.1 media and do not update to xen 4.13
```
2020-02-16 01:21:54+03:00 INSTALL   build/coreboot-4.8.1/qemu-coreboot/coreboot.rom => build/qemu-coreboot/coreboot.rom
2020-02-16 01:21:54+03:00 UNCHANGED build/coreboot-4.8.1/qemu-coreboot/coreboot.rom
29c4cc187ceecc4ca4b2edf0b24f12585295cc6f0c8b66b0f5f7173e84e02181  build/qemu-coreboot/coreboot.rom
29c4cc187ceecc4ca4b2edf0b24f12585295cc6f0c8b66b0f5f7173e84e02181  /home/user/heads/build/qemu-coreboot/coreboot.rom
make[1]: Leaving directory '/home/user/heads'

```

tlaurion, 2020-02-16:
You built for qemu?

-- Sent from /e/ Mail

tlaurion, 2020-02-16:
make BOARD=desired_board
qemu build is the default

-- Sent from /e/ Mail

tlaurion, 2020-02-16:
@0rb677 iomem=relaxed
Why is that?

tlaurion, 2020-02-16:
Please follow the breadcrumbs by starting where it works: 
https://github.com/osresearch/heads/pull/578

0rb677, 2020-02-16:
@tlaurion sorry i wrote `make board=x220` instead `make BOARD=x220`
also fbwhiptail package available only if you connect your vm to vpn in my country.
```
2020-02-16 10:41:32+03:00 MAKE coreboot
2020-02-16 10:42:12+03:00 DONE coreboot
"/home/user/heads/build/coreboot-4.8.1/x220/cbfstool" "/home/user/heads/build/coreboot-4.8.1/x220/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           86884 none
cpu_microcode_blob.bin         0x15480    microcode       25600 none
fallback/ramstage              0x1b900    stage           96557 none
cmos_layout.bin                0x33280    cmos_layout      1816 none
fallback/dsdt.aml              0x33a00    raw             13646 none
fallback/payload               0x36fc0    simple elf    6977487 none
(empty)                        0x6de800   null           954328 none
bootblock                      0x7c7800   bootblock        1968 none
2020-02-16 10:42:12+03:00 INSTALL   build/coreboot-4.8.1/x220/coreboot.rom => build/x220/coreboot.rom
e2c4e3d2de9c434155a1bee6d30741884a9985dfa1698a9c5efbe01c47ee7d1c  build/x220/coreboot.rom
e2c4e3d2de9c434155a1bee6d30741884a9985dfa1698a9c5efbe01c47ee7d1c  /home/user/heads/build/x220/coreboot.rom
make[1]: Leaving directory '/home/user/heads'

```

tlaurion, 2020-02-16:
You looked at opened tickets for x220?

-- Sent from /e/ Mail

0rb677, 2020-02-16:
@tlaurion i dont understund what you mean, 
i use this commit https://github.com/osresearch/heads/pull/578/files#diff-d8b4fe91f08f8822cbab99f7ea3eedf7R23 with `intel_iommu=igfx_off` enabled.

tlaurion, 2020-02-16:
You'll have the 50 seconds bug related to cbfs size in coreboot not being right, detailed in other tickets and PR.

Other then that. Can you kexec in Qubes 4.1?

-- Sent from /e/ Mail

0rb677, 2020-02-16:
@tlaurion please wait 10-15 minutes, waiting for downloading iso.

tlaurion, 2020-02-16:
If yes, please close the issue (kexec qubesos 4.1) and open other/comment in other tickets linked to your issues!

Happy that you like whiptail!

tlaurion, 2020-02-16:
You should probably read this now https://github.com/osresearch/heads/issues/669

-- Sent from /e/ Mail

0rb677, 2020-02-16:
@MrChromebox @tlaurion thanx o lot but it doesn't help. Screen artefacts and boot stuck!
`intel_iommu=igfx_off` is enabled when kexec try to boot.

[image]
[image]

0rb677, 2020-02-16:
@tlaurion also x220 has different spi chips.
`heads/initrd/bin/flash.sh` need update. For example, my x220 tablet  use `-c "MX25L6406E/MX25L6408E"` and x220 use `-c W25Q64.V`

tlaurion, 2020-02-16:
Please leave this ticket open to track newer Xen kexec problems(?)
QubesOS 4.1 bug is probably linked to kexec needed changes to handle newer xen correctly. 
Ping @osresearch @flammit @MrChromebox : if you have time to play with [QubesOS 4.1](https://openqa.qubes-os.org/tests/5493/asset/iso/Qubes-4.1-20200113-x86_64.iso) on qemu (and/or create a small draft on how to properly test with qemu to help others troubleshoot this would be awesome.)

@0rb677 : My intuition on (irrelevant here) internal flash chip definition is happening [here](https://github.com/osresearch/heads/pull/592#discussion_r375282314) 

@0rb677: Please participate under relevant tickets to continue troubleshooting, testing of other PRs. If it works for you, say it there. It will help merging those changes upstream for everyone.
[Universal flash (flash command under board config)](https://github.com/osresearch/heads/pull/592)
[X220/T420 support](https://github.com/osresearch/heads/pull/578)

tlaurion, 2020-02-19:
@osresearch pointed to [this](https://github.com/osresearch/heads/pull/592/files#diff-b05e402723345181f28c3fae60d17e3bR38), that may have reappeared in latest Xen and doesn't permit kexec to launch Xen. @marmarek, can you check?

marmarek, 2020-02-19:
Are you sure that is the correct link? It points at flashrom options, I don't think that has anything to do with kexec.

tlaurion, 2020-02-19:
@marmarek: sorry. link corrected.

tlaurion, 2020-04-08:
@marmarek : have you been able to shed some light to the issue and discussion pointed [here](https://github.com/osresearch/heads/issues/672#issuecomment-588326797)?

marmarek, 2020-04-08:
Not yet

closed 2022-04-06 by tlaurion