linuxboot/heads #674: LUKS KEY SEALING TROUBLESHOOTING LOG AND FIX

tlaurion, 2020-02-16:
Reported by user not on github:

> LUKS KEY SEALING TROUBLESHOOTING LOG AND FIX
> (workaround steps, listed at the end, may be worth posting somewhere)
> 
> Compile heads from osresearch/master commenting out the enable_usb line
> in gui-init. (same build as for testing the ISO verification previously
> mentioned)
> 
> Install Qubes.
> 
> Add a 2nd key to LUKS, so Heads doesn't nuke it and leave me without a
> way to encrypt the drive.
> 
> Set Heads default boot, etc. Choose "y" to add key to LUKS. Do the usual
> routine. Reboot. Choose "default boot."
> 
> Error:
> ```
> New value of PCR[6]: 48...
> Error Illegal index from NV_ReadValue
> Unable to read key from TPM NVRAM
> ```
> 
> Try it again. Reboot, and choose Boot Options | OS boot menu | Qubes |
> Set new default
> 
> reseal disk key to TPM? y
> 
> Messages:
> ```
> /dev/sda2: Adding key
> New value of PCR[6]: 0d8d...
> Error illegal index from NV_WriteValue
> ```
> Enter TPM Owner password, Yubikey PIN.
> 
> Reboot, default boot.
> 
> It works!
> 
> PROCEDURE FOR ADDING LUKS TO TPM (with osresearch/heads master compiled
> without enable_usb in gui-init):
> 0) Ensure there is no valuable data on the disk as you may lose all
> access to your data if anything goes wrong!
> 1) Add an extra keyslot to LUKS from the Heads recovery shell:
> 	cryptsetup luksAddKey /dev/sda2
> 2) give this new keyphrase to Heads when it asks for the recovery
> password (it will nuke this keyslot, as opposed to the original one)
> 3) Reboot
> 4) Go into "boot options" and set the default boot option a second time,
> again choosing "seal key to TPM"
> 5) This time, give Heads the original (still working) LUKS passphrase
> (created during Qubes install, as opposed to the 2nd throwaway
> passphrase created in step 1)
> 6) Reboot and it should work
>

tlaurion, 2020-02-16:
Additional user comment:
`Every time I've tried to seal a LUKS key to the TPM, Heads would delete the original ("recovery") key unless I intentionally created a 2nd key slot first. The result is that the disk becomes impossible to access in the event the LUKS key fails to unseal from the TPM. The solution is to create 2 keyslots initially -- the default keyslot created by the installer, and a 2nd "disposable" or "sacrificial" keyslot for Heads to remove. Provided there are at least 2 keyslots, the original ("recovery") key seems to remain intact.`

tlaurion, 2020-02-16:
cryptsetup is called only in two scripts:
```
user@heads-x230 heads]$ grep -Rn cryptsetup initrd/
grep: initrd/etc/mtab: No such file or directory
initrd/bin/kexec-seal-key:66:    cryptsetup luksKillSlot \
initrd/bin/kexec-seal-key:72:    cryptsetup luksAddKey \
initrd/bin/kexec-save-key:54:    uuid=`cryptsetup luksUUID "$dev" 2>/dev/null` \
initrd/bin/qubes-measure-luks:9:    cryptsetup luksDump $dev \
```

This snippet is responsible for removing slot 1 and inserted random key inside of it:
```
# Remove all the old keys from slot 1
for dev in `cat "$KEY_DEVICES" | cut -d\  -f1`; do
    echo "++++++ $dev: Removing old key slot"
    cryptsetup luksKillSlot \
        --key-file "$RECOVERY_KEY" \
        $dev 1 \
    || warn "$dev: ignoring problem"

    echo "++++++ $dev: Adding key"
    cryptsetup luksAddKey \
        --key-file "$RECOVERY_KEY" \
        --key-slot 1 \
        $dev "$KEY_FILE" \
    || die "$dev: Unable to add key"
done
```

I can't reproduce but your NVRAM issue in the sealing is weird.
The fix for removing enable_usb will be merged in a second.

tlaurion, 2020-02-16:
My explanation is that a reboot should be made after [signing the configurations](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-sign-config#L23) (which calls [enable_usb](https://github.com/osresearch/heads/blob/master/initrd/etc/functions#L137) to use the USB Security dongle, in which [enable_usb](https://github.com/osresearch/heads/blob/master/initrd/etc/functions#L86) is not a simple insmod call, but a [wrapper](https://github.com/osresearch/heads/blob/master/initrd/sbin/insmod)  which modifies PCRS related to the modules that were inserted. 

If from signing boot config, the user continues and proceeds to set a default boot configuration and seal a disk unlock key, measurements in [PCR5](https://github.com/osresearch/heads/blob/master/initrd/sbin/insmod#L15) of the TPM are taken into consideration in some way when sealing (even though we force 0 [here](https://github.com/osresearch/heads/commit/79a09e7424c67fc3168fef95a0352292a9bcf8eb#diff-7056a5d75913d9aa026e63c7ba924ba2R92) when not having a LIBREM_KEY).

The question is why when not having a LIBREM_KEY defined under BOARD config, even though [gui-init now enable_usb in its first step, users have to remove that line](https://github.com/osresearch/heads/blob/master/initrd/bin/gui-init#L132) to be able to seal LUKS secrets. Under heads code, every check looks like this if `[ -x /bin/libremkey_hotp_verification ]; then`, binary that is present only if `LIBREM_KEY ` is present in board configs.

This points to shis snippet:
```
# Librem Key loads USB modules which changes PCR5.
# In the event Librem Key is enabled, skip verification of PCR5
if [ -x /bin/libremkey_hotp_verification ]; then
  pcr_5="X"
else
  pcr_5="0000000000000000000000000000000000000000"
fi
```
Was introduced under https://github.com/osresearch/heads/pull/426/commits/79a09e7424c67fc3168fef95a0352292a9bcf8eb. So in normal heads, without Librem Key, the value of PCR5 is taken into consideration when sealing. But having enable_usb at first stage of gui-init should make them consistent. But it doesn't. Why?

tlaurion, 2020-02-17:
@mrchromebox : So basically, enable_usb should be called only when board config specifies USB_KEYBOARD in gui-init. Else, it should be called when needed, as it is right now on mount-usb, gpg-gui and flash-gui and prior of HOTP verifications.

MrChromebox, 2020-02-17:
@tlaurion I'm good with guarding `enable_usb` with `if (CONFIG_USB_KEYBOARD)`, just wondering if it should be in the main init script vs gui init, so available regardless of GUI use. But familiar with use cases for non-Purism servers to know if that's desired or not

tlaurion, 2020-02-17:
@mrchromebox : The commit shows that enable_usb was added for a server board configuration. That server is using gui-init or generic-init?

I don't understand why it's in gui-init for servers in all case. Should be both guarded with USB_KEYBOARD to me. Enabling usb system wise should be a choice when no ps2 keyboard is available, since it's a security/accessibility tradeoff user should be conscious of in all case.

The whole original idea ok LUKS unsealing with checking of pcr5 was to make sure the system was in an uncompromised case. The user should be aware that by using a a USB device always in at boot, he is choosing that security tradeoff.

In normal heads use case, there is no need to have usb modules present at the moment of releasing the LUKS key. Signing being decoupled and from a precedent boot. From my understanding, that was a choice from @osresearch in his original design.

-- Sent from /e/ Mail

MrChromebox, 2020-02-17:
@tlaurion right, I'm thinking that since we need to load the USB modules to support a usb keyboard, then it needs to be in the init script (not generic or gui-init) in order to support booting directly to a recovery shell. thoughts?

tlaurion, 2020-02-18:
Hmmm.

@MrChromebox I'm not sure I understand the use case i'm sorry. Is it network-init?
What init script are you using?

EDmitry, 2020-02-19:
Is this a dupe of https://github.com/osresearch/heads/issues/603 at this point?

MrChromebox, 2020-02-19:
@EDmitry I'd say so

tlaurion, 2020-02-19:
Fixed with https://github.com/osresearch/heads/commit/8e23a54f284b3847e973b70c4cecbe36acdfd19f

rafaelreis-r, 2020-10-02:
Hey Guys!

Sorry to bring back the closed issue, but I'm having very similar if not the exact same issue on a fresh x230 / Qubes installation. 
I've been a Qubes user for a while and just recently got a x230 to pair it with Heads.

1. Compiled x230-flash and x230 from source 2 days ago - 30-Sep_2020
2. Followed the [Beginner Guide](http://osresearch.net/Beginner-Installation-Guide/) to do the flashing and ME neutering (everything worked perfectly)
3. Flashed 12MB image after boot
4. Extracted pubkey from my **Yubikey 5C** and added it to the USB drive. Added it to BIOS via the menus
5. Followed [this](http://osresearch.net/Installing-Heads/#generic-os-installation) for the OS Installation
6. Not sure if this matters, but I have beefed up the encryption with [these](https://www.qubes-os.org/doc/custom-install/#example-custom-luks-configuration) instructions from Qubes docs.

The GPG key part and TOTP QR Code generation / sync seems to be working fine. I'm able to sign /boot files and Heads seems to be checking it's signature just fine.

TPM is where things get confusing. I've tried multiple times to reseal TPM keys, reset TPM and regenerate TOTP + sign /boot files, tried adding luks keys to different slots on /dev/sda2, tried the exact instructions on this thread, but regardless of what I do I never seem to get the key into the TPM. 

Output when trying default boot (with or without yubikey on the USB port)

[image]

When trying to seal a new key, I get a very brief message from the CLI before going back to the menus. Recorded a video and was able to get a glimpse of the error:

[image]

Another attempt:

[image]

Killed all key slots and left a single key on slot 7. Tried again:

[image]

Got stuck! Did I screw up something?

tlaurion, 2020-10-02:
@rafaelreis-r you screen capture shows 2 things:
- You entered the wrong Disk Recovery Key (LUKS key passphrase from QubesOS installation, where Heads is in US keyboard, maybe?) on two screenshots.
- A bug it seems  present in the actual logic of Heads from your taken screenshots where it seems to ignore the problem (bad passphrase) and doesn't wipe previous setted keyslot 1, while trying to still add a new key where that keyslot is already used.

> Killed all key slots and left a single key on slot 7.
Ouf. You are supposed to have only keyslot 0 (recovery key)and keyslot 1 (Disk Unlock Key released per TPM). I don't understand how you arrived in using more keyslots then that but if playing within the recovery shell. You might have wiped the original keyslot, which I hope is not the case.

Troubleshooting:
- Boot with unsafe boot option in Options -> other boot options -> unsafe boot
It will let qubesos boot ask you for your disk recovery key passphrase, which should work and validate your Disk Recovery Key passphrase.
- You can test the same result from Heads recovery shell by typing `cryptsetup luksOpen /dev/sda2 test` and typing your Disk Recovery Key passphrase. It should return to the recovery shell without error if successful.

If that works, please verify from the Heads recovery shell the output of cryptsetup luksDump and see what's up with your slots. Do you still have a slot 0?

rafaelreis-r, 2020-10-02:
Hey @tlaurion thanks for the reply.

I had indeed messed with the key slots in hope that clearing slots 0 and 1 would make it easier for the PCR sealing logic to work. After reading the [code ](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-seal-key) I see that was not a smart idea.

I reverted it back to what I beleive is a default Qubes installation: only slot 0 in use, and all the remaining slots empty.

[image]

How would I go about setting up Heads from this point on?

Here are some things I tried:

1. OEM factory reset:

It repeatedly fails in "GPG Key automatic keygen failed" (this might be a separate non related issue) - I do reset the yubikey with ykman prior to trying the OEM factory reset.

2. Create the GPG RSA keys [outside](https://github.com/drduh/YubiKey-Guide), add them to the key, export that key to the USB drive, import it and flash it to BIOS in Heads. 

That works but I circle back to the TPM luks key sealing issues. I read on #603 and #677 that using the recovery shell instead of the GUI could work. Haven't tried that though, since I'm not familiar with the commands.

What would be the correct procedure at this point?

tlaurion, 2020-10-05:
@rafaelreis-r : can you please reset 
1- cryptsetup luksOpen /dev/sda2 test
2- It the above works (no error), reboot into recovery
3- TPM reset from Options, TPM, TPM reset (recreate TPM NVRAM space)
4- At this point, if we know 1 works and TPM is reset, Options, Boot Options, Show boot options, select boot default, save to disk should work, asking you to seal Disk Unlock Key released by TPM if both Disk Recovery Key passphrase (1) and TPM is in good state.

rafaelreis-r, 2020-10-05:
Hey @tlaurion. Been doing that over and over during the weekend. Did it again and recorded it. I truly beleive something is wrong either with Heads or my specific x230 machine. 

Here is a detailed reply:

0 - I've rebuilt and reflashed Heads (coreboot.rom) with the latests commits from this weekend. Readded my GPG key to BIOS.

> 1- cryptsetup luksOpen /dev/sda2 test

Works

> 2- It the above works (no error), reboot into recovery
> 3- TPM reset from Options, TPM, TPM reset (recreate TPM NVRAM space)

Works
> 4- At this point, if we know 1 works and TPM is reset, Options, Boot Options, Show boot options, select boot default, save to disk should work, asking you to seal Disk Unlock Key released by TPM if both Disk Recovery Key passphrase (1) and TPM is in good state.

Appears to work, but in fact it doesn't. Here is an (unlisted) video of the process (3min) that I just recorded now. I reboot after each step with ctrl + alt + del to avoid any temporary settings issues.

[Video](https://youtu.be/8AXEbP1AFng)

Despite returning tho the menu, at [2:04](http://www.youtube.com/watch?v=8AXEbP1AFng&t=2m3s) it actually failed to add the key to slot 1. I grabbed a frame:

[image]

which is the exact thing I was pointing out in [my first reply](https://github.com/osresearch/heads/issues/674#issuecomment-702727393). It errors in this [ specific part of the code](https://github.com/osresearch/heads/blob/92e9a24902a34c4a1fa68753ec1f2500a277d9f1/initrd/bin/kexec-seal-key#L76) 

I can still use Heads + Qubes without the TPM Luks Key sealing feature, by going to the "recovery key" on every boot, but it is kind of a bummer. 

I'm 100% available to do further testing / debugging. I would like to contribute however I can. You can hit me @ Slack.

rafaelreis-r, 2020-10-05:
@tlaurion I narrowed it down!

There is indeed a limitation on Heads cryptsetup implementation that prevents it from working with the custom configuration provided by Qubes:

https://www.qubes-os.org/doc/custom-install/#example-custom-luks-configuration

`cryptsetup -v --hash sha512 --cipher aes-xts-plain64 --key-size 512 --use-random --iter-time 10000 --verify-passphrase luksFormat /dev/sda2`

For some reason cryptsetup on Heads doesn't support it. Here is how I found it:

1 - Manually ran lines 72 to 75 on the recovery console https://github.com/osresearch/heads/blob/92e9a24902a34c4a1fa68753ec1f2500a277d9f1/initrd/bin/kexec-seal-key#L72-L76

`no output`

2 - Ran the command above with --debug enabled

`Command failed with code 22: Invalid argument`

This [site ](https://unix.stackexchange.com/questions/597035/cryptsetup-failed-with-code-22-invalid-argument) mentions certain modules needed for some crypt choices. That hinted me that the custom crypto I used from Qubes doc might not be supported by Heads.

3 - Nuked my Qubes installation and installed with defaults. Did the whole sealing process again. 

`Key was sucessfully sealed on TPM`

Guess this should be its own issue, correct? Let me know I'll open it.

tlaurion, 2020-10-05:
@rafaelreis-r 
> cryptsetup -v --hash sha512 --cipher aes-xts-plain64 --key-size 512 --use-random --iter-time 10000 --verify-passphrase luksFormat /dev/sda2

Indeed! That was not tested. That would be linked to crypto and hash modules supported by the linux kernel provided by Head's, since cryptsetup relies on underlying supported linux modules inside of kernel (compiled in, not modules) to be able to unlock the volume.

The question here would be: what QubesOS recommends and implement in their kickstart OS installation and what you did, and what is better, which I have not reviewed here speaking.

tlaurion, 2020-10-13:
@rafaelreis-r :
> `cryptsetup -v --hash sha512 --cipher aes-xts-plain64 --key-size 512 --use-random --iter-time 10000 --verify-passphrase luksFormat /dev/sda2`

The default, as exposed on referred page textually and [here](https://wiki.archlinux.org/index.php/Dm-crypt/Device_encryption) is:
`cryptsetup -v --type luks --cipher aes-xts-plain64 --key-size 256 --hash sha256 --iter-time 2000 --use-urandom --verify-passphrase luksFormat device`

So we expect linux config to have SHA256. Let's verify:
```
user@x230-tests:~/heads/config$ grep SHA256 linux-x230.config 
CONFIG_CRYPTO_SHA256=y
```

And we expect linux config to miss SHA512:
```
user@x230-tests:~/heads/config$ grep SHA512 linux-x230.config 
user@x230-tests:~/heads/config$
```

tlaurion, 2022-02-06:
@rafaelreis-r ? Tag me in to reopen. Works on qubes 4.1 on master as well.

walliams, 2022-06-05:
@tlaurion encountering this issue on x230-maximized build after OEM Factory Reset. also tried following the workaround in OP and not having any luck