linuxboot/heads #692: Documentation: maintainers/testers for faster problems resolution

tlaurion, 2020-03-09:
**Boards:**
x220: @SebastianMcMillan: point others please
t420: @SebastianMcMillan: point others please
t430: @SebastianMcMillan: point others please
x230: @tlaurion @osresearch @merge @jan23
kgpe-d16: @tlaurion others?
server boards: point others please?
librems: @MrChromebox @kylerankin 

**Integration/Test**
Reproducibility expertise: @osresearch @flammit others?
Integration expertise: @tlaurion @osresearch @flammit  others?
qemu: @osresearch @flammit others?
Continuous Integration environments: Who?

Please add where you can help so that you are comfortable being tagged in issues.

SebastianMcMillan, 2020-03-09:
t430: @flawedworld is the owner of the current PR and is the one that actually owns a T430.

flawedworld, 2020-03-09:
@tlaurion happy with the T430, I also have an X230

techge, 2020-03-17:
Actually, I only have two x220 here. Unfortunately, I do not have a T420.

MrChromebox, 2020-03-17:
I have:
* Librem 13v2
* Librem 13v4
* Librem 15v3
* Librem 15v4
* x230

shamen123, 2020-03-17:
x230 and x220 here. Im also waiting for when we roll to next coreboot version to support p8h61m-pro desktop

Tonux599, 2020-03-18:
Happy with kgpe-d16, will be purchasing an additional SPI chip for testing soon as its my main workhorse. Also have an x220 and a FHD modded x230.

Thrilleratplay, 2020-10-16:
I have a dedicated x230 tablet for heads.

Additionally, I have:
* T430
* X220
* X220T
* X230 loose motherboards

and a Carbon X1 Gen1 but haven't have success flashing coreboot on it.

My main machine is a X230 with Nitrocaster FHD mod but that has a fully encrypted drive.

MrChromebox, 2020-10-16:
> Librem 13v4 (skylake): @MrChromebox
> Librem 15v4 (skylake): @MrChromebox  

those two boards are Kabylake, not skylake

tlaurion, 2020-10-17:
@MrChromebox @Thrilleratplay OP modified. Thanks

Thrilleratplay, 2020-10-17:
@tlaurion Can you remove the X1 Carbon Gen 1 from my name.  Until I can successfully coreboot it, I will not be able to provide any assistant with it and do not want to create any confusion.

tlaurion, 2020-11-09:
[On CBFS region, neutering ME, expending BIOS region in IFD after having reduced ME region](https://github.com/osresearch/heads/issues/667#issuecomment-638971582)

bwachter, 2020-11-09:
I'm mainly on the x230 (with several of them). I have a t430 which I'm currently not flashing due to initial setup issues trying to get to a working version - I do have a spare t430 board+CPU, though, but I need to get my hands on a fan for it first. After that it'd be usable for testing with an external display.

blobless, 2020-12-03:
I have an X230 and a KGPE-D16.

irelativism, 2021-01-21:
also have a T60

tlaurion, 2021-01-21:
> also have a T60 and macbook 1.1 and 2.1 boards, this is a diferent chipset though nontheless suported by coreboot.

What are SPI flash sizes please?
We cannot imply users to solder/unsolder. Coreboot supports hardware, not payloads of Heads sizes. Ir I remember well, the T60 is 4mb? maybe even 2mb? And the macbooks?

natterangell, 2021-06-14:
I have the T420 (with Ivy Bridge CPU), and recently successfully built and installed Heads using t420-hotp-maximized (using it with a NitroKey). I bumped Coreboot to version 4.13 in config, and everything works.

tlaurion, 2021-06-15:
> I have the T420 (with Ivy Bridge CPU), and recently successfully built and installed Heads using t420-hotp-maximized (using it with a NitroKey). I bumped Coreboot to version 4.13 in config, and everything works.

@natterangell Added you on list on top post for t420. Thanks!

shamen123, 2021-06-16:
@natterangell  @tlaurion Am I missing something here? but heads wont perform any of the measuring of Ivybridge by just bumping coreboot version in 4.13 as theres no patches for ivy/sandy present. The patch 0030-sandybridge.patch needs to be ported into 4.13 at a minimum . I mean it may compile, install and work, but is heads actually doing anything for @natterangell at this time? 

 [https://github.com/osresearch/heads/tree/master/patches/coreboot-4.13](https://github.com/osresearch/heads/tree/master/patches/coreboot-4.13)

tlaurion, 2021-06-16:
@shamen123 @natterangell : bumping to 4.13 is enough if coreboot config is changed from [CONFIG_MEASURED_BOOT=y](https://github.com/osresearch/heads/pull/949/files#diff-7faabfd7325c578c9d2f0a78e6ece0ad7d2022a1298ee856a957a1258bd674aeL1) to [CONFIG_TPM_MEASURED_BOOT=y](https://github.com/osresearch/heads/pull/949/files#diff-7faabfd7325c578c9d2f0a78e6ece0ad7d2022a1298ee856a957a1258bd674aeR15) to get rid of the 4.8.1 0030-sandybridge.patch

@natterangell : this is what was tested?

shamen123, 2021-06-16:
> @shamen123 @natterangell : bumping to 4.13 is enough if coreboot config is changed from [CONFIG_MEASURED_BOOT=y](https://github.com/osresearch/heads/pull/949/files#diff-7faabfd7325c578c9d2f0a78e6ece0ad7d2022a1298ee856a957a1258bd674aeL1) to [CONFIG_TPM_MEASURED_BOOT=y](https://github.com/osresearch/heads/pull/949/files#diff-7faabfd7325c578c9d2f0a78e6ece0ad7d2022a1298ee856a957a1258bd674aeR15) to get rid of the coreboot 4.8.1 0030-sandybridge.patch
> 
> @natterangell : this is what was tested?

thats my point, the post doesn't seem to imply all steps were done , just a version bump. This may confuse others.

eganonoa, 2021-06-16:
> thats my point, the post doesn't seem to imply all steps were done , just a version bump. This may confuse others.

You are absolutely correct, that caused confusion on my part. @natterangell spurred me to try building on coreboot 4.13 for the x230. Is it right that these are the necessary additional steps to use coreboot 4.13 including on xx30 series?

(1) change the relevant board config file to read "export CONFIG_COREBOOT_VERSION=4.13"; 
(2) change the relevant coreboot config file to read CONFIG_TPM_MEASURED_BOOT=y instead of CONFIG_MEASURED_BOOT=y; 
(3) for xx30 systems run the download_clean_me.sh script in the xx30 blobs folder.

Is there anything else different that should be done, or is that it?

Sorry, I know this isn't necessarily the right place. But it does make me want to reiterate that I'd be happy to contribute to documentation for x220, x230, t430 over the summer (from August).

natterangell, 2021-06-16:
> @natterangell : this is what was tested?

Yes, that is exactly what I did. I apologize for the confusion @shamen123 @eganonoa, I wasn't aware this would trigger an interest :)
For reference, I verified as mentioned [here](https://github.com/osresearch/heads/pull/966#issuecomment-769244811), if this is not as it should be please let me know.

The output from a recovery shell is as follows:
```
pcrs
PCR-00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
PCR-01: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
PCR-02: 61 50 0B 29 97 54 8F 70 6E 8F 55 31 ED 10 58 10 59 F3 22 7F 
PCR-03: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
PCR-04: 8A 6A 96 FD E1 A8 DD 96 27 14 79 DC 40 74 2B 36 AB A3 C2 B3 
PCR-05: FD A6 EA 9A 37 19 BD 61 39 83 38 EC 25 67 B2 F5 A3 31 EC A7 
PCR-06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
PCR-07: 54 F1 99 95 FC 10 D1 23 0A 8E AD 2D B2 0F 1E 13 E0 14 FD D7 
```
```
cbmem -1 | grep -B2 -i "pcr 2 measure"
TPM: Extending digest for FMAP: COREBOOT CBFS: cpu_microcode_blob.bin into PCR 2
TPM: command 0x14 returned 0x0
TPM: Digest of FMAP: COREBOOT CBFS: cpu_microcode_blob.bin to PCR 2 measured
--
TPM: Extending digest for FMAP: COREBOOT CBFS: vbt.bin into PCR 2
TPM: command 0x14 returned 0x0
TPM: Digest of FMAP: COREBOOT CBFS: vbt.bin to PCR 2 measured
--
TPM: Extending digest for FMAP: COREBOOT CBFS: fallback/dsdt.aml into PCR 2
TPM: command 0x14 returned 0x0
TPM: Digest of FMAP: COREBOOT CBFS: fallback/dsdt.aml to PCR 2 measured
--
TPM: Extending digest for FMAP: COREBOOT CBFS: fallback/payload into PCR 2
TPM: command 0x14 returned 0x0
TPM: Digest of FMAP: COREBOOT CBFS: fallback/payload to PCR 2 measured
```
I've attached `cbfs` output too, as well as the coreboot config file I used (feel free to disregard the ASPM and RAMINIT bits, those are there for some other experimentation).
[cbfs.txt](https://github.com/osresearch/heads/files/6664799/cbfs.txt)
[coreboot-t420-hotp-maximized.config.txt](https://github.com/osresearch/heads/files/6664806/coreboot-t420-hotp-maximized.config.txt)
 
If someone else with a T420 can verify, I'd be happy make a PR.

tlaurion, 2021-06-16:
@natterangell A PR while tagging board owners in it (from top post) should do it, not the other way around!

natterangell, 2021-06-17:
Makes sense :) Done!

SebastianMcMillan, 2021-07-04:
@tlaurion Can you remove me from all the devices listed here? I will not have access to those devices for the foreseeable future.

flawedworld, 2021-07-05:
I no longer own any of the hardware mentioned here as it is not getting security coverage from Intel.

tlaurion, 2021-07-05:
> I no longer own any of the hardware mentioned here as it is not getting security coverage from Intel.

@flawedworld : would you improve that section in regard of the choice you took in changing hardware for microcode support from Intel? 
https://osresearch.net/Heads-threat-model/#binary-blobs-microcode-updates-and-transient-execution-vulnerabilities

flawedworld, 2021-07-09:
> @flawedworld : would you improve that section in regard of the choice you took in changing hardware for microcode support from Intel? [...]

If the user does not trust microcode/software/etc from vendor X that they should just not use vendor X. It doesn't make sense to me why a user would continue to use hardware/software from a vendor that they do not trust. I don't consider microcode to be a threat to security as I am already trusting a vendor with their hardware platform and microcode is required for proper secure operation of modern platforms anyway. Microcode updates are part of required maintenance too, and things like the Intel ME or on other Intel platforms, the FSP, also need to be maintained. Even if the ME is neutered or disabled, or whatever method/vocabulary one wishes to use, it is still vulnerable and is a critical part of the firmware that must be maintained. The same applies to any part of a platform be it a Qualcomm mobile phone, an AMD workstation or a POWER9 system. I'm going to be moving to a Microsoft Surface machine partly because of those reasons once I have a need for a portable system again.

tlaurion, 2021-07-22:
@flawedworld have you read [this](https://hardenedvault.net/2021/07/16/ciso-seceng_csme.html) ?
Anything that should be added to https://osresearch.net/Heads-threat-model/#binary-blobs-microcode-updates-and-transient-execution-vulnerabilities to clarify other users choices following yours?

natterangell, 2021-07-22:
I think @flawedworld makes a good argument here from one perspective. It reminds me a little of what I’ve seen Daniel Micay say regarding why people should be cautious with LineageOS on discontinued/unsupported android devices and rather stick with something like GrapheneOS on devices that still receive firmware updates from vendor. 

That being said, I’m not aware of any critical bug on currently supported Thinkpads that cannot be mitigated as well as for any other vendor supported CPU. Not yet at least. But the entire x86 ecosystem is like a wounded animal due to these bugs. It won’t improve much I fear. I hope we can switch to blob free ARM, POWER or RISC-V within a decade but we’re not there yet. So we have to work with what we have.

The industry has been chasing performance gains by relying heavily on speculative execution, and is now paying the price for that. Intel is very much to blame on their part. 

I agree it is rather moot not to trust microcode upgrades for a CPU. But AFAIK the ME firmware can still be upgraded (and then disabled/neutered) even if the platform is not supported in terms of further microcode updates.

Boatloads of end users are still vulnerable to all these bugs because their motherboard OEM doesn’t bother to release BIOS updates with patched microcode or ME updates. Tech savvy users can get around that by getting upgraded ME/CSME from Win-raid or whatever and then patch their own BIOS.

I’m still comfortable with an Ivy Bridge CPU, Coreboot/neutered ME and QubesOS. But if some new bug comes along that cannot be mitigated without a microcode update (which I won’t get from Intel), and said bug can be exploited remotely, or break out from a virtual machine, I’ll have to reconsider like @flawedworld. 

I tell relatives to get a Chromebook or an Apple device. It’s just easier and more secure for most people. But that doesn’t fit my threat model 😊

flawedworld, 2021-07-24:
> @flawedworld have you read [this](https://hardenedvault.net/2021/07/16/ciso-seceng_csme.html) ?

So I've just read this. The case of manufacturing mode opens up the can of worms of OEM's being incompetent, where ultimately the truth is that if you want secure hardware as a consumer, you either buy Apple, Google or Microsoft hardware. It doesn't make sense to disable the ME when it is offering security functionality. Things like TXT, SGX, PTT, Boot Guard and so on all depend on it. A secure platform should taking advantage of all of that, and should not be disabling the ME and having nothing. Really my key thoughts from that article are that awareness in firmware security is very poor across the industry and it's good to see government organizations and vendors like push for improvements across the board (e.g. Microsoft's Project Mu, Defender Firmware Protection and so on).

> Anything that should be added to https://osresearch.net/Heads-threat-model/#binary-blobs-microcode-updates-and-transient-execution-vulnerabilities to clarify other users choices that should follow your decision path?

I think that there should be a clear and to the point statement explaining that once vendor support is gone (support timelines are a whole other topic) the platform is no longer secure and should be abandoned. It's wrong to see a vendor unsupported platform as secure, it means that platform will not be getting security coverage from the vendor, and may not be mentioned in some or all future security reports. Whilst there are hypervisor mitigations possible for some of the public microarchitectural issues such as SRBDS, the boot has got to be firm on this point and state openly that users should move to a more modern, vendor supported platform as soon as they can. A mitigation in the OS/Hypervisor is not a solution and is not even necessarily effective, secure or even possible at the end of the day and the only sustainable approach is to apply microcode patches/blobs/firmware updates/etc and follow vendor advice or move platform.

eganonoa, 2021-07-24:
I would suggest that the decision-making isn't anywhere near as binary as
@flawedworld has said or nearly as easy as saying something like "once
vendor support is gone (support timelines are a whole other topic) the
platform is no longer secure and should be abandoned." First, that
presupposes that the platform that was supported was actually secure in the
first place, which of course raises trust questions re the vendor and the
state in which the vendor is located. Second, (and much more importantly)
you have to think about the people who can best benefit from something like
this and the world that they live in. It is unrealistic to think that
people who are genuinely under threat and can best make use of something
like this are capable of owning devices that are new enough to meet that
standard (cost/access), or in the case of chromebooks have anywhere near
good-enough access to fast internet to make them usable. Put simply, for
very many human rights defenders, human rights organizations and other
threatened people around the world, including ones in rich countries, it is
not possible to easy provision high-cost devices (e.g. Macs, Surface) or
use cloud-centric ones (like Chromebooks).  I speak from personal
experience on this (unfortunately).

eganonoa, 2021-11-29:
@tlaurion As discussed happy to be added to the following as a collaborator/maintainer/tester as appropriate:

- x230
- t430 (non-dGPU)
- t520
- t530 
- w530 

I currently have Heads working on the t520, t530 and w530 (#1059) and would be happy to take the lead on those boards. We have some x230 and t430 (non-dGPU) in reserve that we can use to run tests with.

Please remove me from the x220 list as I no longer have one of those in use with Heads.

natterangell, 2021-12-03:
@tlaurion, FWIW I also have an X230i, might as well add me to the list for that laptop too. It’s pretty beat up and mostly functions as a backup machine these days, but I can test ROMs easily on it.

alex-nitrokey, 2021-12-06:
@tlaurion please remove me from the list (alex-nitrokey) as I am not in possession of an T430 and I am currently not working for NK.

Kind regards

walliams, 2022-03-11:
i'm running heads on a w530-dgpu-K2000m, and i still have the t430-dgpu
feel free to add me to the w530 group

weyounsix, 2022-08-04:
w530-dgpu-K2000m, t430-dgpu, and x230 owner w/ programmer here

jnscmns, 2022-10-17:
> 

w530 and x230 - Feel free to add me to the list. Thanks

tlaurion, 2022-10-25:
@jnscmns w530 is with dgpu if I recall well, which model?

jnscmns, 2022-10-25:
> @jnscmns w530 is with dgpu if I recall well, which model?

K1000M

tlaurion, 2022-11-07:
@srgrint added you as x220 tester. Please tell me if you want to be removed

tlaurion, 2022-12-19:
@3hhh what platforms do you own? Can I add you under OP here as a owner of those platforms?

3hhh, 2022-12-22:
On 12/19/22 21:27, tlaurion wrote:
> @3hhh what platforms do you own? Can I add you under OP here as a owner of those platforms?

T530, feel free to add me.

I'll look into your other posts once I have some more time.

123ahaha, 2022-12-27:
T430, X230 feel free to add me

I also have a w541 i have produced a binary but can't decide myself to flash it because i only have one w541 and it's my main workstation. But i have the blobs, the config file, and the w541 is now under coreboot thanks to chromebox

tlaurion, 2023-02-10:
@techge : just removed you, forgot to do so before it seems. You might want to be check TPM2 qemu support, and if coreboot supports your new board, you might be interested into collaborating into moving things forward for TPM2 support for newer boards (with more blobs dependencies on coreboot's side for x86 of course).

techge, 2023-02-10:
Thank you! And I will for sure try to get things running, if my board will ever be supported by coreboot! It is from an OSS friendly brand, but it does not release coreboot-ready laptops yet - although I thought exactly this when buying the device :grimacing:

tlaurion, 2023-04-20:
@srgrint @akunterkontrolle I added you under t440p under OP. let me know if you want to be removed.

tlaurion, 2023-07-04:
@nestire: added you in OP for x230/t430 legacy boards

tlaurion, 2023-07-04:
Note to all, https://github.com/osresearch/heads/pull/1398 has renamed the following boards prepending UNTESTED before directory and board name.

Consequently, those need to be built with
`make BOARD=UNTESTED_original_board_name`

If any of you reading this is a board owner, please voice yourself:
- UNTESTED_x230-maximized-fhd_edp
- UNTESTED_x230-hotp-maximized-fhd_edp
- UNTESTED_x220-maximized
- UNTESTED_x220-hotp-maximized
- UNTESTED_w530-maximized
- UNTESTED_w530-hotp-maximized
- UNTESTED_w530-dgpu-K2000m-maximized
- UNTESTED_w530-dgpu-K2000m-hotp-maximized
- UNTESTED_w530-dgpu-K1000m-maximized
- UNTESTED_w530-dgpu-K1000m-hotp-maximized
- UNTESTED_t530-maximized
- UNTESTED_t530-hotp-maximized
- UNTESTED_t530-dgpu-maximized
- UNTESTED_t530-dgpu-hotp-maximized
- UNTESTED_t520-maximized
- UNTESTED_t520-hotp-maximized
- UNTESTED_t440p-maximized
- UNTESTED_t440p-hotp-maximized
- UNTESTED_t420-maximized
- UNTESTED_t420-hotp-maximized
- UNTESTED_p8z77-m_pro-tpm1-maximized
- UNTESTED_p8z77-m_pro-tpm1-hotp-maximized

tlaurion, 2023-07-24:
[Add HP Z220 CMT](https://github.com/osresearch/heads/pull/1399#top) added, adding @d-wid for board owner.

tlaurion, 2023-11-10:
@daringer added nv41/ns50 on op and tagged you as tester. Reused those lines under #1522 for testing.

lethedata, 2024-02-03:
I can test x230 (xx30) (x220 keyboard mod) and x230-fhd/edp variant (edp 1080p display)

tlaurion, 2024-02-04:
@lethedata can you detail a little more what Ed mod it is? This is what seems to be confusing for all for fhd/edp mod, even more if there is a reboot issue from OS if only difference is that edp lod, preventing shutdown of power lines on the x230 which was never reported before.

jtf7, 2024-02-05:
@tlaurion 
I would like to be tagged in all things KGPE-D16!

I have 8 of these boards so far and have done extensive research and testing relating to building a stable kgpe-d16 system that is as open source and blob free as possible. That being said, most of what I’ve done is hardware based testing. Programming is definitely not my strong suit but I’m happy to help in whatever ways I can. 

This board is rock solid—once you figure out how to get it running. Based on what I’ve seen in the forums, it seems like a lot of people would like to be using this board (especially with Qubes) but aren’t because of the initial technicalities associated with getting up and running. I think this board (and its KCMA-D8 little brother) have enormous potential and the number of users would grow significantly if there was a dedicated place for users to document how to build a working system.  So...is there a good place for an up-to-date dedicated KGPE-D16 wiki page for this? What would be the best way to go about setting this up?

I would like to download a compiled ROM image of Heads for the KGPE-D16 although I am not familiar with GitHub.  How do I do this? It appears there are numerous flavors of Heads that can be used, such as: Dasharo/Heads, linuxboot/Heads, Nitrokey/Heads, tlaurion/Heads, and many more. The Winbond chips I am currently using I purchased pre-flashed with Heads Workstation PS/2 (Dasharo flavor I think, pull 1565?) so I’m not entirely sure how to obtain ROM images. Assistance with this would be greatly appreciated.

I’m open to testing whichever ROM images seem promising for the KGPE-D16—a “Sampler Pack” would be awesome :)

And most importantly.  Thank you to everyone who has devoted their time towards developing open source software.  I thoroughly believe that open source devices are a very good thing for society and supporting open source helps support things like democracy, free speech, equality, and all the other good stuff.

jtf7, 2024-02-05:
### Hardware Write Protection
A number of users have had issues getting this working so here are some tips and using hardware write protection

_**Hardware write protection is only available on eeprom chips that explicitly support it and typically requires setting specific fuse settings appropriately to enable this function.**_   This information is found in the manufacturer’s documentation for the specific part # of chip you intend to use.

### Winbond W25Q128FVAIG
The Winbond W25Q128FVAIG is a 16MB eeprom chip capable of hardware write protection and is compatible with the KGPE-D16. This chip is physically engraved with the abbreviated text 25Q128FVIG. 

**_Fuse Settings_**
Fuse settings required to activate the hardware write protection feature for this specific part number of Winbond chip are as listed below: 

[S18=WPS=0]
[S14=CMP=0]

[S9=QE=0]
[S8=SRP1=0]
[S7=SRP0=1]

[S6=SEC=0]
[S5=TB=0]
[S4=BP2=1]
[S3=BP1=1]
[S2=BP0=1]

Experts can tweak these settings to fine tune the chip’s configuration however, the settings listed above have worked well for me and should work well for most users.

Connecting pin 3 (/WP) and pin 4 (GND) together with these fuse settings will enable hardware write protection. It may be required to power off the system before connecting/disconnecting these pins in order for the correct write state to be recognized.

Many users have a similar version of Winbond chip with part number W25Q128FVAIQ however, it should be noted that this alternate Q version chip is not capable of hardware write protection. 

**_Additional Fuse Settings: Security Register Lock Bits_**
You may also want to set S11, S12, and S13 as listed below in order to permanently lock the security register as read-only. This is where manufacturer’s info, security related information, and hard coded data for the chip is stored. This is a separate location on the chip from where your ROM image is stored. The vibe I got from reading the Winbond documentation is that this could be a great place for backdoor related shenanigans and it’s probably a good idea to set these fuses in order to lock as read-only. That being said, I’m not entirely sure. I set mine as listed below and it worked for me however, as always, use at your own risk.

[S13=LB3=1]
[S12=LB2=1]
[S11=LB1=1]

@tlaurion I wasn’t sure where to post the info I have. Feel free to cut and paste as necessary or correct me on my improper GitHub posting etiquette.  Thanks

tlaurion, 2024-02-12:
> ### Hardware Write Protection
>
> A number of users have had issues getting this working so here are some tips and using hardware write protection
>
> _**Hardware write protection is only available on [...]

@jtf7 This definitely needs proper documentation, but is more related to documentation as of now and would require automatization and love under Heads #985. Note that flashrom work has been done, and specifically for kgpe-d16 is https://shop.3mdeb.com/shop/adapters/flash-chip-adapters/asus-kgpe-d16-flash-chip-adapter/ with documentation at https://docs.dasharo.com/variants/asus_kgpe_d16/spi-wp/. Desktops with socketed SPI chips offer this capability more easily than on other platforms, limiting applicability of this solution to other platforms.

To be able to do the same with soldered in SPI chips, soldering is currently needed by end users/OEM which is why this solution is not pushed forward as of now without proper gadget easing the process. 

The easier solution that was pushed on Intel for pre-Skylake is PR0 SMI enabled platform locking bits per  https://github.com/linuxboot/heads/pull/1373. This makes Heads the only internal flasher, where OS can read SPI but cannot write to it and where Heads prevent access to recovery shell and booting from USB without GPG detached signed authentication with either USB Security Dongle or USB Thumb drive encrypted backup of key material created at OEM Factory Reset/Re-Ownership wizard per https://github.com/linuxboot/heads/pull/1515.

lethedata, 2024-02-12:
> > I can test x230 (xx30) (x220 keyboard mod) and x230-fhd/edp variant (edp 1080p display)
> 
> @lethedata can you detail a little more what edp mod it is? This is what seems to be confusing for all for fhd/edp mod, even more if there is a reboot issue from OS if only difference is that edp mod, preventing shutdown of power lines on the x230 which was never reported before.

@tlaurion the mod kit I have appears to be an a.gain X220_X230 FHD RevA1.0 . Still looking into things regarding the power issue.

I compiled a small list of different mod kits and information regarding them [here](https://github.com/lethedata/x220_x230-Mod_Kits).

pcm720, 2024-02-12:
@lethedata 
> I compiled a small list of different mod kits and information regarding them [here](https://github.com/lethedata/x220_x230-Mod_Kits).

Are you sure nitrocaster mod doesn't work with the VBT patch?
We have people with this mod using x230-fhd_edp without any issues, and coreboot's Gerrit discussion mentions nothing of the sort.

tlaurion, 2024-02-13:
> @lethedata
>
> Are you sure nitrocaster mod doesn't work with the VBT patch?
> We have people with this mod using x230-fhd_edp without any issues, and coreboot's Gerrit discussion mentions nothing of [...]

@lethedata if that fact came from me and prior comment, that's my fault since I modified prior comment after revising the facts and I'm sorry for misleading you. The vbt patch is supposed to work with the vbt vomit from the coreboot variant selection made from which the edp board under Heads comes from.

tlaurion, 2024-02-13:
> @lethedata
>
> Are you sure nitrocaster mod doesn't work with the VBT patch?
> We have people with this mod using x230-fhd_edp without any issues, and coreboot's Gerrit discussion mentions nothing of [...]

@lethedata if that fact came from me and prior comment, that's my fault since I modified prior comment after revising the facts and I'm sorry for misleading you. The vbt patch is supposed to work with nitro aster, the vbt commt from the coreboot variant selection made from which the edp board under Heads comes from is supposed to be compatible with those mods as well, just as the patch says both upstream and downstream here.

[10 more replies]

closed 2026-03-05 by tlaurion