linuxboot/heads #697: openbsd and heads

0rb677, 2020-03-17:
ping @tlaurion
Has anyone every gotten OpenBSD with FDE to work with heads?

tlaurion, 2020-03-17:
@0rb677 look in closed issues. I think someone did?

tlaurion, 2020-03-17:
Current blocker is newer cryptsetup to support newer LUKS2 format, else I think kexec is able to boot kernel and initrd while really unsure.

0rb677, 2020-03-17:
@tlaurion thx. i haven't lot of time to test it. Hard work :(
Currently i have success boot only Qubes 4.0/Void Linux/Ubuntu (straightforward, also Void need  i915.modeset=1 to GRUB_DEFAULT_LINUX). I use distro+FDE(or unencrypted /boot) by default.
Guixsd need some tricks to boot (untested) or it stucks on kernel boot with Heads firmware
```
(initrd-modules (cons* "i915" %base-initrd-modules)))
```

0rb677, 2020-04-19:
i have whole encrypted disk.  i tried `vi /etc/config` and change boot usb options to `/dev/sda`, but `gui-init` calls `/dev/sda1`. only reflash can help me?

0rb677, 2020-04-19:
@tlaurion how force to boot full encryption disk with heads? (now it trying to boot `/dev/sda1`)

alex-nitrokey, 2020-04-21:
Full disk encryption does not fit into the concept of heads. 

You choose either one of these two. Changes of files in /boot are detected by heads though, so the only disadvantage of choosing heads instead of full disk encryption is minor information leakage (what kernel is in use or similiar).

0rb677, 2020-05-03:
Thanks @alex-nitrokey  :)

FrostKnight, 2021-03-02:
This actually interests me as well. I hope this will be possible in the future too. I am no huge hurry, but it would be cool to see in the future when I get my Nitropad.

tlaurion, 2021-03-02:
@FrostKnight @0rb677 : 
OpenBSD kexec'ing from Linux possible? Answer is [no]( https://web.archive.org/web/20180218032554/https://forums.freebsd.org/threads/kexec-into-freebsd.59123/#post-338575).

tlaurion, 2021-03-02:
If that answers whys not, please close the issue.

FrostKnight, 2021-03-02:
I don't completely understand, but then I don't really know what kexec means... maybe I should look it up.
That being said, Hyperbola plans to move to HyperbolaBSD which will be based on OpenBSD but have many changes to it that are more libre.  Is there anything that the Hyperbola devs could do that would make it possible to use heads to boot HyperbolaBSD?

techge, 2021-03-03:
kexec is a tool to start/change into another linux kernel, thus not for windows, not for BSD or any other non-linux system. This is nothing that is going to be changed by BSD developers and probably nobody else (at least any time soon).

paulmenzel, 2021-03-03:
[Google Engineers Get Windows Booting When Kexec'ed Under Linux](https://www.phoronix.com/scan.php?page=news_item&px=Google-Kexec-Windows-Linux)

techge, 2021-03-03:
Well... not what one expects if they are talking about "booting into Windows", but nice anyway... in a way...

tlaurion, 2021-03-03:
> [Google Engineers Get Windows Booting When Kexec'ed Under Linux](https://www.phoronix.com/scan.php?page=news_item&px=Google-Kexec-Windows-Linux)

"[The engineers involved have been implementing enough of the EFI Boot Services to be able to kexec Windows from Linux. ](https://www.phoronix.com/scan.php?page=news_item&px=Google-Kexec-Windows-Linux)"

[Conference paper going into the details of implementing EFI services](https://2019.osfc.io/talks/windows-boot.html)

FrostKnight, 2021-03-03:
Okay, but are you aware that Hyperbola devs currently have a linux version and are switching to a bsd version derivative in say 2024 or something like that... They did that because they see a better future for security/privacy in making a liberated hard fork of OpenBSD.  They used to work on Parabola GNU/Linux-Libre then they made their own linux forked version of arch with debian patches for stability and in the future they plan to do an OpenBSD hard fork... I tell you all this, because I am not sure your aware, that they are not like the devs at normal bsd derivatives. 

Unless you mean this is completely impossible to do, in which case, I will shut up if you want.

Although, it might not a pressing issue in the near future, and maybe at some point I will have acquired newer hardware that won't be intel anyways.  So,I apologize for wasting your time.

tlaurion, 2021-03-16:
@FrostKnight searched around, and personally hve no idea how linux could kexec into a bsd kernel.

FrostKnight, 2021-03-16:
I mean, I wouldnt know either, but it could be possible if the person developing the bsd, has less bsd mentality and more of say a fusion type thought process.   
Curious, but is the thing that is blocking this idea on the end of heads, or the end of BSD or both?
Not expecting you guys to  do anything, besides hopefully reply.  

I just wondered if this is something that could be fixed if the BSD developer decided to change things ever so slightly.

You said this was the issue way back,  https://github.com/osresearch/heads/issues/697#issuecomment-600069051

I wondered, if that is easy to fix or if its something that is much harder then I realize.  

Aka, if its a gpl rejection issue, or something more problematic.  

Hyperbola devs don't have a problem with GPL or other copyleft licenses. :)

Thanks for checking around though.  I appreciate it.

LiamDGray, 2021-03-16:
> You said this was the issue way back, #697 (comment)

FrostKing: Sounds like the issue is that Heads is not just a bootloader.
It's not a drop-in substitute for BIOS/MBR or UEFI/GPT that another OS
would expect to see. It provides a Linux-specific set of drivers and is, in
fact, Linux.

tlaurion, 2021-03-16:
@FrostKnight If kexec was modified in a way that all memory space is replaced by a BSD kernel, returning into real mode, I do not see why it couldn't be possible.

Kexec is made to put kernel into new address space and switch control to it. That kernel is expected to depend onto its initrd drivers and such to make the switch, which doesn't seem to have been fine between OSes. That would be a kexec project, on which if working, could be integrated into heads. As said previously,  I'm not aware of any work happening to switch from a Linux kernel into BSD.

FrostKnight, 2021-03-16:
Hmm, interesting, is there a kexec repo that I could talk to regarding this somewhere down the road?

I guess I will check...

FrostKnight, 2021-03-17:
Is this related to what you mean?
https://github.com/kexecboot/kexecboot

LiamDGray, 2021-03-17:
@FrostKnight: Pretty sure they mean this:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/kernel

kexec is part of the Linux kernel.  When they say Heads *is* Linux, they
are not kidding.

FrostKnight, 2021-03-17:
Lol, if that's the case, then nothing will change for a very, very long time if at all even.  :(

Oh well, thanks for letting me know the actual truth in detail.  Appreciated that I now understand the actual problem.  

I very much doubt Torvaldo is the kind of person you can get to convince that making it possible to boot anything besides linux is possible.  I might be wrong, but it seems very unlikely. Smh... :P  ---> Me

Oh well... I will just wait for the foreseeable future to see what shows up/If I even will care by 2025.  ARM and RISC-V are growing in popularity, who knows if one of those won't catch my interest by the time this laptop is no longer used for Hyperbola.  And that aside, there is also osboot if I am that desperate to use HyperbolaBSD. ;)

tlaurion, 2021-03-17:
@FrostKnight @LiamDGray 

kexec is a tool aimed at replacing a live kernel with another, permitting to 'reboot' machines without hitting the reset button. 

BSD and Linux have their kexec functions inside both of their kernel implementations.

The problem here is that those kernels are bot interchangeable without a lot of research and development. 

Never said it was impossible. See prior comment here.

There are projects for kexec loading BSD kernels from within BSD. The interest here being to boot BSD from Linux as of now. This is not existing. As usual if kernel people of both universes were talking on kexec space, magic could happen. Its an interesting concept, where BSD people wanting to use Heads to boot inside of BSD without using QubesOS is not CURRENTLY possible without development from those communities. 

Hope this is clearer.

FrostKnight, 2021-03-25:
Yeah, it is more clear, I will ask my Hyperbola buddies, much later, like 2023 or something if they can get involved in this.  For now though, I don't see a big need.  If that makes sense.  Given that, Hyperbola's linux version will be supported till 2025 and its BSD version won't be as good as I want it to be till 2025+

Glad to know it isn't impossible though. I wonder where I can find such projects...

Either way, I will ponder this issue later.

Thanks for the info though!

tlaurion, 2021-09-24:
[Petitboot](https://github.com/glevand/petitboot--petitboot#petitboot---a-kexec-based-bootloader) does it?

"Petitboot is an operating system bootloader based on Linux kexec. It can load any operating system image that supports the Linux kexec re-boot mechanism like Linux and FreeBSD. "

FrostKnight, 2021-09-25:
I guess I will wait and see... for now.

tlaurion, 2021-10-08:
It is possible to kexec to BSD from Linux.

FrostKnight, 2021-10-09:
I wish I knew heh... I am not experienced enough with OpenBSD to do this, but I will be nearby in case, someone knows how, etc...