linuxboot/heads #699: USB-Boot Debian installer

alex-nitrokey, 2020-04-06:
Did anybody successfully boot a Debian installation transferred via dd on a USB drive? For example [this image](https://caesar.ftp.acc.umu.se/debian-cd/current/amd64/iso-cd/debian-10.3.0-amd64-netinst.iso).

It appears to me that one can not kexec into the provided kernel/initram of the iso, neither on heads nor an another system. Is this a known bummer or is there anything one can do about it? Or maybe the system message are just not transferred?

### what I does
trying to usb-boot to a Debian installation USB drive, prepared via dd and the install iso

### what does happen
You see the kexec message that it is booting the kernel, nothing happens afterwards (the screen freezes).

### what I expect
booting into the Debian installer

tlaurion, 2020-04-08:
@alex-nitrokey : My guess is that the kernel you try to kexec in has been made incompatible to the patches applied to kexec for it to boot properly. Probably the same issue that in #672 

Haven't reproduced. Just an insight.

Tonux599, 2020-04-09:
I don't think the Debian installer loads the i915 module and relies on just basic graphics (not sure of the terminology here). I'm not sure if you attempt to launch the graphical installer if this helps at all however I'm pretty sure there's no mode setting. This is why Ubuntu's and Qubes installer work. My experience of kexec is that nothing will be displayed on the screen when kexecing into a system with nomodeset from a system where a direct rendering manager module has been used (i915, nouveau, radeon)

alex-nitrokey, 2020-04-09:
> I don't think the Debian installer loads the i915 module and relies on just basic graphics (not sure of the terminology here). I'm not sure if you attempt to launch the graphical installer if this [...]

This actually might be a good hint here. I can test around a bit, but I am not sure, how to add the i915 module to the initrd in the installer and how to set the correct mode setting for the kernel. Any suggestions? I am not that Debian experienced in this regard... and changing the installer iso might be problematic anyway. In best case we could change the kexec command in a way that it works on the kernel/initram as well :thinking:

Tonux599, 2020-04-09:
I assume the i915 module would have to be one that was built with the same kernel as the installer, if the installer uses the same kernel that is in the Debian repos you should be able to get the module from there.

Unfortunately the other option I see is to not allow heads to do any modesetting (remove i915), however this will probably break fbwhiptail and you would have to rely on generic-init.

Again another option although undesirable is to use another machine to install Debian and make sure i915 is loaded into the initrd.

0rb677, 2020-06-03:
Same problem with GuixSD and all other distros.

alex-nitrokey, 2020-06-04:
> Also heads can't boot in FDE disk and can't boot any BSD. I have no idea how to boot OpenBSD from heads.

I would say it is more or less intended that Heads does not boot non-linux kernel, as the kexec is supposed to ensure a chain of trust and kexec does not boot anything other than linux afair.

bemoody, 2020-07-23:
On the Librem 13 (PureBoot beta11), I found that the following patch to kexec seems to solve the issue:

```
--- a/kexec/arch/i386/x86-linux-setup.c
+++ b/kexec/arch/i386/x86-linux-setup.c
@@ -138,7 +138,8 @@
   goto out;
  if (-1 == ioctl(fd, FBIOGET_VSCREENINFO, &var))
   goto out;
- if (0 == strcmp(fix.id, "VESA VGA")) {
+ if (0 == strcmp(fix.id, "VESA VGA")
+     || 0 == strcmp(fix.id, "inteldrmfb")) {
   /* VIDEO_TYPE_VLFB */
   real_mode->orig_video_isVGA = 0x23;
  } else if (0 == strcmp(fix.id, "EFI VGA")) {
```

It's not clear to me, however, what the difference is between VIDEO_TYPE_VLFB and VIDEO_TYPE_EFI - either value seems to work just as well for the Debian kernel.

alex-nitrokey, 2020-07-24:
sound promising, will try it next week myself!

alex-nitrokey, 2020-07-30:
Works like a charm. Tested Debian installer (graphical and text-based) and booting a Ubuntu Installation.

I do not think that this would break anything, but more testing might be useful.

tlaurion, 2020-07-31:
Awesome news. Will test in my next merging spree. PR?

commit 116787fd (2020-08-01): Merge pull request #788 from Nitrokey/kexec-debian

Fix kexec to work with Debian Installer (fixes #699)
commit 83cac9ed (2020-08-01): Fix kexec to work with Debian Installer (fixes #699)

This patch was reported by @bemoody in issue #699

Tested via `BOARD=x230-hotp-verification` on a Thinkpad x230
closed 2020-08-01 by tlaurion