linuxboot/heads #700: e1000e not detecting network card since musl-cross-make switch?

tlaurion, 2020-04-07:
Can someone replicate
```
insmod /lib/modules/e1000e.ko
dmesg
e1000e: probe of 0000:00:19:0 failed with error -9
```
@osresearch @flammit ?

Tried upgrading to latest stable kernel, same thing. Coreboot patches failing with new musl-cross-make? Attempted to put muslc-cross-make to latest commit id without more luck.

Attempts [here](https://app.circleci.com/pipelines/github/tlaurion/heads)

paulmenzel, 2020-04-08:
@tlaurion, please, you are missing quite a lot of needed information in your report. What Linux version (uname -a)? Both Linux kernels? Please attach the coreboot log and Linux logs.

When was the last time this worked? Does the same Linux kernel work with older coreboot? Does it work with upstream coreboot (and GRUB, SeaBIOS, or … as payload)?

The C library shouldn’t have anything to do with coreboot or the Linux kernel.

Do you have `debug` on the Linux kernel command line?

You can trace `e1000_probe,` in `drivers/net/ethernet/intel/e1000e/netdev.c`.

In your original report the return code was `-9`. Now it’s `-3`. Was that a typo or does it change every time?

tlaurion, 2020-04-08:
@paulmenzel 
> In your original report the return code was `-9`. Now it’s `-3`. Was that a typo or does it change every time?
It was -3 and a corrected typo.

I arrive to the same error with built kernels. Always with coreboot 4.8.1:
linux-4.9.80
linux-4.14.62
linux-4.19.63
linux-4.19.114
linux-5.4.30

I will make cleaner builds to received proper help [here](https://app.circleci.com/pipelines/github/tlaurion/heads) once I figure out how to not strip modules from symbols so you guys can jump in easily and troubleshoot directly from qemu (and give insights #516 #701 ).

Will also add debug from coreboot kernel options for qemu-coreboot in the following builds and report in future comments.

tlaurion, 2020-04-09:
@paulmenzel :

So I'm currently building qemu-coreboot debug builds for kernels: 
4.14.62 (master used kernel for reference) : [Gitlab](https://gitlab.com/tlaurion/heads/-/jobs/505450459) [CircleCi](https://app.circleci.com/pipelines/github/tlaurion/heads/97/workflows/d5fa4389-3f27-48e1-85de-7ffbff4f7545/jobs/106)
4.19.63 : [Gitlab](https://gitlab.com/tlaurion/heads/-/jobs/505453606) [CircleCi](https://app.circleci.com/pipelines/github/tlaurion/heads/98/workflows/8fb3c3d2-0355-4e5e-b4a2-183439c8ea23/jobs/107)
5.4.30: [Gitlab](https://gitlab.com/tlaurion/heads/-/jobs/505457874) [CircleCi](https://app.circleci.com/pipelines/github/tlaurion/heads/99/workflows/c7d9b6a2-fb2c-4f73-9300-5b6644820050/jobs/108)

relevant modules changes to make those builds:
```
diff --git a/modules/linux b/modules/linux
index f881793..f90a1f1 100644
--- a/modules/linux
+++ b/modules/linux
@@ -1,6 +1,6 @@
 modules-y += linux
 
-linux_version := 4.14.62
+linux_version := 5.4.30
 linux_base_dir := linux-$(linux_version)
 
 # TODO: fixup the patch process
@@ -13,12 +13,15 @@ linux_kconfig := $(or $(CONFIG_LINUX_CONFIG),config/linux.config)
 linux_dir := $(linux_base_dir)/$(notdir $(basename $(linux_kconfig)))
 
 linux_tar := linux-$(linux_version).tar.xz
-linux_url := https://cdn.kernel.org/pub/linux/kernel/v4.x/$(linux_tar)
+#linux_url := https://cdn.kernel.org/pub/linux/kernel/v4.x/$(linux_tar)
+linux_url := https://cdn.kernel.org/pub/linux/kernel/v5.x/$(linux_tar)
 
 linux-4.9.38_hash := 76d789d87dd51d2fd58c095727171984fa4a992f5e25b9e3eb1e5fd5cd129074
 linux-4.9.80_hash := 9e2e83ccc0afc3f23340ed5e58a35d8c6300a7c58aa98ca913848de41226477b
 linux-4.14.62_hash := 51ca4d7e8ee156dc0f19bc7768915cfae41dbb0b4f251e4fa8b178c5674c22ab
-
+linux-4.19.63_hash := 75988760b931864b46292dcfad1dd54b3f4da10168041f48ca6d7f6dd4e5d25d
+linux-4.19.114_hash := 1e40a0dc6afc95a259f97b80d5f5ef8f89e2ee49e993ba6844e2bc55de361f0e
+linux-5.4.30_hash := 11dd78f701bce619d90d3b2ee597601716c48087e159c890c1decd7b90349def
 linux_hash := $(linux-$(linux_version)_hash)
 
 # Ensure that touching the config file will force a reconfig/rebuild
@@ -135,10 +151,17 @@ $(module_initrd_lib_dir)/$(notdir $1): $(build)/$(linux_dir)/$1
 	$(call do,INSTALL-MODULE,$1, \
 		$(CROSS)strip \
 			--preserve-dates \
-			--strip-debug \
 			-o "$$@" \
 			"$$<" \
 	)
+#$(module_initrd_lib_dir)/$(notdir $1): $(build)/$(linux_dir)/$1
+#	$(call do,INSTALL-MODULE,$1, \
+#		$(CROSS)strip \
+#			--preserve-dates \
+#			--strip-debug \
+#			-o "$$@" \
+#			"$$<" \
+#	)
 endef
 
 $(call map,linux_module,$(linux_modules-y))
```

Relevant qemu-coreboot board config change:
```
diff --git a/boards/qemu-coreboot/qemu-coreboot.config b/boards/qemu-coreboot/qemu-coreboot.config
index aeb57c7..a16256b 100644
--- a/boards/qemu-coreboot/qemu-coreboot.config
+++ b/boards/qemu-coreboot/qemu-coreboot.config
@@ -39,6 +39,8 @@ CONFIG_LINUX_AHCI=y
 CONFIG_LINUX_USB=y
 CONFIG_LINUX_E1000=y
 
+export CONFIG_BOOT_KERNEL_ADD="debug"
+
 #Uncomment only one BOOTSCRIPT:
 #Whiptail-based init (text-based or FBWhiptail)
 #export CONFIG_BOOTSCRIPT=/bin/gui-init

```

Else, let's remember that those [linux and coreboot patches](https://gitlab.com/tlaurion/heads/-/tree/PrivacyBeast_X230_QubesOS_Certified_bin_modules_not_equal_to_heads_master/patches) are [applied](https://gitlab.com/tlaurion/heads/-/jobs/505453606#L3239) prior of compilation.

To debug, download [artifact](https://gitlab.com/tlaurion/heads/-/jobs/505453606/artifacts/download), extract, and run [qemu against rom per board instruction](https://gitlab.com/tlaurion/heads/-/blob/PrivacyBeast_X230_QubesOS_Certified_bin_modules_not_equal_to_heads_master/boards/qemu-coreboot/qemu-coreboot.config#L56):
```
qemu-system-x86_64 \
		--machine q35 \
		--serial /dev/tty \
		--bios /path/to/downloaded/artifact/coreboot.rom \
	; stty sane
```

From qemu Heads,go to recovery shell:
```
insmod /lib/modules/e1000.ko
dmesg
```
no eth0

As far as Heads code change, neither coreboot nor linux code changed but to fit musl-cross-make changes from this [commit](https://github.com/osresearch/heads/commit/b4a647c485299253715321111461a8577c68661b) up to now. The building toolset changed from musl-cross to musl-croos-pin to musl-cross-make.

@paulmenzel @MrChromebox @jan32 @kylerankin @osresearch @flammit : any help and insight welcome. This behavior is observable on qemu-coreboot and on x230 for what I can tell.

orangecms, 2020-04-09:
Have you tried this?

https://github.com/orangecms/webboot/blob/boot-via-qemu/run-webboot.sh#L12

tlaurion, 2020-04-09:
@orangecms: this is to replicate a problem that is occuring on real hardware: the network devices (covered by e1000e and e1000 drivers) are resulting in probing error. Not sure how this would resolve both qemu and hardware seen error from heads latest commit.

tlaurion, 2020-04-12:
Complementary image:

[image]

tlaurion, 2020-04-13:
> Have you tried this?
> 
> https://github.com/orangecms/webboot/blob/boot-via-qemu/run-webboot.sh#L12

I infirm the problem is not present under qemu image as per @orangecms advice:
`-netdev user,id=u1 -device e1000,netdev=u1` permits the device to be properly detected under qemu. #516

tlaurion, 2020-04-13:
> Complementary image:
> 
> [image]

@osresearch @orangecms:
Same kernel, same coreboot. Different results on hardware then on qemu. The x230 is having that error where probing the ethernet card just doesn't work.

Any insight welcome.
Artifacts for x230 [here](https://app.circleci.com/pipelines/github/tlaurion/heads/110/workflows/808b4fdd-304d-40ad-b0e2-d2872430e246/jobs/119)

tlaurion, 2020-04-21:
@paulmenzel 

Here are the logs from cbmem, lspci and dmesg from[ unmodified current master CI's x230 board artifact](https://app.circleci.com/pipelines/github/osresearch/heads/145/workflows/f07a2f39-eaa6-4d79-b0a8-fe9e472eb361/jobs/152/artifacts):

[x230-master.cbmem_console.txt](https://github.com/osresearch/heads/files/4509714/x230-master.cbmem_console.txt)
[x230-master.dmesg_after_e1000e_insmod.txt](https://github.com/osresearch/heads/files/4509715/x230-master.dmesg_after_e1000e_insmod.txt)
[x230-master.lspci_verbose.txt](https://github.com/osresearch/heads/files/4509716/x230-master.lspci_verbose.txt)

tlaurion, 2020-04-21:
@paulmenzel @osresearch 

Here are the logs from cbmem, lspci and dmesg from modified master CIs (x230-debug) branch to have enough space to include[ qemu-coreboot's debug options](https://gitlab.com/tlaurion/heads/-/commit/e14739165e284df8aed270e6328d3e976cf5b853) inside of x230 board. [Artifact](https://gitlab.com/tlaurion/heads/-/jobs/520181724)

Same result:

tlaurion, 2020-04-21:
Getting my head wrapped around [dynamic-debug]( https://www.kernel.org/doc/html/v4.15/admin-guide/dynamic-debug-howto.html)....

tlaurion, 2020-04-22:
Attempts happening [here](https://app.circleci.com/pipelines/github/tlaurion/heads?branch=x230-debug)

tlaurion, 2020-04-23:
@paulmenzel Ok. We now have a [working x230-debug board](https://app.circleci.com/pipelines/github/tlaurion/heads/128/workflows/406d2ff6-4957-4cb6-8db9-814e07055ab2/jobs/137/artifacts)

Here the interesting output full files:
[x230-debugfs-cbmem_console.txt](https://github.com/osresearch/heads/files/4524499/x230-debugfs-cbmem_console.txt)
[x230-debugfs-dmesg.txt](https://github.com/osresearch/heads/files/4524501/x230-debugfs-dmesg.txt)
[x230-debugfs-dyndebug_output.txt](https://github.com/osresearch/heads/files/4524502/x230-debugfs-dyndebug_output.txt)

tlaurion, 2020-04-23:
Can debug further if needed with some guidance.[ dynamic_debug doc here](https://www.kernel.org/doc/html/latest/admin-guide/dynamic-debug-howto.html)

tlaurion, 2020-04-26:
@osresearch @orangecms @flammit @paulmenzel 

What leads me to think that the problem is linked to ACPI or coreboot being denatured by latest musl-cross-make is that the impact is not just for Heads kernel. When main OS boots, the OS is not able to acquire 00:19 pci device either. Your insights are needed.

paulmenzel, 2020-04-27:
I think, musl does not affect the coreboot build, does it? Did you try upstream coreboot? I am not aware of these issues having been reported to https://github.com/merge/skulls/.

paulmenzel, 2020-04-27:
The issue title still contains *x230*. Does it affect QEMU?

tlaurion, 2020-04-27:
> The issue title still contains _x230_. Does it affect QEMU?

No. 

qemu-coreboot problem was that qemu run command was not binding VM device with host device device. This [PR fixing qemu-coreboot](https://github.com/osresearch/heads/pull/707) with this [line](https://github.com/osresearch/heads/pull/707/files#diff-2c1abc05ab74ad59746fdc5e157bce85R61).

This issue is for x230 with requested debug traces.

tlaurion, 2020-04-27:
> I think, musl does not affect the coreboot build, does it? Did you try upstream coreboot? I am not aware of these issues having been reported to https://github.com/merge/skulls/.

@paulmenzel 
Skulls is not booting a linux kernel, depends on seabios if I recall well and doesnt kexec a linux kernel. 

In Heads present scenario, the problem seems to be (but i'm no expert there) in coreboot making the device available for linux to probe, inside of Heads and in kexec'ed OS booted. 

The fact that the kexeced kernel (OS) has the same behavior has to mean that the coreboot/linux [traces here](https://github.com/osresearch/heads/issues/700#issuecomment-618574556) have to be in cause.

@merge @PatrickRudolph
[from](https://github.com/osresearch/heads/pull/709#issuecomment-619844199) 
`Extracting blobs is required as it contains the MAC of the GbE.` 

Could it be the cause? (generic ifd with extracted minimal ME in this.)

tlaurion, 2020-04-28:
@hughsie: Read your [blog](https://blogs.gnome.org/hughsie/2017/08/08/building-local-firmware-in-fwupd/) post. Is there any way to redistribute gbe?

PatrickRudolph, 2020-04-28:
Will do some tests using cleared and non existing Gbe in IFD.

tlaurion, 2020-05-01:
[Update](https://app.circleci.com/pipelines/github/tlaurion/heads/134/workflows/80b03756-62ff-4796-9f38-a6d7cd757671/jobs/143/artifacts)

Output: 
[dmesg_debug_with_e1000e_debugfs_kernel_opt.txt](https://github.com/osresearch/heads/files/4566375/dmesg_debug_with_e1000e_debugfs_kernel_opt.txt)
[cbmem_debug_with_e1000e_debugfs_kernel_opt.txt](https://github.com/osresearch/heads/files/4566376/cbmem_debug_with_e1000e_debugfs_kernel_opt.txt)

Next step is to add gbe blob and retest....

PatrickRudolph, 2020-05-12:
Added test results here: https://review.coreboot.org/c/coreboot/+/41290
The e1000e returns error -3 if the gbe region was corrupted, for example by writing zeros to it.

tlaurion, 2020-05-12:
@PatrickRudolph : or not including it in coreboot blobs results in the same behavior.
Next step is to see if some randomized data can be put on the mac address section of the blob? Or is there a lot of data being different across gbe.rom?

tlaurion, 2020-05-12:
@PatrickRudolph :
Based on [last commit](https://github.com/osresearch/heads/pull/703) artifacts and MAC associated with that binary:
```
user@x230-external-flash:~/heads$ binwalk -R "\xF0\xDE\xF1\xF7\xE4\xAF" blobs/xx30/gbe.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Raw signature (\xF0\xDE\xF1\xF7\xE4\xAF)
4096          0x1000          Raw signature (\xF0\xDE\xF1\xF7\xE4\xAF)

```

PatrickRudolph, 2020-05-13:
I did a quick search and found the gbe area documented for 

> Intel 6 Series Chipsets

 in the datasheet:

> Intel® 82579 Gigabit Ethernet PHY

Chapter _LAN NVM Format and Contents_.

It's using an A/B partition scheme where only 128bytes are checksummed, but way less are actually used.

tlaurion, 2020-05-20:
@PatrickRudolph : can we set the mac to `00:00:00:00:00:00` on both partition and change checksum and we are all set?!

tlaurion, 2020-06-18:
@PatrickRudolph 
> I did a quick search and found the gbe area documented for
> 
> > Intel 6 Series Chipsets
> 
> in the datasheet:
> 
> > Intel® 82579 Gigabit Ethernet PHY
> 
> Chapter _LAN NVM Format and Contents_.
> 
> It's using an A/B partition scheme where only 128bytes are checksummed, but way less are actually used.

https://notabug.org/libreboot/libreboot/src/master/resources/utilities/ich9deblob/src/gbe/gbe.c

tlaurion, 2020-06-19:
Unfortunately, replacing extracted gbe.bin with generated ifd+gbe, then extracted gbe region inserted in rom under https://github.com/osresearch/heads/pull/703/commits/9f8746320162d2350203f4822681f45760bc5732 causes a brick just per that commit.

contacted @swiftgeek per e-mail to port work to sandy/ivy.

tlaurion, 2020-06-20:
So, to be able to generate gbe blobs (ich9gen wast obsoleted in favor of bincfg), one would need to adapt this to ICH6 series

@9elements @3mdeb anyone?

tlaurion, 2020-06-20:
> Intel 6 Series Chipsets

pdf [here](https://cdrdv2.intel.com/v1/dl/getContent/613456)

From my understanding, [gbe-ich9 .set and .spec files](https://github.com/coreboot/coreboot/blob/master/util/bincfg/gbe-ich9m.spec) should adapted to ich6?

tlaurion, 2020-06-20:
@PatrickRudolph @niftygeek @piotr @pgeorgi @gnutoo (last two authors of the tool)

So in my novice path, to be able to generate a valid gbe.bin, I need to: 
- Read specs
  - http://www.intel.co.uk/content/dam/doc/application-note/i-o-controller-hub-9m-82567lf-lm-v-nvm-map-appl-note.pdf
  - https://communities.intel.com/community/wired/blog/2010/10/14/how-to-basic-eeprom-checksums 
- Understand the mappings that led to the creation of gbe-ich6.spec out of precedent spec
- Test until `user@x230-master:~/coreboot/util/bincfg$ ./bincfg -d gbe-ich6.spec heads/blobs/xx30/gbe.bin gbe-ich6.set` creates a valid gbe-ich6.set file
- Test until `./bincfg gbe-ich6.spec gbe-ich6.set heads/blobs/xx30/gbe_mod.bin` produces a valid gbe_mod.bin that can be included in a bootable rom under https://github.com/osresearch/heads/pull/703 ?

tlaurion, 2020-06-20:
```
00:07.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (Lewisville) (rev 04)
	Subsystem: Lenovo Device 21f3
	Physical Slot: 7
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 64
	Interrupt: pin ? routed to IRQ 71
	Region 0: Memory at f2020000 (32-bit, non-prefetchable) [size=128K]
	Region 1: Memory at f2076000 (32-bit, non-prefetchable) [size=4K]
	Region 2: I/O ports at c200 [size=32]
	Capabilities: [c8] Power Management version 2
		Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [d0] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fee37000  Data: 4300
	Kernel driver in use: e1000e
	Kernel modules: e1000e

```

PatrickRudolph, 2020-06-24:
Yes that's the way to go. Either use bincfg or like coreboot does: convert a hex-dump to binary, apply some bash magic to insert the MAC and then calculate the checksum.

tlaurion, 2020-06-28:
@PatrickRudolph would that make that binary non legal-issue prone?
Where can I see that hex-dumo to binary trick coreboot uses?

closed 2020-11-05 by tlaurion

tlaurion, 2020-11-09:
gbe can now be generated from bincfg https://github.com/osresearch/heads/issues/796