linuxboot/heads #702: Booting Arch Linux

techge, 2020-04-09:
Looking at [this issue](https://github.com/osresearch/heads/issues/699#issuecomment-611274089) in another context, I found out how to fix an issue I had in Arch Linux so far.

### What is happening
When kexec is starting the Arch Linux kernel and initramfs, the screen is not showing anything. I found out that the system is booting normally, if I just blindly type in my luks passphrase.

### How it should look
You should be able to see early messages of the initramfs, instead you only the kernel messages after assuming what is happening.

### How to fix
[Adding i915 module to the initramfs](https://wiki.archlinux.org/index.php/Kernel_mode_setting#Early_KMS_start) fixes this issue.

tlaurion, 2022-02-22:
@techge another user has issue under #1117. Do you have any insight why i915 is not provided per ArchLinux under initrd?
Bug or feature? Should they fix it?

tlaurion, 2022-02-22:
To my eyes, it seems that the initramfs **should** include i915.

techge, 2022-02-23:
As mentioned in [my first post above](https://github.com/osresearch/heads/issues/702#issue-597152200) i915 is included in initrd once `i915` is included in die `MODULES`
part of the `/etc/mkinicpio.conf`. Is it a bug? I don't think so. But then I guess it depends on how you expect a distro to behave...

> https://github.com/osresearch/heads/pull/1120 would like your testing @techge!!!!

Not sure if I get to it any time soon. Actually, I changed my setup and only use my x230 for some tests and specialities and not on a everyday basis any more. Actually I re-flashed with coreboot this weekend and I'd need to flash heads again... Anyway, I switch over to the PR...

tlaurion, 2022-02-25:
@techge so basically, i understand from your experience that adding i915 in initrd, from the booted installer iso, should work around the issue.

techge, 2022-02-28:
> @techge so basically, i understand from your experience that adding i915 in initrd, from the booted installer iso, might work around the issue.

That is correct. The way I see it, this has nothing to do with heads. As I said elsewhere, I used coreboot on my x230 these days and had the same issue. I needed to find out about this *again* :smile:

tlaurion, 2022-07-27:
@lynxred ? #1117 was a duplicate of #702.
Have you built i915 driver to be put in initrd?

tlaurion, 2023-09-24:
It would be interesting to know which modules are packed by default under initrd.

As per recent work, we expect efifb to be able to drive simpledrm and simplefb and efifb

tlaurion, 2026-08-16:
@techge still relevant since https://github.com/linuxboot/heads/pull/2186 ?

techge, 2026-08-20:
Sorry, don't know, not using heads anymore.