linuxboot/heads #711: Swap fn and Ctrl key

Siproqu, 2020-05-02:
I successfully built heads for my T430s (pull requests are coming soon). As the fn and ctrl key are swapped on this model (the ctrl key is not under to the shift key) it is extremely annoying to use Qubes with its ctrl+shift+c/v shortcuts.

Usually people using the nvramcui to swap them. But how should I do it in Heads?
Do I have to rebuilt Heads with nvramcui enabled in the coreboot settings? Or is there another way?

Siproqu, 2020-05-19:
It is possible to patch the cmos layout in the coreboot source. Unfortunately I get lost in the Heads source code and couldn't find out how to add an additional option for the patch.
[Pull request](https://github.com/osresearch/heads/pull/723/commits/d4eed84a5536b9ef5529d04e400a9cafc48d324f)

Siproqu, 2020-05-30:
Solution:
Build Heads with the `CONFIG_USE_OPTION_TABLE=y` option in the coreboot-t430s.config and copy [this](https://github.com/osresearch/heads/pull/723/commits/d4eed84a5536b9ef5529d04e400a9cafc48d324f) patch file into `heads/patches/coreboot-4.8.1/`

closed 2020-05-30 by Siproqu

copyvar, 2022-01-09:
I have the same issue, but I am on x230 with a classic keyboard (https://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads). I was suggested to apply the fn and ctrl swap via nvramtool (see 
https://github.com/osresearch/heads/issues/1087#issuecomment-1008076761). I need some help in applying, as I am not skilled enough to do it on my own. Should I run nvramtool from dom0?

Siproqu, 2022-01-09:
> I was suggested to apply the fn and ctrl swap via nvramtool.

Have you followed the instructions on [this page](https://www.coreboot.org/Nvramtool)?

If not, do this first:
- Build `nvramtool` in a disposable VM (as described in the link under "[Manual Installation](https://www.coreboot.org/Nvramtool#Installation)").
- Copy the `nvramtool` executable to dom0.
- Execute it in dom0 and change your settings (`./nvramtool -h` will explain how to change a setting).

Please keep in mind: **Executing nvramtool in dom0 might be a threat to your system security.**
If you run Qubes simply for the fun and curiosity its fine I guess. But only you know that.

If this did not work, you can still rebuild Heads with the changed cmos settings (swapped fn&ctrl key and the like) and flash it again.
I've wrote a description how to do it here:
https://github.com/Siproqu/heads-wiki/blob/T430s-guides/Installing-and-Configuring/Building-Heads/t430s.md#bios-options

Hope this helps. A more simple step by step explanation takes to much time for me as well unfortunately. The best I can do at the moment is already covered in my wiki linked above.

tlaurion, 2022-01-09:
@Siproqu is there a PR to merge upstream?

Siproqu, 2022-01-09:
> @Siproqu is there a PR to merge upstream?

Yes there is: https://github.com/osresearch/heads-wiki/pull/77
Haven't you seen it? I was under the impression you wait with merging until I tested https://github.com/osresearch/heads/pull/1019 with the T430 blobs as requested (which I'll hopefully be able to do in the next weeks).

tlaurion, 2022-01-11:
@Siproqu yes sorry, just totally forgot about it!

Interesting that you could just hexedit Mac address in both sections. Will test this on x230 soon

copyvar, 2022-01-12:
I did
```
git clone https://github.com/osresearch/heads
git clone https://github.com/coreboot/coreboot.git
nano coreboot/src/mainboard/lenovo/x230/cmos.default 
git diff cmos.default > ~/heads/patches/coreboot-4.14/0061-cmos-settings.patch
cd ~/heads/
./blobs/xx30/download_clean_me.sh -m $(readlink -f ./blobs/xx30/me_cleaner.py)
make BOARD=x230-hotp-maximized
```
what results in 
```
build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1134-gfba79ab-dirty.rom
build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1134-gfba79ab-dirty-bottom.rom
build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1134-gfba79ab-dirty-top.rom
```
Is that correct? I am confused about "dirty".