linuxboot/heads #712: any recommendations how to flash KGPE-D16 ?

0rb677, 2020-05-03:
Is raspberry pi enough to power motherboard or do i need external power supply?
I have built 16MB rom firmware using `make BOARD=kgpe-d16` without rom dumps like i did on x220 (extract.sh)
Do i need connect  2x 8PIN EPS12V cables ?
[image]

Tonux599, 2020-05-04:
Hi @0rb677 

Do you mean W25Q128FVIQ as opposed to W25Q128FVSG? The W25Q128FVIQ is what I use and is a DIP8 chip as opposed to W25Q128FVSG which is a surface mounted chip.

Also check [here](https://www.coreboot.org/Board:asus/kgpe-d16#RAM_HCL) for good RAM modules to use which may be affecting your successful boots. Also you will be required to use more than a single memory modules if you intend on using more than one CPU (I notice two heat sinks but one stick of RAM). Each processor is allocated memory from RAM located above/below it.

0rb677, 2020-05-06:
@Tonux599 hi and thanks :)
`W25Q128FVIQ dip-8` just it ordered on ebay and aliexpress. wait until the chip arrives
How is performance? Do i need to tweak config for acpid onboard graphics? Does the motherboard work well with heads? Can you recommend a memory that is 100% wound up? 
I bought 4x16GB used ram  `MT36KSF2G72PZ-1G6P1NE` but works only two (`A2` and 'C2' slots)
it seems that two of them are just non-working

mikebdp2, 2020-05-08:
@0rb677 **1)** Please consider adding this info to coreboot's wiki HCL - for other people to see. Otherwise, nobody - except the readers of this thread - could know about you experiences: the images you've uploaded _(without retyping the text from them)_ aren't searchable by a RAM model number. Also, have you ever discussed this RAM issue on coreboot's mailing lists or reddit's r/coreboot?

**2)** It could be that RAM has worked in the past and got broken in later coreboot versions. Have you checked the experiences of the other people with this RAM? There could've been a code regression.

**3)** RAM init code is a bit complicated, but it's the opensource - and you could read some of debugging advices at the comments here https://www.reddit.com/r/coreboot/comments/gf9b1m/coreboot_on_an_amd_gseries_kabini_based_hp_thin/ I've used this approach successfully to add the XMP / custom timings support for the coreboot AMD boards with opensource AGESA. These patches could be found on https://review.coreboot.org/ by searching for "AMD_XMP" or just applying the https://review.coreboot.org/c/coreboot/+/40495 . With this patch, you could enforce the custom timings on your RAM by substituting the SPD readings, and maybe that will help to get your RAM working.

tlaurion, 2020-05-08:
> @0rb677 **1)** Please consider adding this info to coreboot's wiki HCL - for other people to see. Otherwise, nobody - except the readers of this thread - could know about you experiences: the images you've uploaded _(without retyping the text from them)_ aren't searchable in the Internet by a RAM model number. Also, have you ever discussed this RAM issue on coreboot's mailing lists or r/coreboot?
> 
> **2)** It could be that RAM has worked in the past and got broken in later coreboot versions. Have you checked the experiences of the other people with this RAM? There could've been a code regression.
> 
> **3)** RAM init code is a bit complicated, but it's the opensource - and you could read some of the RAM debug advices at the comments here https://www.reddit.com/r/coreboot/comments/gf9b1m/coreboot_on_an_amd_gseries_kabini_based_hp_thin/ . I've used this approach successfully to add the XMP / custom timings support for the coreboot AMD boards with opensource AGESA. These patches could be found on https://review.coreboot.org/ by searching for "AMD_XMP" or just applying the https://review.coreboot.org/c/coreboot/+/40495 . With this patch, you could enforce the custom timings on your RAM by substituting the SPD readings, and maybe that will help to get your RAM working.

Some trace of available doc:
https://www.raptorengineering.com/coreboot/kgpe-d16-status.php
https://libreboot.org/docs/hardware/kgpe-d16.html
https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-status.php

Memory chips known to work:
Tired to find doc in history and haven't found quickly... Took note, will search later

tlaurion, 2020-05-08:
@mikebdp2 @0rb677 
make BOARD=kgpe-d16 will produce headless board as it is now, other pull requests under heads have been made to split server and workstation configurations. It is also possible to deactivate AST onboard graphic card, while board's associated linux kernel config will need to be tweaked to support your chosen card. I owuld advise doing a PR when you are successful, in the form:
`kgpe-d16-wks-graphic_card`

Ping me in your progress, I can definitely give a hand and help you document things out in another https://github.com/osresearch/heads-wiki page

This project's documentation needs a lot of love, I know.

0rb677, 2020-05-23:
@tlaurion package on the way. 
Do you need any hardware findings?
Gpu passthrouth works good on proxmox.
Here is iommu configs and my Video cards

# Input Output Memory Management Unit

GPU ROM dumb

```
cd /sys/bus/pci/devices/0000:02:00.0/
echo 1 > rom
cat rom > /usr/share/kvm/<GPURomFileName>.bin
echo 0 > rom
```
```
root@proxmox:~# lsmod | grep vfio
vfio_pci               49152  0
vfio_virqfd            16384  1 vfio_pci
irqbypass              16384  2 vfio_pci,kvm
vfio_iommu_type1       28672  0
vfio                   32768  2 vfio_iommu_type1,vfio_pci
```
```
root@proxmox:~# cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
vfio
vfio_iommu_type1
vfio_pci
vfio_virqfd
```
```
root@proxmox:~# cat /etc/modprobe.d/kvm.conf 
options kvm ignore_msrs=1
```
```
root@proxmox:~# cat /etc/modprobe.d/vfio.conf 
options vfio-pci ids=10de:1c03,10de:10f1 disable_vga=1
```
```
root@proxmox:~# cat /etc/modprobe.d/blacklist.conf 
blacklist radeon
blacklist nouveau
blacklist nvidia
```
```
root@proxmox:~# cat /etc/modprobe.d/iommu_unsafe_interrupts.conf 
options vfio_iommu_type1 allow_unsafe_interrupts=1
```
```
root@proxmox:~# cat /etc/modprobe.d/pve-blacklist.conf 
# This file contains a list of modules which are not supported by Proxmox VE 

# nidiafb see bugreport https://bugzilla.proxmox.com/show_bug.cgi?id=701
blacklist nvidiafb
```

```
root@proxmox:~# lspci -n -s 02:00
02:00.0 0300: 10de:1c03 (rev a1)
02:00.1 0403: 10de:10f1 (rev a1)
root@proxmox:~# lspci -n -s 05:00
05:00.0 0300: 10de:1c02 (rev a1)
05:00.1 0403: 10de:10f1 (rev a1)
```
```
root@proxmox:~# lspci -vnn | grep VGA -A 12
[... 73 lines cut ...]
IOMMU Group 0 00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD/ATI] RD890 Northbridge only dual slot (2x16) PCI-e GFX Hydra part [1002:5a10] (rev 02)
IOMMU Group 1 00:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GFX port 0) [1002:5a16]
IOMMU Group 10 00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 LPC host controller [1002:439d]
IOMMU Group 11 00:14.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 PCI to PCI Bridge [1002:4384]
IOMMU Group 11 01:01.0 VGA compatible controller [0300]: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000] (rev 10)
IOMMU Group 11 01:02.0 FireWire (IEEE 1394) [0c00]: LSI Corporation FW322/323 [TrueFire] 1394a Controller [11c1:5811] (rev 70)
IOMMU Group 12 00:14.5 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI2 Controller [1002:4399]
IOMMU Group 13 05:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106 [GeForce GTX 1060 3GB] [10de:1c02] (rev a1)
IOMMU Group 13 05:00.1 Audio device [0403]: NVIDIA Corporation GP106 High Definition Audio Controller [10de:10f1] (rev a1)
IOMMU Group 14 04:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
IOMMU Group 15 03:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
IOMMU Group 16 02:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] [10de:1c03] (rev a1)
IOMMU Group 16 02:00.1 Audio device [0403]: NVIDIA Corporation GP106 High Definition Audio Controller [10de:10f1] (rev a1)
IOMMU Group 2 00:09.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 4) [1002:5a1c]
IOMMU Group 3 00:0a.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 5) [1002:5a1d]
IOMMU Group 4 00:0b.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] RD890/RD990 PCI to PCI bridge (PCI Express GFX2 port 0) [1002:5a1f]
IOMMU Group 5 00:11.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 SATA Controller [AHCI mode] [1002:4391]
IOMMU Group 6 00:12.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1002:4397]
IOMMU Group 6 00:12.1 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0 USB OHCI1 Controller [1002:4398]
IOMMU Group 6 00:12.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1002:4396]
IOMMU Group 7 00:13.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1002:4397]
IOMMU Group 7 00:13.1 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0 USB OHCI1 Controller [1002:4398]
IOMMU Group 7 00:13.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1002:4396]
IOMMU Group 8 00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 SMBus Controller [1002:4385] (rev 3d)
IOMMU Group 9 00:14.1 IDE interface [0101]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 IDE Controller [1002:439c]

alexmaloteaux, 2020-05-28:
Hi for the ram issue,  i set up a kgpe-d16 server with coreboot 4.10 and raptor openbmc . Im using this registered ecc ram that you can find on ebay for normal price and that is mentionned on the wiki : Micron 16GB server memory Dimm 2Rx4 Ram DDR3 MT36KSF2G72PZ-1G6E1FE PC3L-12800R

I have 32 Gb at the moment and 32 Gb delivered soon, All seems stable so far .

Regarding heads i have 2 questions : 

1) the vga port is unusable right ? I just have a HEADS ascii art logo and then nothing, iiuc one need to recompile OpenBMC with specific head tools ?

2) is there any advantage / possibility to chain heads to boot a bootloader that will boot a freebsd based legacy zfs bootloader ? Or is it possible direclty ?

0rb677, 2020-05-30:
yeah, fucking chip arrived
Wish me good luck :)
```
suxx@lab:~$ sudo flashrom -p ch341a_spi
flashrom v1.2 on Linux 5.4.0-31-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q128.V" (16384 kB, SPI) on ch341a_spi.
No operations were specified.
```

0rb677, 2020-05-31:
@tlaurion i dont understand, after first start i have no video signal. I need connect over serial console and flash BMC? is that right?

0rb677, 2020-05-31:
I dont understand. 

> Current situation takes for granted that you reprogrammed the BMC internally with provided Heads script.

@tlaurion What does it mean? I need to use `Open On-Chip Debugger`, also buy `ARM-USB-TINY` and `FT2232HL` spi programmator for BMC module? 

> The system boots without outputting console to the vga graphic card until the system setups it's own console (both coreboot, heads and password output are sent to serial console and KVM). 
What passwords? I have clean HDD and replaced bios by heads. There is no video signal. I cant boot any system from usb flash.
Also there is no `ASMB4 or ASMB5 KVM firmware module` on this motherboard by default. Do i need to buy it?
Or it can be flashed by `UART` from console use?

alexmaloteaux, 2020-05-31:
> I dont understand.

>     Current situation takes for granted that you reprogrammed the BMC internally with provided Heads script.

check "Transfer Flash image to destination host" here :  [https://raptorengineering.com/coreboot/kgpe-d16-bmc-port-status.php](url)

> @tlaurion @mikebdp2 What does it mean? I need to use Open On-Chip Debugger, also buy ARM-USB-TINY and FT2232HL spi programmator for BMC module?

No, OpenOcd and a 20 pin jtag debugger is if you want to debug/improve OpenBmc Booting process

> There is no video signal. I cant boot any system from usb flash.

By default heads does not input anything until you input /configure through the uart port from within OpenBMC. Actually in my case there is a HEAD logo...
Your system is not booting because i thing you have flashed OpenBMC that is 16Mb too on the Bios Chip and not the IPMI  module

> Also there is no ASMB4 or ASMB5 KVM firmware module on this motherboard by default. Do i need to buy it?

Yes 

> Do i need to buy this https://www.asus.com/com/Commercial-Servers-Workstations/ASMB4iKVM/

I have ASMB5 and it works perfectly

> Or it can be flashed by serial console inside heads?

no you have to flash the module from a linux system running on the board with the raptor flashrom fork

> Any chance to get GUI whiptail like on x220/x230 ?

dont know

> My progress is here https://github.com/0rb677/KGPE-D16-HEADS
> I have built bmc port and flashed new chip use master branch.
> When i start PC, there is nothing on the screen. Can you help me more detail?

I think because you flashed OpenBmc on the Bios chip and not on the ipmi module, reflash head in that case on the Bios

0rb677, 2020-05-31:
@alexmaloteaux big thanks for help. I only built `https://github.com/0rb677/KGPE-D16-HEADS/blob/master/flash-asus-20200530212035`. i dont flash it anywhere.

alexmaloteaux, 2020-05-31:
> @alexmaloteaux big thanks for help. i will try again. How do you build heads? make BOARD=kgpe-d16 without any blobs added? thats all?

Yes but i have myself an issue with the TPM being recognized by coreboot but not by tpm-tools (tag issue ...) :( so I use coreboot without screen purely over OpenBmc console until i have more time to debug

0rb677, 2020-05-31:
@alexmaloteaux If it’s not difficult for you, you can describe the working memory and processors? Or if you have success boot 64GB+ ram.

alexmaloteaux, 2020-05-31:
Ram, : 2x Micron 16GB server memory Dimm 2Rx4 Ram DDR3 MT36KSF2G72PZ-1G6E1FE PC3L-12800R
CPU : 2x Opteron 6378
Asus Pike is coming , im using IBM m1015 atm un hba it Mode.

I dont reset CMOS

Im using coreboot/seabios atm over OpenBmc, and trying to fix an issue with head TPM

0rb677, 2020-05-31:
@alexmaloteaux thanks a lot!!!

tlaurion, 2020-05-31:
I have seen your message but I'm away of computer . You should at least be seeing ASPEED message on screen at boot.

Maybe the video jumper on motherboard is set to external video card? It should be set to internal video card.

There is a PR that splits workstation board from server board.

Ill try to get back to you in the next days.
You can also see the maintainer issue and tag other kgpe-d16 users.

0rb677, 2020-06-01:
@tlaurion 
Without `ASMB5 KVM` and BMC reflashed? Maybe bad chips, but programmator detect them.
`You should at least be seeing ASPEED message on screen at boot.`

tlaurion, 2020-06-01:
@0rb677 @alexmaloteaux I've updated https://github.com/osresearch/heads/pull/472 so that it is rebased on master. It makes two boards: a server (which is the equivalent of master, and takes into account that the OpenBMC KVM chip was reflashed and outputs serial output to it, and a workstation board, which outputs everything to tty0 (instead of server board over ttyS1)

Let me know if you have any question. 
As far as I know, coreboot 4.8.1 was not able to deal correctly with TPM and was left behind with a noTPM option using the libremKey as a less advantageous root of trust. VBOOT+measured boot was never tested by me and the KGPE-D16 board stayed at the stage of PoC.

Which is why #719

tlaurion, 2020-06-01:
@0rb677 @alexmaloteaux 

Note that the workstation board could use gui-init if ASPEED (AST) support was added on workstation kernel config (to not depend on shared linux-kgpe-d16.config).

If you need more help in that direction, let me know.
A discussion is happening under #721 on how we could still support the kgpe-d16 and duplicate coreboot config to 4.11 and make it supported into Heads with proper TPM support. Your involvement would be important, since supporting the kgpe-d16 in its current state without #719 is unfortunately not a priority on me and should be a community sideproject for someone that wants it mosre supported in its 4.11 status.

tlaurion, 2020-06-01:
> I dont understand.
> 
> > Current situation takes for granted that you reprogrammed the BMC internally with provided Heads script.
> 
> @tlaurion @mikebdp2 What does it mean? I need to use `Open On-Chip Debugger`, also buy `ARM-USB-TINY` and `FT2232HL` spi programmator for BMC module?
> 
> > The system boots without outputting console to the vga graphic card until the system setups it's own console (both coreboot, heads and password output are sent to serial console and KVM).
> 
> What passwords do you mean? I have clean HDD and replaced bios by heads (16mb spi chip). There is no video signal. I cant boot any system from usb flash.
> Also there is no `ASMB4 or ASMB5 KVM firmware module` on this motherboard by default. Do i need to buy it?
> Do i need to buy this `https://www.asus.com/com/Commercial-Servers-Workstations/ASMB4iKVM/`
> Or it can be flashed by serial console inside heads?
> Any chance to get GUI whiptail like on x220/x230 ?
> My progress is here https://github.com/0rb677/KGPE-D16-HEADS
> I have built bmc port and flashed new chip use master branch.
> When i start PC, there is nothing on the screen. Can you help me more detail?

Missed that. No you can reprogram internally from https://github.com/osresearch/heads/blob/master/initrd/bin/flashrom-kgpe-d16-openbmc.sh

After having built https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-status.php

Tonux599, 2020-06-01:
As I mentioned in #719 but will reiterate here, Asus 90-C1B0AU-00XBN0VZ is a TPM1.2 module that is compatible with this board and heads. In concerns to usability of this board, with that TPM module and a YubiKey [this](https://github.com/Tonux599/heads/tree/kgpe-d16-workstation) branch provides almost identical functionality that heads dose on my thinkpads. I'll be maintaining it until more concert decisions are made on the future of this board.

alexmaloteaux, 2020-06-02:
> Missed that. You can reprogram from Heads the ASMB4iKVM with that script https://github.com/osresearch/heads/blob/master/initrd/bin/flashrom-kgpe-d16-openbmc.sh

Thats nice :)

Any idea where i can buy a Asus 90-C1B0AU-00XBN0VZ, all major suppliers are out of stock

Tonux599, 2020-06-02:
> Any idea where i can buy a Asus 90-C1B0AU-00XBN0VZ, all major suppliers are out of stock

@alexmaloteaux that's the problem, I took a gamble on a ebay listing titled "Asus TPM" which picture was 20pin but looked slightly different than my Asus TPM2 module. For future reference to anyone looking for this module note that the TPM1.2 module has a capacitor to the left of the chip whereas the TPM2 module does not.

tlaurion, 2020-06-03:
Will buy too

0rb677, 2020-06-03:
@tlaurion are you here? trying your branch.
```
suxx@lab:~/Build$ git clone https://github.com/tlaurion/heads
Cloning into 'heads'...
remote: Enumerating objects: 170, done.
remote: Counting objects: 100% (170/170), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 9895 (delta 95), reused 121 (delta 58), pack-reused 9725
Receiving objects: 100% (9895/9895), 4.39 MiB | 6.78 MiB/s, done.
Resolving deltas: 100% (6745/6745), done.
suxx@lab:~/Build$ cd heads/ && git checkout -b kgpe-d16_current_working
Switched to a new branch 'kgpe-d16_current_working'
suxx@lab:~/Build/heads$ make BOARD=kgpe-d16

```

tlaurion, 2020-06-03:
```
rm -rf heads
git clone https://github.com/tlaurion/heads
cd heads
git checkout kgpe-d16_current_working
make BOARD=kgpe-d16_workstation
```

but then you will hit #733

tlaurion, 2020-06-03:
> @tlaurion are you here? trying your branch and another spi chip. (i have two)
> I still can’t understand what the problem is and why I didn’t have a video signal without any KVM and BMC.

Simply because the master branch takes into account that all output goes to BMC and was subject to debates on past tickets. Main branch didnt seperate boards into workstation (without BMC) and server (headless) and my PoC stayed there without receiving much love since I do use it headless and since #719, I have lost a bit of interest and do not have much time to invest into this board support anymore.

> Also I still need to pick up the console from work and buy ASMB5
> Can you help me to build correctly branch for a workstation with whiptal like in x220? Is it possible? External video cards will be GTX 1060. still look RAM settings in the config , what if it starts up all? I have all hope for you guys.

It would, but AST support should be activated in linux config. You can try that path for the workstation linux config, and modify the workstation board configuration to use gui-init instead of generic-init.

If I recall well, it worked before.

> I’ll check the jumpers now.
> 
> ```
> suxx@lab:~/Build$ git clone https://github.com/tlaurion/heads
> Cloning into 'heads'...
> remote: Enumerating objects: 170, done.
> remote: Counting objects: 100% (170/170), done.
> remote: Compressing objects: 100% (101/101), done.
> remote: Total 9895 (delta 95), reused 121 (delta 58), pack-reused 9725
> Receiving objects: 100% (9895/9895), 4.39 MiB | 6.78 MiB/s, done.
> Resolving deltas: 100% (6745/6745), done.
> suxx@lab:~/Build$ cd heads/ && git checkout -b kgpe-d16_current_working
> Switched to a new branch 'kgpe-d16_current_working'
> suxx@lab:~/Build/heads$ make BOARD=kgpe-d16
> ```
> 
> > (which is the equivalent of master, and takes into account that the OpenBMC KVM chip was reflashed and outputs serial output to it, and a workstation board, which outputs everything to tty0 (instead of server board over ttyS1)
> 
> Is it possible to start heads normally? With shitty resolution and use ASPEED default VGA output?

If terminal output goes to /dev/tty0 and not serial terminal (/dev/ttyS*) your should have output if ASPEED shows to your screen (which has to be a hjumper setting, or as someone else stated before, if you flashed wrong image to the 16Mb flash chip.

@0rb677 Let me know how it goes.

0rb677, 2020-06-03:
@tlaurion thank you very much for spending your free time :) I haven’t received so much information anywhere else.

0rb677, 2020-06-03:
i am too and i see your new issue [#735 ](https://github.com/osresearch/heads/issues/735)
```
make[1]: Leaving directory '/home/suxx/Build/heads/build/linux-4.14.62'
2020-06-03 23:22:49+03:00 MAKE musl-cross
```

0rb677, 2020-06-03:
any point to buy `25Q128FVIQ, 128MBit SPI Flash` instead 16mb?

0rb677, 2020-06-03:
this is VGA jumper
[image]

tlaurion, 2020-06-03:
#735 fixed with https://github.com/tlaurion/heads/commit/18f8265e880f4d199466ab282c9261a0fa9925df merged in master.... continuing...

0rb677, 2020-06-03:
@tlaurion while compiling offtop question :  Any ideas how to make friends openbsd and heads? 
Also heads can't boot in FDE disk.

0rb677, 2020-06-03:
Done.
```
2020-06-04 00:47:45+03:00 MAKE coreboot
2020-06-04 00:48:22+03:00 DONE coreboot
"/home/suxx/Build/heads/build/coreboot-4.8.1/kgpe-d16_workstation/cbfstool" "/home/suxx/Build/heads/build/coreboot-4.8.1/kgpe-d16_workstation/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage          174468 none
config                         0x2aa80    raw               572 none
revision                       0x2ad00    raw               561 none
cmos.default                   0x2af80    cmos_default      256 none
cmos_layout.bin                0x2b0c0    cmos_layout      3524 none
fallback/dsdt.aml              0x2bec0    raw             10010 none
(empty)                        0x2e640    null             5976 none
s3nv                           0x2fdc0    raw             65536 none
fallback/ramstage              0x3fe00    stage           88041 none
fallback/payload               0x55640    simple elf    5441458 none
microcode_amd.bin              0x585e40   microcode       12684 none
microcode_amd_fam15h.bin       0x589040   microcode        7876 none
(empty)                        0x58af80   null         10962392 none
bootblock                      0xfff580   bootblock        2120 none
2020-06-04 00:48:22+03:00 INSTALL   build/coreboot-4.8.1/kgpe-d16_workstation/coreboot.rom => build/kgpe-d16_workstation/coreboot.rom
9087d0a185745bd913fcd61da0400011f5d14e526a26a6fd5cb7406b9d81689c  build/kgpe-d16_workstation/coreboot.rom
9087d0a185745bd913fcd61da0400011f5d14e526a26a6fd5cb7406b9d81689c  /home/suxx/Build/heads/build/kgpe-d16_workstation/coreboot.rom

```
Start to flash.

0rb677, 2020-06-03:
@tlaurion Its boot only if external GPU video card in PCI-E slot.

tlaurion, 2020-06-04:
You added ast support in Linux config.as explained above?

0rb677, 2020-06-04:
@tlaurion forgot. rebuilding.

paulmenzel, 2020-06-04:
Guys, it’s really hard to follow these issues, and in my opinion also rude to expect users to read through the thread. (Unfortunately GitHub comments do not support threading.)

I’d really appreciated it, if you opened one issue with a descriptive title/summary for each problem, so issue discussions are hopefully shorter, and users can more easily search the issue list to find solutions for their problems.

0rb677, 2020-06-04:
@paulmenzel @tlaurion can you add build process for `kgde-d16` on website `http://osresearch.net/` from here? And then we can remove this thread. Also what about prebuilt roms on heads ftp?

Tonux599, 2020-06-04:
@0rb677 modifications to the site can be made by PR [here](https://github.com/osresearch/heads-wiki). The build process for the board will eventually be the same as any other board building with `make BOARD=$BOARD`. [This](https://github.com/osresearch/heads/milestone/1) needs to be satisfied before prebuilt roms are provided.

@tlaurion is there any consideration on making perhaps a IRC channel for first line support or general discussion?

0rb677, 2020-06-04:
Yes?
Also found this. But for AST2400
[Gentoo Kernel Aspeed_AST2400_Graphics ](https://wiki.gentoo.org/wiki/User:Bugalo/Supermicro_A2SDi-4C-HLN4F#Aspeed_AST2400_Graphics)
I have `AST2050` onboard ship.
[image]

tlaurion, 2020-06-05:
> @tlaurion is there any consideration on making perhaps a IRC channel for first line support or general discussion?

Heads channel lives [here](https://u-root.slack.com/)

0rb677, 2020-06-06:
@tlaurion it works with AST enabled.

[image]

0rb677, 2020-06-06:
@tlaurion do you need dmesg/lsusb/lspci/dmidecode?

Memory works. 47,1GB found (but dmesg show some ECC error detected on the NB)
ASPEED works.
Pike works.

alexmaloteaux, 2020-06-06:
@0rb677 i know it is not related but did you flash the pike with the optrom ? i flashed without as i do for all lsi cards and the card is detected but not the disk connected to the pike  sata ports :(

0rb677, 2020-06-07:
@alexmaloteaux  im not try run it 365/24/7. But 3-4 days without shutdown it works good. No im not flash it.
Did you try to enter on controller `ctrl +s` ? Change sata cables? Change sata ports?
I use cheapest Pike 2008 and default 3.5 sata WD 3TB.
[image]

tlaurion, 2020-06-07:
> @tlaurion it works with AST enabled. But it cant initialize gui-init
> 
> [image]
> 
> [image]

https://github.com/osresearch/heads/issues/477#issuecomment-461938492

Attempts continuing over https://github.com/tlaurion/heads/tree/kgpe-d16_current_working

tlaurion, 2020-06-07:
https://github.com/osresearch/heads/issues/560#issue-441766401

tlaurion, 2020-06-07:
https://github.com/osresearch/heads/compare/master...tlaurion:kgpe-d16_staging

0rb677, 2020-06-07:
@tlaurion 
1. re-merge https://github.com/tlaurion/heads/tree/kgpe-d16_current_working
2. trying to add this changes 
https://github.com/osresearch/heads/commit/5c4f415e0927cbe15c847ec57059474b6123a8ab to `heads/boards/kgpe-d16_workstation/kgpe-d16_workstation.config`
and re-flashing

tlaurion, 2020-06-07:
Let me know how it goes.
 
```
#Whiptail
CONFIG_SLANG=y
CONFIG_NEWT=y
```

Will permit you to launch gui-init in text mode instead of FbWhiptail.

For the changes linked to libremkey, I wouldn't advise merging them. Discussion happened [here](https://github.com/osresearch/heads/pull/493)

0rb677, 2020-06-07:
Because AST doesnt work with fbwhiptail ?

tlaurion, 2020-06-07:
> Because AST doesnt work with fbwhiptail ?

@0rb677 : I don't remember. My use case was server.
If you want to go the workstation way and use an external graphic card then the onboard Aspeed, you will have to duplicate the board config and specialize it.

I may have been able to boot with Aspeed and local tty configuration with a newer kernel, on which I depended on my privacybeast branch. If I remember well, the AST support in kernel 4.14.62 was new at that time, and the number of patches that linux applied to that feature since that version are quite important.

Having AST supported in Heads might require the kernel to be upgraded, which requires a lot of work linked to #590 . Upgrading module/kernel cannot just be made by applying the same kernel configuration because of  #732  and #590, and breaks x230-flash and t430-flash because of really limited space in CBFS (4mb). So to upgrade the kernel, the *-flash boards wll need to be reduced in fonctionallities, alse upgrading the kernel would break some boards.

This is why this is PoC

0rb677, 2020-06-07:
@tlaurion workstation way is hard with nvidia cards. because its hard to get working in Qubes (attach to HVM)

tlaurion, 2020-06-07:
> @tlaurion workstation way is hard with nvidia cards. because its hard to get working in Qubes (attach to HVM). i want to try fbwhip again, because i think i merged your modifications without whip package and modified config.
> And then i go your text mode way with SLANG.

That would be having it at install, not in HVM but for dom0. ASpeed not being seen by the whole system because of jumper setting.

tlaurion, 2020-06-07:
@0rb677 : this would change pretty soon with https://www.qubes-os.org/news/2020/03/18/gui-domain/ for AST, VNC setup permitting to control QubesOS remotely. Also note https://github.com/Rudd-O/qubes-network-server/

Let me know how it goes!

0rb677, 2020-06-07:
@tlaurion this is sad :( Maybe set jumper to external card. One Nvidia be for dom0 and Heads. Second for GPU passthrouth to HVM ?

tlaurion, 2020-06-07:
> @tlaurion this is sad :( Maybe set jumper to external card. One Nvidia be for dom0 and Heads. Second for GPU passthrouth to HVM ?

Then in your use case, you would want a functional kgpe-d16_workstation on Aspeed. And make sure you select a GPU that can be passed through (no experience in that).

So the goal here would still be to have gui-init on FBwhiptail for Aspeed onboard configuration.

So you see where to go to make this go forward following your need? 
@Tonux599 any advancement on https://github.com/osresearch/heads/issues/368#issuecomment-592982240 ?

0rb677, 2020-06-07:
@tlaurion do you try to build aspeed VGA driver in dom0 ? Or it doesnt solve issue ?

[98 more replies]

closed 2020-06-19 by tlaurion