linuxboot/heads #716: README specifies that flash chip is

tlaurion, 2020-05-04:
Just a trace.

Observed in the wild to address/verify coherent [statement in README](https://github.com/osresearch/heads/commit/73a3b6d08f6bab0b63a0ebaf97c70138bdade128#diff-04c6e90faac2675aa89e2176d2eec7d8R15) and move doc in heads-wiki altogether:

```
Bottom SPI chips (3):
"EN25QH64" "N25Q064..3E" "MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E"

Top SPI chips (2):
"MX25L3273E" "N25Q032..3E"
```

Thrilleratplay, 2020-09-07:
I think I found where there is a list of comparable chips in the Lenovo BIOS upgrade.  For the x230, these are the steps I took:
```shell
wget https://download.lenovo.com/pccbbs/mobiles/g2uj33us.exe   
innoextract g2uj33us.exe 

cd app/G2ETB7WW/
binwalk -reM '$01D3000.FL1'
cd _\$01D3000.FL1.extracted 
uefitool 239
```

Search for text: "flashpartsmm"

```
EN25QH128FlashPartSmm.efi
EN25QH32FlashPartSmm.efi
EN25QH64FlashPartSmm.efi
MX25L3206EFlashPartSmm.efi
MX25L64XFlashPartSmm.efi
N25Q032FlashPartSmm.efi
N25Q064FlashPartSmm.efi
W25Q32FlashPartSmm.efi
W25Q64FlashPartSmm.efi
```

These names correspond to the following devices on the [flashrom supported hardware page](https://www.flashrom.org/Supported_hardware)

|EFI name|Vendor|Device| size in bytes|
|--|--|--|--|
|EN25QH128 | Eon | EN25QH128 | 16384|
| EN25QH32 | Eon | EN25QH32 | 4096|
| EN25QH64 | Eon| EN25QH64 | 8192|
|MX25L3206E | Macronix| MX25L3206E - MX25L3206E/MX25L3208E | 4096|
|MX25L64X | Macronix |MX25L64X MX25L6405,MX25L6405D, MX25L6406E/MX25L6408E, MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E/MX25L6473F, MX25L6495F | 8192|
|N25Q032 | Micron/Numonyx/ST| N25Q032..1E,  N25Q032..3E | 4096|
|N25Q064 |Micron/Numonyx/ST |N25Q064..1E,  N25Q064..3E | 8192|
|W25Q32 | Winbond | W25Q32.V, W25Q32.W | 4096|
|W25Q64 | Winbond | W25Q64.V, W25Q32.W  | 8192|

The 16M chip is odd, maybe they used them in later versions?

t-anon, 2022-03-04:
Mine has a cFeon QH64-104HIP, is this the same as the Eon EN25QH64?

tlaurion, 2022-03-17:
@t-anon sorry for lack of update, unfortunately flashrom would tell.

It will provide a list of detected chips as per documentation. It has low impact on backup, where write function will differ.

As per doc, it is suggested to backup (-r)  and verify (-v) backups prior flashing (-w)

If really doubtful, just take a couple of backups specifying different chips (-c) and compare those backups with sha256sum or whatever you feel like (*) to reduce anxiety.

And please update us!