linuxboot/heads #718: SeaBIOS as heads payload?

mikebdp2, 2020-05-08:
Compared to a Linux kernel which is currently used by heads, SeaBIOS has a much smaller source code/binaries - means, a significantly smaller attack surface and less space consumed in CBFS - and yet SeaBIOS supports a TPM. Linux kernel flashed into a BIOS chip as a heads payload - is quickly outdated, so the people have to choose between constantly updating it or sacrificing some security.

Also, SeaBIOS supports booting the LZMA-compressed floppy-based hobby OS like KolibriOS or FreeDOS added to CBFS, which provides some really interesting capabilities: i.e. add a KolibriOS https://kolibrios.org/ _(tiny opensource OS written on ASM, with GUI and lots of apps all inside a 1.44MB)_ - and if it supports your Ethernet controller: you can browse the Internet with a simple web browser, or have a secure chat with your friends using a IRCC client with a custom encryption mod like http://board.kolibrios.org/viewtopic.php?t=3734 - all under a simple OS. Which doesn't suffer from the vulnerabilities of a complex OS like Linux, with SystemD _(https://linux.slashdot.org/story/17/07/29/1647255/systemd-named-lamest-vendor-at-pwnie-security-awards)_ and - even if you choose something like Void _(runit)_ or Artix _(OpenRC)_ to avoid a SystemD - still many other issues.

SeaBIOS source code is relatively tiny _(just ~50k code lines, compared to Linux kernel's ~27.8 million or Tianocore's ~1 million)_ and really easy to modify, as I've proved with some custom mods like https://review.coreboot.org/c/coreboot/+/32351

@osresearch Do you think, given that SeaBIOS supports a TPM and consumes much less space inside CBFS, a custom fork of SeaBIOS with some additions could become a better payload for heads?

tlaurion, 2020-05-08:
> Compared to a Linux kernel which is currently used by heads, SeaBIOS has a much smaller source code/binaries - means, a significantly smaller attack surface and less space consumed in CBFS - and yet SeaBIOS supports a TPM. 

As answered in the other tickets you opened, Heads addressed threat model is to use the TPM as early as possible a a root of trust to be able to measure coreboot itself from coreboot romstage cache, where other propositions requires all the chain to be trusted to be able to take any measurements. Here, relying in seabios using TPM to measure anything would be forward, meaning only what is going to be trusted to launch operating system.

> Linux kernel flashed into a BIOS chip as a heads payload - is quickly outdated, so the people have to choose between constantly updating it or sacrificing some security.
> 

Heads is used as a static root of trust in this scenario. No necessary modules are not in the kernel and needs to be loaded prior of use, modifying the TPM measurements by calling a insmod wrapper, which extends TPM measurements to invalidate what is later used to [secure the boot process.](https://github.com/osresearch/heads-wiki/blob/master/Keys.md#tpm-pcrs)

Heads relies on cryptsetup, gpg2, kernel cryptomodules, lvm and kexec basically, to offer what it offers. When a user defines a boot default and decides to create a Disk Unlock Key to unlock LUKS encrypted container, the TPM measurements are used in combination with TPM measured LUKS header, sealed inside of the TPM NVRAM space, to generate a new LUKS key which is added in the LUKS header. The released key stored in password protected NVRAM space can only be released if TPM measurements are valid, LUKS header is valid and Disk Unlock Key typed to unlock NVRAM space is valid. Consequently, the default boot can only launch if the Disk Unlock Key is typed on that computer, protecting the user from video recorded LUKS passphrase being typed in public space, so that a bad actor cannot simply clone SSD in 15 minutes and decrypt content at leasure at a later time.

> Also, SeaBIOS supports booting the LZMA-compressed floppy-based hobby OS like KolibriOS or FreeDOS added to CBFS, which provides some really interesting capabilities: i.e. add a KolibriOS [...]

That may be interesting for #590 

> Which doesn't suffer from the vulnerabilities of a complex OS like Linux, with SystemD _(https://linux.slashdot.org/story/17/07/29/1647255/systemd-named-lamest-vendor-at-pwnie-security-awards)_ and [...]

Yet again, Heads linux payload is secured by the TPM in the sense that once the user decides to drop in recovery shell to go interactive, TPM measurements are extended and TPM measurements cannot be used to launch default boot options either.

> 
> SeaBIOS source code is relatively tiny _(just ~50k code lines, compared to Linux kernel's ~27.8 million or Tianocore's ~1 million)_ and really easy to modify, as I've proved with some custom mods like https://review.coreboot.org/c/coreboot/+/32351
> 
> @osresearch Do you think, given that SeaBIOS supports a TPM and consumes much less space in CBFS, a custom fork of SeaBIOS with some additions could become a better payload for heads?

The current avenue is to drop custom patch and rely on [coreboot VBOOT+Measured boot ](https://github.com/osresearch/heads/pull/709) to rely on upstream coreboot in the future.
Relying on SEABIOS might be interesting if user is willing to drop librem-hotp-validation, Disk Unlock Key usage and verified boot components currently scripted inside of Heads. But the current roadmap, on my side, is a bit different: https://fosdem.org/2020/schedule/event/firmware_hodorateatria/

paulmenzel, 2020-05-09:
@mikebdp2, after reading https://trmm.net/Heads and https://trmm.net/Heads_FAQ, I do not understand your request to use SeaBIOS. Your motivation to use SeaBIOS seems to be to save space and to use it’s boot menu. But that is not the goal of the Heads project at all. Also, the main objective of SeaBIOS (“C-BIOS”) is to provide legacy BIOS services.

> Heads is not just another Linux distribution -- it combines physical hardening of specific hardware
> platforms and flash security features with custom Coreboot firmware and a Linux boot loader in
> ROM.

The main argument for a Linux payload is, that it’s the same code base as the Linux kernel you are going to start, and the GNU/Linux distribution allows to write your own scripts giving you all the flexibility you need in an environment you need. (How do you do two factor authentication with SeaBIOS.) Despite the larger code base, your LOC comparison is unsound, as the rom/boot kernel (what was the naming again?) only uses a fraction of that.

Also, you are mistaken, that it’s required to update the ROM kernel often. The majority of the found security issues found in the Linux kernel, depending on the configuration, do not apply to it, so no updates are needed. Once it works, you are not supposed to update it anymore.

What would be used as the boot loader in your scenario?

Having said that, your use cases are of course valid, but they not align with this project. With the comments here in mind, I recommend to clarify your vision, and recommend to start a new coreboot/payload distribution for your use case.

tlaurion, 2020-05-11:
@paulmenzel @mikebdp2 the rest of the information pertinent is under #717. Feel free to close this ticket if your question has been answered.

mikebdp2, 2020-05-13:
@mikebdp2 Thank you! Please answer at #717 before we could continue there as well _(these discussions are interdependent a bit)_

closed 2024-04-16 by tlaurion