linuxboot/heads #719: Re-upstreaming of the KGPE-D16

tlaurion, 2020-05-08:
There was a proposition to do required work under refused grant application. If there is interest from third party to fund conjointly needed work, board could be maintained and reupstreamed, both under u-bmc and coreboot.

Putting it out there.

tlaurion, 2020-05-08:
Traces:
[u-bmc](https://github.com/u-root/u-bmc/issues/133)
[kgpe-d16 actual board status](https://github.com/osresearch/heads/issues/134)

This board works as a QubesOS server while required work is needed.
Freshly landed [QubesOs-network-server](https://github.com/Rudd-O/qubes-network-server/issues) code

A followup on https://github.com/osresearch/heads/issues/717#issuecomment-625825019

Tonux599, 2020-05-08:
@tlaurion did 3mdeb decide not to take it on? It would be good to have as it's the only completely blob free board I can think of that's qubes compatible. Provided a 6200 series is used.

tlaurion, 2020-05-09:
@Tonux599 3mbeb is absolutely willing to do the work in exchange of paid work.

tlaurion, 2020-05-09:
@Tonux599 

NlNet decided to not take it on (Fund the project).
Q: Will we decide to take it on and make it happen? Let's see. :)

tlaurion, 2020-05-11:
[Applicable](https://ngiatlantic.eu/)?

tlaurion, 2020-05-11:
@pietrushnic

pietrushnic, 2020-05-11:
@tlaurion sorry, for not replying, recently we have quite a lot on my plate. Definitely, we would like to publish here documents that we sent to NLNet while applying for a grant. There is also a slightly modified version that we provided to Insurgo on Slack. Both documents rely on coreboot state in December 2019. @miczyg I think you can push grant content to 3mdeb GitHub repo as PR and link here for review.

miczyg1, 2020-05-12:
This Pull Request contains the application 3mdeb sent to NLNet to obtain the grant:
https://github.com/3mdeb/kgpe-osf/pull/1
Feel free to comment and review.

Tonux599, 2020-05-12:
@pietrushnic @miczyg1 When should we expect an outcome? Also could you briefly outline any ways that members of the community could support you if or if not you get funding?

pietrushnic, 2020-05-12:
@Tonux599 there is no timeline since there is no official commitment from anyone to fully found the effort. We get some good words from various individuals and organizations, but that's all. @tlaurion mentioned that he can sponsor part of the effort, but we are not sure anyone would be happy with half-finished stuff. Vikings Gmbh sent us hardware so we have something to work on. NLNet did not approve of this grant, so I would not expect it will change in the future - I may be the wrong here. To sum the status of this project is "not founded" so we even do not put that on schedule and worry more about founded effort e.g. TrenchBoot and fwupd for Qubes OS.

How the community can help? We are very open here, whatever works for the community.
1. We can split this project and drive as an open and free contribution if there are developers and testers who willing to support the development and debugging. In such a situation we will support whoever will do that as much we can including, coordination review and help in merging code upstream.
2. We can open any channel that the community is comfortable with to try to gather remaining founds. @tlaurion suggested OpenCollective, we can align probably to any suggestion - we have no problem in handling the administrative part.
3. We have a PayPal donation button on our website, although this is probably one of the worst options. 
4. We are fine with commercial assignments if anyone is willing to go that path. It can be even a very small part of the task, but at least it will move us forward.
5. If we can sell a platform in some form or bundle services, which will generate margin, then we can use it to bring back the platform to the mainline.

We would like to help, but without funding, we will do the best effort and it may take very long to bring back this platform. Please note our goal is not just to bring back the platform to mainline, but create a structure in which long term maintenance would not cost a lot. We did that for [PC Engines routers](https://pcengines.github.io/) for over 4 years.

tlaurion, 2020-05-12:
@Tonux599 : For the moment, step would be to get people interested in the port. Let's remember that the KGPE-D16 was funded by [Libreboot and put upstream by RaptorEngineering](https://libreboot.org/docs/hardware/kgpe-d16.html) but previous work did not receive enough attention to be maintained and got dropped in coreboot 4.11 release for lack of maintainership.

Let's also remember that the community crowdfunder the port of the[ BMC chip to OpenBMC ](https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-offer.php)and that work wasn't maintained.

NlNet refused to fund the work because they didn't understand the reasoning of funding old platform.

So the point here would be first to get enough attention so that the story doesn't repeat itself.
Insurgo is interested to fund a big part of the work, but we concluded that we would not do this alone.

Putting a bounty tag for this.

tlaurion, 2020-05-12:
Something really interesting was proposed on this QubesOS ticket: 
https://github.com/QubesOS/qubes-issues/issues/2809#issuecomment-626111758

@pietrushnic, I would love to hear on your reception.

pietrushnic, 2020-05-12:
@tlaurion XVilka suggestion looks interesting, but definitely key problem is orchestration. We would like to avoid the situation when 3 or more teams work on a given feature in parallel and then post results claiming the founds. There should be serious commitment based on proven reputation and trust, otherwise, we have public bidding problem where lower price wins with quality.

Whonix stuff is even better, but we would have to research how it looks from the taxation and administration perspective.

tlaurion, 2020-05-13:
@pietrushnic I'm not sure there is conflict between the two ideas with XVilka avenue. 
If the ticket is clear saying that 3mdeb realizes the work in question (milestone) and funds are released upon proof of work (PR). But you are better placed then I am to see if this would work or not.

mikebdp2, 2020-05-13:
@tlaurion and @pietrushnic : Have you contacted a Free Software Foundation? Since they have a lot of interest in RYF devices and are probably using these KGPE-D16 servers by themselves, if they'd learn about this critical situation - maybe they'll decide to co-fund?

pietrushnic, 2020-05-13:
@mikebdp2 we (as 3mdeb) never contacted FSF. I'm not sure if the situation is critical but it will get worst over time. From experience longer platform lays unmaintained bigger would be effort to bring it back. Do you know anyone who we should reach?

@tlaurion I believe you are in a better position to talk with FSF, then 3mdeb.

pkubaj, 2020-05-14:
22000E isn't that much different from what Raptor wanted for OpenBMC port (at https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-offer.php). Maybe you should set up similar crowdfounding? Currently you only say "we are ready to work" etc. Raptor did the same and received no money, but when they officially started crowdfounding, it turned out there were enough people interested in it to gather the money.

pietrushnic, 2020-05-14:
@pkubaj very good point. It looks like Martin was intermediary gating the process. In case of difference in pricing, please note this was a different time and they are in the US not in PL, their costs are way higher than ours. More to that we are Yocto Participants and have Embedded Linux people on board, OpenBMC uses Yocto/OE build system, so this also simplifies the situation for 3mdeb.

tlaurion, 2020-05-14:
> @mikebdp2 we (as 3mdeb) never contacted FSF. I'm not sure if the situation is critical but it will get worst over time. From experience longer platform lays unmaintained bigger would be effort to [...]

@pietrushnic writing to them.

alexmaloteaux, 2020-05-28:
> @Tonux599 : For the moment, step would be to get people interested in the port. Let's remember that the KGPE-D16:
> 
>     * was funded by [Libreboot and put upstream by RaptorEngineering](https://libreboot.org/docs/hardware/kgpe-d16.html) but previous work did not receive enough attention to be maintained and got dropped in coreboot 4.11 release for lack of maintainership.
> 
>     * the community crowdfunded the port of the[ BMC chip to OpenBMC ](https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-offer.php)and that work wasn't maintained.
> 
>     * NlNet refused to fund the work because they didn't understand the reasoning of funding old platform.
> 
> 
> So the point here would be first to get enough attention so that the story doesn't repeat itself.
> Insurgo is interested to fund a big part of the work, but we concluded that we would not do this alone.
> 
> Some history:
> 
>     * [FOSDEM talk by 3mdeb on AMD coreboot status](https://fosdem.org/2020/schedule/event/coreboot_amd/)
> 
>     * [Reddit thread on announced drop of the KGPE-d16](https://www.reddit.com/r/linux/comments/dz2hlt/boards_like_asus_kgped16_the_most_powerful/)
> 
> 
> Putting a bounty tag for this.

Hi is there a list of the bugs/linux crash that Michal talk about in his fosdem talk ? Do they only happen when iommu is enabled ?

Btw i have a kgpe-d16 with jtag header soldered and an olimex. So if you are starting any group  for dev , put me in, free time is limited but i can always help a bit and test.

tlaurion, 2020-05-30:
@miczyg1 ?

tlaurion, 2020-05-30:
@alexmaloteaux asked:
> Hi is there a list of the bugs/linux crash that Michal talk about in his fosdem talk ? Do they only happen when iommu is enabled ?
@miczyg1 ?

miczyg1, 2020-06-01:
Simply have a look at the coreboot mailing list:
- https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/ME4TULTDZGIYJEMNR2IB2W6P7ONG76EU/#GCJF7CIWGCKZCDWYPWAM2LKQW6APHZRJ (memory problems)
- https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/ZR7LMBXOFKR5FWVQDI22UB6O3LO5WGQ3/#TH57VEGCY357ZH534PU5PKSDSMYLTLFH (S3 suspend/resume problems)
- https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/VWFKH7FDWLO5KGZMIJ7VM67SIUWJRVZL/#OCHSI6UL2XZDSA3HCHOONNN5EX4HLUGE (kernel panic from new microcode)
- https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/H5KHOUVDPQA2NSUE43IBRKSYBZQK4DWN/#H5KHOUVDPQA2NSUE43IBRKSYBZQK4DWN (SLUB memory allocator in Linux panics)
- https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/RDHEDOTBYVD74NWQWYYNLAJWPD7CMDRZ/#RDHEDOTBYVD74NWQWYYNLAJWPD7CMDRZ (KVM/SMV not working)
- https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/AW4DF557YDTTJGKDFRTHMZJBTF65ZJZT/#VOFCIRV24HT3OIG5IDYH4GEXY5VSS3W5 (kernel panics on kernel newer than 4.9)

pietrushnic, 2020-06-01:
In case of what have to be done I believe Arthur email is quite good. @tlaurion you replied to it [here](https://www.mail-archive.com/coreboot@coreboot.org/msg54125.html)

alexmaloteaux, 2020-06-01:
I have read the various mail list entry and i cant reproduce any of them. Actually i have a pretty stable system so far so i will list here my config and test , maybe others can use it as baseline to try to triggers issues and/or star the long 4.12 migration process. please note i have a pure server usage for this board and not a desktop pc.

- Coreboot version : 4.10, commit ea58adddf4  with seabios and sgabios, all input/output redirected to uart 0x2f8 io port for OpenBmc (coreboot, sgabios, os)
- Heads Version : master but cant use atm as im facing TPM issue
- CPU : 2X Opteron 6378 2.4GHz 16M
- Ram : 2x Micron 16GB Ram DDR3 MT36KSF2G72PZ-1G6E1FE PC3L-12800R
- TPM : Asus TPM-L r2.0 TPM 20-1 pin (recognized by seabios but tpm-reset script or tpm-tool have issue ; unknown tag ...)
- Screen : Not using VGA at all, all input/output redirect to OpenBmc uart on io port 0x2f8
- Linux Boot : OK (ubuntu 18.04 / kernel 4.15)
- Freebsd Boot : OK (12.1 stable)
- Linux Boot with IOMMU enabled : OK (iommu=pt amd_iommu=on)
- Freebsd Boot with IOMMU : OK
- Linux install and boot VM in KVM : OK
- Freebsd install and boot VM in bhyve : not yet tested
- Linux PCI-passthrough to VM in KVM : OK (tested with a IBM M1015 flashed in it mode)
- Freebsd PCI-passthrough to VM in Bhyve : not yet tested

Ill try to post a more detail description of the tpm issue i m facing later on

BR

Tonux599, 2020-06-01:
@alexmaloteaux regarding TPM, TPM2 is currently unsupported with heads, see #287 however the Asus 90-C1B0AU-00XBN0VZ is a TPM1.2 module compatible with heads and this board.

pkubaj, 2020-06-01:
I have two of those boards, one is a desktop, the other one is a server. Both run coreboot 4.11 tag.

Both run FreeBSD, the desktop is sometimes booted to Linux.

Both are stable. I didn't test averything that @alexmaloteaux tested. I'm pretty sure PCI passthrough won't work with bhyve due to CPU's missing some required features.

alexmaloteaux, 2020-06-01:
> I have two of those boards, one is a desktop, the other one is a server. Both run coreboot 4.11 tag.
>
> Both run FreeBSD, the desktop is sometimes booted to Linux.
>
> Both are stable. I didn't test [...]

63xx Series supports  AMD-Vi and can support PCI passthrough , except os/coreboot bug

Tonux599, 2020-06-03:
> @pietrushnic writing to them.

@tlaurion any luck?

tlaurion, 2020-06-03:
Waiting for an e-mail reply from FSF

tlaurion, 2020-06-03:
I would suggest anyone interested in the d16 to be reupstreamed writing to FSF too?

Tonux599, 2020-06-03:
> I would suggest anyone interested in the d16 to be reupstreamed writing to FSF too?

@tlaurion any chance of seeing what you wrote or if not guidelines on what would good to include in the correspondence?

tlaurion, 2020-06-03:
> > I would suggest anyone interested in the d16 to be reupstreamed writing to FSF too?
> 
> @tlaurion any chance of seeing what you wrote or if not guidelines on what would good to include in the correspondence?

@Tonux599: 
Contacted them through patron@fsf.org and business@fsf.org.

ryf@fsf.org might be of interest.

```
Hello FSF,

Insurgo here!

We have tried to get NlNet fund the KGPE-D16 revival. As you may know,
coreboot dropped support for that board in coreboot 4.11, leaving the
last RYF platform unmaintained and now rotting.

3mdeb made a really good grant application, but NlNet didn't really
understood the importance of maintaining old hardware, and unfortunately
the ball got dropped.

The details are in this ticket:
https://github.com/osresearch/heads/issues/719

I would love to start a discussion with you guys on what is possible and
if funds were available from you to jump in this subproject.

Thierry

Insurgo Open Technologies
```

alexmaloteaux, 2020-06-03:
How can we write them ? everybody individually pointing at this issue ? or  collectively ?

Best Regards

tlaurion, 2020-06-03:
@alexmaloteaux from my understanding of the [past](https://github.com/osresearch/heads/issues/719#issuecomment-627605335), interest from endusers and company should be expressed individually by the most numerous number of people and why it matters.

tlaurion, 2020-06-07:
@pietrushnic @miczyg1 : took some time to give love to #472 in link of #712. We are getting near of a heads baseline.

tlaurion, 2020-06-07:
RYF contacted and put 3mdeb in contact directly. Let's see where it goes...

tlaurion, 2020-07-02:
@pietrushnic @miczyg1 there is a call preparation with RMS from FSF, vikings.net and tiberiu from tehnoetic.com

I advised that you join the discussion and shared Piotr e-mail address. Will follow through.

ullbeking, 2020-07-14:
@alexmaloteaux 
> Btw i have a kgpe-d16 with jtag header soldered

Alexandre, I was really impressed to read that you got the JTAG header installed on the ASUS KGPE-D16 board :-)  I assume that you also had to remove the solder plugs from the board first?  If you are reading this message, and if you are willing to give a little advice, please could you send me a sign?!  :-)  I have been preparing the board to remove the solder plugs safely, and I would love to compare notes with somebody who has already done it successfully.

Th3Fanbus, 2020-07-14:
@tlaurion about the e-mail to FSF:

> We have tried to get NlNet fund the KGPE-D16 revival. As you may know,
coreboot dropped support for that board in coreboot 4.11, leaving the
last RYF platform unmaintained and now rotting.

The chosen wording makes it seem as if it is the coreboot project's fault that the KGPE-D16 ended up unmaintaned and rotting. However, I have seen no one maintaining KGPE-D16 support in upstream coreboot, only users complaining about things not working. That the code for this mainboard got dropped from coreboot after the 4.11 release is not the reason why the board is no longer maintaned, but the other way around: as there had been no maintenance going on for a rather long time (even though there were plenty of bugs to fix), it had become a maintenance burden for coreboot developers and the code eventually got dropped as a result. Quite surprising, given that several companies are making profit by putting coreboot onto these boards.

Moreover, it is not the last RYF platform. It might be the most powerful x86 platform, but definitely not the only one.

pkubaj, 2020-07-15:
> @tlaurion about the e-mail to FSF:
>
>
> The chosen wording makes it seem as if it is the coreboot project's fault that the KGPE-D16 ended up unmaintaned and rotting. However, I have seen no one [...]

What were those bugs? I have coreboot 4.11 working on two KGPE-D16 boards. It has some issues, but they are easy to workaround. Those boards are quite finicky to set up with RAM compatibility issues etc., but this is also the case with vendor firmware.

I think those "reports" about broken coreboot were just users doing things wrong. And coreboot developers, instead of noticing that it works for other users (I repeatedly reported that) decided to go the easy way - just nuke it.

There's also another thing. I'm not a firmware developer, so I wouldn't be able to fix any issues in coreboot. But it happens that I'm a FreeBSD ports developer and I often fix ports that I don't care about, but I know users care. coreboot developers didn't have that sort of attitude. I wonder why. I just hope that there wasn't anyone sabotaging coreboot, because I would be sad to see that.

paulmenzel, 2020-07-15:
I vote for closing this issue, as too much FUD is spread.

>

I agree that the wording is not optimal.

[…]

> What were those bugs? I have coreboot 4.11 working on two KGPE-D16 boards. It has some issues, but they are easy to workaround. Those boards are quite finicky to set up with RAM compatibility issues [...]

We talked about this in #coreboot@irc.freenode.net quite often: Random boot failures, all of memory not supported, higher energy usage compared to vendor firmware, problems loading latest microcode, ….

> There's also another thing. I'm not a firmware developer, so I wouldn't be able to fix any issues in coreboot. But it happens that I'm a FreeBSD ports developer and I often fix ports that I don't [...]

Hardware development is much harder than software development. You need to test the changes on actual hardware. Chipset changes affect other boards too, so you have to test those too. Also you need to have understanding of the hardware.

You can get active, and push the collecting the money forward by starting some kind of crowdfunding.

pkubaj, 2020-07-15:
> We talked about this in #coreboot@irc.freenode.net quite often: Random boot failures, all of memory not supported, higher energy usage compared to vendor firmware, problems loading latest microcode, ….

Random boot failures - that happens to me maybe in 1% or so of boot attempts (I mean, very rarely). True, vendor firmware doesn't do that, but this is so low priority to me that I literally don't care,
All of memory not supported - isn't that also the case with vendor firmware? That would mean there's a hardware issue, which coreboot can at best workaround. Still there ARE RAM sticks that work properly,
higher energy usage - that's unfortunately true, but that causes energy bills to go up, it doesn't cause any usage issues,
Problems loading latest microcode - I don't have this issue, I have coreboot built with the latest microcode and it works.

Sure, there ARE issues that prevent coreboot from achieving 100% feature parity with vendor firmware. But, this is also the case with other boards. On ASUS F2A85-M I had to manually run fancontrol on Linux because CPU would overheat otherwise and FreeBSD was unusable on it (due to no fancontrol). Yet, I can see F2A85-M is still in upstream coreboot.

On ThinkPad X230, AFAIK the yellow USB port (that can charge devices even when the laptop is off) works like all other USB ports (so it charges only when the laptop is on). X230 is also still in upstream coreboot.

So IMO arguments that there are bugs don't hold as long as it's actually possible to get the board working and users are reporting them working.

paulmenzel, 2020-07-15:
Please let’s talk in IRC, before this issue gets even more convoluted.

You totally misunderstand, why the board was removed from the main branch (but it’s still in the branch `4.11_branch`).

It was *not* about the bugs or an incomplete port, it’s because the developers decided that each platform/board has to support certain features. Please read the [release notes](https://review.coreboot.org/cgit/coreboot.git/tree/Documentation/releases/coreboot-4.12-relnotes.md#n72).

> > We talked about this in #[coreboot@irc.freenode.net](mailto:coreboot@irc.freenode.net) quite often: Random boot failures, all of memory not supported, higher energy usage compared to vendor firmware, problems loading latest microcode, ….
> 
> Random boot failures - that happens to me maybe in 1% or so of boot attempts (I mean, very rarely). True, vendor firmware doesn't do that, but this is so low priority to me that I literally don't care,

You asked, what bugs, and claimed there are none. Come on. That you live with them, is another topic.

> All of memory not supported - isn't that also the case with vendor firmware? That would mean there's a hardware issue, which coreboot can at best workaround. Still there ARE RAM sticks that work properly,

The vendor firmware supports all 256 GB of memory. Where did you get this misinformation from? And again, how is your last statement even related?

> higher energy usage - that's unfortunately true, but that causes energy bills to go up, it doesn't cause any usage issues,
> Problems loading latest microcode - I don't have this issue, I have coreboot built with the latest microcode and it works.

Good for you. I guess you do not use the Linux kernel?

> Sure, there ARE issues that prevent coreboot from achieving 100% feature parity with vendor firmware. But, this is also the case with other boards. On ASUS F2A85-M I had to manually run fancontrol [...]

See my statement at the very top. No idea, where you got your information about the cause for removal from. And people can still use the version from 4.11 (even up to the removal commit). Why do they need the new features, if it works as you said?

Thanks to the payload concept, people can still use newer payloads, which makes the main difference to them.

Anyway, I am out, and kindly ask you to inform yourself more diligently about the whole situation.

ullbeking, 2020-07-15:
@paulmenzel 
> Please let’s talk in IRC, before this issue gets even more convoluted.

If we are going to move the discussion then the mailing list would be vastly better.  It is important to correct the misunderstandings on the record.  IRC is too "whizzy" to be able to absorb the complexities of the problems and the social interactions that led to this problem in the first place.

It's a convoluted issue and I think participants at all levels of experience are still trying to understand what's really going on.  If this had been discussed on a mailing list then I would at least have been aware of the problems.  If on IRC then no chance, because I'm not on IRC nearly as much as I used to be (I just **can't**).

_Personally, I am trying to get much more actively involved in this, especially since I recently realized that the lack of support for the D16 port of coreboot was much less supported than I originally thought.  **It was only by reading the complicated conversation -- which is necessary to tease out the misunderstandings -- that I realized the nature of the problem AT ALL.**_

I don't think that shifting the conversation to IRC is going to help clarify the issue **for me.**  And I REALLY want to get involved with this.  In fact, I think it will make it worse because as my daughter is about to start school and family life is becoming important I am unable to spend sll day every day on IRC.  (Unlike the last several years.)

pietrushnic, 2020-07-15:
@Th3Fanbus 

>  Quite surprising, given that several companies are making profit by putting coreboot onto these boards.

It is easy to write that, but hard to prove `profit` hypothesis. Hardware margins are not so big and I'm pretty sure the volume of those boards does not justify further investment especially when the previous one failed.

@ullbeking 

> In summary, the complexity of this conversation and working through it is actually necessary to understand the nature of the problem, which is just as much a social issue as a technical one. 

I agree with social part, technical is little bit easier for us (3mdeb). There is too much, not needed, misunderstanding and throwing rocks. IMO the community is about trust, responsibility, and eventually reputation. I'm sure we all are acting in goodwill here, but there should be an understanding that no company can afford infinite firmware support without a net profit margin. We believe it is possible to create after-market support driven by the community if it is big enough. That approach is hard like any B2C project.

In an ideal world, we would have enough traction in B2B relations to support the community. There are success stories behind that e.g. [PC Engines](https://pcengines.github.io/), which we support for more than 4 years. There is [Olimex and Allwinner story presented at FOSDEM 2020](https://fosdem.org/2020/schedule/event/olimex_oshw/) that IMO applies here. To drive this success we need people like @tlaurion, coordinators, and organizers not in a chaotic way but well-organized project/product management way. Otherwise, we just have chaos or count on luck that NLNet or FSF will support.

Someone may ask if 3mdeb knows that why it will not organize everything? The answer is that we already delivered as much as we can and we plan to keep our promises to some point, despite time passing by and situation is more and more complex. We also have limited resources, which we have to manage wisely to deliver to the community great new open-source firmware projects/products.

What we can do? We should look at our peers, especially business peers, and figure out how they can benefit from well-maintained KGPE-D16, or maybe you have an idea for a business (even very small one) based on fully open platform. Of course, the goal is to get even small monetary contributions from them and gather the whole amount to reenable KGPE-D16 in upstream coreboot. We open to talk with those peers and negotiate conditions, then inform community about the progress.

We are not experienced with crowdfunding, but eventually will get there, if someone knows that environment we would be glad to talk about that. At some point, @tlaurion pointed to [this campaign](https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-offer.php) which in my opinion is a great example of how things can be organized.

Long run (12 months +) we can have a bug bounty program that will gather founds for maintainership and bug fixing.

Tonux599, 2020-07-15:
In relation to the point above, @tlaurion has Insurgo considered a PrivacyBeast KGPE-D16?

alexmaloteaux, 2020-07-15:
Hi sorry for asking but until there is a final place to ask question , can somebody tell me if there is any solution/workaround for the energy usage.

im using cb 4.10 and i lowered the frequency of my 2x6378 to 1400 with powerd on hardenedbsd and it is mostly ok . But at 2400 Hz i need to have my dynatron g34 heatsink running at 100% all the time and the small room where the server is located is becoming really hot after a few hours of usage.

Thanks

tlaurion, 2020-07-15:
Of course,  but the actual situation of unknown stable sourcing and required work to make the kgpe-d16 maintained and back into coreboot is out of my personal scope.

Coreboot in heads is now.pinned to specific version in master. So the work to make the kgpe-d16 work with coreboot 4.11 can then start by the community, on which I will help, but won't pioneer without funding. 

If someone here wants to help, then it goes back to starting a crowdfunding page, on which Insurgo would jump in and fund part of the work too.

Unfortunately, this is the situation we are in right now. My focus is more on current blob free laptops and future proof platforms. But would gladly work conjointly on bringing back to life the kgpe-d16.

Still no news from the FSF as of now and unfortunately went the farthest I could from a community, unpaid, perspective on my own for this platform. 

Any help/tangible plan welcome on making this funded partially / conjointly would be awesome. Otherwise im a bit overwhelmed by other tasks to lead on this one.

Thierry/Insurgo

agg1, 2020-07-15:
Not that anyone asked but I feel urged to drop a note concerning support for various AMD AM3/G34 boards. AM4 generation and similar are dead already anyway as far as i am concerned. same holds true for anything equipped with ME and similar.
1) branch 4.11 exists and can be contributed to and i welcome the fact it could be considered "stable" now for final review
2) alltogether the quality of what's present for AMD AM3/G34 after a decade of development raises concerns by itself, given this branch reached deprecated state with a quality present which I had considered not ready for production when this gear reached markets 10 years ago. the only difference to vendor bios i see concerning this is vendors do not provide sources. and to prevent a misunderstanding i might add it is certainly not coreboot devs to blame for this situation.
3) some patches for branch 4.11 are worked on to fix some AM3 board (https://github.com/agg1/coreboot/commits/master), which may be a decent alternative to KGPE-D16 and a viable option for a devs who couldn't afford a corebooted workstation for thousands of bucks
4) most importantly i would urge the industry including the FOSS people to rethink their general attitude towards resource requirements, as if it was necessary for amateur hackers and downstream developers to suck kilowatts from the powerwall with a 32core workstation and 256GiB of RAM, meanwhile a Hexacore with 16GiB RAM should do for another decade to come for some system integration tasks and distro maintenance. i can and do maintain a self-made hardened livedvd linux distro and it finished compiling everything from source in less than 24hours and 150Watts power draw only on a corebooted am3 workstation due to some optimizations und cuttings
5) my budget, and even if i could afford a KGPE-D16 i wouldn't want it but instead industry including FOSS people should align their resource requirements so development can be done on commodity hardware such as AM3 boards or quadcore laptops with 16GiGs of RAM with acceptable electricity costs for 24/7 full load operation, which makes me wonder again why up until recently a dual core with 2GB RAM was the only option officially tested with AM3-generation commodity hardware. same holds true for the chaotic nature of documentation available (BIOS, ACPI, chipsets, UEFI) which requires almost a livetime of work to attach some peripherals, initialize memory, etc. it is a nightmare.
6) i could rant day and night about industry habits (20 million LoC for GCC alone). again, rethink if industry could afford to loose another generation of amateur hackers them instead playing video games on threadripper workstations with PSP then

tlaurion, 2020-07-18:
> @tlaurion about the e-mail to FSF:
>
>
> The chosen wording makes it seem as if it is the coreboot project's fault that the KGPE-D16 ended up unmaintaned and rotting.

I am really sorry for the impact those words had. My intention was more to say exactly like everyone here is saying: the KGPE-D16 got dropped because of a lack of consistent maintainsership/testing and other problems IMHO, where others disagree. I think that most of the KGPE-D16 users came from libreboot, which was using a static version of coreboot, which again in my opinion, gave the illusion that those two projects were disjoint. The goal here is to learn from the past, not to blame anybody. When I say that the KGPE-D16 was left rotting, its historicism and I should not do that logic error. The facts are there though, in coreboot port not having enough working status reports in time for it to be maintained, down to crowdfunded OpenBMC port which stayed as a PoC with issues reported, confirmed, fixes provided while not merged for OpenBMC build system; no maintainership was made in time, nor regression testing to prevent dropping. 

This leaves us where we are now, where OpenBMC port was left behind and branched, coreboot port is diverging from upstream from 4.11 (with missing security fixes and incrising costs of putting that board upstream).

It is really a sad story but there needs to be demand and community support at some point.
- [ ] https://github.com/u-root/u-bmc/issues/133

 
> However, I have seen no one maintaining KGPE-D16 support in upstream coreboot, only users complaining about things not working. That the code for this mainboard got dropped from coreboot after the 4.11 release is not the reason why the board is no longer maintained, but the other way around: as there had been no maintenance going on for a rather long time (even though there were plenty of bugs to fix), it had become a maintenance burden for coreboot developers and the code eventually got dropped as a result. Quite surprising, given that several companies are making profit by putting coreboot onto these boards.

> 
> Moreover, it is not the last RYF platform. It might be the most powerful x86 platform, but definitely not the only one.

- There is [3 RYF mainboards](https://ryf.fsf.org/categories/mainboards). 
- Old laptops not supporting virtualization extensions but not having ME per libreboot work.
- coreboot blob-free boards like the [G505S are supported but limited in SPI flash and without a TPM](https://github.com/osresearch/heads/issues/453)... 

Other boards were documented in [earlier posts of this thread](https://github.com/osresearch/heads/issues/719#issuecomment-659024471). 

@miczyg1 made an awesome talk [here](https://fosdem.org/2020/schedule/event/coreboot_amd/) including status of KGPE-D16.
Please read my slides 45+ [here](https://fosdem.org/2020/schedule/event/firmware_hodorateatria/) for a global comprehension of the actual status of blobs in newer boards. RYF on Intel/AMD will not happen for newer boards. I think we should revive the KGPE-D16 and KCMA-D8 as a community but I fought enough on my own and no door opened following my personal attempts to date. I would gladly support someone taking that leadership and moving things forward both with my time and financially but won't lead this myself.

lss4, 2020-07-20:
Back then I posted some of my efforts trying to get KGPE-D16 working on the mailing list. Unfortunately it did not get me anywhere further than that. It's possible that libreboot still works okay for this board (given its codebase), but I'm not as familiar with libreboot's build environment as with coreboot's.

With coreboot 4.11 (4.11_branch), RAM initialization issues on KGPE-D16/KCMA-D8 are hit-or-miss, at least I can get some 100%-bootable RAM combinations on KGPE-D16 for the time being. However, it seems the real issue isn't RAM and lies in somewhere else. For example, something's not right in libpayload with this board as after looking into the code the fatal error messages I'm getting in some places (like trying to run nvramcui/coreinfo) originated from there (mostly USB related). I'm not an expert in this field, but libpayload might be something worth looking into, for finding out what might have caused coreboot to not work as well as before on this board.

For now I can embed and boot some floppy images using coreboot, thanks to Mike Banon's patches, They appear to be stable, just that USB doesn't work. Other than that there's little I could achieve as I need to boot from USB to install an OS and I haven't gotten myself to successfully boot anything from USB yet (either freeze or with error messages, probably because of the aforementioned libpayload issues).

> some patches for branch 4.11 are worked on to fix some AM3 board (https://github.com/agg1/coreboot/commits/master), which may be a decent alternative to KGPE-D16 and a viable option for a devs who couldn't afford a corebooted workstation for thousands of bucks

> my budget, and even if i could afford a KGPE-D16 i wouldn't want it but instead industry including FOSS people should align their resource requirements so development can be done on commodity [...]

One major issue with these workstation/server AMD boards would be the availability of cooling solutions. Nowadays C32/G34 cooling solutions are very hard to find (although someone managed to get some others installed by hard-modding the clips). In comparison, cooling solutions for Intel sockets (115x, 1366, 2011, etc.) and mainstream AMD sockets (AM2/AM3, AM4, FM1, etc.) are much more common. It'd certainly be better if mainstream AMD motherboards of that era could be better supported, I'm looking forward to it.

agg1, 2020-07-23:
Total power consumption with the corebooted AM3 board with a Phenom 2 X6 of mine is far below any of the server/workstation boards: ~140Watts total power consumption under full load (monitor, 4 hard disks in RAID, 16GiG ECC RAM DDR3 400MHz @1.35V seem rock solid and i don't see the benefit of memory I/O at 10GB/s@666MHz instead of 8GB/s@400MHz)
Even if modern systems or the KGPE-D16 in question were slightly faster i had rarely seen a workstation system under full load which had met the power and efficiency target!
Soon i'll test a Phenom2 X6 95Watt CPU (instead of the typical 125W) which may bring down consumption another 10 or 20 Watts, under constant full load that is. Too i am planning to introduce an option for easy undervolting with coreboot, fixing Northbridge frequency at 1.8GHz maybe with reduced voltage and fixing CPU at 1.25Volts which should be stable with all fam10h CPUs in non-TurboCore performance state so no specific low-power CPUs had to be acquired, and the power management chaos may simply be removed entirely, which was the part that contributed to breaking AM3 fam10h revE support, and it is a pain elsewhere too i might add. s2disk is functional still and system can be completely powered off and restored with it, with encrypted swap space even.
Furthermore such an AM3 system can easily be cooled and remains almost dead silent, simply choose a 5Volt adapter with appropriate 120mm ventilators - it is a bliss, if you had ever enjoyed the noise of a workstation/server nearby.
Sadly fam15h support on AM3 allthough present on branch 4.11 couldn't be confirmed, at least the FX 8120 CPU i had tested on the AM3 board with coreboot didn't POST at all.

Other than that i am maintaining a self-made hardened livedvd distribution booting read-only ISO only targetting these "obsolete" x86-64 systems in conjunction with coreboot which has removed certain "features" in kernel and userspace entirely and it is being stabilized after 3 years of development and testing.
Since fam10h CPUs lack accelerated AES encryption for example the distro of mine ships with a crypto stack inspired by Telco scramblers or MIL-STD-188 this easily yields almost 10GiB/s of randomness and integrated with block layer encrypted disk i/o can be had at 1GiB/s per cpu core max instead, maybe with NILFS2 on top of it to be tested still, without any AES hardware acceleration required anymore. so much for the achievable performance and reliability.

Unrelated but since libpayload was mentioned i'll keep an eye on that one too for sure, i had encounted some minor issues recently which i couldn't track down yet.

tlaurion, 2020-08-09:
Community attempts to bring the Heads's KGPE-D16 board config to coreboot 4.11 [here](https://github.com/osresearch/heads/pull/795). Any help welcome.

tlaurion, 2020-12-12:
kgpe-d16 is supported on Heads on coreboot 4.11 with TPM support. That was merged under https://github.com/osresearch/heads/commit/1661e5dcb07a061317581187b15ae553ad8fc3df, thanks to @Tonux599 !

tlaurion, 2020-12-30:
BMC internal flashing to be tested with #943

kuleszdl, 2021-06-04:
Thank you for bringing up this issue (I just stumbled over it). I am one of these "complaining" users who has been running this board in production for several years now and figured out a way to live / workaround around the aforementioned issues. Personally, I still find the board unique due to the arguments presented by @tlaurion, and I find its power consumption (at least with the vendor bios) acceptable for a workstation/server (however, not acceptable when running with coreboot and two cpu packages).

Imho, if 3mdeb is willing to take on this work and just needs funding, this is already a good starting point. Yet, collecting 30k is probably too much if you consider the remaining KGPE-D16 owners. But if the 30k would be split into separate goals it might be possible to collect enough money to reach at least some of them.

So, what could be the goals? Probably (some) of these:

- implement necessary features in fam15h to re-upstream it (from the release notes: relocatable ramstage, postcar stage and C_ENVIRONMENT_BOOTBLOCK)
- fix idle power consumption / sleep state issues
- fix memory training issues
- fix related memory training issues on KCMA-D8 as well (which does not work with ECC memory)
- fix booting with microcode updates loaded by coreboot
- fix other bugs?

@miczyg1 Have you considered starting a campaign on CrowdSupply or a similar platform? It's hard to guess how many people would be willing to back such a campaign and of course there is risk to loose the effort invested in the campaign, but imho it's worth a try.

Tonux599, 2021-06-04:
@kuleszdl If your on the OSWF Slack join #kgpe-d16 where there is discussion on a possible hackathon being organised by 3mdeb. (I can also add you to #kgpe-d16-private where the bulk of the discussion occurred before opting for a public channel)

kuleszdl, 2021-06-09:
@Tonux599 I don't use slack and usually I prefer discussing such issues more in-depth over public issues (such as here) or on mailing lists. Yet, I am very pleased to hear about the planned hackathon which sounds like a very promising idea!

duncancmt, 2021-08-04:
I am Duncan Townsend, CTO of [Immunefi](https://immunefi.com/). We love what you're doing. Immunefi would like to pledge your choice of 17,750 [USDC](https://etherscan.io/token/0xa0b86991c6218b36c1d19d4a2e9eb0ce3606eb48) (approx 15,000 EUR) or 10 Ether (approx 22,700 EUR) towards the completion of this work. You can view the pledged funds at [0xD24F38eCe7084F8bAB4BEc23cbcC9a50566Bc38C](https://etherscan.io/address/0xd24f38ece7084f8bab4bec23cbcc9a50566bc38c) or at [kgped16pledge.immunefi.eth](https://etherscan.io/address/kgped16pledge.immunefi.eth). 

Would you like for us to raise awareness of this need throughout the crypto community? We believe this is something that everybody in crypto should be using and want to make sure this happens. Let us know how we can cross-promote and raise awareness.

Please send an email to team@immunefi.com if further coordination is required.

[44 more replies]